avrdudes/avrdude #725: [patch #9328] ft245r.c: add TPI support (patches 5-7)

avrdude-admin, 2021-12-13:
Thu 04 May 2017 11:13:06 PM UTC

See also [https://bitbucket.org/egauge/avrdude/commits/all](https://savannah.nongnu.orghttps://bitbucket.org/egauge/avrdude/commits/all)

[file #40589: ft245r-add-d2xx-compat.diff](https://savannah.nongnu.org/patch/download.php?file_id=40589)
[file #40588: ft245r-fix-baudrate.diff](https://savannah.nongnu.org/patch/download.php?file_id=40588)
[file #40587: ft245r-eliminate-reader-thread.diff](https://savannah.nongnu.org/patch/download.php?file_id=40587)

This issue was migrated from https://savannah.nongnu.org/patch/?9328

avrdude-admin, 2021-12-13:
Rene Liebscher <rliebscher>
Sun 24 Mar 2019 11:40:33 PM UTC

If all stuff also works with the current libftdi based solution, and only licenses make problems with the d2xx part, so why not separate this d2xx part in a separate patch?

avrdude-admin, 2021-12-13:
Joerg Wunsch <joerg_wunsch>
Wed 24 Nov 2021 09:12:20 AM UTC

I'm not sure about D2XX. The issue came up earlier, it might contradict Savannah's policy to offer "non-free" software hooks. Let's see.

I'm quite impressed about the sheer amount of rework you did here.

Question about that baudrate fixing patch: Would it hurt anyone to offer this as a default? I've tried testing the ft245r driver on my (old) UM245R board, and have the impression I'm affected by that. Since most people probably use AVRDUDE as a binary, the compile-time option wouldn't be much useful to those users.

Alternatively (if enabling it has a negative impact to other features), any chance for a run-time guess whether it needs to be enabled?

avrdude-admin, 2021-12-13:
Joerg Wunsch <joerg_wunsch>
Thu 25 Nov 2021 09:18:31 AM UTC

Except of the D2XX patch, integrated in SVN now (r1488).

Thanks for this major overhaul of the ft245r driver!

mcuee, 2022-05-24:
Maybe this can be closed as the main parts have been integrated.

mcuee, 2022-05-24:
> I'm not sure about D2XX. The issue came up earlier, it might contradict Savannah's policy to offer "non-free" software hooks. Let's see.

Actually the official Windows binary is now using D2XX to give the user a good out-of-box experience so that they do not need to switch driver.

Some discussion and history in the OpenOCD project discussion here.
https://github.com/avrdudes/avrdude/issues/961#issuecomment-1127095993

Probably there is no issue here as the avrdude libftdi custom version is kind of a wrapper of D2XX.

mcuee, 2022-05-29:
> Maybe this can be closed as the main parts have been integrated.

@dl8dtl Please take a look. Thanks.

closed 2022-06-19 by mcuee

mcuee, 2022-12-08:
In the end, I believe avrdude-libftdi is okay and can be considered as a D2XX wrapper.