avrdudes/avrdude #733: [patch #9563] New feature: ability to reset FLIPv2 targets

avrdude-admin, 2021-12-13:
Sat 10 Feb 2018 04:38:09 PM UTC

This patch adds the ability to reset FLIPv2 DFU devices after programming, by adding the "-e reset" command-line option.

The command-line interface and implementation is patterned after the 'par' driver.

[file #43232: avrdude-flip2-reset.patch](https://savannah.nongnu.org/patch/download.php?file_id=43232)

This issue was migrated from https://savannah.nongnu.org/patch/?9563

avrdude-admin, 2021-12-13:
James Cook <mokus>
Mon 12 Feb 2018 04:58:42 AM UTC

Minor correction: the command-line option to enable this is, as in the case of the 'par' plugin, "-E reset" (not "-e reset")

avrdude-admin, 2021-12-13:
Joerg Wunsch <joerg_wunsch>
Tue 09 Nov 2021 09:34:40 PM UTC

Please document the new option, either in avrdude.1 or in avrdude.texi (or in both :).

MCUdude, 2022-03-25:
@dl8dtl I have no experience with FLIP at all, and I'm not even sure I own any FLIP compatible hardware. But if you want I can create a PR where I've collected the bits and pieces from patch 43232 with some added documentation to avrdude.1 and avrdude.texi?

mcuee, 2022-05-28:
If you have Arduino Uno with ATmega16U2, I think that one can be put into DFU mode and use FLIP (v1)
Ref: https://support.arduino.cc/hc/en-us/articles/4410804625682
Ref: https://support.arduino.cc/hc/en-us/articles/4408887452434-Flash-USB-to-serial-firmware-in-DFU-mode

Ref: FLIP V1 for AVR mega 
http://ww1.microchip.com/downloads/en/devicedoc/doc7618.pdf

mcuee, 2022-07-29:
@MCUdude 
Maybe it is good to have the pull request. I am getting an ATxmega32U4 breakout board and hopefully I can test this in another month.

MCUdude, 2022-08-05:
I may receive some Xmega hardware later in August. I'll have a look at creating a PR for this after a bit of testing first.

mcuee, 2022-08-20:
The patch is posted here to make things a bit easier.

```
Index: flip2.c
===================================================================
--- flip2.c	(revision 1425)
+++ flip2.c	(working copy)
@@ -136,6 +136,7 @@
 static void flip2_display(PROGRAMMER* pgm, const char *prefix);
 static int flip2_program_enable(PROGRAMMER* pgm, AVRPART *part);
 static int flip2_chip_erase(PROGRAMMER* pgm, AVRPART *part);
+static int flip2_start_app(PROGRAMMER *pgm);
 static int flip2_read_byte(PROGRAMMER* pgm, AVRPART *part, AVRMEM *mem,
   unsigned long addr, unsigned char *value);
 static int flip2_write_byte(PROGRAMMER* pgm, AVRPART *part, AVRMEM *mem,
@@ -145,6 +146,7 @@
 static int flip2_paged_write(PROGRAMMER* pgm, AVRPART *part, AVRMEM *mem,
   unsigned int page_size, unsigned int addr, unsigned int n_bytes);
 static int flip2_read_sig_bytes(PROGRAMMER* pgm, AVRPART *part, AVRMEM *mem);
+static int flip2_parseexitspecs(PROGRAMMER* pgm, char *s);
 static void flip2_setup(PROGRAMMER * pgm);
 static void flip2_teardown(PROGRAMMER * pgm);
 
@@ -193,6 +195,7 @@
   pgm->read_byte        = flip2_read_byte;
   pgm->write_byte       = flip2_write_byte;
   pgm->read_sig_bytes   = flip2_read_sig_bytes;
+  pgm->parseexitspecs   = flip2_parseexitspecs;
   pgm->setup            = flip2_setup;
   pgm->teardown         = flip2_teardown;
 }
@@ -323,6 +326,10 @@
 void flip2_close(PROGRAMMER* pgm)
 {
   if (FLIP2(pgm)->dfu != NULL) {
+    if (pgm->exit_reset == EXIT_RESET_ENABLED) {
+        flip2_start_app(pgm);
+    }
+    
     dfu_close(FLIP2(pgm)->dfu);
     FLIP2(pgm)->dfu = NULL;
   }
@@ -388,6 +395,22 @@
   return cmd_result;
 }
 
+int flip2_start_app(PROGRAMMER *pgm) {
+    avrdude_message(MSG_INFO, "%s: Starting application\n", progname);
+    
+    struct flip2_cmd cmd = {
+        FLIP2_CMD_GROUP_EXEC, FLIP2_CMD_START_APP, { 0x00, 0, 0, 0 }
+    };
+    
+    // queue command
+    int cmd_result = dfu_dnload(FLIP2(pgm)->dfu, &cmd, sizeof(cmd));
+    
+    // repeat dnload to actually execute
+    dfu_dnload(FLIP2(pgm)->dfu, &cmd, sizeof(cmd));
+    
+    return cmd_result;
+}
+
 int flip2_read_byte(PROGRAMMER* pgm, AVRPART *part, AVRMEM *mem,
   unsigned long addr, unsigned char *value)
 {
@@ -502,6 +525,22 @@
   return (result == 0) ? n_bytes : -1;
 }
 
+/*
+ *  Parse the -E option string
+ */
+int flip2_parseexitspecs(PROGRAMMER* pgm, char *s)
+{
+    if (strcmp(s, "reset") == 0) {
+        pgm->exit_reset = EXIT_RESET_ENABLED;
+    } else if (strcmp(s, "noreset") == 0) {
+        pgm->exit_reset = EXIT_RESET_DISABLED;
+    } else {
+        return -1;
+    }
+    
+    return 0;
+}
+
 int flip2_read_sig_bytes(PROGRAMMER* pgm, AVRPART *part, AVRMEM *mem)
 {
   if (FLIP2(pgm)->dfu == NULL)

```

MCUdude, 2022-09-23:
First, I could not get FLIP2 to work unless I added the USB PID:

```
#------------------------------------------------------------
# flip2
#------------------------------------------------------------

programmer
    id                     = "flip2";
    desc                   = "FLIP for bootloader using USB DFU protocol version 2 (AVR4023)";
    type                   = "flip2";
    prog_modes             = PM_SPM;
    connection_type        = usb;
    usbpid                 = 0x2fec;
;
```

I had to manually apply the patch, but I can confirm it's working like it should. Without it, I'll either have to power cycle or reset the board manually, but with the new `-E reset` flag, it happens automatically after upload. if `-E reset` is not used, the program doesn't start automatically after upload, but it instead stays in bootloader mode.

```
$ ./avrdude -cflip2 -p atxmega256a3bu -Uapplication:w:/Users/hans/Downloads/XMEGA_A3BU_XPLAINED_DEMO1.hex -E reset

avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.00s

avrdude: Device signature = 0x1e9843 (probably x256a3bu)
avrdude: NOTE: "application" memory has been specified, an erase cycle will be performed
         To disable this feature, specify the -D option.
avrdude: erasing chip
avrdude: reading input file /Users/hans/Downloads/XMEGA_A3BU_XPLAINED_DEMO1.hex for application
avrdude: writing 35062 bytes application ...

Writing | ################################################## | 100% 0.42s

avrdude: 35062 bytes of application written
avrdude: verifying application memory against /Users/hans/Downloads/XMEGA_A3BU_XPLAINED_DEMO1.hex

Reading | ################################################## | 100% 0.18s

avrdude: 35062 bytes of application verified
avrdude: Starting application

avrdude done.  Thank you.
```

I'm getting a compiler warning though:

```
flip2.c:192:25: warning: incompatible pointer types assigning to 'int (*)(struct programmer_t *, const char *)' from 'int (PROGRAMMER *, char *)'
      (aka 'int (struct programmer_t *, char *)') [-Wincompatible-pointer-types]
  pgm->parseexitspecs   = flip2_parseexitspecs;
                        ^ ~~~~~~~~~~~~~~~~~~~~
1 warning generated.
```

PR #1102 (2022-10-05): Add support for Flip 2 reset on exit
This allows the application to start immediately after the program has been loaded. Simply use `-E reset` or `-E noreset`. Default is no reset. Tested on my Xplained ATxmega256A3BU board.

I've currently not added anything to the docs. The Flip 2 protocol is hardly even mentioned, and I'm not sure where to put the additional information that Flip2 supports `-E reset`.

Closes #733
commit 80ca8a64 (2022-10-05): Add support for Flip 2 reset on exit
This makes it possible for the application to start immedeatly after the program has been loaded.
Simply use '-E reset' or '-E noreset'. Default is no reset.
Closes #733
closed 2022-10-05 by stefanrueger