avrdudes/avrdude #735: [patch #9573] patch to add atmelice avr32 programming support to avrdude

avrdude-admin, 2021-12-13:
Thu 22 Feb 2018 10:16:28 AM UTC

This patch adds support for programming avr32 devices via the atmelice jtag programmer using the jtag3 driver. The update add a new programmer atmelice_avr32 which is used to access the new feature

[file #43383: avr32-jtag3.patch](https://savannah.nongnu.org/patch/download.php?file_id=43383)
[file #51797: avr32-jtag3.patch](https://savannah.nongnu.org/patch/download.php?file_id=51797)

This issue was migrated from https://savannah.nongnu.org/patch/?9573

avrdude-admin, 2021-12-13:
Bradley Jarvis <bradley_jarvis>
Thu 22 Feb 2018 10:26:54 AM UTC

This patch is in relation to **[bug #43178](https://savannah.nongnu.org/bugs/?43178)**, I have had a few people ask me for the patch lately so want to add it back into avrdude. I just re-applied my patch made on 6.1 which was the original code base I worked with to the latest svn trunk version.

avrdude-admin, 2021-12-13:
Joerg Wunsch <joerg_wunsch>
Sun 20 Sep 2020 08:43:22 PM UTC

Thanks for submitting the patch!

First thing that springs to mind after applying it:

```
gmake[2]: Entering directory '/home/joerg/src/avrdude'
CC       libavrdude_a-jtag3.o
jtag3.c:1031:12: warning: 'jtag3_edbg_recv_frame' defined but not used [-Wunused-function]
static int jtag3_edbg_recv_frame(PROGRAMMER * pgm, unsigned char **msg) {
^~~~~~~~~~~~~~~~~~~~~

```
So why is this function no longer going to be used?

After some closer inspection, I have a number of stylistic issues with the patch before I'd like to apply it.

```
programmer
+  id    = "atmelice_avr32";
+  desc  = "Atmel-ICE (ARM/AVR) in ISP mode";

```
Really in ISP mode?

```
+        if (data[0]==SCOPE_AVR32)
+        {
+            char *str=0;
...
+              case AVR32_FAILURE_RECEIVE_LENGTH: str="Incorrect packet length"; break;
...
+  if (len > (((USBDEV_MAX_XFER_3-4)*16)-4))

```
I would appreciate if the patch keeps the existing style rule of having white space around operators (=, == etc.).

```
-         avrdude_message(MSG_INFO, "\\%03o", data[i]);
+          //avrdude_message(MSG_INFO, "\\%03o", data[i]);
+          avrdude_message(MSG_INFO, "\\%02x", data[i]);
...
+#if 0
/* 4 bytes overhead for CMD, fragment #, and length info */
int max_xfer = pgm->fd.usb.max_xfer;
int nfragments = (len + max_xfer - 1) / max_xfer;
@@ -507,6 +647,7 @@
}
data += this_len;
len -= this_len;
+#endif
}

```
Source code comments and conditional compilation is not the way to keep track of history. This is what SVN (or whatever VCS is used) is for.

Also, as you can see in the first snippet, make sure to not artificially alter the indentation style. Unfortunately, AVRDUDE's code base was never very consistent on whether to use TAB or spaces, but if you change a single existing line, do not convert a TAB to spaces or vice versa.

```
+  //if ((buf = malloc(USBDEV_MAX_XFER_3)) == NULL) {
+  //  avrdude_message(MSG_INFO, "%s: jtag3_edbg_recv(): out of memory\n",
+       //    progname);
+  //  return -1;
+  //}

```
See above. If some piece of code isn't needed anymore, remove it, rather than commenting it out.

```
if ((c == RSP3_FAILED) && ((*resp)[3] == RSP3_FAIL_OCD_LOCKED)) {
avrdude_message(MSG_INFO,
-                     "%s: Device is locked! Chip erase required to unlock.\n",
-                     progname);
+          "%s: Device is locked! Chip erase required to unlock.\n",
+          progname);
} else {
avrdude_message(MSG_INFO, "%s: bad response to %s command: 0x%02x\n",
-                     progname, descr, c);
+          progname, descr, c);

```
Again, random indentation changes, please avoid this. Tell you editor to not touch anything else but what you're actually editing.

```
-  free(resp);
+  if (resp) free(resp);

```
Unneeded. The C standard guarantees free(0) does nothing.

The entire patch looks quite intrusive to the existing jtag3.c file. Thus, it imposes quite a bit of risk of breaking things that are unrelated to the AVR32 addition (see the first warning about a now unused function).

For that reason, I'd like to defer inclusion of the patch until after the next AVRDUDE release. Sorry, I don't have the time and energy of maintaining a separate -stable branch.

However, even though I'm setting it to "postponed" now, please feel encouraged to re-submit the patch with the above things fixed.

avrdude-admin, 2021-12-13:
Bradley Jarvis <bradley_jarvis>
Mon 21 Sep 2020 02:09:36 AM UTC

No problem, I will update the patch following the formatting guides mentioned.

The function jtag3_edbg_recv_frame was only being called from jtag3_recv_frame and it had to be fixed so it was integrated into this function, from memory it was how it was handling fragmented frames. I initially implemented this patch on avrdude 6.1 back in 2015 and I believe in that version the jtag3.c module could not receive fragmented frames(besides I think my version is much clearer/cleaner ;).

I think the ISP comment is a copy/paste error, it should have been JTAG, thanks for picking that up.

I have made the changes and once I get a chance to test it on some hardware and then I will update the patch.

mcuee, 2022-05-24:
Can this patch be adapted for  Atmel JTAG ICE mkII (as well as AVR Dragon) for the AVR32 support?

mcuee, 2022-05-24:
> Can this patch be adapted for Atmel JTAG ICE mkII (as well as AVR Dragon) for the AVR32 support? They do support AVR32.
> 
> Ref: https://www.microchip.com/en-us/development-tool/ATAVRDRAGON https://www.microchip.com/en-us/development-tool/ATJTAGICE2

The patch is here.
* https://github.com/avrdudes/avrdude/issues/239

mcuee, 2022-05-24:
The other thing is that this support can probably extended to other UC3 parts.
Ref: https://www.avrfreaks.net/forum/at32uc3c0512-avrdude-part-definition

mcuee, 2022-06-23:
AVR32 has been de-focused by Microchip/Atmel long ago. So we will close this issue.

closed 2022-06-23 by mcuee

mcuee, 2023-06-11:
Patch:

```
Index: avrdude.conf.in
===================================================================
--- avrdude.conf.in	(revision 1453)
+++ avrdude.conf.in	(working copy)
@@ -1230,6 +1230,15 @@
 ;
 
 programmer
+  id    = "atmelice_avr32";
+  desc  = "Atmel-ICE (ARM/AVR) in JTAG mode";
+  type  = "jtagice3_avr32";
+  connection_type = usb;
+  usbpid = 0x2141;
+;
+
+
+programmer
   id    = "pavr";
   desc  = "Jason Kyle's pAVR Serial Programmer";
   type  = "avr910";
Index: jtag3.c
===================================================================
--- jtag3.c	(revision 1453)
+++ jtag3.c	(working copy)
@@ -70,6 +70,9 @@
   /* Start address of Xmega boot area */
   unsigned long boot_start;
 
+  /* AVR32 flashc address */
+  unsigned long flashc_start;
+
   /* Function to set the appropriate clock parameter */
   int (*set_sck)(PROGRAMMER *, unsigned char *);
 };
@@ -85,13 +88,18 @@
 #define PGM_FL_IS_JTAG          (0x0004)
 #define PGM_FL_IS_EDBG          (0x0008)
 #define PGM_FL_IS_UPDI          (0x0010)
+#define PGM_FL_IS_AVR32         (0x0020)
 
 static int jtag3_open(PROGRAMMER * pgm, char * port);
 static int jtag3_edbg_prepare(PROGRAMMER * pgm);
 static int jtag3_edbg_signoff(PROGRAMMER * pgm);
 static int jtag3_edbg_send(PROGRAMMER * pgm, unsigned char * data, size_t len);
-static int jtag3_edbg_recv_frame(PROGRAMMER * pgm, unsigned char **msg);
 
+static int jtag3_avr32_busy(PROGRAMMER * pgm);
+static int jtag3_avr32_read_register(PROGRAMMER * pgm, Avr32SABAddressMap, unsigned int, unsigned int *);
+static int jtag3_avr32_write_register(PROGRAMMER * pgm, Avr32SABAddressMap, unsigned int, unsigned int);
+static int jtag3_avr32_status_message(unsigned char code, short len, char *msg);
+
 static int jtag3_initialize(PROGRAMMER * pgm, AVRPART * p);
 static int jtag3_chip_erase(PROGRAMMER * pgm, AVRPART * p);
 static int jtag3_read_byte(PROGRAMMER * pgm, AVRPART * p, AVRMEM * mem,
@@ -215,6 +223,10 @@
       avrdude_message(MSG_INFO, "[AVR] ");
       break;
 
+    case SCOPE_AVR32:
+      avrdude_message(MSG_INFO, "[AVR32] ");
+      break;
+
     default:
       avrdude_message(MSG_INFO, "[scope 0x%02x] ", data[0]);
       break;
@@ -231,6 +243,12 @@
       {
[... 1830 lines cut ...]
+    
+enum EDBGSetGetContexts {
+    EDBG_CONTEXT_CONTROL                                = 0x00,
+    EDBG_CONTEXT_CONFIG0                                = 0x10,
+    EDBG_CONTEXT_CONFIG1                                = 0x11,
+}; 
+
+enum EDBGControlContextParameters {
+    EDBG_CONTROL_LED_USAGE                              = 0x00,
+    EDBG_CONTROL_EXT_PROG                               = 0x01,
+    EDBG_CONTROL_TARGET_POWER                           = 0x10
+};
Index: pgm_type.c
===================================================================
--- pgm_type.c	(revision 1453)
+++ pgm_type.c	(working copy)
@@ -81,6 +81,7 @@
         {"jtagice3_updi", jtag3_updi_initpgm, jtag3_updi_desc},
         {"jtagice3_dw", jtag3_dw_initpgm, jtag3_dw_desc},
         {"jtagice3_isp", stk500v2_jtag3_initpgm, stk500v2_jtag3_desc},
+        {"jtagice3_avr32", jtag3_avr32_initpgm, jtag3_avr32_desc},
         {"linuxgpio", linuxgpio_initpgm, linuxgpio_desc},
         {"linuxspi", linuxspi_initpgm, linuxspi_desc},
         {"par", par_initpgm, par_desc},
```