avrdudes/avrdude #737: [patch #9728] Fix UsbTiny programmer

avrdude-admin, 2021-12-13:
Thu 22 Nov 2018 08:27:13 AM UTC

As reported on [https://github.com/arduino/Arduino/issues/8198](https://savannah.nongnu.orghttps://github.com/arduino/Arduino/issues/8198) ,
`**[patch #8924](https://savannah.nongnu.org/patch/?8924)**: Enable TPI for usbtiny` broke the functionality of such programmer for non TPI targets.

The refactor included reworking `usbtiny_initialize`, which used to call `pgm->program_enable` once and, in case of failure, reset the programmer and call it again.

The new code considers the return code successful if >=0, so the first failure is not reported and the programmer is not being reset.

The attached patch restores the previous behaviour by checking if `pgm->program_enable() == 0`

[file #45511: 0001-usbtiny.patch](https://savannah.nongnu.org/patch/download.php?file_id=45511)

This issue was migrated from https://savannah.nongnu.org/patch/?9728

avrdude-admin, 2021-12-13:
Joerg Wunsch <joerg_wunsch>
Sat 05 Jan 2019 11:16:44 PM UTC

Applied in r1429, thanks!

avrdude-admin, 2021-12-13:
Joerg Wunsch <joerg_wunsch>
Sun 07 Nov 2021 03:42:43 PM UTC

Undid the patch due to **[bug #55734](https://savannah.nongnu.org/bugs/?55734)**

Please re-submit, considering the complaints filed in that bug.

MCUdude, 2022-01-10:
This issue can be closed. I can verify that my USBtiny programmer can communicate with TPI and ISP targets with the latest version of Avrdude