avrdudes/avrdude #741: [patch #9757] Fix ATtiny817 Xplained Mini programmer

avrdude-admin, 2021-12-13:
Fri 22 Feb 2019 04:55:14 PM UTC

Hi,

I had already posted to the mailing list, but didn't get a response, so I'll try here as well. This was what I posted to the mailing list:
 [https://lists.nongnu.org/archive/html/avrdude-dev/2019-01/msg00005.html](https://savannah.nongnu.orghttps://lists.nongnu.org/archive/html/avrdude-dev/2019-01/msg00005.html)

I think the error is caused by the way the programmer ID is compared against "xplainedmini_updi". The attached patch seems to fix the issue. However, I don't quite understand why a programmer would need a list of IDs in the first place, so I'd appreciate it if some skeptical eyes looked over my patch.

Thanks,
Sven

[file #46335: patch](https://savannah.nongnu.org/patch/download.php?file_id=46335)

This issue was migrated from https://savannah.nongnu.org/patch/?9757

avrdude-admin, 2021-12-13:
Joerg Wunsch <joerg_wunsch>
Sun 20 Sep 2020 08:03:34 PM UTC

Well, I don't think programmers usually have multiple IDs. However, the avrdude.conf syntax (see config_gram.y) allows for a string_list in that place.

I guess we don't really need to care for more than one ID, but we need to take into account that pgm->id points to the **list** rather than the list's datum.

IMHO, the following patch ought to fix it as well, can you verify?

```
Index: jtag3.c
===================================================================
--- jtag3.c     (revision 1448)
+++ jtag3.c     (working copy)
@@ -867,7 +867,7 @@
/* XplainedMini boards do not need this, and early revisions had a
* firmware bug where they complained about it. */
if (pgm->flag & PGM_FL_IS_EDBG) {
-    if (strcmp(pgm->id, "xplainedmini_updi") != 0) {
+    if (strcmp(ldata(lfirst(pgm->id)), "xplainedmini_updi") != 0) {
if (jtag3_edbg_prepare(pgm) < 0) {
return -1;
}
@@ -1581,7 +1581,7 @@
/* XplainedMini boards do not need this, and early revisions had a
* firmware bug where they complained about it. */
if (pgm->flag & PGM_FL_IS_EDBG) {
-    if (strcmp(pgm->id, "xplainedmini_updi") != 0) {
+    if (strcmp(ldata(lfirst(pgm->id)), "xplainedmini_updi") != 0) {
jtag3_edbg_signoff(pgm);
}
}

```

avrdude-admin, 2021-12-13:
Sven <svenschwermer>
Mon 28 Sep 2020 06:21:00 PM UTC

Hi Jörg,

it took me some time to verify this because between me proposing this patch and your comment, a regression has been introduced: The last revision that my patch works on is r1435. r1436 breaks this programmer again—even with my patch applied. See the extra verbose output for r1436 with my originally suggested patch attached at the end of this post.

On r1435, I tried your patch. It works just as well as my proposed patch. I agree that your patch is cleaner if it's safe to assume that programmers have only up to one ID.

Sven

===

Patched (my original patch) r1436:

./avrdude -p t817 -c xplainedmini_updi -C avrdude.conf -v -v

avrdude: Version 6.3-20171130
Copyright (c) 2000-2005 Brian Dean, [http://www.bdmicro.com/](https://savannah.nongnu.orghttp://www.bdmicro.com/)
Copyright (c) 2007-2014 Joerg Wunsch

System wide configuration file is "avrdude.conf"
User configuration file is "/home/sven/.avrduderc"
User configuration file does not exist or is not a regular file, skipping

Using Port                    : usb
Using Programmer              : xplainedmini_updi
avrdude: jtag3_open_updi()
avrdude: Found CMSIS-DAP compliant device, using EDBG protocol
avrdude: Sending sign-on command: 0x80 (3 bytes msg)
AVR Part                      : ATtiny817
Chip Erase delay              : 0 us
PAGEL                         : P00
BS2                           : P00
RESET disposition             : dedicated
RETRY pulse                   : SCK
serial program mode           : yes
parallel program mode         : yes
Timeout                       : 0
StabDelay                     : 0
CmdexeDelay                   : 0
SyncLoops                     : 0
ByteDelay                     : 0
PollIndex                     : 0
PollValue                     : 0x00
Memory Detail                 :

Block Poll               Page                       Polled
Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
signature      0     0     0    0 no          3    1      0     0     0 0x00 0x00
prodsig        0     0     0    0 no         61   61      0     0     0 0x00 0x00
fuses          0     0     0    0 no          9    1      0     0     0 0x00 0x00
fuse0          0     0     0    0 no          1    1      0     0     0 0x00 0x00
fuse1          0     0     0    0 no          1    1      0     0     0 0x00 0x00
fuse2          0     0     0    0 no          1    1      0     0     0 0x00 0x00
fuse4          0     0     0    0 no          1    1      0     0     0 0x00 0x00
fuse5          0     0     0    0 no          1    1      0     0     0 0x00 0x00
fuse6          0     0     0    0 no          1    1      0     0     0 0x00 0x00
fuse7          0     0     0    0 no          1    1      0     0     0 0x00 0x00
fuse8          0     0     0    0 no          1    1      0     0     0 0x00 0x00
lock           0     0     0    0 no          1    1      0     0     0 0x00 0x00
data           0     0     0    0 no          0    1      0     0     0 0x00 0x00
usersig        0     0     0    0 no         32   32      0     0     0 0x00 0x00
flash          0     0     0    0 no       8192   64      0     0     0 0x00 0x00
eeprom         0     0     0    0 no        128   32      0     0     0 0x00 0x00

Programmer Type : JTAGICE3_UPDI
Description     : Atmel AVR XplainedMini in UPDI mode
avrdude: jtag3_getparm()
avrdude: Sending get parameter (scope 0x01, section 0, parm 0) command: 0x84 (9 bytes msg)
avrdude: Sending get info (serial number) command: 0x81 (23 bytes msg)
ICE hardware version: 0
ICE firmware version: 1.13 (rel. 43)
Serial number   : ATML2658061800003615
avrdude: jtag3_getparm()
avrdude: Sending get parameter (scope 0x01, section 1, parm 0) command: 0x84 (6 bytes msg)
Vtarget         : 5.00 V
avrdude: jtag3_getparm()
avrdude: Sending get parameter (scope 0x12, section 1, parm 32) command: 0x84 (6 bytes msg)
JTAG clock megaAVR/program: 0 kHz
avrdude: jtag3_getparm()
avrdude: Sending get parameter (scope 0x12, section 1, parm 33) command: 0x84 (6 bytes msg)
JTAG clock megaAVR/debug:   0 kHz
avrdude: jtag3_getparm()
avrdude: Sending get parameter (scope 0x12, section 1, parm 48) command: 0x84 (6 bytes msg)
JTAG clock Xmega: 0 kHz
avrdude: jtag3_getparm()
avrdude: Sending get parameter (scope 0x12, section 1, parm 49) command: 0x84 (6 bytes msg)
PDI clock Xmega : 0 kHz

avrdude: jtag3_getparm()
avrdude: Sending get parameter (scope 0x01, section 0, parm 1) command: 0x84 (6 bytes msg)
avrdude: jtag3_setparm()
avrdude: Sending set parameter (scope 0x12, section 0, parm 0) command: 0x80 (3 bytes msg)
avrdude: jtag3_setparm()
avrdude: Sending set parameter (scope 0x12, section 0, parm 1) command: 0x80 (3 bytes msg)
avrdude: jtag3_setparm()
avrdude: Sending set parameter (scope 0x12, section 1, parm 0) command: 0x80 (3 bytes msg)
UPDI SET:
xd->prog_base=0 80
xd->flash_page_size=40
xd->eeprom_page_size=20
xd->nvmctrl=0 10
xd->ocd=80 f
avrdude: jtag3_setparm()
avrdude: Sending set parameter (scope 0x12, section 2, parm 0) command: 0x80 (3 bytes msg)
avrdude: Sending AVR sign-on command: 0x84 (8 bytes msg)
avrdude: Partial Family_ID returned: "tiny"
avrdude: AVR device initialized and ready to accept instructions

Reading |                                                    | 0% 0.00savrdude: jtag3_paged_load(.., signature, 1, 0x0, 1)
mapped to address: 0x1100
avrdude: Sending enter progmode command: 0x80 (3 bytes msg)
avrdude: Sending read memory command: 0x84 (4 bytes msg)
avrdude: Sending read memory command: 0x84 (4 bytes msg)
avrdude: Sending read memory command: 0x84 (4 bytes msg)
[... the last lines repeat indefinitely ...]

avrdude-admin, 2021-12-13:
Joerg Wunsch <joerg_wunsch>
Mon 28 Sep 2020 08:47:28 PM UTC

Thanks for the confirmation, Sven!

avrdude-admin, 2021-12-13:
Andrew Kohlsmith <akohlsmith>
Mon 22 Feb 2021 05:04:57 AM UTC

Hello,

I am using svn trunk on osx with an Atmel Ice. I am running into the exact same problem (endless "avrdude: Sending read memory command: 0x84 (4 bytes msg)") and I am wondering if you've found anything which addresses this regression?

avrdude-admin, 2021-12-13:
Joel Holdsworth <jhol>
Wed 08 Sep 2021 08:36:27 PM UTC

A lot of people are suffering with this issue on new Atmel devices. (For example: [https://github.com/facchinm/avrdude/issues/3](https://savannah.nongnu.orghttps://github.com/facchinm/avrdude/issues/3)). I myself just wasted an afternoon trying to figure out why avrdude couldn't talk to my ATtiny416 Xplained Nano.

Please can we have this patch applied, and a new avrdude release? I think this bug is causing a lot of people problems.

avrdude-admin, 2021-12-13:
Joerg Wunsch <joerg_wunsch>
Tue 23 Nov 2021 06:20:07 AM UTC

Well, my Xplained Mini with the ATtiny817 arrived.

It's got a completely different issue: most of the time, it doesn't accept the sign-on command, returning a response with fragment info 0x00. This triggers an "inconsistent fragment number" complaint, according to the Atmel docs, it means there is no response available.

On an Xplained Mini with the ATmega328PB, that never happens ...

Strange.

avrdude-admin, 2021-12-13:
Joerg Wunsch <joerg_wunsch>
Tue 23 Nov 2021 09:43:30 AM UTC

OK, I added something similar to the patch below.

Will look into the other issue before closing the item.

avrdude-admin, 2021-12-13:
Joerg Wunsch <joerg_wunsch>
Tue 23 Nov 2021 10:30:30 AM UTC

Just tested this:

closed 2021-12-13 by avrdude-admin

mcuee, 2022-08-31:
The last post was not migrated properly. The following run log is missing.

```
% ./avrdude -p t817 -c xplainedmini_updi -q -C avrdude.conf -v -v

avrdude: Version 6.3-20171130
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "avrdude.conf"
         User configuration file is "/home/joerg/.avrduderc"

         Using Port                    : usb
         Using Programmer              : xplainedmini_updi
avrdude: jtag3_open_updi()
avrdude: Found CMSIS-DAP compliant device, using EDBG protocol
avrdude: Sending sign-on command: 0x80 (3 bytes msg)
         AVR Part                      : ATtiny817
         Chip Erase delay              : 0 us
         PAGEL                         : P00
         BS2                           : P00
         RESET disposition             : dedicated
         RETRY pulse                   : SCK
         serial program mode           : yes
         parallel program mode         : yes
         Timeout                       : 0
         StabDelay                     : 0
         CmdexeDelay                   : 0
         SyncLoops                     : 0
         ByteDelay                     : 0
         PollIndex                     : 0
         PollValue                     : 0x00
         Memory Detail                 :

                                  Block Poll               Page                       Polled
           Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           signature      0     0     0    0 no          3    1      0     0     0 0x00 0x00
           prodsig        0     0     0    0 no         61   61      0     0     0 0x00 0x00
           fuses          0     0     0    0 no          9   10      0     0     0 0x00 0x00
           fuse0          0     0     0    0 no          1    1      0     0     0 0x00 0x00
           fuse1          0     0     0    0 no          1    1      0     0     0 0x00 0x00
           fuse2          0     0     0    0 no          1    1      0     0     0 0x00 0x00
           fuse4          0     0     0    0 no          1    1      0     0     0 0x00 0x00
           fuse5          0     0     0    0 no          1    1      0     0     0 0x00 0x00
           fuse6          0     0     0    0 no          1    1      0     0     0 0x00 0x00
           fuse7          0     0     0    0 no          1    1      0     0     0 0x00 0x00
           fuse8          0     0     0    0 no          1    1      0     0     0 0x00 0x00
           lock           0     0     0    0 no          1    1      0     0     0 0x00 0x00
           data           0     0     0    0 no          0    1      0     0     0 0x00 0x00
           usersig        0     0     0    0 no         32   32      0     0     0 0x00 0x00
           flash          0     0     0    0 no       8192   64      0     0     0 0x00 0x00
           eeprom         0     0     0    0 no        128   32      0     0     0 0x00 0x00

         Programmer Type : JTAGICE3_UPDI
         Description     : Atmel AVR XplainedMini in UPDI mode
avrdude: jtag3_getparm()
avrdude: Sending get parameter (scope 0x01, section 0, parm 0) command: 0x84 (9 bytes msg)
avrdude: Sending get info (serial number) command: 0x81 (23 bytes msg)
         ICE hardware version: 0
         ICE firmware version: 1.13 (rel. 43)
         Serial number   : ATML2658061800008969
avrdude: jtag3_getparm()
avrdude: Sending get parameter (scope 0x01, section 1, parm 0) command: 0x84 (6 bytes msg)
         Vtarget         : 5.00 V
avrdude: jtag3_getparm()
avrdude: Sending get parameter (scope 0x12, section 1, parm 32) command: 0x84 (6 bytes msg)
         JTAG clock megaAVR/program: 0 kHz
avrdude: jtag3_getparm()
avrdude: Sending get parameter (scope 0x12, section 1, parm 33) command: 0x84 (6 bytes msg)
[... 24 lines cut ...]
        avrdude: jtag3_setparm()
avrdude: Sending set parameter (scope 0x12, section 2, parm 0) command: 0x80 (3 bytes msg)
avrdude: Sending AVR sign-on command: 0x84 (8 bytes msg)
avrdude: Partial Family_ID returned: "tiny"
avrdude: AVR device initialized and ready to accept instructions
avrdude: jtag3_read_byte(.., signature, 0x0, ...)
          mapped to address: 0x1100
avrdude: Sending enter progmode command: 0x80 (3 bytes msg)
avrdude: Sending read memory command: 0x84 (7 bytes msg)
avrdude: jtag3_read_byte(.., signature, 0x1, ...)
          mapped to address: 0x1101
avrdude: jtag3_read_byte(.., signature, 0x2, ...)
          mapped to address: 0x1102
avrdude: Device signature = 0x1e9320 (probably t817)

avrdude: safemode: Fuses OK (E:FF, H:FF, L:FF)
avrdude: Sending leave progmode command: 0x80 (3 bytes msg)
avrdude: jtag3_close()
avrdude: Sending AVR sign-off command: 0x80 (3 bytes msg)
avrdude: Sending sign-off command: 0x80 (3 bytes msg)

avrdude done.  Thank you.
---verbose---

```

mcuee, 2022-09-01:
> IIRC I haven't had this issue after I updated the firmware on the mEDBG chip. But if your firmware version is 1.13, you most likely have the latest one installed.

The firmware is already the latest. I checked under Microchip Studio and it did not prompt for Firmware Update.

```
PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude -p t817 -c xplainedmini_updi -v

avrdude.exe: Version 7.0-20220831 (dab99c0)
             Copyright (c) Brian Dean, http://www.bdmicro.com/
             Copyright (c) Joerg Wunsch

             System wide configuration file is "C:/work/avr/avrdude_test/avrdude_bin/avrdude.conf"

             Using Port                    : usb
             Using Programmer              : xplainedmini_updi
avrdude.exe: Found CMSIS-DAP compliant device, using EDBG protocol
             AVR Part                      : ATtiny817
             RESET disposition             : dedicated
             RETRY pulse                   : SCK
             Serial program mode           : yes
             Parallel program mode         : yes
             Memory Detail                 :
...

             Programmer Type : JTAGICE3_UPDI
             Description     : Atmel AVR XplainedMini in UPDI mode
             ICE HW version  : 0
             ICE FW version  : 1.13 (rel. 43)
             Serial number   : ATML2658061800009257
             Vtarget         : 5.00 V
avrdude.exe: Partial Family_ID returned: "tiny"
avrdude.exe: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.06s

avrdude.exe: Device signature = 0x1e9320 (probably t817)

avrdude.exe done.  Thank you.

```

mcuee, 2022-09-01:
It seems I have no issues with avrdude now.

The issue is present in Arduino with megatinycore (https://github.com/SpenceKonde/megaTinyCore).

```
:\Users\xiaof\AppData\Local\Arduino15\packages\DxCore\tools\avrdude\6.3.0-arduino17or18/bin/avrdude -CC:\Users\xiaof\AppData\Local\Arduino15\packages\megaTinyCore\hardware\megaavr\2.5.11/avrdude.conf -v -pattiny817 -cxplainedmini_updi -Pusb -e -Ufuse0:w:0x00:m -Ufuse1:w:0x00:m -Ufuse2:w:0x02:m -Ufuse4:w:0x00:m -Ufuse5:w:0b1100{bootloader.resetpinbits}0{bootloader.eesavebit}:m -Ufuse6:w:0x04:m -Ufuse7:w:0x00:m -Ufuse8:w:0x00:m -Uflash:w:C:\Users\xiaof\AppData\Local\Arduino15\packages\megaTinyCore\hardware\megaavr\2.5.11/bootloaders/hex/optiboot_XplainedMini817.hex:i 

avrdude: Version 6.3-20201216
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "C:\Users\xiaof\AppData\Local\Arduino15\packages\megaTinyCore\hardware\megaavr\2.5.11/avrdude.conf"

         Using Port                    : usb
         Using Programmer              : xplainedmini_updi
avrdude: Found CMSIS-DAP compliant device, using EDBG protocol
avrdude: jtag3_edbg_recv(): Inconsistent fragment number; expect 1, got 0

avrdude done.  Thank you.

Error while burning bootloader.

```

mcuee, 2022-09-01:
After running the above Arduino version of avrdude, the FW may go into a bad state. In that case, both avrdude 7.0 release and git main will have issues. Unplug and plug will fix the issue.

```
PS C:\work\avr\avrdude_test\avrdude_binary_archive\avrdude-v7.0-windows-x64> .\avrdude -p t817 -c xplainedmini_updi -vv

avrdude.exe: Version 7.0
             Copyright (c) Brian Dean, http://www.bdmicro.com/
             Copyright (c) Joerg Wunsch

             System wide configuration file is "C:/work/avr/avrdude_test/avrdude_binary_archive/avrdude-v7.0-windows-x64/avrdude.conf"

             Using Port                    : usb
             Using Programmer              : xplainedmini_updi
avrdude.exe: jtag3_open_updi()
avrdude.exe: Found CMSIS-DAP compliant device, using EDBG protocol
avrdude.exe: Sending sign-on command: avrdude.exe: jtag3_edbg_recv(): No response available

avrdude.exe: sign-on command: timeout/error communicating with programmer (status -1)
avrdude.exe: opening programmer "xplainedmini_updi" on port "usb" failed

avrdude.exe done.  Thank you.

PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude -p t817 -c xplainedmini_updi -vv

avrdude.exe: Version 7.0-20220831 (dab99c0)
             Copyright (c) Brian Dean, http://www.bdmicro.com/
             Copyright (c) Joerg Wunsch

             System wide configuration file is "C:/work/avr/avrdude_test/avrdude_bin/avrdude.conf"

             Using Port                    : usb
             Using Programmer              : xplainedmini_updi
avrdude.exe: jtag3_open_updi()
avrdude.exe: HV UPDI support: 1
avrdude.exe: Found CMSIS-DAP compliant device, using EDBG protocol
avrdude.exe: Sending sign-on command: avrdude.exe: jtag3_edbg_recv(): No response available

avrdude.exe: sign-on command: timeout/error communicating with programmer (status -1)
avrdude.exe: opening programmer "xplainedmini_updi" on port "usb" failed

avrdude.exe done.  Thank you.

```

mcuee, 2022-09-01:
Same thing, FW will go bad if I run the old version of avrdude from @mariusgreuel .

```
PS C:\work\avr\avrdude_test\avrdude_binary_archive\avrdude-v6.3.1.1-windows> .\avrdude -p t817 -c xplainedmini_updi

avrdude.exe: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.04s

avrdude.exe: Device signature = 0x1e9320 (probably t817)

avrdude.exe: safemode: Fuses OK (E:FF, H:FF, L:FF)
avrdude.exe: jtag3_edbg_signoff(): unexpected response 0x03, 0x80

avrdude.exe done.  Thank you.

PS C:\work\avr\avrdude_test\avrdude_binary_archive\avrdude-v7.0-windows-x64> .\avrdude -p t817 -c xplainedmini_updi -vv

avrdude.exe: Version 7.0
             Copyright (c) Brian Dean, http://www.bdmicro.com/
             Copyright (c) Joerg Wunsch

             System wide configuration file is "C:/work/avr/avrdude_test/avrdude_binary_archive/avrdude-v7.0-windows-x64/avrdude.conf"

             Using Port                    : usb
             Using Programmer              : xplainedmini_updi
avrdude.exe: jtag3_open_updi()
avrdude.exe: Found CMSIS-DAP compliant device, using EDBG protocol
avrdude.exe: Sending sign-on command: avrdude.exe: jtag3_edbg_recv(): No response available

avrdude.exe: sign-on command: timeout/error communicating with programmer (status -1)
avrdude.exe: opening programmer "xplainedmini_updi" on port "usb" failed

avrdude.exe done.  Thank you.
```

mcuee, 2022-09-01:
All in all, there is no issue with the 7.0 release and current git (at least with the latest FW). Rather the old 6.3 fork may have issues and then the FW will go into a bad state. Unplug and plug will fix the issue.

MCUdude, 2022-09-01:
There are a few commits to jtag3.c committed in late 2021 which may have resolved the issue.

https://github.com/avrdudes/avrdude/commits/5633a6d88ab199b68ba68d53717efc21c0cc5868/jtag3.c?browsing_rename_history=true&new_path=src/jtag3.c&original_branch=main

mcuee, 2022-12-21:
Strange I can not write fuse for the xplainedmini programmer (ISP mode) for the ATmega328PB under Linux.

```
mcuee@UbuntuSwift3:~/build/avr/avrdude_bin$ ./avrdude -c xplainedmini -p m328pb -U lfuse:w:0xE2:m  

avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 0x1e9516 (probably m328pb)
avrdude: reading input file 0xE2 for lfuse
         with 1 byte in 1 section within [0, 0]
avrdude: writing 1 byte lfuse ...
avrdude: 1 byte of lfuse written
avrdude: verifying lfuse memory against 0xE2
avrdude warning: verification mismatch
        device 0xe0 != input 0xe2 at addr 0x0000 (error)
avrdude error: verification mismatch

avrdude done.  Thank you.

mcuee@UbuntuSwift3:~/build/avr/avrdude_bin$ ./avrdude7 -C ./avrdude7.conf  -c xplainedmini -p m328pb -U lfuse:w:0xE2:m  

avrdude7: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.05s

avrdude7: Device signature = 0x1e9516 (probably m328pb)
avrdude7: reading input file "0xE2"
avrdude7: writing lfuse (1 bytes):

Writing | ################################################## | 100% 0.03s

avrdude7: 1 bytes of lfuse written
avrdude7: verifying lfuse memory against 0xE2:

Reading | ################################################## | 100% 0.01s

avrdude7: verification error, first mismatch at byte 0x0000
          0xe0 != 0xe2
avrdude7: verification error; content mismatch

avrdude7 done.  Thank you.

```

MCUdude, 2022-12-21:
> Strange I can not write lfuse for the xplainedmini programmer (ISP mode) for the ATmega328PB under Linux and Windows. Maybe this is specific to the ATmega328PB Xplained Mini board (clock comes from the mEBDG)

@mcuee I'll guess this is caused by the [SUFFER register](http://ww1.microchip.com/downloads/en/devicedoc/50002660a.pdf). For the microUPDI programmer I sell (mEDBG based UPDI programmer), I've actually permanently disabled fuse protection with a custom eep file that's written to the ATmega32U4 when flashed using [microUPDIcore](https://github.com/MCUdude/microUPDIcore).

[image]

http://ww1.microchip.com/downloads/en/devicedoc/50002660a.pdf