avrdudes/avrdude #749: [patch #9867] AVRDUDE.CONF

avrdude-admin, 2021-12-13:
Wed 30 Oct 2019 12:12:43 PM UTC

In the latest AVRDUDE 6.3.0 for Arduino mingw32 the AVRCONF file contains some errors and some items are missing.

It concerns the latest AVR8X series with UPDI interface.
In the common section for the ATTiny and the common section for the ATMega processors the "USERSIG" is always the same size.

This is not true. The UserSig memory is always one extra page of EEprom memory. The correct values for both the USersig Size, PageSize and Readsize are:

For all AtTiny and AtMega 2xx 4xx 8xx, 16xx: 0x20 (32 Bytes)
For all AtTiny and AtMega 32xx and 48xx: 0x40 (64 Bytes)

In the common fields for all types the id bytes are part of the "Sigrow" memory, read-Only 0x100 Bytes at 0x1100, but only 27 bytes are used. The first 3 bytes are defined as "signature", with the ID number. This is correctly mentioned. The remainder is mentioned as "prodsig", also correct, but not verry usefull.

A Meaningfull addition can be: (all fields here ar read-only)
"sernum" 10 bytes 0x1104
"sernum0" 1 Byte  0x1104
"sernum1" 1 byte  0x1105
"sernum2" 1 Byte  0x1106
"sernum3" 1 byte  0x1107
"sernum4" 1 Byte  0x1108
"sernum5" 1 byte  0x1109
"sernum6" 1 Byte  0x110A
"sernum7" 1 byte  0x110B
"sernum8" 1 Byte  0x110C
"sernum9" 1 Byte  0x110D

"osccal"     4 bytes 0x1118
"osccal16"   2 Bytes 0x1118
"ocscal16m0" 1 Byte  0x1118
"osccal16m1" 1 Byte  0x1119
"osccal20"   2 Bytes 0x111A
"osccal20m0" 1 Byte  0x111A
"osccal20m1" 1 Byte  0x111B

"tempsense"  2 Bytes 0x1120
"tempsense0" 1 Byte  0x1120
"tempsense1" 1 Byte  0x1121

"oscerr"     4 Bytes 0x1122
"Osc16err"   2 Bytes 0x1122
"osc16err3v" 1 Byte  0x1122
"osc16ett5v" 1 Byte  0x1123
"osc20err"   2 Bytes 0x1124
"osc20err3v" 1 Byte  0x1124
"osc20err55" 1 byte  0x1125

The last improvement is also to mention the fuses by name.
The fuses are now named as memory "fuses" and the individual fuses are mentioned as "f0" .. f8"
A good addition would be to have duplicate entry's for the fuses
with the official names:
"f0" duplicate "WDTCFG"
"f1" duplicate "BODCFG"
"f2" duplicate "OSCCFG"
"f3" duplicate "FUSE3"
"f4" duplicate "TCD0CFG"
"f5" duplicate "SYSCFG0"
"f6" duplicate "SYSCFG1"
"f7" duplicate "APPEND"
"f8" duplicate "BOOTEND"

hope this contribution is usefull.
Hannes Bulk, Utrecht, Netherlands

This issue was migrated from https://savannah.nongnu.org/patch/?9867

MCUdude, 2022-01-05:
@dl8dtl I'm currently working on a PR for this. However, I stumbled across something weird...

Here is a snippet of the avr8x memory addresses. I've added `sernum` and `sernum0`. The weird thing is that if I remove `sernum0` from avrdude.conf, I can read `sernum` just fine using `-U sernum:r:-:h`. However, If I add `sernum0` and run try to read `sernum` again, I get the following error: `"sernum" memory type not defined for part "ATmega4808"`. Is this a bug in the parser that interprets avrdude.conf?
```
#------------------------------------------------------------
# AVR8X family common values
#------------------------------------------------------------

part
    id		= ".avr8x";
    desc	= "AVR8X family common values";
    has_updi	= yes;
    nvm_base	= 0x1000;
    ocd_base	= 0x0F80;

    memory "signature"
        size		= 3;
        offset		= 0x1100;
        readsize	= 0x3;
    ;

    memory "prodsig"
        size		= 0x3D;
        offset		= 0x1103;
        page_size	= 0x3D;
        readsize	= 0x3D;
    ;

    memory "sernum"
        size		= 10;
        offset		= 0x1104;
        readsize	= 0xA;
    ;

    memory "sernum0"
        size		= 1;
        offset		= 0x1104;
        readsize	= 1;
    ;

...

```

dl8dtl, 2022-01-05:
I guess it's an artefact of the way memory region abbreviations are being handled. For convenience, you could (ever since `-U` was introduced) abbreviate memory regions, so `flash` to `fl`, or `eeprom` to `ee`. I think the case where one memory region name is a substring of another one has never been anticipated.

Have a look at `avr_locate_mem()` in `avrpart.c`.

Still, I wonder whether `sernum` and `sernum0` are actually useful names though.

MCUdude, 2022-01-06:
> Still, I wonder whether sernum and sernum0 are actually useful names though.

What would be useful names then? And should all names be lower case, even fuse names like syscfg0?

dl8dtl, 2022-01-06:
> > Still, I wonder whether sernum and sernum0 are actually useful names though.
> 
> What would be useful names then? And should all names be lower case, even fuse names like syscfg0?

For sernum, I don't think accessing individual fields is much meaningful (so we'd just need `sernum` only). Not sure about whether `osccal` (as a 4-byte entry) is meaningful, otherwise we could just use `osccal16` and `osccal20` (again, without separate single-byte access, it's a 16-bit value anyway).
Of course, it's also not out of the question to change `avr_locate_mem()` to allow for a full exact match, rather than demanding a unique name as it does now.
I'd keep these names all lowercase. It's easier to type for users anyway.

MCUdude, 2022-01-08:
@dl8dtl where in the source code is the list of all memories we can read/write to located?

When adding more memories to the Avrdude.conf file, the table that gets printed in the terminal window with the -v flag gets annoying long. As a start, duplicates can be ignored:

```
$ ./avrdude -C avrdude.conf -patmega4808 -c serialupdi -v -P /dev/cu.usbserial-1410 -Uosccal20:r:-:h

avrdude: Version 6.99-20211218
         Copyright (c) Brian Dean, http://www.bdmicro.com/
         Copyright (c) Joerg Wunsch

         System wide configuration file is "avrdude.conf"
         User configuration file is "/Users/hans/.avrduderc"
         User configuration file does not exist or is not a regular file, skipping

         Using Port                    : /dev/cu.usbserial-1410
         Using Programmer              : serialupdi
         AVR Part                      : ATmega4808
         Chip Erase delay              : 0 us
         PAGEL                         : P00
         BS2                           : P00
         RESET disposition             : dedicated
         RETRY pulse                   : SCK
         serial program mode           : yes
         parallel program mode         : yes
         Timeout                       : 0
         StabDelay                     : 0
         CmdexeDelay                   : 0
         SyncLoops                     : 0
         ByteDelay                     : 0
         PollIndex                     : 0
         PollValue                     : 0x00
         Memory Detail                 :

                                  Block Poll               Page                       Polled
           Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           signature      0     0     0    0 no          3    1      0     0     0 0x00 0x00
           prodsig        0     0     0    0 no         61   61      0     0     0 0x00 0x00
           sernum         0     0     0    0 no         10    1      0     0     0 0x00 0x00
           osccal16       0     0     0    0 no          2    1      0     0     0 0x00 0x00
           osccal20       0     0     0    0 no          2    1      0     0     0 0x00 0x00
           tempsense      0     0     0    0 no          2    1      0     0     0 0x00 0x00
           osc16err       0     0     0    0 no          2    1      0     0     0 0x00 0x00
           osc20err       0     0     0    0 no          2    1      0     0     0 0x00 0x00
           fuses          0     0     0    0 no          9   10      0     0     0 0x00 0x00
           fuse0          0     0     0    0 no          1    1      0     0     0 0x00 0x00
           wdtcfg         0     0     0    0 no          1    1      0     0     0 0x00 0x00
           fuse1          0     0     0    0 no          1    1      0     0     0 0x00 0x00
           bodcfg         0     0     0    0 no          1    1      0     0     0 0x00 0x00
           fuse2          0     0     0    0 no          1    1      0     0     0 0x00 0x00
           osccfg         0     0     0    0 no          1    1      0     0     0 0x00 0x00
           fuse4          0     0     0    0 no          1    1      0     0     0 0x00 0x00
           tcd0cfg        0     0     0    0 no          1    1      0     0     0 0x00 0x00
           fuse5          0     0     0    0 no          1    1      0     0     0 0x00 0x00
           syscfg0        0     0     0    0 no          1    1      0     0     0 0x00 0x00
           fuse6          0     0     0    0 no          1    1      0     0     0 0x00 0x00
           syscfg1        0     0     0    0 no          1    1      0     0     0 0x00 0x00
           fuse7          0     0     0    0 no          1    1      0     0     0 0x00 0x00
           append         0     0     0    0 no          1    1      0     0     0 0x00 0x00
           fuse8          0     0     0    0 no          1    1      0     0     0 0x00 0x00
           bootend        0     0     0    0 no          1    1      0     0     0 0x00 0x00
           lock           0     0     0    0 no          1    1      0     0     0 0x00 0x00
           data           0     0     0    0 no          0    1      0     0     0 0x00 0x00
           usersig        0     0     0    0 no         64   64      0     0     0 0x00 0x00
           flash          0     0     0    0 no      49152  128      0     0     0 0x00 0x00
           eeprom         0     0     0    0 no        256   64      0     0     0 0x00 0x00

         Programmer Type : serialupdi
         Description     : SerialUPDI

avrdude: UPDI link initialization OK
avrdude: NVM type 0: 16-bit, page oriented write
avrdude: Entering NVM programming mode
avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.02s

avrdude: Device signature = 0x1e9650 (probably m4808)
avrdude: reading osccal20 memory:

Reading | ################################################## | 100% 0.01s

avrdude: writing output file "<stdout>"
0x3e,0x3

avrdude: safemode: Fuses OK (E:FF, H:FF, L:FF)
avrdude: Leaving NVM programming mode

avrdude done.  Thank you.
```

dl8dtl, 2022-01-08:
There's no hardcoded list, everything is built up from the descriptions in `avrdude.conf`.
Yes, this list is pretty long for some AVRs.
Unlike `sernum`*<N>* though, the individual `fuse`*<N>* entries probably do make sense, as one might want to modify only those fuses they are actually interested in.
We could try to impose some ordering to the printout.

MCUdude, 2022-01-08:
I found a solution, but I had to modify the print function in avrpart.c

Now it's never printing duplicate memories. Is this a modification you would accept in a PR along with the added memory section and names in avrdude.conf.in?

```
                                  Block Poll               Page                       Polled
           Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           signature      0     0     0    0 no          3    1      0     0     0 0x00 0x00
           prodsig        0     0     0    0 no         61   61      0     0     0 0x00 0x00
           sernum         0     0     0    0 no         10    1      0     0     0 0x00 0x00
           osccal16       0     0     0    0 no          2    1      0     0     0 0x00 0x00
           osccal20       0     0     0    0 no          2    1      0     0     0 0x00 0x00
           tempsense      0     0     0    0 no          2    1      0     0     0 0x00 0x00
           osc16err       0     0     0    0 no          2    1      0     0     0 0x00 0x00
           osc20err       0     0     0    0 no          2    1      0     0     0 0x00 0x00
           fuses          0     0     0    0 no          9   10      0     0     0 0x00 0x00
           fuse0          0     0     0    0 no          1    1      0     0     0 0x00 0x00
           fuse1          0     0     0    0 no          1    1      0     0     0 0x00 0x00
           fuse2          0     0     0    0 no          1    1      0     0     0 0x00 0x00
           fuse4          0     0     0    0 no          1    1      0     0     0 0x00 0x00
           fuse5          0     0     0    0 no          1    1      0     0     0 0x00 0x00
           fuse6          0     0     0    0 no          1    1      0     0     0 0x00 0x00
           fuse7          0     0     0    0 no          1    1      0     0     0 0x00 0x00
           fuse8          0     0     0    0 no          1    1      0     0     0 0x00 0x00
           lock           0     0     0    0 no          1    1      0     0     0 0x00 0x00
           data           0     0     0    0 no          0    1      0     0     0 0x00 0x00
           usersig        0     0     0    0 no         64   64      0     0     0 0x00 0x00
           flash          0     0     0    0 no      49152  128      0     0     0 0x00 0x00
           eeprom         0     0     0    0 no        256   64      0     0     0 0x00 0x00
```

```c
void avr_mem_display(const char * prefix, FILE * f, AVRMEM * m, int type,
                     int verbose)
{
  static unsigned int prev_mem_offset;
  static unsigned char prev_mem_size;
  int i, j;
  char * optr;

  if (m == NULL) {
      fprintf(f,
              "%s                       Block Poll               Page                       Polled\n"
              "%sMemory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack\n"
              "%s----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------\n",
            prefix, prefix, prefix);
  }
  else {
    if (verbose > 2) {
      fprintf(f,
              "%s                       Block Poll               Page                       Polled\n"
              "%sMemory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack\n"
              "%s----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------\n",
              prefix, prefix, prefix);
    }
    if (prev_mem_offset != m->offset || prev_mem_size != m->size) { // Don't print memory aliases
      fprintf(f,
              "%s%-11s %4d %5d %5d %4d %-6s %6d %4d %6d %5d %5d 0x%02x 0x%02x\n",
              prefix, m->desc, m->mode, m->delay, m->blocksize, m->pollindex,
              m->paged ? "yes" : "no",
              m->size,
              m->page_size,
              m->num_pages,
              m->min_write_delay,
              m->max_write_delay,
              m->readback[0],
              m->readback[1]);
      prev_mem_offset = m->offset;
      prev_mem_size = m->size;
    }
    if (verbose > 4) {
      avrdude_message(MSG_TRACE2, "%s  Memory Ops:\n"
                      "%s    Oeration     Inst Bit  Bit Type  Bitno  Value\n"
                      "%s    -----------  --------  --------  -----  -----\n",
                      prefix, prefix, prefix);
      for (i=0; i<AVR_OP_MAX; i++) {
        if (m->op[i]) {
          for (j=31; j>=0; j--) {
            if (j==31)
              optr = avr_op_str(i);
            else
              optr = " ";
          fprintf(f,
                  "%s    %-11s  %8d  %8s  %5d  %5d\n",
                  prefix, optr, j,
                  bittype(m->op[i]->bit[j].type),
                  m->op[i]->bit[j].bitno,
                  m->op[i]->bit[j].value);
          }
        }
      }
    }
  }
}
```

dl8dtl, 2022-01-08:
That looks great to me!

MCUdude, 2022-01-08:
I just found out that it's not 100% ideal for all parts. For "classic" AVRs (ATmega1284 in this case), the list looks now like this:

```
                                  Block Poll               Page                       Polled
           Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           eeprom        65    10   128    0 no       4096    8      0  9000  9000 0xff 0xff
           flash         65    10   256    0 yes    131072  256    512  4500  4500 0xff 0xff
           lock           0     0     0    0 no          1    1      0  9000  9000 0x00 0x00
           signature      0     0     0    0 no          3    1      0     0     0 0x00 0x00
           calibration    0     0     0    0 no          1    1      0     0     0 0x00 0x00
```

refuse, hfuse and lfuse are missing from the list. This is because they don't have a dedicated address we can check on. we probably have to pass `id` and check if the part belongs to `.avr8x`.

MCUdude, 2022-01-09:
@dl8dtl I'm also looking at the table that's printed before all the available memory sections, and there is a lot of irrelevant information for certain parts there.

I've been able to filter out most of them, but a few are still left. For instance _all_ parts I've tried prints `Parallel program mode         : yes`, even though parallell programming isn't supported. How can this be filtered out?

What is it that sets all the flags?

from main.c:

```c
p->flags & AVRPART_PARALLELOK
```

PR #815 (2022-01-21): Add more memory sections to read from
I thought I'll submit this so you can review this @dl8dtl.

This PR adds more memory sections to read on "modern" megaAVRs  and  tinyAVRs, and it also improves the Avrdude output in verbose mode so that duplicate memory entries don't get printed twice.

I ended up using `p->family_id`, since this field is only present on "modern" megaAVRs  and  tinyAVRs.

Tested on "classic" AVRs, "modern" ones, and xmegas.

Closes #749
closed 2022-01-21 by dl8dtl