avrdudes/avrdude #773: AVRDUDE build instructions not up-to-date

kristofmulier, 2021-12-22:
I would like to add a feature to AVRDUDE (see https://github.com/avrdudes/avrdude/issues/770), but I get stuck at the first step - cloning the git repo and building the source code. It looks like the build instructions are no longer up-to-date.

<h2>1. Linux</h2>

First I install all the required packages:
```bash
$ sudo apt-get install build-essential git automake libtool flex bison libelf-dev libusb-dev libftdi1-dev libhidapi-dev
```
and
```bash
$ sudo apt-get install texlive texi2html
```

Next, I clone the repo:
```bash
$ cd ~
$ git clone https://github.com/avrdudes/avrdude
$ cd avrdude
```

Now I want to issue the following commands:
```bash
./bootstrap
./configure
make
```
But that fails. There is no `boostrap` or `configure` file in `~/avrdude`, nor in `~/avrdude/src`. I'm stuck.

&nbsp;<br>
<h2>2. Windows</h2>

The build instructions for Windows are 'under construction'. That's a pity. I'd like to build it on Windows too :-)

*PS: @dl8dtl , I am already consuming a lot of your time, and feel a bit shy about it. Please let me know how to donate to your project ;-)*

dl8dtl, 2021-12-22:
Sorry, we just migrated most of the files into a `src/` subdirectory, but the build instructions have not changed for that so far.
I'll make that change ASAP.
Regarding Windows, that's Marius' current project, he wants to revamp everything in that respect quite a bit. He started to add CMake files for that purpose, but that's still a lot of work in progress.
So far, using "the old way", you could only use something like MinGW, and basically follow the same steps as under unix-like systems.
Btw., texlive & Co you don't strictly need. It would only be needed to build the documentation, but you can probably get away without doing that yourself. (Until release 6.4, I have actually been cross-compiling MinGW-based binaries for Windows under my FreeBSD development machine.)

mariusgreuel, 2021-12-22:
> But that fails. There is no boostrap

bootstrap is clearly there. configure is auto-generated, its expected that it is not in there yet.

> The build instructions for Windows are 'under construction'

The project just moved to GitHub, I will update the documentation in the coming weeks.

dl8dtl, 2021-12-22:
Well, README.md says:
```
### Building

All source code is located in the `src/` subdirectory. Thus all
instructions are relative to that directory.
```

kristofmulier, 2021-12-22:
> *bootstrap is clearly there. configure is auto-generated, its expected that it is not in there yet.*

My apologies. The `boostrap` file was indeed in `~/avrdude/src/bootstrap`. After running it, I do now have `~/avrdude/src/configure`.

> *Regarding Windows, that's Marius' current project, he wants to revamp everything in that respect quite a bit. He started to add CMake files for that purpose, but that's still a lot of work in [...]

Thanks, I will try :-)

dl8dtl, 2021-12-22:
So, I think this has been clarified, closing it then.

closed 2021-12-22 by dl8dtl