avrdudes/avrdude #781: AVRDUDE license

yegorich, 2021-12-24:
The `README.md` states that the project is licensed under GPL-2.0 License but when you look at the boilerplate of the source files, this says GPL-2.0 or  (at your option) any later version. This actually means [GPL-2.0-or-later](https://spdx.org/licenses/GPL-2.0-or-later.html).

As we now "split" the code into library and frontend, this almost implies to license the library as LGPL and the frontend as GPL-2.0-or-later. 

Also, what about sticking to the [REUSE](https://reuse.software/spec/) infrastructure? This means putting license files into the `LICENSES` folder and replacing license boilerplate in the source files with the SPDX identifier. See [libgpiod](https://git.kernel.org/pub/scm/libs/libgpiod/libgpiod.git/tree/) for the reference.

dl8dtl, 2021-12-25:
The `README.md` file simply links the existing license document, rather than providing a kind of statement (like `GPL-2.0-or-later`). The GPL 2 has always had that clause that includes "or any later version" in the wording, so I don't think there is a difference at all.
The library split has been done many years ago. Relicensing the existing backend files under a different (less restrictive) license would require the consent of all relevant authors and contributors – I don't think this is really worth the effort, in particular since by now, I'm not aware of anyone else using the backend library but AVRDUDE's `main.c` itself.
I'm shy to perform any cosmetic changes (like moving out the license statement into a separate directory) just for the sake of performing that cosmetic change: it spams any later diff crossing that border with a large number of irrelevant change lines, thus potentially hiding the actual code changes. (The same holds for gratuitous whitespace changes.)

yegorich, 2021-12-27:
In both cases the license text for GPL 2.0 is the same but the declaration differs. For the GPL 2.0 only it is:

> Copyright (C) yyyy name of author
> 
> This program is free software; you can redistribute it and/or modify it under the terms of the GNU General Public License as published by the Free Software Foundation; version 2.
> 
> This program is distributed in the hope that it will be useful, but WITHOUT ANY WARRANTY; without even the implied warranty of MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE. See the GNU General Public License for more details.
> 
> You should have received a copy of the GNU General Public License along with this program; if not, write to the Free Software Foundation, Inc., 51 Franklin Street, Fifth Floor, Boston, MA 02110-1301, USA. 

For GPL 2.0 or later:

> <one line to give the program's name and an idea of what it does.>
> Copyright (C) <yyyy> <name of author>
> 
> This program is free software; you can redistribute it and/or modify it under the terms of the GNU General Public License as published by the Free Software Foundation; either version 2 of the License, or (at your option) any later version.
> 
> This program is distributed in the hope that it will be useful, but WITHOUT ANY WARRANTY; without even the implied warranty of MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE. See the GNU General Public License for more details.
> 
> You should have received a copy of the GNU General Public License along with this program; if not, write to the Free Software Foundation, Inc., 51 Franklin Street, Fifth Floor, Boston, MA 02110-1301, USA.

Linux distros like [Buildroot](https://buildroot.org/) track this difference when generating a license information for the used packages. Hence, I would just change the text in `README.md` as just to be more precise about the license:

> AVRDUDE is licensed under the [GPL-2.0-or-later](https://spdx.org/licenses/GPL-2.0-or-later.html) license.

As AVRDUDE uses only one license, the other changes are not that important.

dl8dtl, 2021-12-30:
I just removed that section from `README.md`. The `LICENSE` file ought to be enough.
Impending changes (like the addition of the `whereami` code) will make the license not so uniform anymore anyway.

yegorich, 2021-12-30:
@dl8dtl adding `whereami` would make `LICENSES` folder, that REUSE suggests, really useful as it will hold two license files.

mariusgreuel, 2022-01-04:
@dl8dtl: I think we should revert commit [Hint about possibly differing licensing terms](https://github.com/avrdudes/avrdude/commit/fa079bec8c233bbec793b31d883eba698fff2cb3).

As the COPYING file is now a modified GPL-2.0, GitHub can no longer automatically infer that the project license is indeed GPL-2.0, i.e. on the front page is states now "View license" instead of "GPL-2.0 License"

IMHO, this discussion is superfluous. We are mixing up _project license_ vs. _contribution license_. Both do not need to be the same. The project is licensed via GPL-2.0, the individual contributions may be licensed via a compatible license, such as GPL-2.0 or GPL-2.0-or-later. I see no reason to mention this, and since the project is very consistent, we do not need a LICENSE folder or anything like this.

If anyone needs to know the license(s) of the individual contributions, that person can easily do a license scan over the whole project.

dl8dtl, 2022-01-04:
Fine with me.

dl8dtl, 2022-01-04:
I reverted that commit, agreeing with your reasoning.
I wasn't aware of that kind of automatisms.

mariusgreuel, 2022-01-11:
@yegorich : Can we close this one?

yegorich, 2022-01-11:
Yes.

closed 2022-01-11 by yegorich