avrdudes/avrdude #785: CMake: Recognize more than just bison

dl8dtl, 2021-12-28:
Right now, the CMake project files require `bison` as a parser generator.
However, the grammar of `avrdude.conf` is generic enough to also work with the standard `yacc` or `byacc` (Berkeley yacc) generators.
In order for CMake to become a fully functional replacement of for autoconf/automake, this should be fixed.

mariusgreuel, 2021-12-28:
Unfortunately, finding yacc/byacc is not built into cmake. While we could easily add it, it is probably a page of code
https://gitlab.kitware.com/cmake/cmake/-/blob/master/Modules/FindBISON.cmake
which makes me think: Rather then cluttering the project, is it too much to ask to install bison?

Why do you think we should support yacc? Is there a specific OS that you have in mind that does not have a bison package?

dl8dtl, 2021-12-28:
> Why do you think we should support yacc?

Because non-Linux Unix systems all ship with `lex` and `yacc` by default, so no additional package is needed at all.
That is, Solaris/Illumos (they don't even ship `flex` but their native `lex`), and all BSDs.

Small is beautiful ;-), and less prerequisites is always better.

(Maybe it would also worth to ask CMake to officially recognize these tools.)

mariusgreuel, 2022-01-03:
While I think keeping the cmake project clean outweights the requirement for bison, I added some code that uses yacc/byacc as a fallback. Tested in Linux and FreeBSD.

commit aa2f132b (2022-01-03): Merge pull request #793 from mariusgreuel/pr-yacc

Use yacc/byacc as an alternative to bison, closes #785
closed 2022-01-03 by dl8dtl