avrdudes/avrdude #790: Curiosity Nano and terminal mode

MCUdude, 2021-12-30:
I was just experimenting with my ATmega4809 Curiosity Nano in terminal mode when I discovered a few issues.
I'm building my Avrdude binary from the upstream source code, and I'm using MacOS. I start the session with the following command:

```
./avrdude -C avrdude.conf -v -patmega4809 -cpkobn_updi -t
```

### eeprom

First, I'm having issues writing to EEPROM. It seems like I can't write, but reading works.

```
avrdude> write eeprom 0 1 2 3 4
>>> write eeprom 0 1 2 3 4 
avrdude (write): error writing 0x01 at 0x00000 cell=0x00
avrdude (write): error writing 0x02 at 0x00001 cell=0x00
avrdude (write): error writing 0x03 at 0x00002 cell=0x00
avrdude (write): error writing 0x04 at 0x00003 cell=0x00

avrdude> dump eeprom 0 4
>>> dump eeprom 0 4 
0000  00 00 00 00                                       |....            |

avrdude> 
```

(It turns out that I can't write hex files to the eeprom either.)

<details>

```
$ cat ee.hex 
:20000000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF00
:20002000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFE0
:20004000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFC0
:20006000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFA0
:20008000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF80
:2000A000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF60
:2000C000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF40
:2000E000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF20
:00000001FF

$ ./avrdude -C avrdude.conf -v -patmega4809 -cpkobn_updi -Ueeprom:w:ee.hex:i

...

Writing | ################################################## | 100% 0.10s

avrdude: 256 bytes of eeprom written
avrdude: verifying eeprom memory against ee.hex:
avrdude: load data eeprom data from input file ee.hex:
avrdude: input file ee.hex contains 256 bytes
avrdude: reading on-chip eeprom data:

Reading | ################################################## | 100% 0.04s

avrdude: verifying ...
avrdude: verification error, first mismatch at byte 0x0000
         0x00 != 0xff
avrdude: verification error; content mismatch

avrdude: safemode: Fuses OK (E:FF, H:FF, L:FF)

avrdude done.  Thank you.
```

</details>

## pgm

When sending a PGM command, I get a segmentation fault and Avrdude crashes:

```
avrdude> pgm
>>> pgm 
Segmentation fault: 11
```

## spi

When sending an SPI command, I get a segmentation fault and Avrdude crashes:

```
avrdude> spi
>>> spi 
Segmentation fault: 11
```

### vtarg

Even though the Curiosiy Nano supports voltage adjustments, I'm not able to set the operating voltage:
```
avrdude> vtarg 3.3
>>> vtarg 3.3 
avrdude (vtarg): the JTAGICE3_UPDI programmer cannot set V[target]
```

dl8dtl, 2021-12-30:
Strange. I've got a Curiosity Nano board with an AVR128DA48, and much to my surprise, even single-bytre writes in terminal mode do work there (for paged memory devices, they are usually the most critical part). File updates (`-U ee:w:…`) do work as well.
No idea about voltage adjustmentst – do you have any documentation for them? (As I don't see any adjustable voltage regulator on the board, I have some doubts it is possible.)

MCUdude, 2021-12-31:
> Strange. I've got a Curiosity Nano board with an AVR128DA48, and much to my surprise, even single-bytre writes in terminal mode do work there (for paged memory devices, they are usually the most critical part). File updates (-U ee:w:…) do work as well.

My curiosity nano is quite old. I bought it in 2018 I think. It may be firmware related perhaps? I'll see if I can have the firmware upgraded.

> No idea about voltage adjustmentst – do you have any documentation for them? (As I don't see any adjustable voltage regulator on the board, I have some doubts it is possible.)

It does have a voltage regulator, have a look in the [schematics](https://ww1.microchip.com/downloads/en/DeviceDoc/ATmega4809_Curiosity_Nano_Schematics.pdf). It can be adjusted using Atmel Studio or pymcuprog.

[Pymcuprog](https://github.com/microchip-pic-avr-tools/pymcuprog) is open source, so the implementation could be borrowed from there i think.

dl8dtl, 2021-12-31:
Yes, I can see the voltage regulator.
Would be cool if Microchip documented their protocols, like Atmel did before.
I can confirm the crashes for `pgm` and `spi` commands; they should probably just be disabled for UPDI programmers.

MCUdude, 2021-12-31:
So I have another, newer Curiosity Nano board, one with an ATtiny3217 attached.

This is able to write to the EEPROM:

```
$ ./avrdude -C avrdude.conf -v -pattiny3217 -cpkobn_updi -t

avrdude: Version 6.99-20211218
         Copyright (c) Brian Dean, http://www.bdmicro.com/
         Copyright (c) Joerg Wunsch

         System wide configuration file is "avrdude.conf"
         User configuration file is "/Users/hans/.avrduderc"
         User configuration file does not exist or is not a regular file, skipping

         Using Port                    : usb
         Using Programmer              : pkobn_updi
avrdude: Found CMSIS-DAP compliant device, using EDBG protocol
         AVR Part                      : ATtiny3217
         Chip Erase delay              : 0 us
         PAGEL                         : P00
         BS2                           : P00
         RESET disposition             : dedicated
         RETRY pulse                   : SCK
         serial program mode           : yes
         parallel program mode         : yes
         Timeout                       : 0
         StabDelay                     : 0
         CmdexeDelay                   : 0
         SyncLoops                     : 0
         ByteDelay                     : 0
         PollIndex                     : 0
         PollValue                     : 0x00
         Memory Detail                 :

                                  Block Poll               Page                       Polled
           Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           signature      0     0     0    0 no          3    1      0     0     0 0x00 0x00
           prodsig        0     0     0    0 no         61   61      0     0     0 0x00 0x00
           fuses          0     0     0    0 no          9   10      0     0     0 0x00 0x00
           fuse0          0     0     0    0 no          1    1      0     0     0 0x00 0x00
           fuse1          0     0     0    0 no          1    1      0     0     0 0x00 0x00
           fuse2          0     0     0    0 no          1    1      0     0     0 0x00 0x00
           fuse4          0     0     0    0 no          1    1      0     0     0 0x00 0x00
           fuse5          0     0     0    0 no          1    1      0     0     0 0x00 0x00
           fuse6          0     0     0    0 no          1    1      0     0     0 0x00 0x00
           fuse7          0     0     0    0 no          1    1      0     0     0 0x00 0x00
           fuse8          0     0     0    0 no          1    1      0     0     0 0x00 0x00
           lock           0     0     0    0 no          1    1      0     0     0 0x00 0x00
           data           0     0     0    0 no          0    1      0     0     0 0x00 0x00
           usersig        0     0     0    0 no         32   32      0     0     0 0x00 0x00
           flash          0     0     0    0 no      32768  128      0     0     0 0x00 0x00
           eeprom         0     0     0    0 no        256   64      0     0     0 0x00 0x00

         Programmer Type : JTAGICE3_UPDI
         Description     : Curiosity nano (nEDBG) in UPDI mode
         ICE hardware version: 0
         ICE firmware version: 1.21 (rel. 37)
         Serial number   : MCHP3333021800001022
         Vtarget         : 3.31 V
         JTAG clock megaAVR/program: 0 kHz
         JTAG clock megaAVR/debug:   0 kHz
         JTAG clock Xmega: 0 kHz
         PDI clock Xmega : 100 kHz

avrdude: Partial Family_ID returned: "tiny"
avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.08s

avrdude: Device signature = 0x1e9522 (probably t3217)
avrdude> write eeprom 0 1 2 3 4
>>> write eeprom 0 1 2 3 4 

avrdude> dump eeprom 0 4
>>> dump eeprom 0 4 
0000  01 02 03 04                                       |....            |
```

For reference, here is the information about the nEDBG chip found on my 4809 Curiosity Nano:

```
         Programmer Type : JTAGICE3_UPDI
         Description     : Curiosity nano (nEDBG) in UPDI mode
         ICE hardware version: 255
         ICE firmware version: 1.02 (rel. 262)
         Serial number   : ATML3094051800001520
         Vtarget         : 3.30 V
         JTAG clock megaAVR/program: 0 kHz
         JTAG clock megaAVR/debug:   0 kHz
         JTAG clock Xmega: 0 kHz
         PDI clock Xmega : 750 kHz
```

dl8dtl, 2022-01-01:
OK, these `spi` and `pgm` commands are only meaningful on bitbang programmers.
I disabled these for all programmers that do not implement a `setpin` method.
Thus, the only remaining issue here is the lack of voltage regulator support, so I'm moving this from "bug" to "enhancement".

MCUdude, 2022-01-03:
I happened to be in contact with someone that knows these undocumented protocols much better than I do. Here's his response regarding the protocols not being documented and the target voltage setting:

> Unfortunately the docs are far from up to date. Setting the supply voltage is done (just like powerdebugger): https://github.com/microchip-pic-avr-tools/pymcuprog/blob/main/pymcuprog/utils.py#L117

PR #795 (2022-01-04): Add target voltage adjustment for Curiosity Nano boards
I figured I might as well go down the rabbit hole and get familiar with the Avrdude source code.
The Powerdebugger also supports setting target voltage, but since I don't own a Powerdebugger, I didn't add support for it. 

Tested on my ATtiny3217 Curiosity Nano.

Closes #790
closed 2022-01-04 by dl8dtl