avrdudes/avrdude #797: How to make a release from the AVRDUDE build artifacts?

kristofmulier, 2022-01-04:
I found the build artifacts that were produced after the merge of my pull request in GitHub Actions:

https://github.com/avrdudes/avrdude/actions/runs/1651208443

Thanks a lot!

We'll bring out a new release of Embeetle IDE soon in which we need the new system for finding `'avrdude.conf'`. So I'll use the build artifacts mentioned above. I noticed however that the builds contain a lot of source files and other files I've never seen before in an AVRDUDE distribution. Also, it looks like a few files are missing. I'm a bit confused - how precisely should I distribute this?

&nbsp;<br>
<h2>1. The case for Linux</h2>

All AVRDUDE distributions I've seen so far on Linux (which I used to extract from Arduino IDE) look like this:

```
# Arduino IDE build:
# ------------------
📁 avrdude_6.3.0
 ├─ 📁 bin
 │   └─ 🗋 avrdude
 └─ 📁 etc
     └─ 🗋 avrdude.conf
```

The Linux build I downloaded from the GitHub Actions looks like this:

```
# GitHub Actions build:
# ---------------------
📁 avrdude_6.99
 ├─ 📁 CMakeFiles
 ├─ 🗋 ac_cfg.h
 ├─ 🗋 avrdude
 ├─ 🗋 avrdude.conf
 ├─ 🗋 avrdude.conf.in
 ├─ 🗋 avrdude.spec
 ├─ 🗋 cmake_install.cmake
 ├─ 🗋 CMakeCache.txt
 ├─ 🗋 config_gram.c
 ├─ 🗋 config_gram.h
 ├─ 🗋 lexer.c
 ├─ 🗋 liblibavrdude.a
 └─ 🗋 Makefile
```

I suppose I can delete all files and only keep `🗋 avrdude` and `🗋 avrdude.conf`, right? What about `liblibavrdude.a`? It looks important.

&nbsp;<br>
<h2>2. The case for Windows</h2>

The Windows builds I used to extract from Arduino IDE look like this:

```
# Arduino IDE build:
# ------------------
📁 avrdude_6.3.0
 ├─ 📁 bin
 │   ├─ 🗋 avrdude.exe
 │   ├─ 🗋 giveio.sys
 │   ├─ 🗋 install_giveio.bat
 │   ├─ 🗋 libftdi1-config
 │   ├─ 🗋 libusb0.dll
 │   ├─ 🗋 loaddrv.exe
 │   ├─ 🗋 remove_giveio.bat
 │   └─ 🗋 status_giveio.bat
 └─ 📁 etc
     └─ 🗋 avrdude.conf
```

The Windows build I just downloaded from GitHub Actions is different:

```
# GitHub Actions build:
# ---------------------
📁 avrdude_6.99
 ├─ 📁 CMakeFiles
 ├─ 🗋 ac_cfg.h
 ├─ 🗋 avrdude.exe
 ├─ 🗋 avrdude.conf
 ├─ 🗋 avrdude.conf.in
 ├─ 🗋 avrdude.spec
 ├─ 🗋 cmake_install.cmake
 ├─ 🗋 CMakeCache.txt
 ├─ 🗋 config_gram.c
 ├─ 🗋 config_gram.h
 ├─ 🗋 lexer.c
 ├─ 🗋 libavrdude.dll.a
 ├─ 🗋 liblibavrdude.a
 ├─ 🗋 Makefile
 └─ 🗋 windows.rc
```

Some notable differences:

1. The Arduino IDE distribution has a few `giveio`-related files that are not present in the build from GitHub Actions: `giveio.sys`, `install_giveio.bat`, `remove_giveio.bat` and `status_giveio.bat`. What are these files for? Why are they missing in the GitHub Actions build?

2. The Arduino IDE distribution has also the following files that are not present in the GitHub Actions build: `libftdi1-config`, `libusb0.dll` and `loaddrv.exe`.

3. The GitHub Actions build has the following files that are not present in the Arduino IDE distribution: `libavrdude.dll.a`, `liblibavrdude.a`, ...

&nbsp;<br>
<h2>3. Conclusion</h2>

I'm confused. How precisely should I distribute this build to our users? My goal is to pack it in a zip-folder - one for Windows and one for Linux - that Embeetle IDE downloads to the user's computer. Embeetle will *not* install anything or add anything to the `$PATH`. It just drops the tool in a location (potentially chosen by the user) and unzips it there. Embeetle then invokes AVRDUDE by its absolute path.

All necessary dll's should be packed with AVRDUDE, such that it just works regardless of the user's local settings.

Thank you very much for your help @dl8dtl and @mariusgreuel . I appreciate a lot all of the time you have invested to help us ^_^

dl8dtl, 2022-01-04:
Is it prossible you first need to run `make install DESTDIR=/some/place`?

(So far, I haven't tried the CMake infrastructure much.)

dl8dtl, 2022-01-04:
`giveio.sys` is gone.
It was only required to drive parallel-port attached programming dongles on Windows, it was a gross hack anyway, and wouldn't work on any modern Windows. We simply decided to drop support for parallel-port attached programmers on the Windows platform. They have been in fashion some 20 years ago. If anyone is really keen to use one these days, apart from the difficulty to even find a computer with a parallel port, they are better off using one of the unixoid platforms that support parallel port access natively through an OS driver.

libusb is a 3rd party project, not part of AVRDUDE. You can probably ship it along with Embeetle.

mariusgreuel, 2022-01-04:
> How precisely should I distribute this build to our users?

The build artifacts are not intended to be used by end-users. The .zip file currently just reflects the CMake build folder, and the structure may change at any time.

You should use the release artifacts. As for GitHub, this is still work in progress, and much has not been defined yet. As you can see, there are no releases yet at GitHub:
https://github.com/avrdudes/avrdude/releases

The current releases can be found here:
http://download.savannah.gnu.org/releases/avrdude/

The Un*x release binaries are available through the package manager.

mariusgreuel, 2022-01-04:
> The Arduino IDE distribution has also the following files that are not present in the GitHub Actions build

Windows binaries may require additional DLLs, depending on which libraries they are linked against. The Savannah release for instance does not run unless you download libusb0.dll manually. The GitHub Windows builds are linked against the static libraries in order to fix these issues, but it is still work in progress.

dl8dtl, 2022-01-04:
The idea was to get Kristof some kind of prerelease that he could use …
I could of course always roll one here the traditional way, but I thought using the Github build artifacts is going to be the path to future releases.

kristofmulier, 2022-01-04:
Thank you @dl8dtl and @mariusgreuel for your replies 👍 

> [Jörg] `giveio.sys` is gone. It was only required to drive parallel-port attached programming dongles on Windows [..]

You're right. Parallel ports on computers are a dying breed. No need to invest time in that.

&nbsp;<br>
> [Jörg] `libusb` is a 3rd party project, not part of AVRDUDE. You can probably ship it along with Embeetle.

> [Marius] The Savannah release for instance does not run unless you download libusb0.dll manually. The GitHub Windows builds are linked against the static libraries in order to fix these issues, but it is still work in progress.

Probably the distribution in Arduino IDE is the Savannah release. I just did a quick test on Windows. When using AVRDUDE from Arduino IDE, the `libusb0.dll` is indeed required. Without it, I get a cryptic error message:

```bash
Error -1073741515
```

As for the GitHub Actions build, the `libusb0.dll` seems not required. I just successfully flashed my Arduino Uno board without the dll. It's great you link the GitHub Windows builds against static libraries! That makes life so much easier for the users. More dll's == more problems.

What about `libftdi1-config`? Is this a dll for FTDI chips?

&nbsp;<br>
> [Marius] The build artifacts are not intended to be used by end-users. The .zip file currently just reflects the CMake build folder, and the structure may change at any time. You should use the release artifacts. As for GitHub, this is still work in progress, and much has not been defined yet. As you can see, there are no releases yet at GitHub:

Jörg already replied to this. I indeed need some kind of pre-release for our next Embeetle IDE version.

dl8dtl, 2022-01-04:
> What about `libftdi1-config`? Is this a dll for FTDI chips?

Nope, that's just a tool (i.e., an executable) that can tell (to a development environment) where the libftdi1 include and library files are located.
```
$ libftdi1-config --libs
-L/usr/local/lib -lftdi1
$ libftdi1-config --cflags
-I/usr/local/include/libftdi1
```
No idea why they are shipping it with Arduino.

kristofmulier, 2022-01-04:
> [Jörg] Nope, that's just a tool (i.e., an executable) that can tell (to a development environment) where the libftdi1 include and library files are located.

Is the situation here comparable with the `libusb0.dll`? What I mean is: The Savannah/Arduino build requires a local `libusb0.dll` while the GitHub Actions build is linked against the static library and therefore no longer needs the dll. So I was thinking - maybe it's the same situation for the FTDI dll?

dl8dtl, 2022-01-04:
I'm not sure, if they ship the FTDI DLL, yes, but that still doesn't explain why they also ship that development tool.
Arduino applies a lot of local hacks to AVRDUDE. I think they do build their own binaries, but it's quite possible they include libftdi1 on purpose (and for sure, they do want to include libusb).

kristofmulier, 2022-01-04:
Hi @facchinm [Martino Facchin], can you please join the discussion?

@dl8dtl and @mariusgreuel , [Martino Facchin](https://github.com/facchinm) is the maintainer of the AVRDUDE fork for Arduino IDE.

mariusgreuel, 2022-01-04:
@kristofmulier: The main differences between the builds are the different USB and FTDI libraries. Until I get the GitHub Windows releases ready, you may want to stick with the Arduino release. You can also grab stuff from the build artifacts, however, you should not link to them or make any assumptions. You just need avrdude.exe and avrdude.conf.

Here is a little reading on libusb:
https://github.com/avrdudes/avrdude/wiki/The-story-on-libusb-for-Windows

The Arduino build uses libusb, Jörgs MinGW build uses libusb-win32, the GitHub actions use libusb by default, and optionally libusb-win32. I have a [releases here](https://github.com/mariusgreuel/avrdude/releases/tag/v6.3.1.1-windows), which uses a custom USB and FTDI library that work well on Windows (AFAIK). My intend is do get some of my patches into the upstream path, but that will takes a few more weeks.

kristofmulier, 2022-01-04:
Hi @mariusgreuel ,

> Until I get the GitHub Windows releases ready, you may want to stick with the Arduino release.

The problem is that I really need the new system for locating `avrdude.conf` (see https://github.com/avrdudes/avrdude/pull/780 ). So sticking with the Arduino release is not an option.

> The Arduino build uses libusb, Jörgs MinGW build uses libusb-win32, the GitHub actions use libusb by default, and optionally libusb-win32.

So if I understand correctly:
 - The Arduino/Savannah build uses **libusb** and requires the presence of `libusb0.dll` next to `avrdude.exe`.
 - Jörgs MinGW build uses **libusb-win32**. Does it also require a dll for that? Or is it statically linked?
 - The GitHub actions use **libusb** by default. As I understand, it is statically linked into the executable, which is absolutely awesome :-)

> My intend is do get some of my patches into the upstream path, but that will takes a few more weeks.

That's great! When you complete that, I will certainly update the AVRDUDE tool on our server (for the Embeetle IDE users). In the meantime, I would like to put an AVRDUDE tool there that uses the new system for locating `avrdude.conf` (perhaps in a few days or so, when we bring out a new release). What do you suggest me to do?

facchinm, 2022-01-11:
Hi @kristofmulier ,
Arduino build statically links against libusb-win32 (see here https://github.com/arduino/avrdude-build-script/blob/staging/libusb-compat-0.1.5.build.bash#L23-L35 ). Some older versions ship libusb dll but it was only needed as a dependency for libftdi.

We'd be glad to contribute to the mainline packaging effort since we've been doing it since forever :slightly_smiling_face:  

Also, expect some patches to be submitted in the next few weeks! @per1234

mariusgreuel, 2022-01-12:
@kristofmulier : In addition to the CMake build artifacts, I packed up the exes and conf that are used by the  deploy actions. I also added nightly.links here, that you can use for your distribution:
https://github.com/avrdudes/avrdude/wiki/Getting-Nightly-Builds-for-AVRDUDE

The build artifacts have a retention period of 90 days. Once available, you should grab the release artifacts.

closed 2022-01-12 by mariusgreuel

kristofmulier, 2022-01-13:
Super, thank you @mariusgreuel !

kristofmulier, 2022-01-13:
@facchinm ,
You should get in touch with @mariusgreuel . He is organizing the builds for AVRDUDE.