avrdudes/avrdude #799: --enable-libusb_1_0 doesn't disable old libusb

dmacks, 2022-01-04:
Building on avrdude-6.4 on OS X 10.13 with a ton of libraries installed via fink, I see:
```
checking for usb_get_string_simple in -lusb... yes
checking usb.h usability... yes
checking usb.h presence... yes
checking for usb.h... yes
checking lusb0_usb.h usability... no
checking lusb0_usb.h presence... no
checking for lusb0_usb.h... no
checking for libusb_init in -lusb-1.0... yes
checking libusb-1.0/libusb.h usability... yes
checking libusb-1.0/libusb.h presence... yes
checking for libusb-1.0/libusb.h... yes
checking libusb.h usability... no
checking libusb.h presence... no
checking for libusb.h... no
checking for libusb_init in -lusb... no
checking for libusb.h... (cached) no
[...]
DO HAVE    libusb
DO HAVE    libusb_1_0
```
and it compiles/links using both versions' headers and libraries. I get the same results even if I pass `--enable-libusb_1_0` (which is documented to use *only* -1.0 (given that it is available). If I pass `--disable-libusb_1_0`, then only the older version's headers and libraries are used. The logic in configure.ac looks pretty convoluted, and also hard-codes/second-guesses flags based on platform rather than using pkg-config to use what libusb* publishes as platform-correct.

A simplified method might be (semi-pseudocode):
```
HAVE_LIBUSB1=0
HAVE_LIBUSB=0
PKG_PROG_PKG_CONFIG
if (! --disable-libusb_1_0) {
  PKG_CHECK_MODULES([LIBUSB1], [libusb-1.0], [HAVE_LIBUSB1=1], [])
}
if (! HAVE_LIBUSB1) {
  PKG_CHECK_MODULES([LIBUSB], [libusb], [HAVE_LIBUSB=1])
}
```
at which point if ./configure has not aborted you have exactly one of HAVE_LIBUSB1 or HAVE_LIBUSB set (useful for setting AC_DEFINE) and flags in LIBUSB1_CFLAGS LIBUSB1_LIBS LIBUSB_CFLAGS LIBUSB1_CFLAGS (useful for propagating via AC_SUBST).

mariusgreuel, 2022-01-04:
I do not have access to a Mac, but a few hints: On Windows and Linux, it is OK to have both libraries enabled. Libusb-0.1 sometimes uses libusb-1.0 through a compatibility layer, such as libusb-compat. FreeBSD (which may be similar to Darwin?) ships a library, that has both APIs included in a single library.

The new CMake project should behave similar, as it allows you to use both API simultaneously. If you don't mind, you may try [this procedure](https://github.com/avrdudes/avrdude/wiki/Building-AVRDUDE-for-macOS) and provide feedback.

dl8dtl, 2022-01-04:
Actually, I just grepped through the source code: most of our USB-based implementations actually only use the libusb-0.1 API. It would be a tremendous work to rewrite all these for libusb-1.0.
USBasp prefers libusb-1.0 API over 0.1 if both are there.

closed 2022-01-04 by dl8dtl