avrdudes/avrdude #812: ATmega328P reported as lgt8fx328p

mariusgreuel, 2022-01-08:
When connecting to popular boards such as the Arduino Uno, avrdude 6.4 reports an ATmega328P signature as "lgt8fx328p":

```
avrdude.exe -c arduino -p m328p -P com5
...
avrdude.exe: Device signature = 0x1e950f (probably lgt8fx328p)
```

I think it would be nice if avrdude was to print the _expected_ part.

Further, when providing the wrong signature, I think it would be nice if avrdude was to print the _most likely_ part.

```
avrdude.exe -c arduino -p m328 -P com5
...
avrdude.exe: Device signature = 0x1e950f (probably lgt8fx328p)
avrdude.exe: Expected signature for ATmega328 is 1E 95 14
```

I guess all this would not bother me that much if it weren't for the popular ATmega328P part.

I suggest two things:
- If the correct signature is specified, print the part that was specified, and not the one from the lookup table.
- For parts with ambiguous signatures, mark the more popular part as "preferred" and print that one.

MCUdude, 2022-01-08:
Semi off-topic: What are the parts called lgt8fx* in avrdude.conf? AFAIK, no parts have the x in their names. There are lgt8f88, lgt8f168 and lgt8f328. There are *P and *d suffixes. I believe all these have the same device signature

dl8dtl, 2022-01-08:
The names came in from an external contribution.
I think the confusion is since the datasheet mentions the device family as "LGT8XM", where the "X" is apparently kind of a wildcard for the memory size:
http://www.lgtic.com/upload/lgt8fx8p/LGT8FX8P_databook_v1.0.4.pdf
They have only been in there since AVRDUDE 6.4, we can rename them for the next version.

dl8dtl, 2022-01-08:
I think that "preferred" can somehow be arranged by the order within `avrdude.conf` (possibly with changes in the search algorithm for printout).
Introducing a new keyword in `avrdude.conf` appears to be a bit of overkill to me.

dl8dtl, 2022-01-10:
I also changed the device names for these Logic Green devices.