avrdudes/avrdude #813: No error message when connecting to a not-connected AVRISP mkII programmer

mariusgreuel, 2022-01-08:
When I run the following command without a programmer attached, I would expect some sort of error message (without -v):
```
$ avrdude -c avrisp2 -p m328p

avrdude done.  Thank you.

$ avrdude -c avrisp2 -p m328p -v

avrdude: Version 6.99-20220108 (e3338c4)
         Copyright (c) Brian Dean, http://www.bdmicro.com/
         Copyright (c) Joerg Wunsch

         System wide configuration file is "/usr/local/bin/../etc/avrdude.conf"
         User configuration file is "/home/marius/.avrduderc"

         Using Port                    : usb
         Using Programmer              : avrisp2
avrdude: usbdev_open(): did not find any USB device "usb" (0x03eb:0x2104)

avrdude done.  Thank you.
```

MCUdude, 2022-01-08:
I can confirm that it happens on my computer too, but not on any other programmer I've tested.
Seems to me an AVRISPmkII related issue

MCUdude, 2022-01-08:
With the -v flag set I get this output when using other USB-based programmers:

```
avrdude: Error: Could not find USBtiny device (0x1781/0xc9f)
avrdude: error: could not find USB device with vid=0x16c0 pid=0x5dc vendor='www.fischl.de' product='USBasp'
avrdude: ERROR: Could not find device with Teensy bootloader (16C0:0478)
avrdude: ERROR: Could not find device with Micronucleus bootloader (16D0:0753)
```

However, with the AVRISPmkII **And AVR Dragon** I get this:
```
avrdude: usbdev_open(): did not find any USB device "usb" (0x03eb:0x2104) # AVRISPmkII
avrdude: usbdev_open(): did not find any USB device "usb" (0x03eb:0x2107) # AVR Dragon
```

Without the -v flag I can get the correct error by adding `-P USB` to the Avrdude command

dl8dtl, 2022-01-08:
Actually, there are quite a number of programmers that simply return -1 if opening the device failed.
Apparently, it has always been the assumption that the upstream code (i.e., `main.c`) handles that situation, and tells the user about it.
See PR #814

commit 5746bb26 (2022-01-09): Merge pull request #814 from dl8dtl/notify-open-failure

Notify user about open failure

Closes #813 (also closes #471)
closed 2022-01-09 by dl8dtl