avrdudes/avrdude #819: Using libavrdude as a standalone library

yegorich, 2022-01-10:
So far `libavrdude` will be built internally as a static library. For the common usage it has to provide the following features:

* static or dynamic build (there is a related CMake option)
* install target for both the library and header file
* the header file must not include local headers like `ac_cfg.h`

dl8dtl, 2022-01-10:
> the header file must not include local headers like `ac_cfg.h`

See PR #820 - you're welcome to review it.

dl8dtl, 2022-01-11:
Btw., with PR #826 merged, I notice it's now *either* a shared or a static library.
This is probably OK, but with autoconf/automake, both used to be built.

yegorich, 2022-01-11:
Yes, this is normal CMake procedure. But there are tricks to make both like what `libftdi1` [makes](http://developer.intra2net.com/git/?p=libftdi;a=blob;f=src/CMakeLists.txt;h=17b36173621bd9337b03e07a56f468d598ac208b;hb=HEAD#l41).

Generally, it is not a problem. When you package avrdude, you would run the build twice with different options and use these artefacts. This is what Debian rules do for many CMake-based packages

dl8dtl, 2022-01-11:
I just opened issue #827 - libavrdude.so is built but not installed.

mcuee, 2023-06-21:
Related issue:

#1414 

As of now, it is not possible to build libavrdude.dll dynamic library under Windows.