avrdudes/avrdude #822: jtagice3_tpi programmers not supported by Avrdude

MCUdude, 2022-01-10:
I have an old Xplained Nano ATtiny104 lying around, and to my surprise, I can't use the onboard mEDBG to program the target. This is because JTAG3 programmers in TPI mode has never been added, which means that neither the Pickit4, SNAP or the Atmel ICE can't be used to program TPI compatible AVRs.

Here's a relevant Avrfreaks discussion where @janegilruud has the same conclusion:
 https://www.avrfreaks.net/forum/need-help-using-attiny104-xplained-nano-mac
> I did a quick look, and it seems TPI support is missing for all Atmel/Microchip tools, so that needs to be added. When that is done there will probably be a new type called "jtagice3_tpi", but for now that is missing so it won't help adding it to avrdude.conf.

Here's a discussion I found in a mailing list: https://www.mail-archive.com/avrdude-dev@nongnu.org/msg04906.html

MCUdude, 2022-01-11:
AFAIK, this is the public information Microchip provides regarding the jtag3 protocol and how TPI is implemented.
From my understanding, TPI support has to be implemented through the stk500v2 protocol, similar to how ISP is already implemented on the jtag3.

https://onlinedocs.microchip.com/pr/GUID-33422CDF-8B41-417C-9C31-E4521ADAE9B4-en-US-2/index.html

dl8dtl, 2022-01-11:
> From my understanding, TPI support has to be implemented through the stk500v2 protocol, similar to how ISP is already implemented on the jtag3.

This seems correct to me.

The AVRISPmkII already correctly handles that. Inside `stk500v2.c`, this is called `xprog` as that's the PDI protocol name on the STK600 (which first implemented PDI) was dubbed. TPI is handled similarly.

MCUdude, 2022-01-16:
I stumbled across this document. I'll leave it here so I don't loose it. Might come in handy if I or someone else are going to give this a go some time:

http://ww1.microchip.com/downloads/en/DeviceDoc/50002630A.pdf

dl8dtl, 2022-01-16:
Good. I guess it's basically the same what we have in the `EDBG` subdirectory of our `atmel-docs`, but back then, they only provided it as HTML.

cederom, 2022-04-09:
I just got my hands on ATtiny104XNANO in order to program ATtiny10.. and noticed that TPI is not available :-)

cederom, 2022-04-10:
Update: I made TPI work with ATtiny10 on FT2232H based interface, see https://github.com/avrdudes/avrdude/issues/928 :-)

janegilruud, 2022-07-05:
I've been a little preoccupied these last few days, but I'll find some time and take a look.

MCUdude, 2022-07-05:
@janegilruud that would be great, thank you!
Please let me know if there's anything I can do to help. I have an Attiny10 and an Attiny104 I can test with, and I own a PK4, a SNAP and an ATtiny104 Xplained Nano board.

mcuee, 2022-10-23:
@janegilruud 

Just wondering if we have some updates here.

MCUdude, 2022-11-08:
A few weeks ago, [Pymcuprog](https://github.com/microchip-pic-avr-tools/pymcuprog) and [Pyedbglib](https://github.com/microchip-pic-avr-tools/pyedbglib) were updated to support more targets (thanks @xedbg!). Now we have preliminary support for TPI devices through the JTAG3/EDBG protocol. This is precisely what we'd like to replicate here.

I spent a few hours last night trying to replicate how pymcuprog initializes the programmer, but I didn't succeed.
The obstacle I couldn't get across was that [jtag3_getsync() in stk500v2.c returns -1](https://github.com/avrdudes/avrdude/blob/4c92030e3a486cfbaeb36e298f6f2929e8e031eb/src/stk500v2.c#L3509-L3514) no matter what I do.

mcuee, 2022-11-08:
Just a guess, what if you skip that check?

xedbg, 2022-11-08:
Which error message comes out (from down in the stack) at the point of failure?

MCUdude, 2022-11-09:
@xedbg I just updated the Xplained Mini ATtiny104 to the latest FW Studio 7 has to offer (1.07 IIRC).

I've pushed my test branch here: https://github.com/MCUdude/avrdude/tree/jtag3-tpi

Here's the output I'm getting when [I leave jtag3_getsync()](https://github.com/MCUdude/avrdude/blob/c0ec6a57384842cbb521b837dad78408c187a6cf/src/stk500v2.c#L3516) in there:
```
$ ./avrdude -cxplainedmini -pattiny104 -v

avrdude: Version 7.0-20220508
         Copyright (c) Brian Dean, http://www.bdmicro.com/
         Copyright (c) Joerg Wunsch

         System wide configuration file is /Users/hans/Downloads/avrdude/src/avrdude.conf
         User configuration file is /Users/hans/.avrduderc

         Using Port                    : usb
         Using Programmer              : xplainedmini
avrdude: found CMSIS-DAP compliant device, using EDBG protocol
avrdude: jtag3_edbg_recv(): no response available
stk500v2_jtag3_open() rv: -1
avrdude stk500v2_jtag3_open() error: unable to sync with the JTAGICE3 in ISP mode
avrdude main() error: unable to open programmer xplainedmini on port usb

avrdude done.  Thank you.
```

Here's the output when I'm bypassing it altogether:

```
$ ./avrdude -cxplainedmini -pattiny104 -v

avrdude: Version 7.0-20220508
         Copyright (c) Brian Dean, http://www.bdmicro.com/
         Copyright (c) Joerg Wunsch

         System wide configuration file is /Users/hans/Downloads/avrdude/src/avrdude.conf
         User configuration file is /Users/hans/.avrduderc

         Using Port                    : usb
         Using Programmer              : xplainedmini
avrdude: found CMSIS-DAP compliant device, using EDBG protocol
         AVR Part                      : ATtiny104
         RESET disposition             : dedicated
         RETRY pulse                   : SCK
         Serial program mode           : yes
         Parallel program mode         : yes
         Memory Detail                 :

                                           Block Poll               Page                       Polled
           Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           fuse                    0     0     4    0 no          1   16      0     0     0 0x00 0x00
           lockbits                0     0     0    0 no          1   16      0     0     0 0x00 0x00
           signature               0     0     0    0 no          3   16      0     0     0 0x00 0x00
           calibration             0     0     0    0 no          1   16      0     0     0 0x00 0x00
           flash                   0     0   128    0 no       1024   16      0     0     0 0x00 0x00

         Programmer Type : JTAG3_ISP
         Description     : Atmel AVR XplainedMini in ISP mode
avrdude: jtag3_edbg_recv(): no response available
avrdude: jtag3_edbg_recv(): no response available
avrdude stk500v2_jtag3_recv() error: unable to receive

avrdude: stk500v2_jtag3_initialize() start
avrdude: jtag3_edbg_recv(): no response available
avrdude main() error: initialization failed, rc=-1
        - double check the connections and try again
        - use -B to set lower ISP clock frequency, e.g. -B 200kHz
        - use -F to override this check

avrdude: jtag3_edbg_recv(): no response available
avrdude stk500v2_jtag3_recv() error: unable to receive
avrdude: jtag3_edbg_recv(): no response available
avrdude: jtag3_edbg_recv(): no response available

avrdude done.  Thank you.
```

I've added `#define SCOPE_AVR_TPI              0x14` to jtag3_private.h, but I'm not familiar enough with the protocol to properly initialize the programmer for TPI use.

MCUdude, 2022-11-09:
So it fails very early in the process. [`stk500v2_jtag3_open()`](https://github.com/MCUdude/avrdude/blob/c0ec6a57384842cbb521b837dad78408c187a6cf/src/stk500v2.c#L3505) is the where the connection is established, so it fails before anything is set up properly. Maybe the issue is in [`jtag3_open_common()`](https://github.com/MCUdude/avrdude/blob/c0ec6a57384842cbb521b837dad78408c187a6cf/src/jtag3.c#L1445)?

mcuee, 2022-11-11:
I just got the ATtiny104 Xplained Nano from Mouser Singapore (along with some other boards and chips).

The following is the output from git main.
```
PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude -cxplainedmini -pattiny104 -v

avrdude: Version 7.0-20221101 (4c92030)
         Copyright (c) Brian Dean, http://www.bdmicro.com/
         Copyright (c) Joerg Wunsch

         System wide configuration file is C:\work\avr\avrdude_test\avrdude_bin\avrdude.conf

         Using Port                    : usb
         Using Programmer              : xplainedmini
avrdude: found CMSIS-DAP compliant device, using EDBG protocol
         AVR Part                      : ATtiny104
         RESET disposition             : dedicated
         RETRY pulse                   : SCK
         Serial program mode           : yes
         Parallel program mode         : yes
         Memory Detail                 :

                                           Block Poll               Page                       Polled
           Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           fuse                    0     0     4    0 no          1   16      0     0     0 0x00 0x00
           lockbits                0     0     0    0 no          1   16      0     0     0 0x00 0x00
           signature               0     0     0    0 no          3   16      0     0     0 0x00 0x00
           calibration             0     0     0    0 no          1   16      0     0     0 0x00 0x00
           flash                   0     0   128    0 no       1024   16      0     0     0 0x00 0x00

         Programmer Type : JTAG3_ISP
         Description     : Atmel AVR XplainedMini in ISP mode
         Vtarget         : 5.00 V
avrdude: jtag3_edbg_recv(): no response available
avrdude: jtag3_edbg_recv(): no response available
avrdude stk500v2_jtag3_recv() error: unable to receive

avrdude stk500v2_jtag3_initialize() error: part ATtiny104 has no ISP interface
avrdude main() error: initialization failed, rc=-1
        - double check the connections and try again
        - use -B to set lower ISP clock frequency, e.g. -B 200kHz
        - use -F to override this check

avrdude: jtag3_edbg_recv(): no response available
avrdude stk500v2_jtag3_recv() error: unable to receive
avrdude: jtag3_edbg_recv(): no response available
avrdude: jtag3_edbg_recv(): no response available

avrdude done.  Thank you.
```

With @MCUdude's test branch.
```
PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude_tpi_jtag3 -cxplainedmini -pattiny104 -vv

avrdude_tpi_jtag3: Version 7.0-20221109 (c0ec6a5)
                   Copyright (c) Brian Dean, http://www.bdmicro.com/
                   Copyright (c) Joerg Wunsch

                   System wide configuration file is C:\work\avr\avrdude_test\avrdude_bin\avrdude.conf

                   Using Port                    : usb
                   Using Programmer              : xplainedmini
avrdude_tpi_jtag3: stk500v2_jtag3_open()
avrdude_tpi_jtag3: found CMSIS-DAP compliant device, using EDBG protocol
avrdude_tpi_jtag3: sending sign-on command: 0x80 (3 bytes msg)
                   AVR Part                      : ATtiny104
                   RESET disposition             : dedicated
                   RETRY pulse                   : SCK
[... 23 lines cut ...]
avrdude_tpi_jtag3 stk500v2_jtag3_recv() error: unable to receive

avrdude_tpi_jtag3: stk500v2_jtag3_initialize() start
avrdude_tpi_jtag3: jtag3_setparm()
avrdude_tpi_jtag3: sending set parameter (scope 0x12, section 0, parm 0) command: avrdude_tpi_jtag3: jtag3_edbg_recv(): no response available

avrdude_tpi_jtag3: set parameter (scope 0x12, section 0, parm 0) command: timeout/error communicating with programmer (status -1)
avrdude_tpi_jtag3 main() error: initialization failed, rc=-1
                  - double check the connections and try again
                  - use -B to set lower ISP clock frequency, e.g. -B 200kHz
                  - use -F to override this check

avrdude_tpi_jtag3: jtag3_edbg_recv(): no response available
avrdude_tpi_jtag3 stk500v2_jtag3_recv() error: unable to receive
avrdude_tpi_jtag3: stk500v2_jtag3_close()
avrdude_tpi_jtag3: jtag3_close()
avrdude_tpi_jtag3: sending AVR sign-off command: avrdude_tpi_jtag3: jtag3_edbg_recv(): no response available

avrdude_tpi_jtag3: AVR sign-off command: timeout/error communicating with programmer (status -1)
avrdude_tpi_jtag3: sending sign-off command: avrdude_tpi_jtag3: jtag3_edbg_recv(): no response available

avrdude_tpi_jtag3: sign-off command: timeout/error communicating with programmer (status -1)

avrdude_tpi_jtag3 done.  Thank you.
```

xedbg, 2022-11-11:
For a start, this output here:
   get parameter (scope 0x12, section 1, parm 32)
indicates that it is attempting to read the "mega programming clock"
  this is not relevant to TPI and will probably return FAILURE_NOT_SUPPORTED

MCUdude, 2022-11-11:
@xedbg thanks for the hint! I'll look into it as soon as the kids are in bed.

It would be amazing if we could resolve this before Avrdude 7.1 gets released (right before Christmas I think)!
I'll try my best to make it happen

xedbg, 2022-11-11:
Yes, indeed :/
There are housekeeping calls which need to be made on the general-purpose debuggers, but many of those are optional on the mEDBG on kits.  So working with a Xplained kit should make for a simpler start.
But the bulk of the work needs to be done on the TPI protocol envelope (0x14) level.  There are only a handful of config parameters (https://github.com/microchip-pic-avr-tools/pyedbglib/blob/main/pyedbglib/protocols/tpiprotocol.py#L63) and there are defaults in place for the Xplained kit, so those are also optional
I realise the pymcuprog stub code only reads out the device ID, but it does so using a memory read (https://github.com/microchip-pic-avr-tools/pymcuprog/blob/eda95bdcc4a4cb4e4f442aa59a2d49c423dfd730/pymcuprog/nvmtpi.py#L37).

MCUdude, 2022-11-11:
It seems like the Xplainedmini TPI becomes unresponsive (instead of returning -1) when it receives a call/command that it doesn't recognize. The Xplainedmini ISP isn't like this, is it?

In order to match the messages sent in pymcuprog/pyedbglib, it seems like I can't use `jtag3_setparm()`, since this carries too much data, but instead use `jtag3_command()`. But this is again different than what's done to initialize the other JTAG3 interfaces (ISP, JTAG, PDI, UPDI)

xedbg, 2022-11-11:
Well, technically these are three separate protocols in parallel: ISP on 0x11, TPI on 0x14 and all-the-rest on 0x12... so something is fishy in there, probably as a result of the onions of protocols stacked up here :|
A set to (0x12, 0, 0) is really only relevant to operations in 0x12, not TPI on 0x14.  The standalone debuggers arguably will be ok with this regardless, but the mEDBG was stripped down to aggressively to fit in that the host really unfortunately has to know that it (in this case) is a TPI medbg instance, which needs only housekeeping and TPI functionality (so nothing on 0x11 or 0x14). 
32k flash has its limitations... (specially while someone is holding one's beer).
To find this out, it should really use the DISCOVERY QUERY call for a command_handler list, and see which one it contains...

janegilruud, 2022-11-30:
Created PR #1205

MCUdude, 2022-12-19:
#1205 is now merged! 🎉

closed 2022-12-19 by MCUdude

cederom, 2022-12-19:
WOW! CONGRATZ! I WILL TAKE A LOOK IN A FREE MOMENT :-)

when a release? :-)

MCUdude, 2022-12-19:
> when a release? :-)

Avrdude v7.1 will probably be released by the end of December

cederom, 2022-12-20:
Initial testing results with today git pull main:

```
% uname -a
FreeBSD octagon 13.1-STABLE FreeBSD 13.1-STABLE #1 stable/13-n253256-09072186eec1: Fri Dec  9 22:47:42 CET 2022     root@octagon:/usr/obj/usr/src/amd64.amd64/sys/GENERIC amd64

% git log -5 --oneline
159d126 (HEAD -> main, origin/main, origin/HEAD) Update NEWS
95363a1 Terminal read improvements (#1209)
bbd3ed2 Merge pull request #1208 from stefanrueger/urclock-updi
b108546 Consider libedit only in interactive mode (#1207)
72f0975 Merge pull request #1206 from stefanrueger/nsafety
```

Cannot get to terminal on AtTiny104 XplainedNano:

```
% ./avrdude -c xplainedmini_tpi -p t104 -vv -t

avrdude: Version 7.0-20221214 (159d126)
         Copyright the AVRDUDE authors;
         see https://github.com/avrdudes/avrdude/blob/main/AUTHORS

         System wide configuration file is /XXX/avrdude.git/build_freebsd/src/avrdude.conf
         User configuration file is /XXX/.avrduderc
avrdude yywarning() warning: programmer ktlink overwrites previous definition /XXX/avrdude.git/build_freebsd/src/avrdude.conf:758. [/XXX/.avrduderc:376]

         Using Port                    : usb
         Using Programmer              : xplainedmini_tpi
avrdude: jtag3_open_tpi()
avrdude: found CMSIS-DAP compliant device, using EDBG protocol
         AVR Part                      : ATtiny104
         RESET disposition             : dedicated
         RETRY pulse                   : SCK
         Serial program mode           : yes
         Parallel program mode         : yes
         Memory Detail                 :

                                           Block Poll               Page                       Polled
           Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           fuse                    0     0     4    0 no          1   16      0     0     0 0x00 0x00
           lockbits                0     0     0    0 no          1   16      0     0     0 0x00 0x00
           signature               0     0     0    0 no          3   16      0     0     0 0x00 0x00
           calibration             0     0     0    0 no          1   16      0     0     0 0x00 0x00
           flash                   0     0   128    0 no       1024   16      0     0     0 0x00 0x00
           sigrow                  0     0     0    0 no         16   16      0     0     0 0x00 0x00

         Programmer Type : JTAGICE3_TPI
         Description     : Atmel AVR XplainedMini in TPI mode
avrdude: jtag3_getparm()
avrdude: sending get parameter (scope 0x01, section 0, parm 0) command: 0x84 (9 bytes msg)
avrdude: sending get info (serial number) command: 0x81 (23 bytes msg)
         ICE HW version  : 0
         ICE FW version  : 1.07 (rel. 10)
         Serial number   : ATML2678020200009059
avrdude: jtag3_enable_tpi() is empty. No action necessary.
avrdude: jtag3_initialize_tpi() start
avrdude jtag3_command_tpi() error: [TPI] command Enter Progmode FAILED! Status: 0x03
avrdude main() error: initialization failed, rc=-1
        - double check the connections and try again
        - use -B to set lower ISP clock frequency, e.g. -B 200kHz
        - use -F to override this check

avrdude jtag3_command_tpi() error: [TPI] command Leave Progmode FAILED! Status: 0x01
avrdude: jtag3_close_tpi() is empty. No action necessary.

avrdude done.  Thank you.
```

[... 54 lines cut ...]
avrdude: sending get info (serial number) command: 0x81 (23 bytes msg)
         ICE HW version  : 0
         ICE FW version  : 1.13 (rel. 43)
         Serial number   : ATML2795042700003537
avrdude: jtag3_enable_tpi() is empty. No action necessary.
avrdude: jtag3_initialize_tpi() start
avrdude: jtag3_edbg_recv(): no response available
avrdude jtag3_recv_tpi() error: jtag3_recv_tpi(): unable to receive

avrdude: TPI Enter Progmode command: timeout/error communicating with programmer (status -1)
avrdude main() error: initialization failed, rc=-1
        - double check the connections and try again
        - use -B to set lower ISP clock frequency, e.g. -B 200kHz
        - use -F to override this check

avrdude: jtag3_edbg_recv(): no response available
avrdude jtag3_recv_tpi() error: jtag3_recv_tpi(): unable to receive

avrdude: TPI Leave Progmode command: timeout/error communicating with programmer (status -1)
avrdude: jtag3_close_tpi() is empty. No action necessary.

avrdude done.  Thank you.
```

Any hints welcome :-)

janegilruud, 2022-12-20:
ATtiny416 uses UPDI, not TPI. Wrong interface.

mcuee, 2022-12-20:
No issues here under a FreeBSD VM.

```
mcuee@freebsdx64vm:~/build $ git clone https://github.com/avrdudes/avrdude.git
Cloning into 'avrdude'...
remote: Enumerating objects: 14847, done.
remote: Counting objects: 100% (753/753), done.
remote: Compressing objects: 100% (312/312), done.
remote: Total 14847 (delta 523), reused 646 (delta 437), pack-reused 14094
Receiving objects: 100% (14847/14847), 7.98 MiB | 12.82 MiB/s, done.
Resolving deltas: 100% (10402/10402), done.

mcuee@freebsdx64vm:~/build $ cd avrdude

mcuee@freebsdx64vm:~/build/avrdude $ ./build.sh 
-- The C compiler identification is Clang 11.0.1
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Found Git: /usr/local/bin/git (found version "2.35.2") 
-- Found FLEX: /usr/bin/flex (found version "2.6.4") 
-- Found YACC: /usr/bin/yacc
-- Looking for libelf.h
-- Looking for libelf.h - found
-- Looking for libelf/libelf.h
-- Looking for libelf/libelf.h - not found
-- Looking for usb.h
-- Looking for usb.h - found
-- Looking for lusb0_usb.h
-- Looking for lusb0_usb.h - not found
-- Looking for libusb.h
-- Looking for libusb.h - found
-- Looking for libusb-1.0/libusb.h
-- Looking for libusb-1.0/libusb.h - not found
-- Looking for hidapi/hidapi.h
-- Looking for hidapi/hidapi.h - found
-- Looking for ftdi_tcioflush
-- Looking for ftdi_tcioflush - found
-- Configuration summary:
-- ----------------------
-- DO HAVE    libelf
-- DO HAVE    libusb
-- DO HAVE    libusb_1_0
-- DO HAVE    libhidapi
-- DON'T HAVE libftdi
-- DO HAVE    libftdi1
-- DO HAVE    libreadline
-- DISABLED   doc
-- DISABLED   parport
-- DISABLED   linuxgpio
-- DISABLED   linuxspi
-- ----------------------
-- Configuring done
-- Generating done
-- Build files have been written to: /home/mcuee/build/avrdude/build_freebsd
[  1%] Generating avrdude.conf
[  1%] Built target conf
[  3%] [FLEX][Parser] Building scanner with flex 2.6.4
[  4%] [YACC][Parser] Building parser with yacc
[  6%] Building C object src/CMakeFiles/libavrdude.dir/arduino.c.o
[  7%] Building C object src/CMakeFiles/libavrdude.dir/avr.c.o
[  9%] Building C object src/CMakeFiles/libavrdude.dir/avr910.c.o
[ 10%] Building C object src/CMakeFiles/libavrdude.dir/avrcache.c.o
[ 12%] Building C object src/CMakeFiles/libavrdude.dir/avrftdi.c.o
[ 13%] Building C object src/CMakeFiles/libavrdude.dir/avrftdi_tpi.c.o
[ 15%] Building C object src/CMakeFiles/libavrdude.dir/avrpart.c.o
[ 16%] Building C object src/CMakeFiles/libavrdude.dir/bitbang.c.o
[ 18%] Building C object src/CMakeFiles/libavrdude.dir/buspirate.c.o
[... 144 lines cut ...]
         Serial number   : ATMLxxxxxxxxxxxxxxxxx
avrdude: jtag3_enable_tpi() is empty. No action necessary.
avrdude: jtag3_initialize_tpi() start
avrdude: jtag3_getparm()
avrdude: sending get parameter (scope 0x01, section 1, parm 0) command: 0x84 (6 bytes msg)
Vtarget         : 5.00 V
avrdude: AVR device initialized and ready to accept instructions

Reading |                                                    | 0% 0.00 s 
avrdude: jtag3_read_byte_tpi(.., signature, 0x0, ...)

avrdude: jtag3_read_byte_tpi(.., signature, 0x1, ...)
Reading | #################                                  | 33% 0.04 s 
avrdude: jtag3_read_byte_tpi(.., signature, 0x2, ...)
Reading | ################################################## | 100% 0.06 s 

avrdude: device signature = 0x1e900b (probably t104)
avrdude> quit
avrdude: jtag3_close_tpi() is empty. No action necessary.

avrdude done.  Thank you.

mcuee@freebsdx64vm:~/build/avrdude/build_freebsd/src $ uname -a
FreeBSD freebsdx64vm 13.0-RELEASE FreeBSD 13.0-RELEASE #0 releng/13.0-n244733-ea31abc261f: Fri Apr  9 04:24:09 UTC 2021     root@releng1.nyi.freebsd.org:/usr/obj/usr/src/amd64.amd64/sys/GENERIC  amd64
```

mcuee, 2022-12-20:
@cederom 

As mentioned by other reply, you have to use UPDI for ATtiny416. The following is with the ATtiny416-XNANO Kit right out of the box (newly bought).

```
mcuee@freebsdx64vm:~/build/avrdude/build_freebsd/src $ sudo ./avrdude -c xplainedmini_updi -p t416 -t
Password:

         Vtarget                      : 5.00 V
avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 0x1e9221 (probably t416)
avrdude> dump fuses

Reading | ################################################## | 100% 0.15 s 

0000  00 00 02 ff 00 f4 07 00  00                       |.........       |

avrdude> quit
avrdude> 
avrdude done.  Thank you.

mcuee@freebsdx64vm:~/build/avrdude/build_freebsd/src $ sudo ./avrdude -c xplainedmini_updi -p t416
 -U flash:r:factory_t416_xnano_readback.hex:i

         Vtarget                      : 5.00 V
avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 0x1e9221 (probably t416)
avrdude: reading flash memory ...

Reading | ################################################## | 100% 1.67 s 

avrdude: writing output file factory_t416_xnano_readback.hex

avrdude done.  Thank you.

```

cederom, 2022-12-20:
Sorry for my bad interface selection :-) True 416 works fine in UPDI :-) Congratz and Thank You! :-)

```
% ./avrdude -c xplainedmini_updi -p t416 -vv -t

avrdude: Version 7.0-20221214 (159d126)
         Copyright the AVRDUDE authors;
         see https://github.com/avrdudes/avrdude/blob/main/AUTHORS

         System wide configuration file is /XXX/avrdude.git/build_freebsd/src/avrdude.conf
         User configuration file is /XXX/.avrduderc
avrdude yywarning() warning: programmer ktlink overwrites previous definition /XXX/avrdude.git/build_freebsd/src/avrdude.conf:758. [/XXX/.avrduderc:376]

         Using Port                    : usb
         Using Programmer              : xplainedmini_updi
avrdude: jtag3_open_updi()
avrdude: HV UPDI support: 1
avrdude: found CMSIS-DAP compliant device, using EDBG protocol
avrdude: sending sign-on command: 0x80 (3 bytes msg)
         AVR Part                      : ATtiny416
         RESET disposition             : dedicated
         RETRY pulse                   : SCK
         Serial program mode           : yes
         Parallel program mode         : yes
         Memory Detail                 :

                                           Block Poll               Page                       Polled
           Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           fuse0       wdtcfg      0     0     0    0 no          1    1      0     0     0 0x00 0x00
           fuse1       bodcfg      0     0     0    0 no          1    1      0     0     0 0x00 0x00
           fuse2       osccfg      0     0     0    0 no          1    1      0     0     0 0x00 0x00
           fuse4       tcd0cfg     0     0     0    0 no          1    1      0     0     0 0x00 0x00
           fuse5       syscfg0     0     0     0    0 no          1    1      0     0     0 0x00 0x00
           fuse6       syscfg1     0     0     0    0 no          1    1      0     0     0 0x00 0x00
           fuse7       append      0     0     0    0 no          1    1      0     0     0 0x00 0x00
           fuse8       bootend     0     0     0    0 no          1    1      0     0     0 0x00 0x00
           fuses                   0     0     0    0 no          9   10      0     0     0 0x00 0x00
           lock                    0     0     0    0 no          1    1      0     0     0 0x00 0x00
           tempsense               0     0     0    0 no          2    1      0     0     0 0x00 0x00
           signature               0     0     0    0 no          3    1      0     0     0 0x00 0x00
           prodsig                 0     0     0    0 no         61   61      0     0     0 0x00 0x00
           sernum                  0     0     0    0 no         10    1      0     0     0 0x00 0x00
           osccal16                0     0     0    0 no          2    1      0     0     0 0x00 0x00
           osccal20                0     0     0    0 no          2    1      0     0     0 0x00 0x00
           osc16err                0     0     0    0 no          2    1      0     0     0 0x00 0x00
           osc20err                0     0     0    0 no          2    1      0     0     0 0x00 0x00
           data                    0     0     0    0 no          0    1      0     0     0 0x00 0x00
           userrow     usersig     0     0     0    0 no         32   32      0     0     0 0x00 0x00
           eeprom                  0     0     0    0 no        128   32      0     0     0 0x00 0x00
           flash                   0     0     0    0 no       4096   64      0     0     0 0x00 0x00

         Programmer Type : JTAGICE3_UPDI
         Description     : Atmel AVR XplainedMini in UPDI mode
avrdude: jtag3_getparm()
avrdude: sending get parameter (scope 0x01, section 0, parm 0) command: 0x84 (9 bytes msg)
avrdude: sending get info (serial number) command: 0x81 (23 bytes msg)
         ICE HW version  : 0
         ICE FW version  : 1.13 (rel. 43)
         Serial number   : ATML2795042700003537
avrdude: jtag3_getparm()
avrdude: sending get parameter (scope 0x01, section 0, parm 1) command: 0x84 (6 bytes msg)
avrdude: jtag3_setparm()
avrdude: sending set parameter (scope 0x12, section 0, parm 0) command: 0x80 (3 bytes msg)
avrdude: jtag3_setparm()
avrdude: sending set parameter (scope 0x12, section 0, parm 1) command: 0x80 (3 bytes msg)
avrdude: jtag3_setparm()
avrdude: sending set parameter (scope 0x12, section 1, parm 0) command: 0x80 (3 bytes msg)
avrdude: jtag3_getparm()
avrdude: sending get parameter (scope 0x01, section 1, parm 0) command: 0x84 (6 bytes msg)
[... 113 lines cut ...]
0280  87 e1 9e e3 0e 94 03 01  88 23 41 f0 80 e2 80 93  |.........#A.....|
0290  25 04 8a e4 90 e0 0e 94  90 00 63 c0 62 e4 70 e0  |%.........c.b.p.|
02a0  87 e1 9e e3 0e 94 03 01  88 23 49 f0 80 91 01 0f  |.........#I.....|
02b0  0e 94 ee 00 80 e0 9e e3  0e 94 7b 00 52 c0 6b e3  |..........{.R.k.|
02c0  70 e0 87 e1 9e e3 0e 94  03 01 88 23 89 f0 80 91  |p..........#....|
02d0  00 11 0e 94 ee 00 80 91  01 11 0e 94 ee 00 80 91  |................|
02e0  02 11 0e 94 ee 00 80 e0  9e e3 0e 94 7b 00 39 c0  |............{.9.|
02f0  64 e3 70 e0 87 e1 9e e3  0e 94 03 01 88 23 69 f1  |d.p..........#i.|
0300  80 91 03 11 0e 94 ee 00  80 91 04 11 0e 94 ee 00  |................|
0310  80 91 05 11 0e 94 ee 00  80 91 06 11 0e 94 ee 00  |................|
[... cut ...]
0390  b5 00 88 23 e1 f3 0e 94  21 01 f9 cf f8 94 ff cf  |...#....!.......|
03a0  00 00 30 31 32 33 34 35  36 37 38 39 41 42 43 44  |..0123456789ABCD|
03b0  45 46 00 00 ff ff ff ff  ff ff ff ff ff ff ff ff  |EF..............|
03c0  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
03d0  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
03e0  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
03f0  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
```

cederom, 2022-12-20:
Mystery solved, turns out I did remove three 1k resistors between interface and target chip, no wonder it did not work heh, when these programming resistors are back all works fine! THANK YOU AND CONGRATULATIONS!!! :-)

```
% ./avrdude -c xplainedmini_tpi -p t104 -t
avrdude warning: programmer ktlink overwrites previous definition /XXX/avrdude.git/build_freebsd/src/avrdude.conf:758. [/XXX/.avrduderc:376]

Vtarget                      : 5.00 V
avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 0x1e900b (probably t104)
avrdude> dump flash 0 1024

Reading | -------------------------------------------------- | 0% 10.00 s

avrdude error: short read, read only 0 out of 64 bytes
0000  0f c0 1e c0 1d c0 1c c0  1b c0 1a c0 19 c0 18 c0  |................|
0010  17 c0 16 c0 15 c0 14 c0  13 c0 12 c0 11 c0 10 c0  |................|
0020  11 27 1f bf cf e5 d0 e0  de bf cd bf 20 e0 a0 e4  |.'.......... ...|
0030  b0 e0 01 c0 1d 93 aa 34  b2 07 e1 f7 4a d0 5c c0  |.......4....J.\.|
0040  df cf 44 e0 45 b9 1a b9  43 e3 49 b9 48 e1 4d b9  |..D.E...C.I.H.M.|
0050  4e e0 4c b9 42 e4 50 e0  e0 e4 f0 e0 41 93 50 83  |N.L.B.P.....A.P.|
0060  08 95 75 9b fe cf 88 b9  08 95 cf 93 c8 2f 80 e3  |..u........../..|
0070  f8 df 88 e7 f6 df 8c 2f  82 95 8f 70 8a 30 18 f0  |......./...p.0..|
0080  89 5c ef df 02 c0 80 5d  ec df 8c 2f 8f 70 8a 30  |.\.....].../.p.0|
0090  18 f0 89 5c e6 df 02 c0  80 5d e3 df 8d e0 e1 df  |...\.....]......|
00a0  8a e0 df df cf 91 08 95  81 17 91 07 69 f4 08 95  |............i...|
00b0  00 00 00 00 00 00 00 00  41 50 51 0b c9 f7 00 00  |........APQ ....|
00c0  81 50 91 0b 19 f4 08 95  77 ee 63 e0 47 2f 56 2f  |.P. ....w.c.G/V/|
00d0  ef cf 48 ed 4c bf 16 bf  45 e7 49 bf 0d 9a 39 9a  |..H.L...E.I. .9.|
00e0  31 9a af df 80 e0 c1 e0  c8 0f bf df 05 9a 88 ee  |1...............|
00f0  93 e0 da df 8c 2f f7 cf  f8 94 ff cf ff ff ff ff  |...../..........|
0100  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0110  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0120  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0130  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0140  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0150  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0160  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0170  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0180  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0190  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
01a0  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
01b0  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
01c0  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
01d0  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
01e0  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
01f0  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0200  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0210  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0220  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0230  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0240  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0250  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0260  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0270  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0280  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0290  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
02a0  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
02b0  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
02c0  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
02d0  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
02e0  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
02f0  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0300  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0310  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0320  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0330  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0340  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0350  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0360  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0370  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
[... 162 lines cut ...]
02b0  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
02c0  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
02d0  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
02e0  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
02f0  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
0300  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
0310  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
[... cut ...]
03b0  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
03c0  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
03d0  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
03e0  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
03f0  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|

```

I noticed that after leaving terminal mode I had to reconnect the board in order to flash etc but that is not a big problem. Thank you folks YOU ROX!! :-)