avrdudes/avrdude #824: CMake compiled binaries lack debug symbols

dl8dtl, 2022-01-10:
Autoconf/automake always call the compiler with -g.
Apparently, CMake-generated Makefiles don't.
At least on Unix-like systems,. there hasn't been a reason to turn off debugging symbols even when optimization is enabled for decades now; both settings do not contradict each other. Stripping debugging symbols on installation is easy, adding debugging symbols later is impossible.
So, that "classic" distinction between "Release" vs. "Debug" is basically irrelevant on Unix-like systems. Best practice is to *always* enable optimizations (nobody wants pessimized code), and *always* compile with debug symbols.

mariusgreuel, 2022-01-10:
Have you tried CMAKE_BUILD_TYPE=RelWithDebInfo?

See https://github.com/avrdudes/avrdude/blob/main/.github/workflows/build.yml

dl8dtl, 2022-01-10:
> Have you tried CMAKE_BUILD_TYPE=RelWithDebInfo?

No.
Can we somehow make that the default?

dl8dtl, 2022-01-10:
Well, as a first shot, it would be useful to document that in the Wiki instead of "Release".
As nobody can memorize that stuff, people are probably going to copy&paste it from the Wiki instructions anyway. ;-)

*****

OK, works.

dl8dtl, 2022-01-11:
https://www.kitware.com/cmake-and-the-default-build-type/

Any opinions on using that to set the default build type to RelWithDebInfo?

mariusgreuel, 2022-01-11:
> Any opinions on using that to set the default build type to RelWithDebInfo?

I don't know.

- This code from the blog works only for built-systems such as make.
- The build type is a preference (I prefer Debug), so I would rather not change the defaults.
- There are probably other options you need (see the build action).

On Linux, I set up an alias for building avrdude, on Windows I use a json configuration file to switch between the build types and architectures.

dl8dtl, 2022-01-11:
OK, two suggestions:

* for Unix-like sytems, we can safely the Wiki instructions to use RelWithDebInfo; doesn't lose any performance but allows for debugging
* also, we could provide a .sh script assisting the build – after all, it's a bit more to memorize than the old `./configure; make` now (I can handle that)
* no idea whether you'd like to add something similar for Windows builds

mariusgreuel, 2022-01-13:
> we could provide a .sh script assisting the build

I like that solution, because we do not need to change the CMake defaults then. Actually, that is what I am doing, too, when building in Windows. I always work on a branch that has extra files and changes, that I simply rebase on top of the avrdude main branch.

I think a shell script for building in Un*x will be useful for others too, plus you can add other configuration options, too.

dl8dtl, 2022-01-14:
As the Wiki has already been updated, this one can be closed.

PR #835 will then provide a Unix build script as well.

closed 2022-01-14 by dl8dtl