avrdudes/avrdude #830: CMake: split CMakeLists.txt into common and source related

yegorich, 2022-01-12:
@mariusgreuel what about splitting the `CMakeLists.txt` and moving the common one to the project's root directory? This would simplify integration into different Linux distros and IDEs as, by default, they all expect `CMakeLists.txt` to be in the root folder.

mariusgreuel, 2022-01-12:
I have not really thought about this, but I am open for changes. I tried to keep the current CMake project simple.

> what about splitting the CMakeLists.txt

In which components? Executable and libavrdude? What do you mean with 'common'?

yegorich, 2022-01-12:
I suggest to place one `CMakeLists.txt` into the root folder. This file will define the project, take care of finding dependencies and it will include the `src/CMakeLists.txt` that takes care of `libavrdude` and `avrdude`.

See our [repo](https://github.com/visionsystemsgmbh/libonrisc) for the reference.

mariusgreuel, 2022-01-12:
OK, I think I understand what you are proposing. I don't understand why you want to make this change. Also, what is the benefit of splitting the files? You seem to have some other plans that you have not shared yet? :-)

I am aware of the pattern where root makefiles include makefiles in sub-directories. But then, the project in the sub-directory can still be built by itself. In your companies repo you seem to have lost that functionality because the sub-folders cannot be built by themselves?

yegorich, 2022-01-12:
This approach will be proposed by CMake itself in its [tutorials](https://cmake.org/cmake/help/latest/guide/tutorial/Adding%20a%20Library.html).

Other projects use it as well. See [cmocka](https://git.cryptomilk.org/projects/cmocka.git/tree/).

In the root `CMakeLists.txt` you handle options and can then decide which subfolder to include/build. For example, if you enable tests, you include tests folder etc.

I'd like to create some tests for `libavrdude` using `cmocka`.

yegorich, 2022-01-13:
I'll cook up a PR and then we can discuss it in detail.

mariusgreuel, 2022-01-13:
> I'd like to create some tests for libavrdude

Ah, that is what you have in mind.

I'm sold! It just happens that I am a huge fan of automated testing, hardware virtualization, etc. I am stopping here before I get carried away. Anyways, I think a complex project such as AVRDUDE could really use some tests. I am looking forward to see what you come up with!

yegorich, 2022-01-15:
@mariusgreuel so far, I have only tried cmocka. Do you have any preferences concerning the test framework?

mariusgreuel, 2022-01-15:
I never tried cmocka. I tried http://cunit.sourceforge.net and a few others, but found not enough value to actually use it.

Professionally, I ended up cooking up a new framework, however, the c-unit test stuff (asserts, mocks, reports) makes up a very small percentage of the whole framework.

Personally, I like the light-weight frameworks that are header-file-only, as they do not force any structure upon me, such as providing a new build system. Also, IMHO, some go overboard and do pre-processing, or nasty linker hacks that only work with GCC. I like GoogleTest, which also works well when testing C - to me it provides a lot of value.

My only requirement would be that it integrates well with the existing code and build system. Other than that, fell free to use what you are comfortable with.

Not sure what kind of testing you are planning, but you may cook up your own testing infrastructure. The unit test stuff is probably the smallest part of it (creating an array of tests and a few assert macros is really not much work).

Perhaps you have already some ideas, but here is some food for thought: I suggested to Jörg once to do some I/O level testing, i.e. implement the transport layer, and perhaps record and compare traces. A more advanced approach I had in mind would be to virtualize the hardware, such as transport layer (serial/USB, ...), programmer-protocol (stk500, ...), programming-protocol (spi, tpi, ...), and the AVRs memory (attiny, atmega, ...). Then you could do permutations on the protocols and parts, and really shake the bugs out ;-).

mariusgreuel, 2022-01-15:
One thing that comes to mind: The smallest "C unit test framework" I have seen was written by a colleague: 5 lines of makefile code that compiles and runs an executable, while piping the stdout into a file. Then it does a "diff -w test_actual.txt test_expected.txt". This simple scheme works amazingly well, plus it allows you do format the output the way you need it (hex dumps, bitmaps with named bits, etc.). You simply use a graphical diff to see the differences if diff fails.

If you are undecided, this is what I would start with.

PR #852 (2022-01-27): CMake: split configuration in two files
The main `CMakeLists.txt` file in the project's root directory takes
care of the main project settings like project name and version,
handling the options, finding dependencies, etc.

The `src/CMakeLists.txt` handles options that are necessary to build
libavrdude library and avrdude binary.

Fixes #830
closed 2022-01-27 by dl8dtl