avrdudes/avrdude #831: Port Linux sysfs GPIO feature to libgpiod

yegorich, 2022-01-13:
According to [libgpiod](https://git.kernel.org/pub/scm/libs/libgpiod/libgpiod.git/about/) description, sysfs interface is deprecated since 4.8. Hence, to enable compatibility with future kernel versions porting to `libgpiod` is required.

wutje, 2022-02-06:
Note that the linux_spi code already has the GPIO v2 infrastructure implemented.
See init code starting here
https://github.com/avrdudes/avrdude/blob/1c0b70da899628f1cad90e21cba8463a59a4744f/src/linuxspi.c#L176
and actual usage starting here:
https://github.com/avrdudes/avrdude/blob/1c0b70da899628f1cad90e21cba8463a59a4744f/src/linuxspi.c#L110

So instead of a new dependency one could extract this code (init and set GPIO) and reuse it for linux_gpio.

Just my 2 cents.

mcuee, 2022-07-11:
From here.
* https://github.com/avrdudes/avrdude/pull/917#issuecomment-1179769709

> > Moving to libgpiod may be the ultimate solution. #831
> 
> Absolutely! Now that libgpiod is a part of the linux kernel, it looks like libgpiod is available under /linux/gpio.h, which is how `linuxspi` interacts with the GPIO it uses as reset pin.
> 
> But I have no idea how to port the existing linuxgpio code over to libgpiod. Maybe the person who write linuxgpio in the first place would be interested to look into it?

Hopefully a developer will look into this.

OpenOCD libgpiod codes can be a potential reference. They have both libgpiod and sysfs based codes in the repo.
https://github.com/openocd-org/openocd/blob/master/src/jtag/drivers/linuxgpiod.c
https://github.com/openocd-org/openocd/blob/master/src/jtag/drivers/sysfsgpio.c

PR #1299 (2023-03-21): Add optional libgpiod support to linuxgpio
This PR updates the linuxgpio backend to try both the old sysfs interface and the new (since Linux 4.8, so not *that* new) libgpiod interface.
* If libgpiod is not available at build time, it builds only the old sysfs support.
* If libgpiod *is* available at build time, the linuxgpio driver will support both sysfs and libgpiod.
* At runtime, linuxgpio will try the libgpiod interface first, and fall back to sysfs if it doesnt work.

I tested this on a Raspberry Pi 4 (running Debian Bookworm arm64), flashing a little attiny2313.  I used the same config  [... cut]
closed 2023-03-21 by stefanrueger