avrdudes/avrdude #833: make build issues

MCUdude, 2022-01-13:
I just found out that I'm not able to build using make anymore doe to a recent change in configure.ac. I don't know what's causing it, but when there's no avrdude.conf and no avrdude.conf.tmp file present, I'm getting this error when running `make all`

```
$ make all
/Library/Developer/CommandLineTools/usr/bin/make  all-recursive
Making all in .
make[2]: *** No rule to make target `avrdude.conf', needed by `all-am'.  Stop.
make[1]: *** [all-recursive] Error 1
make: *** [all] Error 2
```

I believe fc54ef5e59ee2947b8b48adccf4d2e89b519aefe is what's causing the issue.

If I revert back to commit f29b2a283a921b067aa9804feaabadf8eddb64c8, it compiles fine and generates a working avrdude.conf file.

dl8dtl, 2022-01-13:
Strange because that commit didn't touch anything related to the creation of `avrdude.conf`.
However, that entire `avrdude.conf` handling used to be a hack merely.

dl8dtl, 2022-01-13:
Works for me (on MacOS – guess that's what you are interested in).
But well, keep in mind, `avrdude.conf` is *not* generated during `make all`, it's derived from `avrdude.conf.in` by `configure`.
So you have to re-run `configure` it seems.

closed 2022-01-13 by MCUdude

mariusgreuel, 2022-01-13:
@MCUdude : Speaking of building on macOS: Would you mind trying the new CMake project? The procedure can be found in the wiki at https://github.com/avrdudes/avrdude/wiki/Building-AVRDUDE-for-macOS

I wonder if the procedure is correct, and I do not have a Mac to try it. Any feedback would be much appreciated. Feel free to edit the page directly if you find any issues.

dl8dtl, 2022-01-13:
> I wonder if the procedure is correct, and I do not have a Mac to try it.

Works for me (on MacOS 12.0.1).

mariusgreuel, 2022-01-13:
Cool, thank you!

MCUdude, 2022-01-14:
@mariusgreuel it does work on my Mac too (MacOS 10.14), but the install instructions lack a `brew` installation procedure. Also, I'm pretty sure `git` comes pre-installed, so that doesn't need to be installed by brew or macports

dl8dtl, 2022-01-14:
Well, it's a Wiki, just go ahead, and update it. ;-)

You're right, just checked it, git is available from the base system for me as well.

MCUdude, 2022-01-14:
> Well, it's a Wiki, just go ahead, and update it. ;-)

Will do!

> You're right, just checked it, git is available from the base system for me as well.

How about the other packages? I'm not currently at home, and I'm not sure all of them really need to be installed? For instance, I _think_ `bison` is already present? I'm not sure. I haven't formatted my Mac in years, and I have all sorts of programs like these already installed, so my assumptions may be wrong.

dl8dtl, 2022-01-14:
Yeah right, `bison` is also there by default (and wouldn't even be needed, as `yacc` will do as well).

dl8dtl, 2022-01-14:
… and `flex` is already around as well (no surprise, has already been there when they forked the userland from FreeBSD).