avrdudes/avrdude #840: Rework Users Guide

mariusgreuel, 2022-01-15:
We are planning on modernizing the AVRDUDE documentation.  Currently, the repo contains two documents, the Un*x man page file `src/avrdude.1`, and a manual in Texinfo format `src/doc/avrdude.texi` that can be used for HTML and PDF generation. Texinfo is an old format that should probably be replaced by a modern text format such as Markdown or reStructuredText.

We have also the repos [avrdude README.md](https://github.com/avrdudes/avrdude), and some [developer documents in the avrdude Wiki](https://github.com/avrdudes/avrdude/wiki) - just mentioning this, I am not suggesting to merge that with the documentation.

We would like to have a man page, and a manual as well. Ideally, though not a requirement, it would be possible to generate the man page out of a single source. First and foremost it should be simple to understand and maintain.

There are various options that are up for discussion, such as GitHub Wiki, GitHub Pages, readthedocs.io, and many more.

For testing purposes, I ran the [texi document through a rst converter](https://github.com/mariusgreuel/avrdude/tree/docs) and [generated a readthedocs page](https://avrdude.readthedocs.io/).

ElectronicsArchiver, 2022-01-15:
The most easy to use variant would be markdown + github pages I'd say.

Simply have all docs in `/docs/` or `/documentation/` 
with subfolders for the different topics.

I would also include the github wiki, considering 
the mentioned problems with committing to it as
an 'outsider' compared to normal repository commits.

ElectronicsArchiver, 2022-01-15:
On that note, it would be beneficial to use the `project` 
tab  to track the progress on this documention refactoring.

ElectronicsArchiver, 2022-01-15:
An additional strategy to consider is creating a separate
repository for the documentation. Consider that the 
`avrdudes` organization is almost solely focused on, well,
**AVRDUDE**, it won't have a problem adding another
repository (Considering some orgs have thousands).

This should be considered thinking about the amount
of documentation that **AVRDUDE** currently has,
the differrent types as well as the amount of commits
that will be done in a refactoring like this.

Going with a soft transition, simply starting from zero
and slowly converting the information from this repo
over instead of directly moving everthing at once,
would also help keep track and do it properly.

ElectronicsArchiver, 2022-01-15:
With the separate repository way the following
folder structure could be utilized:

avrdudes `Organization`
 ↳Documentation `Repository`
  ↳Pages `Github Pages In Markdown Format`
   ↳README.md `Documentation Home ( User )`
  ↳Resources `Images,..`
  ↳README.md `Documentation Home ( Editor )`

I would also like to suggest to use visual media
wherever applicable, after all it's the best medium
to convey information with.

dl8dtl, 2022-01-16:
> The most easy to use variant would be markdown + github pages I'd say.

Curious, how would that get as a man page?

ElectronicsArchiver, 2022-01-16:
*Linux Man Pages*, it won't, nor do I think it should.

While it would probably be helpful to store the man
pages data in the same location as the proper docs,
writing one version that satisfies both perfectly, is
close to impossible, considering the vastly different
formats of the two.

There may be some markdown -> man pages
conversion tools, however, I think it would be 
best to just write a dedicated version that
tries to be as short / direct as possible instead.

Like I mentioned before, considering the separate
repository approach with a soft conversion would
make this a simple 'project on trial' that wouldn't
hinder the main development effort, AVRDUDE.

mcuee, 2023-03-26:
@mariusgreuel and @dl8dtl 

Is this still an issue? Now we can generate documentation using CMake, the man page and HTML/PDF documentations are all fine and hosted by github.
https://avrdudes.github.io/avrdude/

mcuee, 2023-06-03:
Ping @dl8dtl and @mariusgreuel. Thanks.

stefanrueger, 2023-06-03:
I am happy with the man pages on the one hand and with a pdf and html pages created from texinfo on the other hand.

True, changes to the documentation needs to be done in two places, but then the two sources are of different granularity and for different purpose, so I am cool with that.

mcuee, 2023-06-05:
I will close this for now. We can re-open if necessary.

closed 2023-06-05 by mcuee