avrdudes/avrdude #856: Compiler warnings

MCUdude, 2022-01-29:
FYI, I'm getting a few compiler warnings after #850 was merged. Would be great if this could be fixed. Perhaps the CI could catch something like this in the furure?

```
 CCLD     avrdude
ld: warning: directory not found for option '-FCoreFoundation'
  CC       libavrdude_la-ft245r.lo
  CC       libavrdude_la-stk500.lo
  CC       libavrdude_la-usbasp.lo
usbasp.c:71:9: warning: type specifier missing, defaults to 'int' [-Wimplicit-int]
        static msg[30];
        ~~~~~~ ^
usbasp.c:122:12: warning: incompatible pointer types passing 'int [30]' to parameter of type 'char *'
      [-Wincompatible-pointer-types]
                snprintf(msg, sizeof msg, "Unknown libusb error: %d", result);
                         ^~~
/Library/Developer/CommandLineTools/SDKs/MacOSX10.14.sdk/usr/include/secure/_stdio.h:57:29: note: expanded from macro 'snprintf'
  __builtin___snprintf_chk (str, len, 0, __darwin_obsz(str), __VA_ARGS__)
                            ^~~
usbasp.c:123:10: warning: incompatible pointer types returning 'int [30]' from a function with result type 'const char *'
      [-Wincompatible-pointer-types]
                return msg;
                       ^~~
3 warnings generated.

```

dl8dtl, 2022-01-29:
> ld: warning: directory not found for option '-FCoreFoundation'

I don't see that one though.

Please review PR #857 - I think it fixes the issue.

dl8dtl, 2022-01-30:
Closed by PR #857

closed 2022-01-30 by dl8dtl