avrdudes/avrdude #864: [Question] How to install/configure 'libelf', 'libusb', 'libhidapi', 'libftdi' on Windows using Visual Studio?

water5, 2022-02-04:
I using Visual Studio on Windows to build AVRDUDE, this workflows `.yml` file not include instructions for install / configure `libelf`, `libusb`, `libhidapi`, `libftdi` for Visual Studio, and how to use this `.yml` file? Just view only?

[https://github.com/avrdudes/avrdude/blob/main/.github/workflows/build.yml](https://github.com/avrdudes/avrdude/blob/main/.github/workflows/build.yml)

mariusgreuel, 2022-02-05:
I suppose you refer to the build procedure (Building AVRDUDE for Windows using Visual Studio)[https://github.com/avrdudes/avrdude/wiki/Building-AVRDUDE-for-Windows-using-Visual-Studio].

> Just view only?

Yes. The build.yml file is a GitHub Actions workflow that cannot be used in Visual Studio. It does, however, contain the build matrix for AVRDUDE. The Wiki only contains only a procedure, which may be incorrect/incomplete/out-of-date. So, if you are interested in the exact set of CMake options being used to build AVRDUDE, you can refer to the GitHub action.

water5, 2022-02-05:
>I suppose you refer to the build procedure (Building AVRDUDE for Windows using Visual Studio)[https://github.com/avrdudes/avrdude/wiki/Building-AVRDUDE-for-Windows-using-Visual-Studio].

I already read this, but not more detail for install / configure `libelf`, `libusb`, `libhidapi`, `libftdi` for `Windows`, only mention: 
>https://github.com/avrdudes/avrdude/blob/main/.github/workflows/build.yml. See the msvc job for details.

When I open AVRDUDE project, rise error below:
>Error		CMake Error at C:/Program Files (x86)/Microsoft Visual Studio/2019/Community/Common7/IDE/CommonExtensions/Microsoft/CMake/CMake/share/cmake-3.20/Modules/ExternalProject.cmake:2633 (message):
  error: `could not find git for clone of libelf-populate`		C:/Program Files (x86)/Microsoft Visual Studio/2019/Community/Common7/IDE/CommonExtensions/Microsoft/CMake/CMake/share/cmake-3.20/Modules/ExternalProject.cmake	2633	

>Error		CMake Error at C:/Program Files (x86)/Microsoft Visual Studio/2019/Community/Common7/IDE/CommonExtensions/Microsoft/CMake/CMake/share/cmake-3.20/Modules/FetchContent.cmake:1000 (message):
  `CMake step for libelf failed`: 1	CrossCompileExample	C:/Program Files (x86)/Microsoft Visual Studio/2019/Community/Common7/IDE/CommonExtensions/Microsoft/CMake/CMake/share/cmake-3.20/Modules/FetchContent.cmake	1000

mariusgreuel, 2022-02-05:
> error: could not find git

It's probably what the message says. I have no idea about all the different ways CMake can find Git on Windows, but one way is probably though the PATH variable. If you do have Git installed, make sure you can access Git through the Windows PATH variable.

Also, I found this:
https://developercommunity.visualstudio.com/t/bundled-git-installation-not-usable-with-cmake-ope/1144595

If you did not install Git (i.e. you use the **bundled** Git with VS) it may not work. If this is the case, you should provide feedback to Microsoft.

closed 2022-02-12 by mariusgreuel