avrdudes/avrdude #865: Unknown CMake command "FLEX_TARGET" and how I fixed it.

merlinblack, 2022-02-05:
Fedora Linux 35
Cmake 3.22.2

I ran into the error "Unknown CMake command "FLEX_TARGET"" when following the Linux instructions.

I fixed the issue by adding the following to CMakeLists.txt:
```
project(avrdude)
cmake_minimum_required(VERSION 3.5)
find_package(BISON)
find_package(FLEX)
```
The first two lines are not strictly required but remove some warnings.

I'll make a PR once I have avrdude up and running.

Also for reference, here is a Fedora line to install the pre-reqs besides cmake and gcc etc:
```
sudo dnf install flex bison elfutils-libelf-devel libusb-devel libftdi-devel libhid-devel
```

mariusgreuel, 2022-02-05:
The CMake project was recently reworked. Now, you need to run CMake from the avrdude folder, not from the src folder. I will update the Wiki.

merlinblack, 2022-02-05:
Ah ha! Cool.
 
Also my pre-reqs installing was a bit off - I needed hidapi-devel instead of libhid-devel

Almost there - for some reason the avrdude.conf file is not being generated.

merlinblack, 2022-02-05:
This is a bit weird too - it's compiling but without USB support - even when Cmake finds the libs.
```
-- Configuration summary:
-- ----------------------
-- DO HAVE    libelf
-- DO HAVE    libusb
-- DO HAVE    libusb_1_0
-- DO HAVE    libhidapi
-- DON'T HAVE libftdi
-- DO HAVE    libftdi1
-- DISABLED   doc
-- DISABLED   parport
-- DISABLED   linuxgpio
-- DISABLED   linuxspi
-- ----------------------
-- Configuring done
-- Generating done
-- Build files have been written to: /home/nigel/prog/avr/avrdude/build_linux
[  1%] [FLEX][Parser] Building scanner with flex 2.6.4
[  3%] [BISON][Parser] Building parser with bison 3.7.6
[  4%] Building C object src/CMakeFiles/libavrdude.dir/arduino.c.o
[  6%] Building C object src/CMakeFiles/libavrdude.dir/avr.c.o
[  8%] Building C object src/CMakeFiles/libavrdude.dir/avr910.c.o
[  9%] Building C object src/CMakeFiles/libavrdude.dir/avrftdi.c.o
In file included from /home/nigel/prog/avr/avrdude/src/avrftdi.c:41:
/home/nigel/prog/avr/avrdude/src/avrftdi_private.h:21:2: warning: #warning No libftdi or libusb support. Install libftdi1/libusb-1.0 or libftdi/libusb and run configure/make again. [-Wcpp]
   21 | #warning No libftdi or libusb support. Install libftdi1/libusb-1.0 or libftdi/libusb and run configure/make again.
      |  ^~~~~~~
[ 11%] Building C object src/CMakeFiles/libavrdude.dir/avrftdi_tpi.c.o
In file included from /home/nigel/prog/avr/avrdude/src/avrftdi_tpi.c:14:
/home/nigel/prog/avr/avrdude/src/avrftdi_private.h:21:2: warning: #warning No libftdi or libusb support. Install libftdi1/libusb-1.0 or libftdi/libusb and run configure/make again. [-Wcpp]
   21 | #warning No libftdi or libusb support. Install libftdi1/libusb-1.0 or libftdi/libusb and run configure/make again.
      |  ^~~
```

mariusgreuel, 2022-02-05:
> Almost there - for some reason the avrdude.conf file is not being generated.

Yes, there is PR #858 that is supposed to fix this.

mariusgreuel, 2022-02-05:
> This is a bit weird too - it's compiling but without USB support - even when Cmake finds the libs

That is indeed odd. Its actually the FTDI programmer that issues this message, but the HAVE_LIBFTDI1 and HAVE_LIBUSB_1_0 #defines should be defined, as the configuration summary indicates. Perhaps CMake got confused when you installed the libraries after it ran? Can you remove the `build_linux` folder and try the whole thing again? In build_linux/src/ac_cfg.h, what are the  two #defines present?

merlinblack, 2022-02-07:
I pulled the latest master as off a few minutes ago - and it now seems to be working as far as compiling in USB support. :-)

It's not talking properly to my UPDI programmer but that's for another day/issue. Thanks for your help!

closed 2022-02-07 by merlinblack