avrdudes/avrdude #870: USB support not being compiled in (apparently).

merlinblack, 2022-02-11:
Tried building on Fedora 35 and also a Ubuntu 20.04 docker container.
Pulled code from main branch today.

When building on Fedora 35 and Ubuntu 20.04 with the required libraries installed - USB support seemingly does not get compiled in.

For the Ubuntu build, I followed the wiki instructions for installing the pre-requisites, and added 'libreadline-dev'. It does not compile without libreadline.

Here's the ac_cfg.h from my attempt with Ubuntu.
```
/*
 * avrdude - A Downloader/Uploader for AVR device programmers
 * Copyright (C) 2021 Marius Greuel
 *
 * This program is free software; you can redistribute it and/or modify
 * it under the terms of the GNU General Public License as published by
 * the Free Software Foundation; either version 2 of the License, or
 * (at your option) any later version.
 *
 * This program is distributed in the hope that it will be useful,
 * but WITHOUT ANY WARRANTY; without even the implied warranty of
 * MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
 * GNU General Public License for more details.
 *
 * You should have received a copy of the GNU General Public License
 * along with this program. If not, see <http://www.gnu.org/licenses/>.
 */

#ifdef _MSC_VER
#include "msvc/msvc_compat.h"
#endif

#define VERSION ""

/* Options */

/* Linux sysfs GPIO support enabled */
/* #undef HAVE_LINUXGPIO */

/* Linux SPI support enabled */
/* #undef HAVE_LINUXSPI */

/* Parallel port access enabled */
/* #undef HAVE_PARPORT */

/* ----- Functions ----- */

/* Define if lex/flex has yylex_destroy */
#define HAVE_YYLEX_DESTROY 1

/* Define if ftdi1 has 'ftdi_tcioflush' function. */
/* #undef HAVE_FTDI_TCIOFLUSH */

/* ----- Libraries and Headers ----- */

/* Define to 1 if the system has the type `uint_t'. */
/* #undef HAVE_UINT_T */

/* Define to 1 if the system has the type `ulong_t'. */
/* #undef HAVE_ULONG_T */

/* Define if ELF support is enabled via libelf */
/* #undef HAVE_LIBELF */
#define HAVE_LIBELF

/* Define to 1 if you have the <libelf.h> header file. */
/* #undef HAVE_LIBELF_H */
#define HAVE_LIBELF_H 1

/* Define to 1 if you have the <libelf/libelf.h> header file. */
/* #undef HAVE_LIBELF_LIBELF_H */

/* Define if USB support is enabled via libusb */
/* #undef HAVE_LIBUSB */

/* Define if USB support is enabled via a libusb-1.0 compatible libusb */
/* #undef HAVE_LIBUSB_1_0 */
#define HAVE_LIBUSB_1_0

/* Define if USB support is enabled via a libusb-win32 compatible libusb */
/* #undef HAVE_LIBUSB_WIN32 */

/* Define to 1 if you have the <usb.h> header file. */
/* #undef HAVE_USB_H */
#define HAVE_USB_H 1

/* Define to 1 if you have the <lusb0_usb.h> header file. */
/* #undef HAVE_LUSB0_USB_H */

/* Define to 1 if you have the <libusb.h> header file. */
/* #undef HAVE_LIBUSB_H */

/* Define to 1 if you have the <libusb-1.0/libusb.h> header file. */
/* #undef HAVE_LIBUSB_1_0_LIBUSB_H */
#define HAVE_LIBUSB_1_0_LIBUSB_H 1

/* Define if HID support is enabled via the Win32 DDK */
/* #undef HAVE_LIBHID */
#define HAVE_LIBHID

/* Define if HID support is enabled via libhidapi */
/* #undef HAVE_LIBHIDAPI */
#define HAVE_LIBHIDAPI

/* Define to 1 if you have the <hidapi/hidapi.h> header file. */
/* #undef HAVE_HIDAPI_HIDAPI_H */
#define HAVE_HIDAPI_HIDAPI_H 1

/* Define if FTDI support is enabled via libftdi */
/* #undef HAVE_LIBFTDI */

/* Define if FTDI support is enabled via libftdi1 */
/* #undef HAVE_LIBFTDI1 */
#define HAVE_LIBFTDI1

/* Define if libftdi supports FT232H, libftdi version >= 0.20 */
/* #undef HAVE_LIBFTDI_TYPE_232H */
#define HAVE_LIBFTDI_TYPE_232H

/* Define to 1 if you have the `readline' library (-lreadline). */
/* #undef HAVE_LIBREADLINE */
#define HAVE_LIBREADLINE 1
```
Here's the result when trying to use the new binary:
```
./avrdude -v -cusbtiny -patmega328p

avrdude: Version 
         Copyright (c) Brian Dean, http://www.bdmicro.com/
         Copyright (c) Joerg Wunsch

         System wide configuration file is "/home/nigel/prog/avr/avrdude/build_linux/src/avrdude.conf"
         User configuration file is "/home/nigel/.avrduderc"
         User configuration file does not exist or is not a regular file, skipping

         Using Port                    : usb
         Using Programmer              : usbtiny
avrdude: error: no usb support. Please compile again with libusb installed.
avrdude: opening programmer "usbtiny" on port "usb" failed

avrdude done.  Thank you.
```

I get the same result when compiling either from Fedora 35 or with a Ubuntu container.

merlinblack, 2022-02-11:
I think the following is most likely related too:
```
./avrdude -v -cxplainedmini_updi -patmega4809

avrdude: Version 
         Copyright (c) Brian Dean, http://www.bdmicro.com/
         Copyright (c) Joerg Wunsch

         System wide configuration file is "/home/nigel/prog/avr/avrdude/build_linux/src/avrdude.conf"
         User configuration file is "/home/nigel/.avrduderc"
         User configuration file does not exist or is not a regular file, skipping

         Using Port                    : usb
         Using Programmer              : xplainedmini_updi
avrdude: Found CMSIS-DAP compliant device, using EDBG protocol
         AVR Part                      : ATmega4809
         RESET disposition             : dedicated
         RETRY pulse                   : SCK
         Serial program mode           : yes
         Parallel program mode         : yes
         Memory Detail                 :

                                           Block Poll               Page                       Polled
           Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           signature               0     0     0    0 no          3    1      0     0     0 0x00 0x00
           prodsig                 0     0     0    0 no         61   61      0     0     0 0x00 0x00
           sernum                  0     0     0    0 no         10    1      0     0     0 0x00 0x00
           osccal16                0     0     0    0 no          2    1      0     0     0 0x00 0x00
           osccal20                0     0     0    0 no          2    1      0     0     0 0x00 0x00
           tempsense               0     0     0    0 no          2    1      0     0     0 0x00 0x00
           osc16err                0     0     0    0 no          2    1      0     0     0 0x00 0x00
           osc20err                0     0     0    0 no          2    1      0     0     0 0x00 0x00
           fuses                   0     0     0    0 no          9   10      0     0     0 0x00 0x00
           fuse0                   0     0     0    0 no          1    1      0     0     0 0x00 0x00
           fuse1                   0     0     0    0 no          1    1      0     0     0 0x00 0x00
           fuse2                   0     0     0    0 no          1    1      0     0     0 0x00 0x00
           fuse4                   0     0     0    0 no          1    1      0     0     0 0x00 0x00
           fuse5                   0     0     0    0 no          1    1      0     0     0 0x00 0x00
           fuse6                   0     0     0    0 no          1    1      0     0     0 0x00 0x00
           fuse7                   0     0     0    0 no          1    1      0     0     0 0x00 0x00
           fuse8                   0     0     0    0 no          1    1      0     0     0 0x00 0x00
           lock                    0     0     0    0 no          1    1      0     0     0 0x00 0x00
           data                    0     0     0    0 no          0    1      0     0     0 0x00 0x00
           usersig                 0     0     0    0 no         64   64      0     0     0 0x00 0x00
           flash                   0     0     0    0 no      49152  128      0     0     0 0x00 0x00
           eeprom                  0     0     0    0 no        256   64      0     0     0 0x00 0x00

         Programmer Type : JTAGICE3_UPDI
         Description     : Atmel AVR XplainedMini in UPDI mode
         ICE hardware version: 0
         ICE firmware version: 1.19 (rel. 57)
         Serial number   : MICROUPDIPROGRAMMERX
         Vtarget         : 3.30 V
         JTAG clock megaAVR/program: 0 kHz
         JTAG clock megaAVR/debug:   0 kHz
         JTAG clock Xmega: 0 kHz
         PDI clock Xmega : 0 kHz

avrdude: Short read, read only 0 out of 64 bytes
avrdude: jtag3_edbg_recv(): Unexpected response 0x73
avrdude: retrying with external reset applied
avrdude: jtag3_edbg_send(): Unexpected response 0x81, 0x11
avrdude: jtag3_edbg_recv(): Unexpected response 0x80
avrdude: retrying with external reset applied
avrdude: JTAGEN fuse disabled?
avrdude: initialization failed, rc=-1
         Double check connections and try again, or use -F to override
         this check.

avrdude: jtag3_edbg_send(): Unexpected response 0x81, 0x11
avrdude: jtag3_edbg_recv(): Unexpected response 0x80
avrdude: jtag3_edbg_send(): Unexpected response 0x81, 0x11
avrdude: jtag3_edbg_recv(): Unexpected response 0x80

avrdude done.  Thank you.
```

dl8dtl, 2022-02-11:
> /* Define if USB support is enabled via libusb */
> /* #undef HAVE_LIBUSB */

I guess that's the issue. You have libusb-1.0 (headers) but not libusb-0.1. Essentially, both are required.

On an (old) Ubuntu, my configure (or CMake configuration) output looks like:

```
-- Configuration summary:
-- ----------------------
-- DO HAVE    libelf
-- DO HAVE    libusb
-- DO HAVE    libusb_1_0
-- DO HAVE    libhidapi
-- DO HAVE    libftdi
-- DON'T HAVE libftdi1
-- DISABLED   doc
-- DISABLED   parport
-- DISABLED   linuxgpio
-- DISABLED   linuxspi
-- ----------------------
```
With that, USBtiny is working:
```
% ./build_linux/src/avrdude -v -cusbtiny -patmega16

avrdude: Version 6.99-20220210 (452f673)
         Copyright (c) Brian Dean, http://www.bdmicro.com/
         Copyright (c) Joerg Wunsch

         System wide configuration file is "/home/j/src/avrdude/build_linux/src/avrdude.conf"
         User configuration file is "/home/j/.avrduderc"

         Using Port                    : usb
         Using Programmer              : usbtiny
avrdude: usbdev_open(): Found USBtinyISP, bus:device: 001:009
         AVR Part                      : ATmega16
         Chip Erase delay              : 9000 us
         PAGEL                         : PD7
         BS2                           : PA0
         RESET disposition             : dedicated
         RETRY pulse                   : SCK
         Serial program mode           : yes
         Parallel program mode         : yes
         Timeout                       : 200
         StabDelay                     : 100
         CmdexeDelay                   : 25
         SyncLoops                     : 32
         PollIndex                     : 3
         PollValue                     : 0x53
         Memory Detail                 :

                                           Block Poll               Page                       Polled
           Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           eeprom                  4    10   128    0 no        512    4      0  9000  9000 0xff 0xff
           flash                  33     6   128    0 yes     16384  128    128  4500  4500 0xff 0xff
           lock                    0     0     0    0 no          1    1      0  9000  9000 0x00 0x00
           lfuse                   0     0     0    0 no          1    1      0  9000  9000 0x00 0x00
           hfuse                   0     0     0    0 no          1    1      0  9000  9000 0x00 0x00
           signature               0     0     0    0 no          3    1      0     0     0 0x00 0x00
           calibration             0     0     0    0 no          4    1      0     0     0 0x00 0x00

         Programmer Type : USBtiny
         Description     : USBtiny simple USB programmer, https://learn.adafruit.com/usbtinyisp
avrdude: programmer operation not supported

avrdude: Using SCK period of 10 usec
avrdude: AVR device initialized and ready to accept instructions

[... 49 lines cut ...]
           usersig                 0     0     0    0 no         32   32      0     0     0 0x00 0x00
           flash                   0     0     0    0 no       8192   64      0     0     0 0x00 0x00
           eeprom                  0     0     0    0 no        128   32      0     0     0 0x00 0x00

         Programmer Type : JTAGICE3_UPDI
         Description     : Atmel AVR XplainedMini in UPDI mode
         ICE hardware version: 0
         ICE firmware version: 1.13 (rel. 43)
         Serial number   : ATML2658061800009235
         Vtarget         : 5.00 V
         JTAG clock megaAVR/program: 0 kHz
         JTAG clock megaAVR/debug:   0 kHz
         JTAG clock Xmega: 0 kHz
         PDI clock Xmega : 0 kHz

avrdude: Partial Family_ID returned: "tiny"
avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.10s

avrdude: Device signature = 0x1e9320 (probably t817)

avrdude done.  Thank you.
```
What surprises me a bit is that you don't see a version number at the top of the banner.

merlinblack, 2022-02-11:
My config summary on fedora looks like this:
```
-- Configuration summary:
-- ----------------------
-- DO HAVE    libelf
-- DO HAVE    libusb
-- DO HAVE    libusb_1_0
-- DO HAVE    libhidapi
-- DON'T HAVE libftdi
-- DO HAVE    libftdi1
-- DISABLED   doc
-- DISABLED   parport
-- DISABLED   linuxgpio
-- DISABLED   linuxspi
-- ----------------------
```
and the ac_cfg.h
```
﻿/*
 * avrdude - A Downloader/Uploader for AVR device programmers
 * Copyright (C) 2021 Marius Greuel
 *
 * This program is free software; you can redistribute it and/or modify
 * it under the terms of the GNU General Public License as published by
 * the Free Software Foundation; either version 2 of the License, or
 * (at your option) any later version.
 *
 * This program is distributed in the hope that it will be useful,
 * but WITHOUT ANY WARRANTY; without even the implied warranty of
 * MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
 * GNU General Public License for more details.
 *
 * You should have received a copy of the GNU General Public License
 * along with this program. If not, see <http://www.gnu.org/licenses/>.
 */

#ifdef _MSC_VER
#include "msvc/msvc_compat.h"
#endif

#define VERSION "6.99-20220210 (452f673)"

/* Options */

/* Linux sysfs GPIO support enabled */
/* #undef HAVE_LINUXGPIO */

/* Linux SPI support enabled */
/* #undef HAVE_LINUXSPI */

/* Parallel port access enabled */
/* #undef HAVE_PARPORT */

/* ----- Functions ----- */

/* Define if lex/flex has yylex_destroy */
#define HAVE_YYLEX_DESTROY 1

/* Define if ftdi1 has 'ftdi_tcioflush' function. */
#define HAVE_FTDI_TCIOFLUSH 1

/* ----- Libraries and Headers ----- */

/* Define to 1 if the system has the type `uint_t'. */
/* #undef HAVE_UINT_T */

/* Define to 1 if the system has the type `ulong_t'. */
/* #undef HAVE_ULONG_T */

/* Define if ELF support is enabled via libelf */
#define HAVE_LIBELF 1

/* Define to 1 if you have the <libelf.h> header file. */
#define HAVE_LIBELF_H 1

/* Define to 1 if you have the <libelf/libelf.h> header file. */
/* #undef HAVE_LIBELF_LIBELF_H */

/* Define if USB support is enabled via libusb */
#define HAVE_LIBUSB 1

/* Define if USB support is enabled via a libusb-1.0 compatible libusb */
#define HAVE_LIBUSB_1_0 1

/* Define if USB support is enabled via a libusb-win32 compatible libusb */
/* #undef HAVE_LIBUSB_WIN32 */

/* Define to 1 if you have the <usb.h> header file. */
#define HAVE_USB_H 1

/* Define to 1 if you have the <lusb0_usb.h> header file. */
/* #undef HAVE_LUSB0_USB_H */

/* Define to 1 if you have the <libusb.h> header file. */
/* #undef HAVE_LIBUSB_H */

/* Define to 1 if you have the <libusb-1.0/libusb.h> header file. */
#define HAVE_LIBUSB_1_0_LIBUSB_H 1

/* Define if HID support is enabled via the Win32 DDK */
#define HAVE_LIBHID 1

/* Define if HID support is enabled via libhidapi */
#define HAVE_LIBHIDAPI 1

/* Define to 1 if you have the <hidapi/hidapi.h> header file. */
#define HAVE_HIDAPI_HIDAPI_H 1

/* Define if FTDI support is enabled via libftdi */
/* #undef HAVE_LIBFTDI */

/* Define if FTDI support is enabled via libftdi1 */
#define HAVE_LIBFTDI1 1

/* Define if libftdi supports FT232H, libftdi version >= 0.20 */
#define HAVE_LIBFTDI_TYPE_232H 1

/* Define to 1 if you have the `readline' library (-lreadline). */
#define HAVE_LIBREADLINE 1
```

You mentioned the lack of version number - it's in the ac_cgf.h but does not get printed, when using the '-v' flag.
It seems we have the same version. I'm going to have a dig for the code that prints out "avrdude: error: no usb support. Please compile again with libusb installed." and see what I can see.

merlinblack, 2022-02-11:
Ah-ha! There was an 'ac_cfg.h' in the src directory that was shadowing the generated ac_cfg.h in build_linux/src.
Deleting it solved the issue, I now have USB functionality. I know the generated ac_cfg is now being used as I put an  `#error` in there to see if it was being pulled in. I'm not sure how it got to the src directory - it's not in git.

Now to figure out why my UPDI is not working. @dl8dtl Thanks for the help :-)

closed 2022-02-11 by merlinblack

mariusgreuel, 2022-02-11:
In Git, you can run a `git clean -xdf` to blow away everything that is not under source control - do `git clean -xdn` to do a dry-run first.

merlinblack, 2022-02-13:
Ah, that's a good git command to know. I did do a git status and it did not show up as a change or an untracked file. The .gitognore is set up to ignore ac_cfg.h - even when it's in the wrong place. `git clean -xdn` shows it. :-)