avrdudes/avrdude #871: AVR128DB28 attempting change a fuse setting using (otherwise working) serialUPDI adapter (2/11/2022)

wb0gaz, 2022-02-11:
SerialUPDI adapter was built using FTDI232 module and is working properly to read flash memory, write flash memory, and read fuse value with target AVR128DB28 (supply voltage 3.6VDC from USB serial adapter.)

Issue: Writing/changing fuse value (in this case, BOOTSIZE fuse at 0x1058 per avrdude.conf is to be changed from factory default value of 0x20 to user-specified value of 0x00) fails to verify (and does not change value when subsequently read on a different UPDI adapter device type.)

Command:

_>avrdude -c serialupdi -p avr128db28 -P com7 -U bootsize:w:0x00:m_

Response:

avrdude: UPDI link initialization OK
avrdude: NVM type 2: 24-bit, word oriented write
avrdude: Entering NVM programming mode
avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.05s

avrdude: Device signature = 0x1e970e (probably avr128db28)
avrdude: reading input file "0x00"
avrdude: writing bootsize (1 bytes):

Writing | ################################################## | 100% 0.03s

avrdude: 1 bytes of bootsize written
avrdude: verifying bootsize memory against 0x00:

Reading | ################################################## | 100% 0.01s

avrdude: verification error, first mismatch at byte 0x0000
0x20 != 0x00
avrdude: verification error; content mismatch

avrdude: safemode: Fuses OK (E:FF, H:FF, L:FF)
avrdude: Leaving NVM programming mode

avrdude done. Thank you.

MCUdude, 2022-02-11:
Have you cloned the upstream Avrdude repo recently? We have been doing some fuse related work lately (proper support for fuse name aliases).

Can you make sure you have the latest upstream version, re-build and run the following command (with the -v flag)? I'd like to see the output of the table that's being generated.

```
avrdude -c serialupdi -v -p avr128db28 -P com7 -U bootsize:w:0x00:m
```

wb0gaz, 2022-02-11:
MCUdude --- thank you for the quick reply!

I obtained the executable from:

https://github.com/mariusgreuel/avrdude/releases/tag/v7.0-rc1

When I encountered the problem, I opened the same issue at:

https://github.com/mariusgreuel/avrdude/issues/8

however I was advised to move the issue here (so that version was closed and this version was open.)

I'm not equipped to re-build from source at this point (I'm functioning as an end-user on Windows 10 64-bit), however, I'd be glad to test given access to a binary/installation.

Here's a print with -v using 6.99 version (the version I obtained above about two days ago; it's current as of there as the last update was 9 days ago.) I modified your test command to try to write 0x20 to fuse (as writing 0x00 didn't produce an error, given that I had changed the fuse value using a borrowed programmer.)

I'm an absolute beginner with github, so I am sorry if the formatting below is not ideal. If there's a better way to provide needed information, please let me know, I'll be glad to follow any instructions given.

###

_>avrdude -c serialupdi -v -p avr128db28 -P com7 -U bootsize:w:0x20:m_

avrdude: Version 6.99
         Copyright (c) Brian Dean, http://www.bdmicro.com/
         Copyright (c) Joerg Wunsch

         System wide configuration file is "C:/Users/user/Desktop/avrdude-v7.0-rc1-windows-x64/avrdude.conf"

         Using Port                    : com7
         Using Programmer              : serialupdi
avrdude: Found device at port 'COM7'
         AVR Part                      : AVR128DB28
         RESET disposition             : dedicated
         RETRY pulse                   : SCK
         Serial program mode           : yes
         Parallel program mode         : yes
         Memory Detail                 :

                                           Block Poll               Page                       Polled
           Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           signature               0     0     0    0 no          3    1      0     0     0 0x00 0x00
           prodsig                 0     0     0    0 no        125  125      0     0     0 0x00 0x00
           tempsense               0     0     0    0 no          2    1      0     0     0 0x00 0x00
           sernum                  0     0     0    0 no         16    1      0     0     0 0x00 0x00
           fuses                   0     0     0    0 no          9   16      0     0     0 0x00 0x00
           fuse0                   0     0     0    0 no          1    1      0     0     0 0x00 0x00
           wdtcfg                  0     0     0    0 no          1    1      0     0     0 0x00 0x00
           fuse1                   0     0     0    0 no          1    1      0     0     0 0x00 0x00
           bodcfg                  0     0     0    0 no          1    1      0     0     0 0x00 0x00
           fuse2                   0     0     0    0 no          1    1      0     0     0 0x00 0x00
           osccfg                  0     0     0    0 no          1    1      0     0     0 0x00 0x00
           fuse4                   0     0     0    0 no          1    1      0     0     0 0x00 0x00
           tcd0cfg                 0     0     0    0 no          1    1      0     0     0 0x00 0x00
           fuse5                   0     0     0    0 no          1    1      0     0     0 0x00 0x00
           syscfg0                 0     0     0    0 no          1    1      0     0     0 0x00 0x00
           fuse6                   0     0     0    0 no          1    1      0     0     0 0x00 0x00
           syscfg1                 0     0     0    0 no          1    1      0     0     0 0x00 0x00
           fuse7                   0     0     0    0 no          1    1      0     0     0 0x00 0x00
           codesize                0     0     0    0 no          1    1      0     0     0 0x00 0x00
           append                  0     0     0    0 no          1    1      0     0     0 0x00 0x00
           fuse8                   0     0     0    0 no          1    1      0     0     0 0x00 0x00
           bootsize                0     0     0    0 no          1    1      0     0     0 0x00 0x00
           bootend                 0     0     0    0 no          1    1      0     0     0 0x00 0x00
           lock                    0     0     0    0 no          4    1      0     0     0 0x00 0x00
           userrow                 0     0     0    0 no         32   32      0     0     0 0x00 0x00
           data                    0     0     0    0 no          0    1      0     0     0 0x00 0x00
           flash                   0     0     0    0 no     131072  512      0     0     0 0x00 0x00
           eeprom                  0     0     0    0 no        512    1      0     0     0 0x00 0x00

         Programmer Type : serialupdi
         Description     : SerialUPDI

avrdude: UPDI link initialization OK
avrdude: NVM type 2: 24-bit, word oriented write
avrdude: Entering NVM programming mode
avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.05s

avrdude: Device signature = 0x1e970e (probably avr128db28)
avrdude: reading input file "0x20"
avrdude: writing bootsize (1 bytes):

Writing | ################################################## | 100% 0.03s

avrdude: 1 bytes of bootsize written
avrdude: verifying bootsize memory against 0x20:

Reading | ################################################## | 100% 0.01s

avrdude: verification error, first mismatch at byte 0x0000
         0x00 != 0x20
avrdude: verification error; content mismatch

avrdude: safemode: Fuses OK (E:FF, H:FF, L:FF)
avrdude: Leaving NVM programming mode

avrdude done.  Thank you.

MCUdude, 2022-02-11:
> I'm not equipped to re-build from source at this point (I'm functioning as an end-user on Windows 10 64-bit), however, I'd be glad to test given access to a binary/installation.

Well, I only know how to build for macOS, so I can't help you there. But you don't have a binary that's built of the latest version. I don't own any AVR-DB hardware, but I can confirm that the upstream version works flawlessly with AVR-DAs.

Can you try running the following command? `fues8` is the original name for `bootsize`, and may possibly work.

```
avrdude -c serialupdi -v -p avr128db28 -P com7 -U fuse8:w:0x20:m
```

wb0gaz, 2022-02-11:
Thank you, MCUdude.

I have verified that using the argument fuse8 works properly (I have changed the fuse from 0x20 to 0x00 and back to 0x20 and back to 0x00, each time inspecting memory location (using a simple program I wrote to read/write memory locations) 0x1058 where the fuse data is stored, and seeing it change. 

I then repeated the test using bootsize as the name, and each time I tried to change the value, I got the "verification error; content mismatch" problem.

I'm happy to do further testing, but I can confirm version 6.99 works properly provided I use the fuse8 identifier rather than bootsize.

mariusgreuel, 2022-02-11:
@wb0gaz You may want to try the latest CI builds from here: https://github.com/avrdudes/avrdude/wiki/Getting-Nightly-Builds-for-AVRDUDE, these are a little newer than the build I did.

MCUdude, 2022-02-11:
Great to heat `fuse8` works for you! The issue has most likely already been fixed in the upstream version in #868

wb0gaz, 2022-02-11:
Thanks mariusgreuel!

I was not aware of nightly builds (I am still total beginner at Github world). I now see how to reach those versions.

As there is not an operational problem now (using "fuse8" tag), I will be happy to test if that would help, otherwise I will wait for your next release version and proceed with that.

Thanks again!

Dave

(should I now use the "close" option?)

MCUdude, 2022-02-11:
> (should I now use the "close" option?)

I suggest we wait until someone has tested that it actually works on AVR-DB hardware with a SerialUPDI programmer

dl8dtl, 2022-02-11:
I've got a Curiosity Nano board with an AVR128DB48. Modifying `fuse8` works there, but obviously, it's not SerialUPDI but the onboard EDBG.

dbuchwald, 2022-02-17:
> I've got a Curiosity Nano board with an AVR128DB48. Modifying `fuse8` works there, but obviously, it's not SerialUPDI but the onboard EDBG.

I could test it with my breakout board for AVR128DB64 if you guys tell me exactly what I'm supposed to test - like the exact procedure.

MCUdude, 2022-02-17:
Does this work?

```
# Write bootsize
./avrdude -C avrdude.conf -c serialupdi -P /dev/cu.usb* -p avr128db64 -U bootsize:w:0x20:m

# Read bootsize
./avrdude -C avrdude.conf -c serialupdi -P /dev/cu.usb* -p avr128db64 -U bootsize:r:-:h

# Write fuse8 (bootsize)
./avrdude -C avrdude.conf -c serialupdi -P /dev/cu.usb* -p avr128db64 -U fuse8w:0x00:m

# Read fuse8 (bootsize)
./avrdude -C avrdude.conf -c serialupdi -P /dev/cu.usb* -p avr128db64 -U fuse8:r:-:h
```

dbuchwald, 2022-02-17:
Sure, I will look into it tonight.

dbuchwald, 2022-02-17:
Confirmed to work correctly:

```
./avrdude -C avrdude.conf -c serialupdi -P /dev/ttyUSB0 -p avr128db64 -U bootsize:w:0x20:m

avrdude: UPDI link initialization OK
avrdude: NVM type 2: 24-bit, word oriented write
avrdude: Entering NVM programming mode
avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.01s

avrdude: Device signature = 0x1e970b (probably avr128db64)
avrdude: reading input file "0x20"
avrdude: writing fuse8/bootsize (1 bytes):

Writing | ################################################## | 100% 0.02s

avrdude: 1 bytes of fuse8/bootsize written
avrdude: verifying fuse8/bootsize memory against 0x20:

Reading | ################################################## | 100% 0.00s

avrdude: 1 bytes of fuse8/bootsize verified
avrdude: Leaving NVM programming mode

avrdude done.  Thank you.

./avrdude -C avrdude.conf -c serialupdi -P /dev/ttyUSB0 -p avr128db64 -U bootsize:r:-:h

avrdude: UPDI link initialization OK
avrdude: NVM type 2: 24-bit, word oriented write
avrdude: Entering NVM programming mode
avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.01s

avrdude: Device signature = 0x1e970b (probably avr128db64)
avrdude: reading fuse8/bootsize memory:

Reading | ################################################## | 100% 0.00s

avrdude: writing output file "<stdout>"
0x20
avrdude: Leaving NVM programming mode

avrdude done.  Thank you.

./avrdude -C avrdude.conf -c serialupdi -P /dev/ttyUSB0 -p avr128db64 -U fuse8:w:0x00:m

avrdude: UPDI link initialization OK
avrdude: NVM type 2: 24-bit, word oriented write
avrdude: Entering NVM programming mode
avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.01s

avrdude: Device signature = 0x1e970b (probably avr128db64)
avrdude: reading input file "0x00"
avrdude: writing fuse8/bootsize (1 bytes):

Writing | ################################################## | 100% 0.02s

avrdude: 1 bytes of fuse8/bootsize written
avrdude: verifying fuse8/bootsize memory against 0x00:

Reading | ################################################## | 100% 0.00s

avrdude: 1 bytes of fuse8/bootsize verified
avrdude: Leaving NVM programming mode

avrdude done.  Thank you.

./avrdude -C avrdude.conf -c serialupdi -P /dev/ttyUSB0 -p avr128db64 -U fuse8:r:-:h

avrdude: UPDI link initialization OK
avrdude: NVM type 2: 24-bit, word oriented write
avrdude: Entering NVM programming mode
avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.01s

avrdude: Device signature = 0x1e970b (probably avr128db64)
avrdude: reading fuse8/bootsize memory:

Reading | ################################################## | 100% 0.00s

avrdude: writing output file "<stdout>"
0x0
avrdude: Leaving NVM programming mode

avrdude done.  Thank you.
```