avrdudes/avrdude #874: Can't get serialupdi to work

dl8dtl, 2022-02-11:
So, for a test, I tried to attach a serialupdi to my Curiosity Nano with an AVR128DB48.
Just in case, I cut the connection between the onboard debugger and the UPDI pin on the CNano.
I then attached a PL2303 to all that.
Circuitry is a diode between the PL2303's TxD and RxD (cathode on TxD), and RxD going straight to UPDI. The PL2303 device has 68 Ω resistors in both, RxD and TxD lines – shouldn't be too high I guess.
I also tried a 470 Ω resistor in series with the diode.
[image]
I can't get SerialUPDI to work, anyway.
```
% ./build_freebsd/src/avrdude -c serialupdi -P /dev/cuaU4 -p avr128db48 -B1

avrdude: UPDI link initialization OK
avrdude: initialization failed, rc=-1
         Double check connections and try again, or use -F to override
         this check.

avrdude: Leaving NVM programming mode

avrdude done.  Thank you.

[ve] j@uriah 658% ./build_freebsd/src/avrdude -c serialupdi -P /dev/cuaU4 -p avr128db48 -vvvv
executable_abspath = /home/joerg/src/avrdude-official/build_freebsd/src/avrdude
executable_abspath_len = 58
executable_dirpath = /home/joerg/src/avrdude-official/build_freebsd/src
executable_dirpath_len = 50
sys_config = /home/joerg/src/avrdude-official/build_freebsd/src/avrdude.conf
sys_config_found = true

avrdude: Version 6.99-20220211 (7b79b72)
         Copyright (c) Brian Dean, http://www.bdmicro.com/
         Copyright (c) Joerg Wunsch

         System wide configuration file is "/home/joerg/src/avrdude-official/build_freebsd/src/avrdude.conf"
         User configuration file is "/home/joerg/.avrduderc"

         Using Port                    : /dev/cuaU4
         Using Programmer              : serialupdi
avrdude: Opening serial port...
avrdude: Sending double break
avrdude: Send: . [00] 
avrdude: ser_recv(): programmer is not responding
avrdude: Send: . [00] 
avrdude: Recv: . [00] 
         AVR Part                      : AVR128DB48
         RESET disposition             : dedicated
         RETRY pulse                   : SCK
         Serial program mode           : yes
         Parallel program mode         : yes
         Memory Detail                 :

                                           Block Poll               Page                       Polled
           Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           signature               0     0     0    0 no          3    1      0     0     0 0x00 0x00
                                           Block Poll               Page                       Polled
           Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           prodsig                 0     0     0    0 no        125  125      0     0     0 0x00 0x00
                                           Block Poll               Page                       Polled
           Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           tempsense               0     0     0    0 no          2    1      0     0     0 0x00 0x00
                                           Block Poll               Page                       Polled
           Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           sernum                  0     0     0    0 no         16    1      0     0     0 0x00 0x00
                                           Block Poll               Page                       Polled
           Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           fuses                   0     0     0    0 no          9   16      0     0     0 0x00 0x00
                                           Block Poll               Page                       Polled
[... 121 lines cut ...]
avrdude: Received 1 bytes [0x80]
avrdude: UDPI init OK
avrdude: initialization failed, rc=-1
         Double check connections and try again, or use -F to override
         this check.

avrdude: Leaving NVM programming mode
avrdude: Sending reset request
avrdude: STCS 0x59 to address 0x08
avrdude: Sending 3 bytes [0x55, 0xc8, 0x59]
avrdude: Send: U [55] . [c8] Y [59] 
avrdude: Recv: U [55] . [c8] Y [59] 
avrdude: Sending release reset request
avrdude: STCS 0x00 to address 0x08
avrdude: Sending 3 bytes [0x55, 0xc8, 0x00]
avrdude: Send: U [55] . [c8] . [00] 
avrdude: Recv: U [55] . [c8] . [00] 
avrdude: STCS 0x0C to address 0x03
avrdude: Sending 3 bytes [0x55, 0xc3, 0x0c]
avrdude: Send: U [55] . [c3] . [0c] 
avrdude: Recv: U [55] . [c3] . [0c] 

avrdude done.  Thank you.
```
Is there any obvious mistake I made?

dl8dtl, 2022-02-11:
@dbuchwald I'm interested in any opinions from your end …

MCUdude, 2022-02-11:
Are you using a Schottky diode between RX and TX? I'm using a BAT54J and a 470R resistor between RX and TX on my programmer (that I'm planning to sell someday). I've never had any issues with this, and @dbuchwald's SerialUPDI implementation has proven itself to be very reliable, at least for me.

[image]

[image]  [image]

dl8dtl, 2022-02-12:
No, it's a standard p-n silicon diode.
Will try a Schottky one later.

dbuchwald, 2022-02-12:
Ok, sorry to be a little late to the party, but I needed a while to do some analysis and testing on my end.

First of all, I don't have good news as in "this is bug in the code, and here's the fix". I can't say I replicated the issue either. Let's start at the beginning:

```
avrdude: STCS 0x08 to address 0x03
avrdude: Sending 3 bytes [0x55, 0xc3, 0x08]
avrdude: Send: U [55] . [c3] . [08] 
avrdude: Recv: . [00] U [55] . [c3] 
avrdude: STCS 0x80 to address 0x02
avrdude: Sending 3 bytes [0x55, 0xc2, 0x80]
avrdude: Send: U [55] . [c2] . [80] 
avrdude: Recv: . [08] U [55] . [c2] 
avrdude: LDCS from 0x00
avrdude: Sending 2 bytes [0x55, 0x80]
avrdude: Send: U [55] . [80] 
avrdude: Recv: . [80] U [55] 
avrdude: Recv: . [80] 
avrdude: Received 1 bytes [0x80]
avrdude: UDPI init OK
avrdude: UPDI link initialization OK
```

Problem starts at the first Recv line (avrdude: Recv: . **[00]** U [55] . [c3]) - basically, with the RX/TX connected, you expect to have perfect echo here, like so:

```
avrdude: STCS 0x08 to address 0x03
avrdude: Sending 3 bytes [0x55, 0xc3, 0x08]
avrdude: Send: U [55] . [c3] . [08]
avrdude: Recv: U [55] . [c3] . [08]
avrdude: STCS 0x80 to address 0x02
avrdude: Sending 3 bytes [0x55, 0xc2, 0x80]
avrdude: Send: U [55] . [c2] . [80]
avrdude: Recv: U [55] . [c2] . [80]
avrdude: LDCS from 0x00
avrdude: Sending 2 bytes [0x55, 0x80]
avrdude: Send: U [55] . [80]
avrdude: Recv: U [55] . [80]
avrdude: Recv: 0 [30]
avrdude: Received 1 bytes [0x30]
avrdude: UDPI init OK 
```

This initial "0x00" might come from serial buffer or electrical issue with your configuration, but it breaks the transmission, resulting in misinterpretation of whatever is coming back. Part of the echo (0x80) is being interpreted as "UPDI OK" signal. This can happen normally and further operation reveals that connection is not as "OK" as initially assumed, so double break is happening to reset the operation:

```
avrdude: Sending double break
avrdude: Send: . [00] 
avrdude: ser_recv(): programmer is not responding
avrdude: Send: . [00] 
avrdude: Recv: . [00] 
```

Again, you expect something else, this break should be properly echoed, but it's not.

When I looked at it for the first time, I thought it was the onboard programmer/debugger of the Curiosity Nano board, but I checked:

[image]

It's not it. I can work with this board, even if it's powered via the debugger. I have also tried powering it directly via VTG, bypassing the debugger itself:

[image]

but it still works:

```
src git:(main) ./avrdude -C ./avrdude.conf -p avr128da48 -b 230400 -c serialupdi -U signature:r:-:h -P /dev/tty.usbserial-0001

avrdude: UPDI link initialization OK
avrdude: NVM type 2: 24-bit, word oriented write
avrdude: Entering NVM programming mode
avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.00s

avrdude: Device signature = 0x1e9708 (probably avr128da48)
avrdude: reading signature memory:

Reading | ################################################## | 100% 0.00s

avrdude: writing output file "<stdout>"
0x1e,0x97,0x8
avrdude: Leaving NVM programming mode

avrdude done.  Thank you.
```

So with all that long version of "weird, it works for me", these are the things I would like to point out:

1. I'm also using Schottky diode (BAT85) in my build,
2. As you can see in the pictures, I have slightly different version of the Curiosity Nano board (AVR128DA48),
3. I'm using different USB->UART adapter (based on CP2102).

I would suggest the following:

1. Start with using the original Microchip's **pymcuprog** utility - if it works, the problem is in AVRDUDE software, if it doesn't, it will be in your hardware. In the former case, please run **pymcuprog** in verbose mode to see the difference. I made point of replicating their logging model so the outputs are easy to compare,
2. Try using the VTG instead of Curiosity Nano programmer for power - chances are that my programmer produces "stronger" LOW signal than the onboard programmer, but yours is not, so bypassing it might help,
3. Try Schottky diode, but I don't know if it matters,
4. Test it against other AVR UPDI chip. What I noticed is that the choice of series resistors is important - my CP2102 doesn't work if there is 470R between it and the diode, but FT232RL does require it. Or something, I'm not entirely sure I remember the details correctly, but you get the point - the thing is that the resistors setup does matter and you have to get it right to work. That being said, I tried different options with and without the resistor, but none of the setups produced the output you were seeing, so I can't say it's that.

Sorry, I wish I could be more helpful, but this is all I can do at the moment with the hardware I have...

dl8dtl, 2022-02-12:
I even disconnected the onboard debugger, just in case …
I doubt this is a counterfeit chip, it's a pretty old adapter.
The above was on FreeBSD, on Linux, it doesn't look that much better:
```
% ./build_linux/src/avrdude -c serialupdi -P /dev/ttyUSB0 -p avr128db48 -vvvv
executable_abspath = /home/j/src/avrdude/build_linux/src/avrdude
executable_abspath_len = 43
executable_dirpath = /home/j/src/avrdude/build_linux/src
executable_dirpath_len = 35
sys_config = /home/j/src/avrdude/build_linux/src/avrdude.conf
sys_config_found = true

avrdude: Version 6.99-20220210 (452f673)
         Copyright (c) Brian Dean, http://www.bdmicro.com/
         Copyright (c) Joerg Wunsch

         System wide configuration file is "/home/j/src/avrdude/build_linux/src/avrdude.conf"
         User configuration file is "/home/j/.avrduderc"

         Using Port                    : /dev/ttyUSB0
         Using Programmer              : serialupdi
avrdude: Opening serial port...
avrdude: Sending double break
avrdude: Send: . [00] 
avrdude: Recv: . [00] 
avrdude: Send: . [00] 
avrdude: Recv: . [00] 
         AVR Part                      : AVR128DB48
         RESET disposition             : dedicated
         RETRY pulse                   : SCK
         Serial program mode           : yes
         Parallel program mode         : yes
         Memory Detail                 :

                                           Block Poll               Page                       Polled
           Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           signature               0     0     0    0 no          3    1      0     0     0 0x00 0x00
                                           Block Poll               Page                       Polled
           Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           prodsig                 0     0     0    0 no        125  125      0     0     0 0x00 0x00
                                           Block Poll               Page                       Polled
           Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           tempsense               0     0     0    0 no          2    1      0     0     0 0x00 0x00
                                           Block Poll               Page                       Polled
           Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           sernum                  0     0     0    0 no         16    1      0     0     0 0x00 0x00
                                           Block Poll               Page                       Polled
           Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           fuses                   0     0     0    0 no          9   16      0     0     0 0x00 0x00
                                           Block Poll               Page                       Polled
           Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           fuse0                   0     0     0    0 no          1    1      0     0     0 0x00 0x00
                                           Block Poll               Page                       Polled
           Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           wdtcfg                  0     0     0    0 no          1    1      0     0     0 0x00 0x00
                                           Block Poll               Page                       Polled
           Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           fuse1                   0     0     0    0 no          1    1      0     0     0 0x00 0x00
                                           Block Poll               Page                       Polled
           Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           bodcfg                  0     0     0    0 no          1    1      0     0     0 0x00 0x00
[... 117 lines cut ...]
avrdude: UPDI link initialization failed
avrdude: initialization failed, rc=-1
         Double check connections and try again, or use -F to override
         this check.

avrdude: Leaving NVM programming mode
avrdude: Sending reset request
avrdude: STCS 0x59 to address 0x08
avrdude: Sending 3 bytes [0x55, 0xc8, 0x59]
avrdude: Send: U [55] . [c8] Y [59] 
avrdude: ser_recv(): programmer is not responding
avrdude: Sending release reset request
avrdude: STCS 0x00 to address 0x08
avrdude: Sending 3 bytes [0x55, 0xc8, 0x00]
avrdude: Send: U [55] . [c8] . [00] 
avrdude: ser_recv(): programmer is not responding
avrdude: STCS 0x0C to address 0x03
avrdude: Sending 3 bytes [0x55, 0xc3, 0x0c]
avrdude: Send: U [55] . [c3] . [0c] 
avrdude: ser_recv(): programmer is not responding

avrdude done.  Thank you.
```
For "normal" serial stuff, that PL2303 has never caused me any trouble though.
I'll check with an FT232 tonight.

dl8dtl, 2022-02-12:
With an FT232B, it works on Linux.
On FreeBSD, I still see that one-byte offset:
```
% ./build_freebsd/src/avrdude -c serialupdi -P /dev/cuaU4 -p avr128db48 -vvvv
executable_abspath = /home/joerg/src/avrdude-official/build_freebsd/src/avrdude
executable_abspath_len = 58
executable_dirpath = /home/joerg/src/avrdude-official/build_freebsd/src
executable_dirpath_len = 50
sys_config = /home/joerg/src/avrdude-official/build_freebsd/src/avrdude.conf
sys_config_found = true

avrdude: Version 6.99-20220211 (7b79b72)
         Copyright (c) Brian Dean, http://www.bdmicro.com/
         Copyright (c) Joerg Wunsch

         System wide configuration file is "/home/joerg/src/avrdude-official/build_freebsd/src/avrdude.conf"
         User configuration file is "/home/joerg/.avrduderc"

         Using Port                    : /dev/cuaU4
         Using Programmer              : serialupdi
avrdude: Opening serial port...
avrdude: Sending double break
avrdude: Send: . [00] 
avrdude: ser_recv(): programmer is not responding
avrdude: Send: . [00] 
avrdude: Recv: . [00] 
         AVR Part                      : AVR128DB48
         RESET disposition             : dedicated
         RETRY pulse                   : SCK
         Serial program mode           : yes
         Parallel program mode         : yes
         Memory Detail                 :
…
         Programmer Type : serialupdi
         Description     : SerialUPDI

avrdude: STCS 0x08 to address 0x03
avrdude: Sending 3 bytes [0x55, 0xc3, 0x08]
avrdude: Send: U [55] . [c3] . [08] 
avrdude: Recv: . [00] U [55] . [c3] 
avrdude: STCS 0x80 to address 0x02
avrdude: Sending 3 bytes [0x55, 0xc2, 0x80]
avrdude: Send: U [55] . [c2] . [80] 
avrdude: Recv: . [08] U [55] . [c2] 
avrdude: LDCS from 0x00
avrdude: Sending 2 bytes [0x55, 0x80]
avrdude: Send: U [55] . [80] 
avrdude: Recv: . [80] U [55] 
avrdude: Recv: . [80] 
avrdude: Received 1 bytes [0x80]
avrdude: UDPI init OK
avrdude: UPDI link initialization OK
avrdude: LDCS from 0x0B
avrdude: Sending 2 bytes [0x55, 0x8b]
avrdude: Send: U [55] . [8b] 
avrdude: Recv: 0 [30] U [55] 
avrdude: Recv: . [8b] 
avrdude: Received 1 bytes [0x8b]
avrdude: Device is locked
avrdude: Device in NVM programming state, leaving programming mode
avrdude: Sending reset request
…
```

dbuchwald, 2022-02-12:
OK, I need to clarify some things here, so that it's easier to understand what is going on.

This kind of output suggests serious electrical issue:

```
avrdude: Sending 2 bytes [0x55, 0x80]
avrdude: Send: U [55] . [80] 
avrdude: ser_recv(): programmer is not responding
avrdude: ser_recv(): programmer is not responding

```

There are two lines of failed recv() operation, and here is what these are: first one is echo, and this echo is not introduced by connection parameters. This echo happens because of the diode between RX and TX - basically, whenever TX pulls line low (transmitting bit of data from USB to AVR UPDI), it should pull also RX line low. If this doesn't work, then you have something pulling RX high which is much stronger than TX pulling low. Or, which is also possible, your diode doesn't work (it's wrong way around).

Second failed recv() is the actual message being sent from AVR UPDI - as you can see, it's not responding, or the response is being pulled high too strong to be detected by the USB adapter.

So yeah, that's a different story, but given how the first recv() just has to happen (it will happen even if you disconnect the UPDI line from AVR target!), it shows that the SerialUPDI adapter (whatever it is) is not functioning correctly. I would bet on the diode/resistors configuration.

To back it up with actual evidence, here is what happens when you try to use the SerialUPDI protocol with the diode pointing the right direction, but not connected to any AVR (UPDI line is floating):

```
         Programmer Type : serialupdi
         Description     : SerialUPDI

avrdude: STCS 0x08 to address 0x03
avrdude: Sending 3 bytes [0x55, 0xc3, 0x08]
avrdude: Send: U [55] . [c3] . [08]
avrdude: Recv: U [55] . [c3] . [08]
avrdude: STCS 0x80 to address 0x02
avrdude: Sending 3 bytes [0x55, 0xc2, 0x80]
avrdude: Send: U [55] . [c2] . [80]
avrdude: Recv: U [55] . [c2] . [80]
avrdude: LDCS from 0x00
avrdude: Sending 2 bytes [0x55, 0x80]
avrdude: Send: U [55] . [80]
avrdude: Recv: U [55] . [80]
avrdude: ser_recv(): programmer is not responding
avrdude: serialupdi_recv(): programmer is not responding
avrdude: Check failed
avrdude: Datalink not active, resetting...
avrdude: Sending double break
avrdude: Send: . [00]
avrdude: Recv: . [00]
avrdude: Send: . [00]
avrdude: Recv: . [00]
avrdude: STCS 0x08 to address 0x03
avrdude: Sending 3 bytes [0x55, 0xc3, 0x08]
avrdude: Send: U [55] . [c3] . [08]
avrdude: Recv: U [55] . [c3] . [08]
avrdude: STCS 0x80 to address 0x02
avrdude: Sending 3 bytes [0x55, 0xc2, 0x80]
avrdude: Send: U [55] . [c2] . [80]
avrdude: Recv: U [55] . [c2] . [80]
avrdude: LDCS from 0x00
avrdude: Sending 2 bytes [0x55, 0x80]
avrdude: Send: U [55] . [80]
avrdude: Recv: U [55] . [80]
avrdude: ser_recv(): programmer is not responding
avrdude: serialupdi_recv(): programmer is not responding
avrdude: Check failed
avrdude: Restoring datalink failed
avrdude: UPDI link initialization failed
avrdude: initialization failed, rc=-1
         Double check connections and try again, or use -F to override
         this check.
```

Now, the same thing, still no AVR, but with reversed diode:

```
[... 179 lines cut ...]
  serial_recv(&pgm->fd, buffer, 1);

  usleep(100*1000);

  buffer[0] = UPDI_BREAK;

  serial_send(&pgm->fd, buffer, 1);
  serial_recv(&pgm->fd, buffer, 1);

  if (serial_setparams(&pgm->fd, pgm->baudrate? pgm->baudrate: 115200, SERIAL_8E2) < 0) {
    return -1;
  }

  updi_set_rtsdtr_mode(pgm);

  /*
   * drain any extraneous input
   */
  serial_drain(&pgm->fd, 0);

  return 0;
}
```

It shouldn't have any adverse effects on the operation, but I would introduce it only if absolutely needed. After all - this hasn't been tested yet.

dl8dtl, 2022-02-12:
> So yeah, I'm pretty positive you either have the diode pointing the wrong way or extra resistor that blocks the returning signal.

I'm almost sure the diode had the correct polarity, I can try reproducing that later on.
The PL2303 adapter has two 68 Ω resistors in both, the RxD and the TxD lines, so I only added the diode outside.
I can also try a different diode.

> Now, since I can't replicate this behaviour on my machines, all I can suggest is you add the following snippet to `updi_link.c` at line 163:

Yes, that helps:
```
% ./build_freebsd/src/avrdude -c serialupdi -P /dev/cuaU4 -p avr128db48 

avrdude: UPDI link initialization OK
avrdude: Device is in SLEEP mode
avrdude: NVM type 2: 24-bit, word oriented write
avrdude: Entering NVM programming mode
avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.05s

avrdude: Device signature = 0x1e970c (probably avr128db48)
avrdude: Leaving NVM programming mode

avrdude done.  Thank you.
```
So I guess it would make sense to have this in general?

dl8dtl, 2022-02-12:
Well no, the PL2303 one doesn't work, even with the ser_drain().

MCUdude, 2022-02-12:
Does it work with pymcuprog?

dl8dtl, 2022-02-12:
> Does the PL2303 work with pymcuprog?

Seems so:
```
% pymcuprog -t uart -u /dev/cuaU4 -d avr128db48 ping
pymcuprog.serialupdi.link - WARNING - Check failed
Pinging device...
Ping response: 1E970C
Done.
% pymcuprog -t uart -u /dev/cuaU4 -d avr128db48 ping
Pinging device...
Ping response: 1E970C
Done.
```
And that's even under FreeBSD.

dbuchwald, 2022-02-12:
> Seems so:
>
> ```
> % pymcuprog -t uart -u /dev/cuaU4 -d avr128db48 ping
> pymcuprog.serialupdi.link - WARNING - Check failed
> Pinging device...
> Ping response: 1E970C
> Done.
> % pymcuprog -t uart -u /dev/cuaU4 [...]

Can you, please, run the same commands with `-v debug` and post output? And immediately afterwards run the `avrdude` session with `-vvvv` so I can compare outputs?

Thanks in advance,
Dawid

dl8dtl, 2022-02-12:
```
% pymcuprog -t uart -u /dev/cuaU4 -d avr128db48 -v debug ping
pymcuprog.programmer - INFO - Setting up programming session for 'avr128db48'
pymcuprog.deviceinfo.deviceinfo - INFO - Looking for device avr128db48
pymcuprog.serialupdi.physical - INFO - Opening port '/dev/cuaU4' at '115200' baud
pymcuprog.serialupdi.physical - DEBUG - send : [0x0]
pymcuprog.serialupdi.link - DEBUG - STCS to 0x03
pymcuprog.serialupdi.physical - DEBUG - send : [0x55, 0xc3, 0x8]
pymcuprog.serialupdi.link - DEBUG - STCS to 0x02
pymcuprog.serialupdi.physical - DEBUG - send : [0x55, 0xc2, 0x80]
pymcuprog.serialupdi.link - DEBUG - LDCS from 0x00
pymcuprog.serialupdi.physical - DEBUG - send : [0x55, 0x80]
pymcuprog.serialupdi.physical - DEBUG - receive : [0x30]
pymcuprog.serialupdi.link - INFO - UPDI init OK
pymcuprog.serialupdi.physical - DEBUG - send : [0x55, 0xe6]
pymcuprog.serialupdi.application - INFO - SIB: 'AVR     P:2D:1-3M2 (A5.KV00D.0)'
pymcuprog.serialupdi.application - INFO - Device family ID: 'AVR'
pymcuprog.serialupdi.application - INFO - NVM interface: 'P:2'
pymcuprog.serialupdi.application - INFO - Debug interface: 'D:1'
pymcuprog.serialupdi.application - INFO - PDI oscillator: '3M2'
pymcuprog.serialupdi.application - INFO - Extra info: '(A5.KV00D.0)'
pymcuprog.serialupdi.application - INFO - NVM type 2: 24-bit, word oriented write
pymcuprog.serialupdi.link - DEBUG - STCS to 0x03
pymcuprog.serialupdi.physical - DEBUG - send : [0x55, 0xc3, 0x8]
pymcuprog.serialupdi.link - DEBUG - STCS to 0x02
pymcuprog.serialupdi.physical - DEBUG - send : [0x55, 0xc2, 0x80]
pymcuprog.serialupdi.link - DEBUG - LDCS from 0x00
pymcuprog.serialupdi.physical - DEBUG - send : [0x55, 0x80]
pymcuprog.serialupdi.physical - DEBUG - receive : [0x30]
pymcuprog.serialupdi.link - INFO - UPDI init OK
pymcuprog.serialupdi.link - DEBUG - LDCS from 0x00
pymcuprog.serialupdi.physical - DEBUG - send : [0x55, 0x80]
pymcuprog.serialupdi.physical - DEBUG - receive : [0x30]
pymcuprog.serialupdi.application - INFO - PDI revision = 0x03
pymcuprog.serialupdi.link - DEBUG - LDCS from 0x0B
pymcuprog.serialupdi.physical - DEBUG - send : [0x55, 0x8b]
pymcuprog.serialupdi.physical - DEBUG - receive : [0x92]
pymcuprog.serialupdi.link - DEBUG - LDCS from 0x0B
pymcuprog.serialupdi.physical - DEBUG - send : [0x55, 0x8b]
pymcuprog.serialupdi.physical - DEBUG - receive : [0x92]
pymcuprog.serialupdi.application - INFO - Entering NVM programming mode
pymcuprog.serialupdi.link - DEBUG - Writing key
pymcuprog.serialupdi.physical - DEBUG - send : [0x55, 0xe0]
pymcuprog.serialupdi.physical - DEBUG - send : [0x20, 0x67, 0x6f, 0x72, 0x50, 0x4d, 0x56, 0x4e]
pymcuprog.serialupdi.link - DEBUG - LDCS from 0x07
pymcuprog.serialupdi.physical - DEBUG - send : [0x55, 0x87]
pymcuprog.serialupdi.physical - DEBUG - receive : [0x10]
pymcuprog.serialupdi.application - DEBUG - Key status = 0x10
pymcuprog.serialupdi.application - INFO - Apply reset
pymcuprog.serialupdi.link - DEBUG - STCS to 0x08
pymcuprog.serialupdi.physical - DEBUG - send : [0x55, 0xc8, 0x59]
pymcuprog.serialupdi.application - INFO - Release reset
pymcuprog.serialupdi.link - DEBUG - STCS to 0x08
pymcuprog.serialupdi.physical - DEBUG - send : [0x55, 0xc8, 0x0]
pymcuprog.serialupdi.link - DEBUG - LDCS from 0x0B
pymcuprog.serialupdi.physical - DEBUG - send : [0x55, 0x8b]
pymcuprog.serialupdi.physical - DEBUG - receive : [0x38]
pymcuprog.serialupdi.link - DEBUG - LDCS from 0x0B
pymcuprog.serialupdi.physical - DEBUG - send : [0x55, 0x8b]
pymcuprog.serialupdi.physical - DEBUG - receive : [0x8]
pymcuprog.serialupdi.application - DEBUG - Now in NVM programming mode
Pinging device...
pymcuprog.programmer - INFO - Reading device ID...
pymcuprog.serialupdi.physical - DEBUG - send : [0x55, 0xe6]
pymcuprog.serialupdi.application - INFO - SIB: 'AVR     P:2D:1-3M2 (A5.KV00D.0)'
pymcuprog.serialupdi.application - INFO - Device family ID: 'AVR'
pymcuprog.serialupdi.application - INFO - NVM interface: 'P:2'
pymcuprog.serialupdi.application - INFO - Debug interface: 'D:1'
pymcuprog.serialupdi.application - INFO - PDI oscillator: '3M2'
pymcuprog.serialupdi.application - INFO - Extra info: '(A5.KV00D.0)'
[... 143 lines cut ...]
avrdude: Check failed
avrdude: Restoring datalink failed
avrdude: UPDI link initialization failed
avrdude: initialization failed, rc=-1
         Double check connections and try again, or use -F to override
         this check.

avrdude: Leaving NVM programming mode
avrdude: Sending reset request
avrdude: STCS 0x59 to address 0x08
avrdude: Sending 3 bytes [0x55, 0xc8, 0x59]
avrdude: Send: U [55] . [c8] Y [59] 
avrdude: ser_recv(): programmer is not responding
avrdude: Sending release reset request
avrdude: STCS 0x00 to address 0x08
avrdude: Sending 3 bytes [0x55, 0xc8, 0x00]
avrdude: Send: U [55] . [c8] . [00] 
avrdude: ser_recv(): programmer is not responding
avrdude: STCS 0x0C to address 0x03
avrdude: Sending 3 bytes [0x55, 0xc3, 0x0c]
avrdude: Send: U [55] . [c3] . [0c] 
avrdude: ser_recv(): programmer is not responding

avrdude done.  Thank you.
```

dl8dtl, 2022-02-12:
> Well, I don't see how this could break anything, and in fact this is how this has been implemented before I started playing with RTS/DTR manipulation, so I guess we could try. I would suggest to test it with various devices before we make it official.

I just created PR #876 for this.

dl8dtl, 2022-02-12:
> You will notice that in most cases there is additional resistor between diode anode and UPDI input on the target board.

Btw., I've now got:
[image]

dl8dtl, 2022-02-12:
Well, it did work with pymcuprog with the previous schematic, and it also works after changing it.

dl8dtl, 2022-02-12:
> (Python using different serial driver than whatever avrdude is linked against at runtime)

I don't think they are. After all, it's a Unix, so both pyserial as well as AVRDUDE have to use the same Posix-style tty driver.

dl8dtl, 2022-02-12:
I still think we must be doing something wrong (most likely not in updi_serial.c but in the old ser_posix.c).
By increasing the serial timeout, I once could read the signature, but it's not reproducible.

dbuchwald, 2022-02-13:
> I don't think they are. After all, it's a Unix, so both pyserial as well as AVRDUDE have to use the same Posix-style tty driver.

Yeah, that is what I thought too, but when investigating RTS/DTR stuff I noticed some discrepancies in behaviour. It's possible that in the end the same POSIX driver library is used, but in a different way, and this is how we get to...

> I still think we must be doing something wrong (most likely not in updi_serial.c but in the old ser_posix.c). By increasing the serial timeout, I once could read the signature, but it's not reproducible.

Well, I know for a fact that this is true. I'm not an expert on anything POSIX, including serial code, but during my investigation I found something like that:

```
  termios.c_iflag = IGNBRK;
  termios.c_oflag = 0;
  termios.c_lflag = 0;
  termios.c_cflag = translate_flags(cflags);
  termios.c_cc[VMIN]  = 1;
  termios.c_cc[VTIME] = 0;
```

The part with `translate_flags` is mine, but I haven't changed the underlying behaviour here, it was always assigning arbitrary value here, and according to documentation I found on that, this is not the correct way to do it. You are supposed to use bitwise operators to toggle flags you are interested in and leave everything else intact. It makes sense, doesn't it?

Now, I considered fixing it, but gave up this idea - serial channel is used for many different devices and I wasn't keen on breaking compatibility with them.

Bottom line: before I get the device and replicate the issue myself, I don't want to speculate as to the reason for it to occur. Maybe there is a simple fix for it, but most likely it will require potentially breaking changes and we will have to decide whether it even makes sense to add support for one "exotic" serial adapter at the risk of breaking support for several others...

MCUdude, 2022-02-13:
I can confirm that there is an issue with PL2303. I can't get it to work with either Avrdude or pymcuprog. I can confirm that the adapter itself is actually working by performing a simple echo test by wiring RX and TX together.

I've tried the exact same setup (same wires and everything) with a CP2102, CP2104, FT232, CH340C and a CH340N. All adapters works flawlessly except for the PL2303. I'm using a Mac, and is running macOS 10.14

Here's my setup (the protoboard holds the BAT54J diode and 470R resistor):
[image]

dbuchwald, 2022-02-13:
> I can confirm that there is an issue with PL2303. I can't get it to work with either Avrdude or pymcuprog. I can confirm that the adapter itself is actually working by performing a simple echo test by wiring RX and TX together.

I think I had one of these as well, but couldn't get it to run correctly on MacOS and just tossed it in trash, thinking it's broken. Will find out soon enough :)

dl8dtl, 2022-02-13:
So yep, it seems the PL2303 isn't really "best of class" … (I guess that one goes to FTDI anyway). However, as pymcuprog can handle it, I think we should be able to handle it as well.

I guess I gotta learn DTrace now …

dl8dtl, 2022-02-14:
I'm still fighting a bit with my DTrace script regarding tracing of read and write operations, but I can record all ioctls:
```
# ~j/src/dtrace/tty-ioctl.d
dtrace: script '/home/joerg/src/dtrace/tty-ioctl.d' matched 39 probes
CPU     ID                    FUNCTION:NAME
  1   7891              ttydev_ioctl:return        python3.7 TIOCGETA -> iflag 2b02 oflag 3 cflag cb00 lflag 5cb ispeed 9600
  1   7891              ttydev_ioctl:return        python3.7 TIOCGETA -> iflag 2b02 oflag 3 cflag cb00 lflag 5cb ispeed 9600
  1   7890               ttydev_ioctl:entry        python3.7 TIOCSETA iflag 2002 oflag 0 cflag df00 lflag 0 ispeed 115200
  1   7890               ttydev_ioctl:entry        python3.7 TIOCMBIS flags 2
  1   7890               ttydev_ioctl:entry        python3.7 TIOCMBIS flags 4
  1   7890               ttydev_ioctl:entry        python3.7 TIOCFLUSH
  2   7891              ttydev_ioctl:return        python3.7 TIOCGETA -> iflag 2b02 oflag 3 cflag cb00 lflag 5cb ispeed 9600
  2   7891              ttydev_ioctl:return        python3.7 TIOCGETA -> iflag 2b02 oflag 3 cflag cb00 lflag 5cb ispeed 9600
  2   7890               ttydev_ioctl:entry        python3.7 TIOCSETA iflag 2002 oflag 0 cflag db00 lflag 0 ispeed 300
  2   7890               ttydev_ioctl:entry        python3.7 TIOCMBIS flags 2
  2   7890               ttydev_ioctl:entry        python3.7 TIOCMBIS flags 4
  2   7890               ttydev_ioctl:entry        python3.7 TIOCFLUSH
  0   7891              ttydev_ioctl:return        python3.7 TIOCGETA -> iflag 2b02 oflag 3 cflag cb00 lflag 5cb ispeed 9600
  0   7891              ttydev_ioctl:return        python3.7 TIOCGETA -> iflag 2b02 oflag 3 cflag cb00 lflag 5cb ispeed 9600
  0   7890               ttydev_ioctl:entry        python3.7 TIOCSETA iflag 2002 oflag 0 cflag df00 lflag 0 ispeed 115200
  0   7890               ttydev_ioctl:entry        python3.7 TIOCMBIS flags 2
  0   7890               ttydev_ioctl:entry        python3.7 TIOCMBIS flags 4
  0   7890               ttydev_ioctl:entry        python3.7 TIOCFLUSH
  2   7891              ttydev_ioctl:return          avrdude TIOCGETA -> iflag 2b02 oflag 3 cflag cb00 lflag 5cb ispeed 9600
  2   7891              ttydev_ioctl:return          avrdude TIOCGETA -> iflag 2b02 oflag 3 cflag cb00 lflag 5cb ispeed 9600
  2   7890               ttydev_ioctl:entry          avrdude TIOCSETA iflag 1 oflag 0 cflag 9f00 lflag 0 ispeed 115200
  2   7890               ttydev_ioctl:entry          avrdude FIONBIO flags 0
  2   7890               ttydev_ioctl:entry          avrdude FIOASYNC flags 0
  0   7891              ttydev_ioctl:return          avrdude TIOCGETA -> iflag 1 oflag 0 cflag 9f00 lflag 0 ispeed 115200
  0   7891              ttydev_ioctl:return          avrdude TIOCGETA -> iflag 1 oflag 0 cflag 9f00 lflag 0 ispeed 115200
  0   7890               ttydev_ioctl:entry          avrdude TIOCSETA iflag 1 oflag 0 cflag 9b00 lflag 0 ispeed 300
  0   7890               ttydev_ioctl:entry          avrdude FIONBIO flags 0
  0   7890               ttydev_ioctl:entry          avrdude FIOASYNC flags 0
  1   7891              ttydev_ioctl:return          avrdude TIOCGETA -> iflag 1 oflag 0 cflag 9b00 lflag 0 ispeed 300
  1   7891              ttydev_ioctl:return          avrdude TIOCGETA -> iflag 1 oflag 0 cflag 9b00 lflag 0 ispeed 300
  1   7890               ttydev_ioctl:entry          avrdude TIOCSETA iflag 1 oflag 0 cflag 9f00 lflag 0 ispeed 115200
  1   7890               ttydev_ioctl:entry          avrdude FIONBIO flags 0
  1   7890               ttydev_ioctl:entry          avrdude FIOASYNC flags 0
  0   7891              ttydev_ioctl:return          avrdude TIOCGETA -> iflag 1 oflag 0 cflag 9f00 lflag 0 ispeed 115200
  0   7891              ttydev_ioctl:return          avrdude TIOCGETA -> iflag 1 oflag 0 cflag 9f00 lflag 0 ispeed 115200
  0   7890               ttydev_ioctl:entry          avrdude TIOCSETA iflag 1 oflag 0 cflag 9b00 lflag 0 ispeed 300
  0   7890               ttydev_ioctl:entry          avrdude FIONBIO flags 0
  0   7890               ttydev_ioctl:entry          avrdude FIOASYNC flags 0
  1   7891              ttydev_ioctl:return          avrdude TIOCGETA -> iflag 1 oflag 0 cflag 9b00 lflag 0 ispeed 300
  1   7891              ttydev_ioctl:return          avrdude TIOCGETA -> iflag 1 oflag 0 cflag 9b00 lflag 0 ispeed 300
  1   7890               ttydev_ioctl:entry          avrdude TIOCSETA iflag 1 oflag 0 cflag 9f00 lflag 0 ispeed 115200
  1   7890               ttydev_ioctl:entry          avrdude FIONBIO flags 0
  1   7890               ttydev_ioctl:entry          avrdude FIOASYNC flags 0
  2   7891              ttydev_ioctl:return          avrdude TIOCMGET -> 7
  2   7890               ttydev_ioctl:entry          avrdude TIOCMSET flags 1
  2   7890               ttydev_ioctl:entry          avrdude TIOCSETAW iflag 2b02 oflag 3 cflag cb00 lflag 5cb ispeed 9600
```

dbuchwald, 2022-02-14:
Sorry for delay in reply, but I don't have much to add here until I get my PL2303 adapter (hopefully tomorrow). In the meantime I can point out some of the things I wrote before that I can see in your DTrace:

```
  1   7891              ttydev_ioctl:return        python3.7 TIOCGETA -> iflag 2b02 oflag 3 cflag cb00 lflag 5cb ispeed 9600
  1   7890               ttydev_ioctl:entry        python3.7 TIOCSETA iflag 2002 oflag 0 cflag df00 lflag 0 ispeed 115200
  2   7891              ttydev_ioctl:return          avrdude TIOCGETA -> iflag 2b02 oflag 3 cflag cb00 lflag 5cb ispeed 9600
  2   7890               ttydev_ioctl:entry          avrdude TIOCSETA iflag 1 oflag 0 cflag 9f00 lflag 0 ispeed 115200
```

This is what I meant about overwriting contents of the iflags/oflags/lflags/cflags by avrdude serial implementation. Python doesn't do that, it just disables the things it doesn't need, preserving some defaults. Problem is that from this specific dump I can't see anything really problematic - AVRDUDE fails to preserve HUPCL on CFLAG or IMAXBEL/BRKINT on IFLAG (on the other hand - maybe that's the issue? after all it's the break signal input that's failing...).

What I'm trying to figure out is why the hell we need these FIONBIO/FIOASYNC calls in AVRDUDE - they don't seem to do anything particularly useful, but probably not breaking anything either.

One other notable difference is that Python is closing connections before switching to 300 baud, and we are not doing it, just changing the baud rate. This might also cause issues, but it was introduced on purpose, to eliminate unnecessary spikes on RTS/DTR lines. We can always put it back in if needed.

Bottom line: after reading lots and lots of documentation and playing with some simple routines I feel we could attempt adding second POSIX serial driver to the mix - one based on the PySerial library - with the intention of replacing the old one some day. Thoughts?

MCUdude, 2022-02-14:
> One other notable difference is that Python is closing connections before switching to 300 baud, and we are not doing it, just changing the baud rate. This might also cause issues, but it was [...]

I hope a potential fix for this doesn't ruin the current behavior of the DTR/RTS lines, as the programmer shown in one of my earlier posts relies on know behavior. Would it be possible to detect that the USB to serial adapter is in fact a PL2303, and do the necessary "changes" at runtime to make this work? Maybe some USB VID/PID related check?

I did a quick Google search to read what people think about the PL2303, and those who write about it have mostly negative things to say. Apparently, the market was littered with clones, and Profilic decided to brick the counterfeit ones, leaving countless unusable devices left in the field. Personally, I prefer to use Silabs CP2102N's or CH340's since they're cheap and have good driver support on all platforms. How much should be sacrificed to get a not-that-good USB to serial adapter to work as a SerialUPDI programmer?

dl8dtl, 2022-02-14:
Well, so far I thought that the Windows driver of the PL2303 is just that crappy (I've bought three of these devices relatively cheap once as the guy selling it was upset with them), but maybe the hardware is "special", too. It does any "normal" job (character IO on serial line) without any notable problems here on FreeBSD though.
Hard to believe that counterfeit chips would be even worse. ;-)

MCUdude, 2022-02-14:
> Maybe it's really time for an overhaul of our Posix serial implementation.

Could maybe, just maybe an overhaul make arbitrary baud rates work with Avrdude? I know MacOS by default is limited to regular baud rates where 230400 is the highest, but I can confirm that "non-standard" ones work in the Arduino IDE terminal, such as 250000, 500000, 1M, and 2M.

dbuchwald, 2022-02-14:
Hey, I couldn't wait for the PL2303 hardware to arrive and decided to try and change the code anyway and see if that helps with anything. Could you try to clone and build AVRDUDE from my repo?

https://github.com/dbuchwald/avrdude

I tested it against my devices on Linux and it seems to work OK, but I'm curious whether it addresses any of the PL2303 issues.

Thanks in advance,
Dawid

MCUdude, 2022-02-14:
> Could you try to clone and build AVRDUDE from my repo?

I've tried your fork, but I can't get my PL2303 to work. Bear in mind that I wasn't able to get it working with pymcuprog either, so I don't really trust the PL2303 chip I currently have.

MCUdude, 2022-02-14:
> Are you using MacOS?

Yes, MacOS 10.14. I actually had to install the driver (from Profilic's website) for it to even show up as a serial device. I just tested it using [Coolterm](https://freeware.the-meiers.org), and I'm able to manually toggle DTR and RTS (tested with my oscilloscope). To my surprise, I can also use "custom baudrates" as well.

dbuchwald, 2022-02-14:
Ok, so you have PL2303 working under MacOS, but both AVRDUDE and pymcuprog don't seem to work with it? This is really interesting. Can't wait to get my hands on one of these weird animals :)

What about my version of AVRDUDE and your own adapter? Does it work, including the RTS/DTR operation? I tested CP2102 and it seems to work just fine, including the extra options...

MCUdude, 2022-02-14:
> What about my version of AVRDUDE and your own adapter? 

Do you mean the blue PCB? It works perfectly with your fork. All the way from 2400 to 230400 baud! And when I switch the programmer over to "AUTO" mode, I can use `-x rtsdtr=high` to force UPDI operation, while `-x rtsdtr=high` doesn't work, as expected, since this forces UART mode.

dbuchwald, 2022-02-14:
> Do you mean the blue PCB? It works perfectly with your fork. All the way from 2400 to 230400 baud! And when I switch the programmer over to "AUTO" mode, I can use `-x rtsdtr=high` to force UPDI operation, while `-x rtsdtr=high` doesn't work, as expected, since this forces UART mode.

Awesome, thank you for checking that! I will wait for Joerg's tests of PL2303 (or my own - whichever comes first) then. I really hope this issue gets sorted out :)

dl8dtl, 2022-02-14:
> Could you try to clone and build AVRDUDE from my repo?

No change for the PL2303 though.

```
% pymcuprog -t uart -u /dev/cuaU4 -d avr128db48 ping
Pinging device...
Ping response: 1E970C
Done.
% ./build_freebsd/src/avrdude -c serialupdi -P /dev/cuaU4 -p avr128db48 -t -vvvv
executable_abspath = /tmp/avrdude/build_freebsd/src/avrdude
executable_abspath_len = 38
executable_dirpath = /tmp/avrdude/build_freebsd/src
executable_dirpath_len = 30
sys_config = /tmp/avrdude/build_freebsd/src/avrdude.conf
sys_config_found = true

avrdude: Version 6.99-20220214 (5e81d90)
         Copyright (c) Brian Dean, http://www.bdmicro.com/
         Copyright (c) Joerg Wunsch

         System wide configuration file is "/tmp/avrdude/build_freebsd/src/avrdude.conf"
         User configuration file is "/home/joerg/.avrduderc"

         Using Port                    : /dev/cuaU4
         Using Programmer              : serialupdi
avrdude: Opening serial port...
avrdude: Sending double break
avrdude: Send: . [00] 
avrdude: ser_recv(): programmer is not responding
avrdude: Send: . [00] 
avrdude: Recv: . [00] 
         AVR Part                      : AVR128DB48
         RESET disposition             : dedicated
         RETRY pulse                   : SCK
         Serial program mode           : yes
         Parallel program mode         : yes
         Memory Detail                 :

[omitted]
         Programmer Type : serialupdi
         Description     : SerialUPDI

avrdude: STCS 0x08 to address 0x03
avrdude: Sending 3 bytes [0x55, 0xc3, 0x08]
avrdude: Send: U [55] . [c3] . [08] 
avrdude: ser_recv(): programmer is not responding
avrdude: STCS 0x80 to address 0x02
avrdude: Sending 3 bytes [0x55, 0xc2, 0x80]
avrdude: Send: U [55] . [c2] . [80] 
avrdude: ser_recv(): programmer is not responding
avrdude: LDCS from 0x00
avrdude: Sending 2 bytes [0x55, 0x80]
avrdude: Send: U [55] . [80] 
avrdude: ser_recv(): programmer is not responding
avrdude: ser_recv(): programmer is not responding
avrdude: serialupdi_recv(): programmer is not responding
avrdude: Check failed
avrdude: Datalink not active, resetting...
avrdude: Sending double break
avrdude: Send: . [00] 
avrdude: ser_recv(): programmer is not responding
avrdude: Send: . [00] 
avrdude: Recv: . [00] 
avrdude: STCS 0x08 to address 0x03
avrdude: Sending 3 bytes [0x55, 0xc3, 0x08]
avrdude: Send: U [55] . [c3] . [08] 
avrdude: ser_recv(): programmer is not responding
avrdude: STCS 0x80 to address 0x02
avrdude: Sending 3 bytes [0x55, 0xc2, 0x80]
avrdude: Send: U [55] . [c2] . [80] 
avrdude: Recv: . [00] U [55] . [c2] 
avrdude: LDCS from 0x00
avrdude: Sending 2 bytes [0x55, 0x80]
avrdude: Send: U [55] . [80] 
avrdude: Recv: . [80] U [55] 
avrdude: Recv: . [80] 
avrdude: Received 1 bytes [0x80]
avrdude: UDPI init OK
avrdude: UPDI link initialization OK
avrdude: LDCS from 0x0B
avrdude: Sending 2 bytes [0x55, 0x8b]
avrdude: Send: U [55] . [8b] 
avrdude: Recv: U [55] . [8b] 
avrdude: ser_recv(): programmer is not responding
avrdude: serialupdi_recv(): programmer is not responding
avrdude: Sending reset request
avrdude: STCS 0x59 to address 0x08
avrdude: Sending 3 bytes [0x55, 0xc8, 0x59]
avrdude: Send: U [55] . [c8] Y [59] 
avrdude: Recv: U [55] . [c8] Y [59] 
avrdude: Sending release reset request
avrdude: STCS 0x00 to address 0x08
avrdude: Sending 3 bytes [0x55, 0xc8, 0x00]
avrdude: Send: U [55] . [c8] . [00] 
avrdude: Recv: U [55] . [c8] . [00] 
avrdude: STCS 0x08 to address 0x03
avrdude: Sending 3 bytes [0x55, 0xc3, 0x08]
avrdude: Send: U [55] . [c3] . [08] 
avrdude: Recv: U [55] . [c3] . [08] 
avrdude: STCS 0x80 to address 0x02
avrdude: Sending 3 bytes [0x55, 0xc2, 0x80]
avrdude: Send: U [55] . [c2] . [80] 
avrdude: Recv: U [55] . [c2] . [80] 
avrdude: LDCS from 0x00
avrdude: Sending 2 bytes [0x55, 0x80]
avrdude: Send: U [55] . [80] 
avrdude: Recv: U [55] . [80] 
avrdude: ser_recv(): programmer is not responding
avrdude: serialupdi_recv(): programmer is not responding
avrdude: Check failed
avrdude: Datalink not active, resetting...
avrdude: Sending double break
avrdude: Send: . [00] 
avrdude: ser_recv(): programmer is not responding
avrdude: Send: . [00] 
avrdude: Recv: . [00] 
avrdude: STCS 0x08 to address 0x03
avrdude: Sending 3 bytes [0x55, 0xc3, 0x08]
avrdude: Send: U [55] . [c3] . [08] 
avrdude: Recv: . [00] U [55] . [c3] 
avrdude: STCS 0x80 to address 0x02
avrdude: Sending 3 bytes [0x55, 0xc2, 0x80]
avrdude: Send: U [55] . [c2] . [80] 
avrdude: Recv: . [08] U [55] . [c2] 
avrdude: LDCS from 0x00
avrdude: Sending 2 bytes [0x55, 0x80]
avrdude: Send: U [55] . [80] 
avrdude: Recv: . [80] U [55] 
avrdude: Recv: . [80] 
avrdude: Received 1 bytes [0x80]
avrdude: UDPI init OK
avrdude: initialization failed, rc=-1
         Double check connections and try again, or use -F to override
         this check.

avrdude: Leaving NVM programming mode
avrdude: Sending reset request
avrdude: STCS 0x59 to address 0x08
avrdude: Sending 3 bytes [0x55, 0xc8, 0x59]
avrdude: Send: U [55] . [c8] Y [59] 
avrdude: Recv: U [55] . [c8] Y [59] 
avrdude: Sending release reset request
avrdude: STCS 0x00 to address 0x08
avrdude: Sending 3 bytes [0x55, 0xc8, 0x00]
avrdude: Send: U [55] . [c8] . [00] 
avrdude: Recv: U [55] . [c8] . [00] 
avrdude: STCS 0x0C to address 0x03
avrdude: Sending 3 bytes [0x55, 0xc3, 0x0c]
avrdude: Send: U [55] . [c3] . [0c] 
avrdude: Recv: U [55] . [c3] . [0c] 

avrdude done.  Thank you.
```
I'll see to do some more investigations tomorrow.

dbuchwald, 2022-02-15:
> No change for the PL2303 though.

Crap. I was really hoping that cleaning up the initialisation code would help, but apparently it’s not that easy. 

What I find the most disturbing is that your adapter seems to have skipped several messages as if it was literally offline, then woke up and started echoing data before and after the disruption.

And all that just works when called from Python…

dl8dtl, 2022-02-16:
> And all that just works when called from Python…

That's would puzzles me as well. Otherwise, I wouldn't shed a tear for the PL2303, and just document it's unsuitable.

dbuchwald, 2022-02-16:
> That's would puzzles me as well.

Wait until you hear this. Last night I finally received my own PL2303 adapter. What surprised me was that it had USB-C connector on one end, which is uncommon for such devices, but didn't think much of it.

Turns out, there are different versions of PL2303 chips, they are still being manufactured and new versions introduced. Mine is based on PL2303GS, and that would have been fine, but the revision number is 6.05 - and this is not yet supported by Linux kernel (I'm using 5.13, but even 5.17-rc4 doesn't). 

https://github.com/torvalds/linux/blob/master/drivers/usb/serial/pl2303.c#L421

The funny part is that dmesg told me to send e-mail to specific address with the details of the unrecognised device and I got automatic reply that the message was not delivered because I'm not allowed to send e-mails to it. I mean, seriously...

I could change the module and recompile the kernel, but it feels like the rabbit hole it "a bit" too deep...

I ordered two more devices online, hoping that at least one of them would work. As in "being recognised by kernel, working with pymcuprog, but not working with avrdude".

Seriously, this saga is getting ridiculous.

dbuchwald, 2022-02-16:
> > Could you try to clone and build AVRDUDE from my repo?
> 
> I've tried your fork, but I can't get my PL2303 to work. Bear in mind that I wasn't able to get it working with pymcuprog either, so I don't really trust the PL2303 chip I currently have.

Can you try removing the resistor from between the diode and the UPDI pin on the target device? What I mean is that you just connect diode between RX and TX, and then RX to UPDI and that's it? This is the setup I got PL2303 work for me (both under Linux and MacOS BigSur).

MCUdude, 2022-02-16:
> Can you try removing the resistor from between the diode and the UPDI pin on the target device? What I mean is that you just connect diode between RX and TX, and then RX to UPDI and that's it? This is the setup I got PL2303 work for me (both under Linux and MacOS BigSur).

Yes, I can try this later this evening. When I looked at the PL2303 data lines on the scope a few days ago I noticed that the voltage levels weren't 5 or 5.2V as my mac outputs, but rather something closer to 3.9V instead. It may be that the PL2303's TX pins have a really high output impedance.

MCUdude, 2022-02-16:
Nope, my PL2303 can't be used as a SerialUPDI programmer, even with the 470R resistor shorted.

dbuchwald, 2022-02-16:
> Nope, my PL2303 can't be used as a SerialUPDI programmer, even with the 470R resistor shorted.

Interesting. Thanks for checking that. Do you also happen to know which specific revision of the device it is? Maybe we should put this information in the avrdude notes?

dl8dtl, 2022-02-16:
> Turns out, there are different versions of PL2303 chips, they are still being manufactured and new versions introduced. Mine is based on PL2303GS, and that would have been fine, but the revision [...]

Oops.

> EDIT: @dl8dtl could you post here the output of lsusb -vv for the device you are using? It will make it easier for me to track down specific version of the adapter, because it probably matters...

Currently, it's on my FreeBSD machine. `usbconfig -v` output is:

```
# usbconfig -d4.6 -v
ugen4.6: <Prolific Technology Inc. USB-Serial Controller> at usbus4, cfg=0 md=HOST spd=FULL (12Mbps) pwr=ON (100mA)
ugen4.6.0: uplcom0: <Prolific Technology Inc. USB-Serial Controller, class 0/0, rev 1.10/3.00, addr 6>

  bLength = 0x0012 
  bDescriptorType = 0x0001 
  bcdUSB = 0x0110 
  bDeviceClass = 0x0000  <Probed by interface class>
  bDeviceSubClass = 0x0000 
  bDeviceProtocol = 0x0000 
  bMaxPacketSize0 = 0x0040 
  idVendor = 0x067b 
  idProduct = 0x2303 
  bcdDevice = 0x0300 
  iManufacturer = 0x0001  <Prolific Technology Inc.>
  iProduct = 0x0002  <USB-Serial Controller>
  iSerialNumber = 0x0000  <no string>
  bNumConfigurations = 0x0001 
…
```

Is that sufficient?

> EDIT: Last update for now - after removing the resistor between diode and UPDI pin on the device I can now use both pymcuprog and avrdude successfully:

Hmm, that's the resistor in the Tx line, is it?

dl8dtl, 2022-02-16:
Well, if I dig around a bit, I should be able to find another of those PL2303 here. I remember I once bought three of them.

dbuchwald, 2022-02-16:
> Oops.

Actually it turned out fine, I created my own version of the kernel module and it works now just fine. It does work with avrdude too, unfortunately, so I can't proceed with investigation until I get another device.

> ```
>   bcdUSB = 0x0110 
>   bcdDevice = 0x0300 
> ```
> 
> Is that sufficient?

Yes, it is. According to the kernel module this would be PL2303HX version - exactly as the one I ordered. I just hope this is what I will receive in couple of days :)

> Hmm, that's the resistor in the Tx line, is it?

Depends on how you look at it :) For me UART is connected to UPDI via the RX line and TX line is connected only to the diode, like so:

```
--------------------                                 To Target device
                DTR|                                  __________________
                Rx |--------------,------------------| UPDI----------------->
                Tx |-------|<|---'          .--------| Gnd
                Vcc|---------------------------------| Vcc
                CTS|                     .`          |__________________
                Gnd|--------------------'
--------------------
```

MCUdude, 2022-02-16:
This is my adapter:

```
$ lsusb -vv

...

Bus 001 Device 009: ID 067b:2303 Prolific Technology, Inc. PL2303 Serial Port
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               1.10
  bDeviceClass            0 (Defined at Interface level)
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x067b Prolific Technology, Inc.
  idProduct          0x2303 PL2303 Serial Port
  bcdDevice            3.00
  iManufacturer           1 
  iProduct                2 
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength           39
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0x80
      (Bus Powered)
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           3
      bInterfaceClass       255 Vendor Specific Class
      bInterfaceSubClass      0 
      bInterfaceProtocol      0 
      iInterface              0 
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x000a  1x 10 bytes
        bInterval               1
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x02  EP 2 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x83  EP 3 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               0
```

dbuchwald, 2022-02-16:
> This is my adapter:
> 
> ```
> $ lsusb -vv
> 
>   bcdUSB               1.10
>   bcdDevice            3.00
> ```

Amazing, it's exactly the same device that @dl8dtl is using. The only difference is that you are trying to use it under MacOS, correct?

MCUdude, 2022-02-16:
> The only difference is that you are trying to use it under MacOS, correct?

Correct. I used a Linux based media box just to run `lsusb`. I tried to install pymcuprog on it just to see if it would work, but for some reason I'm not able to install it:

```
    /home/chromebox/.local/lib/python3.6/site-packages/setuptools/command/install.py:37: SetuptoolsDeprecationWarning: setup.py install is deprecated. Use build and pip and other standards-based tools.
      setuptools.SetuptoolsDeprecationWarning,
    running build
    running build_ext
    building 'hid' extension
    error: unknown file type '.pxd' (from 'chid.pxd')
    
    ----------------------------------------
Command "/usr/bin/python3 -u -c "import setuptools, tokenize;__file__='/tmp/pip-build-qi3u61vm/hidapi/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" install --record /tmp/pip-q0w7gqy1-record/install-record.txt --single-version-externally-managed --compile --user --prefix=" failed with error code 1 in /tmp/pip-build-qi3u61vm/hidapi/
```

Anyways, I connected it to a Windows laptop, found some really old drivers and eventually got it to show up as a COM port. The latest Profilic drivers just made the adapter unusable, saying `PL2303HXA PHASED OUT SINCE 2012. PLEASE CONTACT YOUR CUSTOMER SUPPLIER` in the Windows device manager. After installed a driver dated from 2008 I got it working, and I can confirm that the same adapter that my Mac doesn't like **works fine in Windows** using a really old driver. What a terrible mess...

dl8dtl, 2022-02-16:
> and I can confirm that the same adapter that my Mac doesn't like **works fine in Windows** using a really old driver

:-o

dl8dtl, 2022-02-16:
After learning quite a bit about DTrace, I could finally get some more useful trace out of it.
[avrdude.txt](https://github.com/avrdudes/avrdude/files/8083641/avrdude.txt)
[pymcuprog.txt](https://github.com/avrdudes/avrdude/files/8083642/pymcuprog.txt)
What puzzles me is that pymcuprog looks really straightforward, I don't even see that speed change to kick out a BREAK condition there. The just open the port, fiddle with the tty flags, and then send their stuff right away, getting the correct responses.

dl8dtl, 2022-02-16:
OK, when tracing a session where their first initialization fails, they also set the baud rate to 300, send a '\0', read it back, but then indeed reopen the device. They also set VMIN to 0 (we don't touch it, so it remains at 1), but changing that doesn't help either.

By bumping the overal serial timeout in `updi_link.c` to 200 ms, I could always correctly read back the echo of the BREAK:
```
avrdude: Opening serial port...
avrdude: Sending double break
avrdude: Send: . [00] 
avrdude: Recv: . [00] 
avrdude: Send: . [00] 
avrdude: Recv: . [00] 
```
With just 100 ms, the first read always returned a "failed".

With that, I could *once* even read the signature …

dl8dtl, 2022-02-16:
The latter tests have all been done with your code, @dbuchwald

dl8dtl, 2022-02-16:
Also, with the increased read timeout, the "shifted" replies are gone I think:
```
% ./build_freebsd/src/avrdude -c serialupdi -P /dev/cuaU4 -p avr128db48 -vvvv
executable_abspath = /tmp/avrdude/build_freebsd/src/avrdude
executable_abspath_len = 38
executable_dirpath = /tmp/avrdude/build_freebsd/src
executable_dirpath_len = 30
sys_config = /tmp/avrdude/build_freebsd/src/avrdude.conf
sys_config_found = true

avrdude: Version 6.99-20220214 (5e81d90)
         Copyright (c) Brian Dean, http://www.bdmicro.com/
         Copyright (c) Joerg Wunsch

         System wide configuration file is "/tmp/avrdude/build_freebsd/src/avrdude.conf"
         User configuration file is "/home/joerg/.avrduderc"

         Using Port                    : /dev/cuaU4
         Using Programmer              : serialupdi
avrdude: Opening serial port...
avrdude: Sending double break
avrdude: Send: . [00] 
avrdude: Recv: . [00] 
avrdude: Send: . [00] 
avrdude: Recv: . [00] 
         AVR Part                      : AVR128DB48
         RESET disposition             : dedicated
         RETRY pulse                   : SCK
         Serial program mode           : yes
         Parallel program mode         : yes
         Memory Detail                 :[omitted]
         Programmer Type : serialupdi
         Description     : SerialUPDI

avrdude: STCS 0x08 to address 0x03
avrdude: Sending 3 bytes [0x55, 0xc3, 0x08]
avrdude: Send: U [55] . [c3] . [08] 
avrdude: ser_recv(): programmer is not responding
avrdude: STCS 0x80 to address 0x02
avrdude: Sending 3 bytes [0x55, 0xc2, 0x80]
avrdude: Send: U [55] . [c2] . [80] 
avrdude: ser_recv(): programmer is not responding
avrdude: LDCS from 0x00
avrdude: Sending 2 bytes [0x55, 0x80]
avrdude: Send: U [55] . [80] 
avrdude: ser_recv(): programmer is not responding
avrdude: ser_recv(): programmer is not responding
avrdude: serialupdi_recv(): programmer is not responding
avrdude: Check failed
avrdude: Datalink not active, resetting...
avrdude: Sending double break
avrdude: Send: . [00] 
avrdude: Recv: . [00] 
avrdude: Send: . [00] 
avrdude: Recv: . [00] 
avrdude: STCS 0x08 to address 0x03
avrdude: Sending 3 bytes [0x55, 0xc3, 0x08]
avrdude: Send: U [55] . [c3] . [08] 
avrdude: ser_recv(): programmer is not responding
avrdude: STCS 0x80 to address 0x02
avrdude: Sending 3 bytes [0x55, 0xc2, 0x80]
avrdude: Send: U [55] . [c2] . [80] 
avrdude: ser_recv(): programmer is not responding
avrdude: LDCS from 0x00
avrdude: Sending 2 bytes [0x55, 0x80]
avrdude: Send: U [55] . [80] 
avrdude: ser_recv(): programmer is not responding
avrdude: ser_recv(): programmer is not responding
avrdude: serialupdi_recv(): programmer is not responding
avrdude: Check failed
avrdude: Restoring datalink failed
avrdude: UPDI link initialization failed
avrdude: initialization failed, rc=-1
         Double check connections and try again, or use -F to override
         this check.

avrdude: Leaving NVM programming mode
avrdude: Sending reset request
avrdude: STCS 0x59 to address 0x08
avrdude: Sending 3 bytes [0x55, 0xc8, 0x59]
avrdude: Send: U [55] . [c8] Y [59] 
avrdude: ser_recv(): programmer is not responding
avrdude: Sending release reset request
avrdude: STCS 0x00 to address 0x08
avrdude: Sending 3 bytes [0x55, 0xc8, 0x00]
avrdude: Send: U [55] . [c8] . [00] 
avrdude: ser_recv(): programmer is not responding
avrdude: STCS 0x0C to address 0x03
avrdude: Sending 3 bytes [0x55, 0xc3, 0x0c]
avrdude: Send: U [55] . [c3] . [0c] 
avrdude: ser_recv(): programmer is not responding

avrdude done.  Thank you.
```

dbuchwald, 2022-02-17:
Hey, I just got information that the new adapter should be delivered today, so I might be able to do some testing tonight/tomorrow. Hopefully, of course. Oh, and I'm working from the office today, so I don't have access to any of my hardware.

In the meantime, @dl8dtl, could you shortly explain the procedure for obtaining the DTrace output? I looked at the files you provided and I must say I can't make sense out of it, like here:

```
  0  49689                     write:return write, 3
             0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f  0123456789abcdef
         0: 55 c3 08                                         U..

  0  49687                      read:return read, 3
             0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f  0123456789abcdef
         0: 00 c3 08    
```

Like why the hell is it echoing 0x00 instead of 0x55? I also need to understand why it sets the `iflags` to 2102 where it should have set it to 2002.

As for the different flow for **pymcuprog** you are observing - it's actually intentional. See, NVM programmer is a simple state machine, and **pymcuprog** assumes certain stable starting state. However, as I noticed in my testing, if you interrupt NVM programming session (killing **pymcuprog** during EEPROM write for instance), it will not be able to recover until you unplug the device and plug it back in.

The only way around this was to start **every session** with reset operation, and this is why you can see these drops to 300 baud. To be fair, we shouldn't be doing that either, just use POSIX send serial break operation, but I played with it for a while and it turns out that the MacOS is the oddball here, ignoring duration parameter. It works, but is just very slow as a result, freezing for 2 seconds each time it has to reset the connection - and it does it every time.

Bottom line: it's like a perfect storm of counterfeit abandoned chips with strange drivers, weird POSIX libraries, surprising Python implementation, Microchip's hacks and my "less-than-optimal" implementation...

dl8dtl, 2022-02-17:
> In the meantime, @dl8dtl, could you shortly explain the procedure for obtaining the DTrace output?

Well, I don't really trust that much either.
I copied a bit from dtruss, but dtruss cannot trace the ioctls.
Maybe it's a better idea to "plug up" dtruss, and teach it ioctls instead of hacking my own.

dbuchwald, 2022-02-17:
@dl8dtl some good news, finally. I received PL2303HX and it's the same device you have. It also **does fail** in the same way you described it - failing with avrdude and working with pymcuprog.

Actually, I have some more good news: I just pushed another commit to my repo which fixes the problem, and I can now use PL2303HX with avrdude:

```
lsusb -d 067b:2303 -vv

Bus 001 Device 009: ID 067b:2303 Prolific Technology, Inc. PL2303 Serial Port / Mobile Action MA-8910P
Couldn't open device, some information will be missing
Device Descriptor:
  bcdUSB               1.10
  idVendor           0x067b Prolific Technology, Inc.
  idProduct          0x2303 PL2303 Serial Port / Mobile Action MA-8910P
  bcdDevice            3.00

```

```
./avrdude -C avrdude.conf -c serialupdi -P /dev/ttyUSB0 -p avr128db64 -U eeprom:r:-:h

avrdude: UPDI link initialization OK
avrdude: NVM type 2: 24-bit, word oriented write
avrdude: Entering NVM programming mode
avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.01s

avrdude: Device signature = 0x1e970b (probably avr128db64)
avrdude: reading eeprom memory:

Reading | ################################################## | 100% 0.98s

avrdude: writing output file "<stdout>"
0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff
avrdude: Leaving NVM programming mode

avrdude done.  Thank you.
```

Can you, please, confirm that the fix works for you too?

If it does (and I really hope it does), don't open the champagne just yet. We still have one more issue I need to fix. See, if you interrupt the NVM session (Ctrl+C) avrdude will not work until you disconnect the power from the AVR, or run ping with pymcuprog again - and this is the part that bothers me: how comes pymcuprog seems to be able to "wake up" AVR using it's "double break" while avrdude can't do it? So, more investigation for me, but that's for tomorrow, I'm really tired now...

MCUdude, 2022-02-17:
It still doesn't work on my mac, but I expect the drivers is causing this...

dl8dtl, 2022-02-17:
> Can you, please, confirm that the fix works for you too?

I still need to bump the timeout:
```
diff --git a/src/updi_link.c b/src/updi_link.c
index 37989c2..ecb7b39 100644
--- a/src/updi_link.c
+++ b/src/updi_link.c
@@ -53,7 +53,7 @@ static void updi_set_rtsdtr_mode(PROGRAMMER* pgm)
 
 static int updi_physical_open(PROGRAMMER* pgm, int baudrate, unsigned long cflags)
 {
-  serial_recv_timeout = 100;
+  serial_recv_timeout = 200;
   union pinfo pinfo;
 
   pinfo.serialinfo.baud = baudrate;
```
But then, it works:
```
% ./build_freebsd/src/avrdude -c serialupdi -P /dev/cuaU4 -p avr128db48

avrdude: UPDI link initialization OK
avrdude: Device is in SLEEP mode
avrdude: NVM type 2: 24-bit, word oriented write
avrdude: Entering NVM programming mode
avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.01s

avrdude: Device signature = 0x1e970c (probably avr128db48)
avrdude: Leaving NVM programming mode

avrdude done.  Thank you.
```

Congrats!

dl8dtl, 2022-02-17:
Btw., the serial timeout stuff is due for a major overhaul. In my opinion, before releasing 7.0.
I've also got that patch for the WiFi programmer around that needs a longer timeout, but I don't like the way that patch introduces just another crock around the actual problem.

dbuchwald, 2022-02-19:
I think I have final version of the serialupdi code sorted out, ready for your testing in my repo. I tested it with FT230RL, CP2102, PL2303HX and PL2303GS. Each adapter works and avrdude can recover from previously failed session.

I need your help, @dl8dtl, @MCUdude and @mariusgreuel - can you guys test drive my version from here:

https://github.com/dbuchwald/avrdude

And let me know if you have any issues with your hardware?

And, just for the record:

```
./avrdude -C avrdude.conf -c serialupdi -P /dev/ttyUSB0 -p avr128db64 -U eeprom:r:-:h

avrdude: UPDI link initialization OK
avrdude: NVM type 2: 24-bit, word oriented write
avrdude: Entering NVM programming mode
avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.01s

avrdude: Device signature = 0x1e970b (probably avr128db64)
avrdude: reading eeprom memory:

Reading | ################                                   | 32% 0.32s^C

./avrdude -C avrdude.conf -c serialupdi -P /dev/ttyUSB0 -p avr128db64 -U eeprom:r:-:h

avrdude: UPDI link initialization OK
avrdude: Device in NVM programming state, leaving programming mode
avrdude: NVM type 2: 24-bit, word oriented write
avrdude: Entering NVM programming mode
avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.01s

avrdude: Device signature = 0x1e970b (probably avr128db64)
avrdude: reading eeprom memory:

Reading | ################################################## | 100% 1.00s

avrdude: writing output file "<stdout>"
0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff
avrdude: Leaving NVM programming mode

avrdude done.  Thank you.

lsusb
Bus 002 Device 003: ID 0bda:8153 Realtek Semiconductor Corp. RTL8153 Gigabit Ethernet Adapter
Bus 002 Device 002: ID 0424:5807 Microchip Technology, Inc. (formerly SMSC) Hub
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 003: ID 8087:0a2b Intel Corp. Bluetooth wireless interface
Bus 001 Device 009: ID 046d:085c Logitech, Inc. C922 Pro Stream Webcam
Bus 001 Device 008: ID 413c:2113 Dell Computer Corp. KB216 Wired Keyboard
Bus 001 Device 007: ID 05e3:0610 Genesys Logic, Inc. Hub
Bus 001 Device 006: ID 0bda:4014 Realtek Semiconductor Corp. USB Audio
Bus 001 Device 016: ID 77a1:01a2 Kingst Kingst Logic Analyzer
Bus 001 Device 023: ID 067b:2303 Prolific Technology, Inc. PL2303 Serial Port / Mobile Action MA-8910P
Bus 001 Device 002: ID 0424:2807 Microchip Technology, Inc. (formerly SMSC) Hub
Bus 001 Device 005: ID 1bcf:2b96 Sunplus Innovation Technology Inc. Integrated_Webcam_HD
Bus 001 Device 017: ID 046d:c52b Logitech, Inc. Unifying Receiver
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

dbuchwald, 2022-02-19:
One more update - when I finally got it to work I started wondering how much of the fix was actually necessary to get it to work, and as it turns out, no changes to ser_posix.c were required. This it the simplified version of the fix for which I have created PR already:

https://github.com/avrdudes/avrdude/pull/885

```
diff --git a/src/updi_link.c b/src/updi_link.c
index 684b6b8..541a70a 100644
--- a/src/updi_link.c
+++ b/src/updi_link.c
@@ -53,7 +53,7 @@ static void updi_set_rtsdtr_mode(PROGRAMMER* pgm)
 
 static int updi_physical_open(PROGRAMMER* pgm, int baudrate, unsigned long cflags)
 {
-  serial_recv_timeout = 100;
+  serial_recv_timeout = 1000;
   union pinfo pinfo;
 
   pinfo.serialinfo.baud = baudrate;
@@ -155,6 +155,8 @@ static int updi_physical_send_double_break(PROGRAMMER * pgm)
   serial_send(&pgm->fd, buffer, 1);
   serial_recv(&pgm->fd, buffer, 1);
 
+  serial_drain(&pgm->fd, 0);
+
   if (serial_setparams(&pgm->fd, pgm->baudrate? pgm->baudrate: 115200, SERIAL_8E2) < 0) {
     return -1;
   }
@@ -196,10 +198,14 @@ int updi_physical_sib(PROGRAMMER * pgm, unsigned char * buffer, uint8_t size)
 
 int updi_link_open(PROGRAMMER * pgm) 
 {
+  char init_buffer[1];
+
   if (updi_physical_open(pgm, pgm->baudrate? pgm->baudrate: 115200, SERIAL_8E2) < 0) {
     return -1;
   }
-  return updi_physical_send_double_break(pgm);
+
+  init_buffer[0]=UPDI_BREAK;
+  return updi_physical_send(pgm, init_buffer, 1);
 }
 
 void updi_link_close(PROGRAMMER * pgm)
```

[4 more replies]

closed 2022-02-20 by dl8dtl