avrdudes/avrdude #875: UPDI via EDBG: fails when no target connected

dl8dtl, 2022-02-12:
So well, OK, obviously no target communication can happen in that situation. ;-)
However, parameter setup on the EDBG itself ought to work when using the `-F` option – that way it could e.g. be done on an STK500 without a target connection.
I was trying to adjust Vtarget on an Curiosity Nano where the actual target is disconnected.
```
% ./build_freebsd/src/avrdude -c pkobn_updi -p avr128db48 -tF

         Vtarget         : 3.31 V
         PDI/UPDI clock Xmega/megaAVR : 100 kHz

avrdude: Short read, read only 0 out of 64 bytes
avrdude: jtag3_edbg_recv(): Unexpected response 0x12
avrdude: retrying with external reset applied
avrdude: jtag3_edbg_send(): Unexpected response 0x81, 0x11
avrdude: jtag3_edbg_recv(): Unexpected response 0x80
avrdude: retrying with external reset applied
avrdude: JTAGEN fuse disabled?
avrdude: initialization failed, rc=-1
avrdude: AVR device initialized and ready to accept instructions
avrdude: Device signature = 0x000000 (retrying)
avrdude: Device signature = 0x000000 (retrying)
avrdude: Device signature = 0x000000
avrdude: Yikes!  Invalid device signature.
avrdude: Expected signature for AVR128DB48 is 1E 97 0C
avrdude> vtarg 4
>>> vtarg 4 
avrdude: jtag3_edbg_send(): Unexpected response 0x81, 0x11
avrdude: jtag3_edbg_recv(): Unexpected response 0x80
avrdude: jtag3_set_vtarget(): cannot obtain V[aref]
avrdude (vtarg): failed to set V[target] (rc = -1)
avrdude> q
>>> q 
avrdude: jtag3_edbg_send(): Unexpected response 0x81, 0x11
avrdude: jtag3_edbg_recv(): Unexpected response 0x80
avrdude: jtag3_edbg_send(): Unexpected response 0x81, 0x11
avrdude: jtag3_edbg_recv(): Unexpected response 0x80
avrdude: jtag3_edbg_signoff(): unexpected response 0x81, 0x11
avrdude: jtag3_edbg_signoff(): unexpected response 0x01, 0x00

avrdude done.  Thank you.
```

MCUdude, 2022-02-12:
Just a thought, maybe the verbosity should be reduced a little? Is it necessary to print all these `jtag3_edbg_send()`, `jtag3_edbg_recv()`, `jtag3_edbg_signoff()` messages when no verbosity flag is set? 

Maybe set these messages to `MSG_NOTICE` instead of `MSG_INFO`? I can submit a PR for this if you like. But this doesn't fix the initial issue though.

dl8dtl, 2022-02-12:
> Just a thought, maybe the verbosity should be reduced a little?

Agreed.

MCUdude, 2022-02-14:
Are you able to set the target voltage using pymcuprog without actually having a target connected?

dl8dtl, 2022-02-14:
```
pyedbglib.pyedbglib_errors.PyedbglibNotSupportedError: System 'freebsd' not implemented for library 'hidapi'
```
:-(

Have to try it with Linux or MacOS.
Later.

closed 2022-02-18 by dl8dtl