avrdudes/avrdude #879: Make the terminal 'write' functionality more useful

MCUdude, 2022-02-17:
The terminal mode is really handy if you have a target where you want to just check something or do minor changes. However, I feel like the 'write' functionality has more potential.

The current syntax is this:
`write <memtype> <addr> <byte1> <byte2> ... <byteN>`

Features that would be neat to have:

```
# Fill user defined memory section with a known value (three dots as part of the syntax)
# write <memtype> <start addr> <...> <end addr> <byte to write to all memory locations>

# example
write eeprom 0x40 ... 0x80  0x00
```

```
# Write strings with \0 at the end
# write <memtype> <start addr> <string in quotes>

# Example (writes Hello World!\0 to EEPROM)
write eeprom 0x80 "Hello World!"
```

Any thoughts?

closed 2022-02-21 by dl8dtl

dl8dtl, 2022-02-21:
I'll try it on MacOS now.

dl8dtl, 2022-02-21:
Hmm, my Mac ports' `makeinfo` is version 4.8, from 2004 (!). :-/
My FreeBSD's `makeinfo` is version 6.8, from 2021.

I guess that explains it …

PR #880 (2022-02-21): Avrdude terminal write improvements
I've been working one some improvements regarding the Avrdude terminal write functionality. I have not yet updated any docks; I'd like to get some feedback first.

With this PR it's now possible to fill a memory section, write 8, 16 and 32-bit integers, characters and floats to memory. Negative numbers works too, and everything is stored in little endian.

Any thoughts? I'm all ears!

I have yet to add support for strings in double quotes, but this is quite a bit of work I believe.

#38 related
Closes #879 

```
avrdude: Device signature = 0x1e9650 (probably [... cut]