avrdudes/avrdude #883: Rework PicKit2 programmer to use libhidapi instead of libhid

mariusgreuel, 2022-02-18:
Assuming the PicKit2 programmer is a HID device, I suggest to rework the code to use libhidapi, which is available on all platforms. Then we can drop the remaining libhid support for Windows, including the CMake/Automake support.

#882 already removed libhid support for ser_avrdoper, so the pickit.c code is the last one to reference libhid.

mcuee, 2022-05-08:
This will be great.

We at libusb project recommend to use hidapi for HID device.

mcuee, 2022-05-09:
PICKit 2 is an HID device.

mcuee, 2022-06-22:
Not so sure if the following enhancement patch can be of some reference value here.
* https://github.com/avrdudes/avrdude/issues/692
> mcprog uses the MCP2210 USB-to-SPI interface.
> This code was successfully tested on Linux and MAC.
> 
> The patch started as a copy of the pickit2 code already shipped with avrdude, but as the libusb backend seems to be dysfunctional on MAC for quite some time, I replaced libusb by libhidapi.

MCUdude, 2022-07-07:
+1!

I was thinking I could help out with #1004, but my PICkit2 doesn't work with MacOS + Avrdude:

```
$ sudo ./avrdude -cpickit2 -patmega328p
Password:
Could not claim interface. Error code -13, Permission denied
You may need to run avrdude as root or set up correct usb port permissions.
pickit2_write_report failed (ec -2). No such file or directory
avrdude: initialization failed, rc=-1
         Double check connections and try again, or use -F to override
         this check.

avrdude done.  Thank you.
```

I'm not capable of doing this job by myself, but I can help out with testing if somebody is ready for a task like this!

mcuee, 2023-03-11:
Related request after this is done.
* https://github.com/avrdudes/avrdude/issues/958

mcuee, 2024-03-03:
Close this one as not planned until someone is actibely working on this issue.

This issue is also linked here.
https://github.com/avrdudes/avrdude/discussions/1710

closed 2024-03-03 by mcuee