avrdudes/avrdude #884: List of signing keys?

fauxpark, 2022-02-19:
Whilst updating the MSYS2 package to 6.4, the CI encountered a new PGP key which was not in the list of valid keys defined in the PKGBUILD (https://github.com/msys2/MINGW-packages/pull/10803). I am wondering, is there a list of canonical signing keys that will be used to sign avrdude? Can the other two keys be removed?

mariusgreuel, 2022-02-19:
Jörg can answer your question regarding the v6.4 release. 

@fauxpark , @DavidEGrayson Perhaps we can discuss how the upcoming releases should look like as well. 

Not sure if you have seen it, but I created a Github build action for MinGW that you can use for future reference: <https://github.com/avrdudes/avrdude/blob/main/.github/workflows/build.yml>

The automake config is still in the sources, but is going to be removed eventually in favor of CMake.

I also create a simple [deploy action](https://github.com/avrdudes/avrdude/blob/main/.github/workflows/deploy.yml), however, this one just creates the MSVC packages so far. Question for me is what you need in this deploy action so that you can do your job.

Looking at your [PKGBUILD](https://github.com/msys2/MINGW-packages/blob/master/mingw-w64-avrdude/PKGBUILD), you just need the sources, and a signature file, correct? Would you care to be notified when a new release is made?

@dl8dtl As creating the signature file can also be done in the Github action, I think it makes sense to create a new keypair that we use exclusively for the avrdude(s) deploy action. I have never tried it, but I assume you just store the private key as base64 in a Github secret (GPG_PRIVATE_KEY), and I just use that in the deploy action.

fauxpark, 2022-02-19:
> Not sure if you have seen it, but I created a Github build action for MinGW that you can use for future reference: [...]

Good to know, that will probably be useful when the next release is made with the CMake stuff in.

The build itself is not a problem at the moment, the step it was failing at was the verification of the signature which was solved by simply adding the new key to the list in the PKGBUILD. My question was more about whether this will be the only key used going forward (so the existing ones can be removed) and where this is documented.

> I also create a simple [deploy action](https://github.com/avrdudes/avrdude/blob/main/.github/workflows/deploy.yml?rgh-link-date=2022-02-19T11%3A37%3A56Z), however, this one just creates the MSVC packages so far. Question for me is what you need in this deploy action so that you can do your job.

MSYS2 uses Pacman and has [its own infrastructure for building and publishing packages](https://packages.msys2.org) - all it needs is the PKGBUILD.

> Looking at your [PKGBUILD](https://github.com/msys2/MINGW-packages/blob/master/mingw-w64-avrdude/PKGBUILD?rgh-link-date=2022-02-19T11%3A37%3A56Z), you just need the sources, and a signature file, correct? Would you care to be notified when a new release is made?

Correct, although the .sig file is not particularly required, it's recommended where possible.

I'm not part of the MSYS2 team, just a user and occasional contributor :)

DavidEGrayson, 2022-02-19:
This isn't really about automation.  Someone on your team is taking the trouble to make source tarballs, sign them with a PGP key, and upload them to:

DavidEGrayson, 2022-02-19:
This question really isn't really about automation.  The main question is who on the AVRDUDE team is trusted to make new source releases and upload them to https://download.savannah.gnu.org/releases/avrdude/ and how can we verify that the releases we download from there comes from one of those people, and not an attacker.

You provide PGP signatures on that site, presumably because you wanted to provide this kind of assurance.  However, if an attacker hacks into download.savannah.gnu.org and uses their own PGP key to sign a new release of AVRDUDE, and everyone just blindly trusts that key, then the PGP signatures actually provide no security.

It's good to think about how many different websites or accounts would need to be hacked in order for an attacker to make a non-suspicious release of AVRDUDE code.  One way to increase that number would be to have a list of trusted PGP keys in the AVRDUDE source code, or somewhere else on GitHub where it can be monitored by the leaders of AVRDUDE.

dl8dtl, 2022-02-19:
This is my key, for sure. I recently (well, some years ago now) upgraded my key to a stronger one, that's why you probably see a different signature than for older releases.
You'll find my keys on the usual keyservers.
I don't think we'll provide future releases on Savannah anymore.

mariusgreuel, 2022-02-19:
> This question isn't about automation

I brought up automation because it make sense to do future signing in the deploy action, meaning the keys are likely to change again. I may be jumping ahead here, but if you need to change fingerprints, we might as well provide you with the one used for future signing, plus the information you need to adapt your PGKBUILD file for future releases.

dl8dtl, 2022-02-20:
To finish this up, my current GPG key ID is 30D34EB (rsa4096/56628323218C669FF578705C7E9EADC3030D34EB in full), and the fingerprint is 5E84 F980 C3CA FD4B B584  1070 F48C A81B 69A8 5873.
Of course, this message here is at the same risk of being faked as the savannah download folders. ;-)

closed 2022-02-20 by dl8dtl

fauxpark, 2022-02-20:
I wasn't particularly worried about the authenticity of the new key. The issue that still has yet to be resolved is
> whether this will be the only key used going forward (so the existing ones can be removed) and where this is documented.

dl8dtl, 2022-02-20:
> whether this will be the only key used going forward

We simply haven't decided yet how future release will be distributed.

If we are going to continue to distribute them on Savannah (and I think, we'll at least do that in parallel to anything here on Github for a while), I'll sign them with my key. Savannah omly ever propagates uploaded files if they are accompanied by a detached sign of the respective developer(s).