avrdudes/avrdude #890: AVR-Eclipse plugin broken by missing -u commandline option

user-name42, 2022-02-23:
First, thanks to the developers for this tool.  My issue is really an avr-eclipse bug, but there doesn't seem to be much activity around that project lately.  My copy was broken by a recent upgrade to avrdude 6.99.  I fixed it by restoring the -u option like this: 

```
--- ./main.c.old	2022-02-23 01:26:32.097096344 -0500
+++ ./main.c.new	2022-02-23 01:30:04.319662520 -0500
@@ -614,6 +614,10 @@
         exit(0);
         break;
 
+      case 'u': /* when this option went missing, it broke the AVREclipse plugin  */
+	avrdude_message( MSG_INFO, "%s: -u isn't really a thing any more.  Ignoring.", progname, ch );
+	break;
+
       default:
         avrdude_message(MSG_INFO, "%s: invalid option -%c\n\n", progname, ch);
         usage();
```

dl8dtl, 2022-02-23:
Well, that's why the next release is going to get a new major version number: it intentionally contains not backwards compatible changes. The removal of the safemode feature is one of those.

Once 7.0 is released (hopefully not too far away now), the Eclipse plugin simply needs to adopt that change, and drop the -u option as well.

MCUdude, 2022-02-23:
@dl8dtl so it's not a viable option to print a slightly alloying "!!WARNING!!" saying that the -u flag is no longer valid, but then carry on with the programming as usual?

dl8dtl, 2022-02-23:
Well, we could, but there are other incompatible changes as well, e.g. in the config file. So if any user has a config file that tries to permanently disable safemode, this becomes invalid by the next release as well.

MCUdude, 2022-02-23:
> Well, we could, but there are other incompatible changes as well, e.g. in the config file. So if any user has a config file that tries to permanently disable safemode, this becomes invalid by the next release as well.

I think it would be in everyones interest that the transition from Avrdude 6.x to 7.x is as easy and plainless as possible. Remember, Avrdude is built into a lot of different tools out there, such as Arduino IDE and AVR-eclipse. Couldn't the "permanently disable safemode" option in the conf file just be ignored all together instead of throwing an error? I'm afraid the alternative is that other vendors settles with Avrdude 6.x because it's "good enough" and doesn't require any extra work.

dl8dtl, 2022-02-23:
hmm

dl8dtl, 2022-02-23:
scratching head … this might lead to endlessly keeping old stuff

MCUdude, 2022-03-20:
@dl8dtl we should take a stand on what to do.

For reference, the erase cycle counter, removed in 6.0 is still "present", since the -y flag is still reserved and doesn't terminate Avrdude if present:

```
$ ./avrdude  -C avrdude.conf -p atmega4808 -c pickit4_updi -t -y
avrdude: erase cycle counter no longer supported

         Vtarget                      : 5.05 V
         JTAG clock megaAVR/program   : 1000 kHz
         JTAG clock megaAVR/debug     : 100 kHz
         PDI/UPDI clock Xmega/megaAVR : 100 kHz

avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.05s

avrdude: Device signature = 0x1e9650 (probably m4808)
avrdude>
```

commit 981a6f1f (2022-03-20): Restore the -u option as a dummy.

Closes #890
closed 2022-03-20 by dl8dtl