avrdudes/avrdude #892: Pickit4 UPDI is writing at offset 0x4000 into flash instead of 0x0000.

merlinblack, 2022-02-24:
Compiled from latest source,on Linux.
Have a small test program for a ATmega4809 that I've exported the hex file from Arduino IDE.
Writing it using avrdude and a pickit4 in UPDI mode does not work as expected. The program is written but in the wrong place in flash.

Here's my command line and output:
```
#> ./avrdude -v -cpickit4_updi -pm4809 -Uflash:w:blinky.hex:i

avrdude: Version 6.99-20220221 (64cc54a)
         Copyright (c) Brian Dean, http://www.bdmicro.com/
         Copyright (c) Joerg Wunsch

         System wide configuration file is "/home/nigel/prog/avr/avrdude/build_linux/src/avrdude.conf"
         User configuration file is "/home/nigel/.avrduderc"
         User configuration file does not exist or is not a regular file, skipping

         Using Port                    : usb
         Using Programmer              : pickit4_updi
avrdude: Found CMSIS-DAP compliant device, using EDBG protocol
         AVR Part                      : ATmega4809
         RESET disposition             : dedicated
         RETRY pulse                   : SCK
         Serial program mode           : yes
         Parallel program mode         : yes
         Memory Detail                 :

                                           Block Poll               Page                       Polled
           Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           signature               0     0     0    0 no          3    1      0     0     0 0x00 0x00
           prodsig                 0     0     0    0 no         61   61      0     0     0 0x00 0x00
           sernum                  0     0     0    0 no         10    1      0     0     0 0x00 0x00
           osccal16                0     0     0    0 no          2    1      0     0     0 0x00 0x00
           osccal20                0     0     0    0 no          2    1      0     0     0 0x00 0x00
           tempsense               0     0     0    0 no          2    1      0     0     0 0x00 0x00
           osc16err                0     0     0    0 no          2    1      0     0     0 0x00 0x00
           osc20err                0     0     0    0 no          2    1      0     0     0 0x00 0x00
           fuses                   0     0     0    0 no          9   10      0     0     0 0x00 0x00
           fuse0       wdtcfg      0     0     0    0 no          1    1      0     0     0 0x00 0x00
           fuse1       bodcfg      0     0     0    0 no          1    1      0     0     0 0x00 0x00
           fuse2       osccfg      0     0     0    0 no          1    1      0     0     0 0x00 0x00
           fuse4       tcd0cfg     0     0     0    0 no          1    1      0     0     0 0x00 0x00
           fuse5       syscfg0     0     0     0    0 no          1    1      0     0     0 0x00 0x00
           fuse6       syscfg1     0     0     0    0 no          1    1      0     0     0 0x00 0x00
           fuse7       append      0     0     0    0 no          1    1      0     0     0 0x00 0x00
           fuse8       bootend     0     0     0    0 no          1    1      0     0     0 0x00 0x00
           lock                    0     0     0    0 no          1    1      0     0     0 0x00 0x00
           data                    0     0     0    0 no          0    1      0     0     0 0x00 0x00
           usersig     userrow     0     0     0    0 no         64   64      0     0     0 0x00 0x00
           flash                   0     0     0    0 no      49152  128      0     0     0 0x00 0x00
           eeprom                  0     0     0    0 no        256   64      0     0     0 0x00 0x00

         Programmer Type : JTAGICE3_UPDI
         Description     : MPLAB(R) PICkit 4 in UPDI mode
         ICE HW version  : 4
         ICE FW version  : 1.13 (rel. 236)
         Serial number   : 
         Vtarget         : 4.84 V
         JTAG clock megaAVR/program   : 1000 kHz
         JTAG clock megaAVR/debug     : 100 kHz
         PDI/UPDI clock Xmega/megaAVR : 100 kHz

avrdude: Partial Family_ID returned: "mega"
avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.05s

avrdude: Device signature = 0x1e9651 (probably m4809)
avrdude: NOTE: "flash" memory has been specified, an erase cycle will be performed
         To disable this feature, specify the -D option.
avrdude: erasing chip
avrdude: reading input file "blinky.hex"
avrdude: writing flash (3048 bytes):

Writing | ################################################## | 100% 1.27s

avrdude: 3048 bytes of flash written
avrdude: verifying flash memory against blinky.hex:

Reading | ################################################## | 100% 0.65s

avrdude: 3048 bytes of flash verified

avrdude done.  Thank you.
```
From the above, it all looks like it's worked perfectly - but the LED is not blinking.

I tried using "pymcuprog" which did work, so I tried using pymcuprog to read back what avrdude does using:
```
#> pymcuprog -datmega4809 read -m flash -f dump.bin
Connecting to anything possible
Connected to MPLAB PICkit 4 CMSIS-DAP from Microchip Technology Incorporated (serial number BUR213372088)
Debugger firmware version 1.13.236
Debugger hardware revision 4
Pinging device...
Ping response: 1E9651
Reading...
Data written to binary file: 'dump_flash.bin'
Done.
```
And a dump of the file from hexdump looks like:
```
#> hexdump -C dump_flash.bin | head -n20
00000000  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
*
00004000  0c 94 70 00 0c 94 98 00  0c 94 98 00 0c 94 98 00  |..p.............|
00004010  0c 94 98 00 0c 94 98 00  0c 94 98 00 0c 94 98 00  |................|
*
00004040  0c 94 98 00 0c 94 9b 03  0c 94 75 03 0c 94 98 00  |..........u.....|
00004050  0c 94 98 00 0c 94 98 00  0c 94 98 00 0c 94 98 00  |................|
00004060  0c 94 98 00 0c 94 c9 03  0c 94 98 00 0c 94 98 00  |................|
00004070  0c 94 98 00 0c 94 98 00  0c 94 98 00 0c 94 98 00  |................|
*
000040a0  43 6f 6d 70 6c 69 65 64  3a 20 46 65 62 20 32 34  |Complied: Feb 24|
000040b0  20 32 30 32 32 2c 20 32  33 3a 32 31 3a 32 39 00  | 2022, 23:21:29.|
000040c0  42 6f 6f 74 69 6e 67 20  46 61 6e 63 79 20 42 6c  |Booting Fancy Bl|
000040d0  69 6e 6b 00 4d 69 6c 6c  69 73 3a 20 00 00 00 05  |ink.Millis: ....|
000040e0  11 24 1f be cf ef cd bf  df e3 de bf 18 e2 a0 e0  |.$..............|
000040f0  b8 e2 e5 ee fb e0 02 c0  05 90 0d 92 a2 30 b1 07  |.............0..|
00004100  d9 f7 28 e2 a2 e0 b8 e2  01 c0 1d 92 a8 3a b2 07  |..(..........:..|
00004110  e1 f7 10 e0 c0 e7 d0 e0  04 c0 21 97 fe 01 0e 94  |..........!.....|
00004120  71 05 cf 36 d1 07 c9 f7  0e 94 f1 03 0c 94 91 05  |q..6............|
00004130  0c 94 00 00 fc 01 50 8d  41 8d 25 2f 30 e0 84 2f  |......P.A.%/0../|
```
So you can see my program did get written - only at offset 0x4000. A dump from when it is working shows my program written at the beginning of the flash, i.e. offset 0x0000.

The offset in avrdude.conf for 'flash' for an ATmega4809 is 0x4000. Perhaps somehow this offset is being apply twice or something similar.

MCUdude, 2022-02-24:
Can you paste the contests of the hex file you're trying to load? Maybe there's an addressing issue in there.

merlinblack, 2022-02-25:
I thought that might be the case too, however it looks ok to me. If anyone wants to try it, this should toggle PA7 with a pattern, and output elapsed Millis on USART0. (On a Atmega4809 PDIP40)
```
:100000000C9470000C9498000C9498000C94980038
:100010000C9498000C9498000C9498000C94980000
:100020000C9498000C9498000C9498000C949800F0
:100030000C9498000C9498000C9498000C949800E0
:100040000C9498000C949B030C9475030C949800EA
:100050000C9498000C9498000C9498000C949800C0
:100060000C9498000C94C9030C9498000C9498007C
:100070000C9498000C9498000C9498000C949800A0
:100080000C9498000C9498000C9498000C94980090
:100090000C9498000C9498000C9498000C94980080
:1000A000436F6D706C6965643A2046656220323436
:1000B00020323032322C2032333A32313A32390067
:1000C000426F6F74696E672046616E637920426C7F
:1000D000696E6B004D696C6C69733A200000000515
:1000E00011241FBECFEFCDBFDFE3DEBF18E2A0E0DB
:1000F000B8E2E5EEFBE002C005900D92A230B10738
:10010000D9F728E2A2E0B8E201C01D92A83AB207EE
:10011000E1F710E0C0E7D0E004C02197FE010E94A3
:100120007105CF36D107C9F70E94F1030C949105F0
:100130000C940000FC01508D418D252F30E0842F60
:1001400090E0821B930B541710F0CF9608950197FF
:100150000895FC0196898789981761F0A789AE0FE9
:10016000BF2FB11D5D968C9197899F5F9F73978B71
:1001700090E008958FEF9FEF0895FC01968987899D
:10018000981731F08789E80FF11D858D90E008956B
:100190008FEF9FEF0895FC0196892789892F90E0C2
:1001A000805C9F4F821B91098F7399270895CF938D
:1001B000DF93EC01E881F9810680F781E02D099554
:1001C000EC85FD8586818F738683EC85FD85858131
:1001D0008F7585838F898E8B1D8ADF91CF910895CE
:1001E0008F929F92AF92BF92EF92FF920F931F93C5
:1001F000CF93DF93EC014A015B01E22E8D898823C6
:1002000039F0E881F9810688F789E02DCE0109955A
:100210000C8910E0FFB6F89460E070E281EA97E0A4
:10022000A50194010E94280549015A018FEF881AFF
:10023000980AA80AB80AB694A79497948794EC856C
:10024000FD8586818D7F8683EC85FD85868186830D
:100250001D8AA09123110A2E000CBB0BA50194014D
:100260000E94690520E034E040E050E00E944A0529
:10027000EC85FD85820E931EA41EB51E8086918698
:10028000EC85FD85E782EC85FD858681806C868323
:10029000EC85FD858581806885839091E2058B8959
:1002A000809589239801220F331FF901E00FF11F78
:1002B000EC0FFD1F9089892B8093E20586858132A2
:1002C00028F590E0FC01EF53F44B40814F3FF1F0F3
:1002D000FC01E056F44BA081E0E2AE9FD001112476
:1002E000BC5F8158944BFC015081F0E0E0E05F3F3F
:1002F00029F0CD014096FC01E50FF11D9FB7F89460
:1003000012964C938081886080839FBFF901E00F33
:10031000F11FEC0FFD1F8785813208F590E0FC018D
:10032000EF53F44B70817F3FD1F0FC01E25AF44B64
:10033000E081EE2331F0E130B1F118F0E63008F45D
:1003400046C0FC01E056F44BE08150E2E59FF0012D
:100350001124FC5F4081472309F443C07583020FD9
:10036000131FC00FD11F8F85813288F490E0FC01EC
:10037000EF53F44B20812F3F51F08056944BDC011A
:10038000EC91B0E2EB9FF0011124FC5F2183FFBEF2
:10039000DF91CF911F910F91FF90EF90BF90AF90A1
:1003A0009F908F900895FC01E158F44B6081633079
:1003B00008F06F5FE091010A41E050E001C0440F96
:1003C0006A95EAF740954E234093010ABACF40E17F
:1003D000E49FF0011124E05AF54F41814F7E4183A3
:1003E000B0CF8158944BDC018C91309761F08F3FF6
:1003F00051F07096E80FF11D9FB7F89480818860E6
:1004000080839FBFACCFF0E0E0E0F6CFFC0161302D
:1004100019F4648B81E00895148A6623D9F380E08F
:100420000895DC0154961C92549754968C915497DD
:100430008230A0F4282F30E0F901EE0FFF1FE20F09
[... 101 lines cut ...]
:100A9000CF010895052E97FB1EF400940E94610576
:100AA00057FD07D00E94280507FC03D04EF40C9494
:100AB000610550954095309521953F4F4F4F5F4FC1
:100AC000089590958095709561957F4F8F4F9F4FBA
:100AD0000895B7FF0C9477050E947705821B930B4E
:100AE0000895EE0FFF1F0590F491E02D09940E94E8
:100AF0008205A59F900DB49F900DA49F800D911D20
:100B000011240895A29FB001B39FC001A39F700D4F
:100B1000811D1124911DB29F700D811D1124911D05
:060B20000895F894FFCFD8
:100B260000000000790235039A00E502CB00A90017
:100B3600BD00110206026403F000D7006203FA004A
:100B4600FA00FA00E80332003200320032003200C6
:100B560032003200D0070000000000000000000054
:100B66000101010101010000000000000000000079
:100B76000000000000000203000001020304050655
:100B8600070001020304050001020304050607002D
:100B96000102030001020304050600000000000034
:100BA600000002020202020203030303030303031B
:100BB60004040404050505050505050102040810DD
:100BC6002040800102040810200102040810204081
:0F0BD6008001020408010204081020400D0A00EB
:020BE50001000D
:00000001FF
```

MCUdude, 2022-02-26:
I'm currently on winter vacation, but I'll see if I can reproduce the issue when I get home!

merlinblack, 2022-02-26:
That would be great. I did try stepping through the code to see what I could find, however, nothing really stuck out. I thought perhaps the jtag3.c code at line 2467 might be suspect, but this same code works when reading the signature memory.  

While hacking around, at one point for whatever reason MPLab IPE (on Linux) started writing to flash beginning at offset 0x2000. Power cycling the pickit did not fix it, but after using pymcuprog to erase and write it was back to normal. Perhaps there is a 'command' to set where in flash the Pickit4 operates, and pymcuprog sets this to 0x0000.

MCUdude, 2022-02-28:
@merlinblack thanks for reporting! I can confirm that this is an issue on the Pickit4, SNAP and the Xplained Mini UPDI. probably all JTAG3 compatible programmers...

I can also confirm that this was not an issue with Avrdude 6.3

MCUdude, 2022-02-28:
@merlinblack you're absolutely right about your assumptions, it's line 2467 that's causing the issue, and to be honest, it's my fault, sorry for that... However, I have no idea why this only applies to the flash memory and not the other ones!

https://github.com/avrdudes/avrdude/blob/64cc54ac3513fec863c439c9f87381fc45fc3706/src/jtag3.c#L2464-L2476

@dl8dtl do you know what the flash memory needs to be treated differently than the other memories?

This is a quick fix that works, but isn't very elegant:

```c
  /*
   * Non-Xmega device.
   */
  if (p->flags & AVRPART_HAS_UPDI) {
    if (m->size == 1) {
      addr = m->offset;
    }
    else if (m->size > 1 && strcmp(m->desc, "flash") != 0) {
      addr += m->offset;
    }
  }
  return addr;
}
```

merlinblack, 2022-03-01:
@MCUdude Yeah the whole only affecting flash is weird. Or perhaps it just does not affect the signature memory. I haven't tried writing fuses.

merlinblack, 2022-03-01:
I think is **only** flash that is affected. I've just tested wring to EEPROM and it works as expected. I used pymcuprog to verify the contents of EEPROM.

MCUdude, 2022-03-01:
It's very strange that the fix didn't work for you, since this is was exactly how the code worked before I broke it in a previous PR.

Maybe @dl8dtl knows?

merlinblack, 2022-03-01:
That error looks similar (perhaps the same) to what I was getting with a microUPDI setup. I'll have to try it again...
_sounds of tinkering_
No, different error I think.
```
./avrdude -v -cxplainedmini_updi -pm4809

avrdude: Version 6.99-20220221 (64cc54a)
         Copyright (c) Brian Dean, http://www.bdmicro.com/
         Copyright (c) Joerg Wunsch

         System wide configuration file is "/home/nigel/prog/avr/avrdude/build_linux/src/avrdude.conf"
         User configuration file is "/home/nigel/.avrduderc"
         User configuration file does not exist or is not a regular file, skipping

         Using Port                    : usb
         Using Programmer              : xplainedmini_updi
avrdude: Found CMSIS-DAP compliant device, using EDBG protocol
         AVR Part                      : ATmega4809
         RESET disposition             : dedicated
         RETRY pulse                   : SCK
         Serial program mode           : yes
         Parallel program mode         : yes
         Memory Detail                 :

                                           Block Poll               Page                       Polled
           Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           signature               0     0     0    0 no          3    1      0     0     0 0x00 0x00
           prodsig                 0     0     0    0 no         61   61      0     0     0 0x00 0x00
           sernum                  0     0     0    0 no         10    1      0     0     0 0x00 0x00
           osccal16                0     0     0    0 no          2    1      0     0     0 0x00 0x00
           osccal20                0     0     0    0 no          2    1      0     0     0 0x00 0x00
           tempsense               0     0     0    0 no          2    1      0     0     0 0x00 0x00
           osc16err                0     0     0    0 no          2    1      0     0     0 0x00 0x00
           osc20err                0     0     0    0 no          2    1      0     0     0 0x00 0x00
           fuses                   0     0     0    0 no          9   10      0     0     0 0x00 0x00
           fuse0       wdtcfg      0     0     0    0 no          1    1      0     0     0 0x00 0x00
           fuse1       bodcfg      0     0     0    0 no          1    1      0     0     0 0x00 0x00
           fuse2       osccfg      0     0     0    0 no          1    1      0     0     0 0x00 0x00
           fuse4       tcd0cfg     0     0     0    0 no          1    1      0     0     0 0x00 0x00
           fuse5       syscfg0     0     0     0    0 no          1    1      0     0     0 0x00 0x00
           fuse6       syscfg1     0     0     0    0 no          1    1      0     0     0 0x00 0x00
           fuse7       append      0     0     0    0 no          1    1      0     0     0 0x00 0x00
           fuse8       bootend     0     0     0    0 no          1    1      0     0     0 0x00 0x00
           lock                    0     0     0    0 no          1    1      0     0     0 0x00 0x00
           data                    0     0     0    0 no          0    1      0     0     0 0x00 0x00
           usersig     userrow     0     0     0    0 no         64   64      0     0     0 0x00 0x00
           flash                   0     0     0    0 no      49152  128      0     0     0 0x00 0x00
           eeprom                  0     0     0    0 no        256   64      0     0     0 0x00 0x00

         Programmer Type : JTAGICE3_UPDI
         Description     : Atmel AVR XplainedMini in UPDI mode
         ICE HW version  : 0
         ICE FW version  : 1.19 (rel. 57)
         Serial number   : MICROUPDIPROGRAMMERX
         Vtarget         : 3.30 V
avrdude: Short read, read only 0 out of 64 bytes
avrdude: jtag3_edbg_recv(): Unexpected response 0x9c
avrdude: retrying with external reset applied
avrdude: jtag3_edbg_send(): Unexpected response 0x81, 0x11
avrdude: jtag3_edbg_recv(): Unexpected response 0x80
avrdude: retrying with external reset applied
avrdude: initialization failed, rc=-1
         Double check connections and try again, or use -F to override
         this check.

avrdude: jtag3_edbg_send(): Unexpected response 0x81, 0x11
avrdude: jtag3_edbg_recv(): Unexpected response 0x80
avrdude: jtag3_edbg_send(): Unexpected response 0x81, 0x11
avrdude: jtag3_edbg_recv(): Unexpected response 0x80

avrdude done.  Thank you.
```
It does work with the avrdude that comes with Arduino IDE - which led me to test the pickit4 with the arduino ide version of avrdude, but it doesn't work.
```
#> ./avrdude -v -cpickit4_updi -pm4809
avrdude: Version 6.3-20201216
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "../etc/avrdude.conf"
         User configuration file is "/home/nigel/.avrduderc"
         User configuration file does not exist or is not a regular file, skipping

         Using Port                    : usb
         Using Programmer              : pickit4_updi
avrdude: usbdev_open(): Found MPLAB PICkit 4 CMSIS-DAP, serno: BUR213372088
avrdude: usbdev_open(): WARNING: failed to set configuration 1: Device or resource busy
avrdude: Found CMSIS-DAP compliant device, using EDBG protocol
avrdude: usbdev_send(): wrote -5 out of 912 bytes, err = Input/output error
avrdude: jtag3_edbg_prepare(): failed to send command to serial port

avrdude done.  Thank you.
```
Doing the same (./avrdude -v -cpickit4_updi -pm4809) with from source avrdude still works - just not the flashing.

Despite the Arduino version not working with my pickit, it does correctly flash my 4809 using my microUPD, so at least it's jtag updi code seems to work in that version.

dl8dtl, 2022-03-01:
I thin PR #895 closes this as well.

closed 2022-03-01 by dl8dtl

merlinblack, 2022-03-01:
Wonderfull! That works for me. It's only subtlely different to MCUdudes fix above, returning the addr unchanged regardless of m->size. Basically an off-by-one type error?

Anyhow - thanks all!

commit 0d58adb1 (2022-03-01): Close #892 #895