avrdudes/avrdude #913: Terminal write mode doesn't support string input (yet)

MCUdude, 2022-04-01:
This is something I'd wanted to implement for a while, but haven't really figured out how without completely rewriting the existing implementation. The main problem is that the current implementation is designed to handle one argument at a time, but a string (with spaces in between words) will be "parsed as multiple arguments.

But maybe @dl8dtl can help bring some light into this.

This simple C example does exactly want I want it to. The space-separated string is interpreted as a single argc argument:

```c
#include <stdio.h>
#include <string.h>

int main(int argc, const char * argv[])
{
  for(int i = 0; i < argc; i++)
    printf("Arg%2d, Len %2lu, Str: %s\n", i, strlen(argv[i]), argv[i]);
}
```

Output:
```
$ ./arg_test 1234 "This is a space separated string"
Arg 0, Len 10, Str: ./arg_test
Arg 1, Len  4, Str: 1234
Arg 2, Len 32, Str: This is a space separated string
```

However, the terminal write implementation is somehow different. Here, the string (in quotes) is divided into six separate arguments.

```c
static int cmd_write(PROGRAMMER * pgm, struct avrpart * p,
		     int argc, char * argv[])
{
  for(int i = 0; i < argc; i++)
    avrdude_message(MSG_INFO,"Arg%2d, Len %2lu, Str: %s\n", i, strlen(argv[i]), argv[I]);

//...
}
```

Output:
```
avrdude> write eeprom 0 0x10 1234 "This is a space separated string"
>>> write eeprom 0 0x10 1234 "This is a space separated string" 
Arg 0, Len  5, Str: write
Arg 1, Len  6, Str: eeprom
Arg 2, Len  1, Str: 0
Arg 3, Len  4, Str: 0x10
Arg 4, Len  4, Str: 1234
Arg 5, Len  5, Str: "This
Arg 6, Len  2, Str: is
Arg 7, Len  1, Str: a
Arg 8, Len  5, Str: space
Arg 9, Len  9, Str: separated
Arg10, Len  7, Str: string"
```

If the string ended up as a single argument like the C code example shown above, implementing this would be trivial. So how can we get the same behavior in Avrdude terminal write mode as with the C code example?
Thanks!

MCUdude, 2022-04-01:
From a brief overview, it looks like `nexttok` is the only function that needs rework in order to support strings inside quotes.

https://github.com/avrdudes/avrdude/blob/e069871c8e4356c3b742d53903c8504b0de356e3/src/term.c#L120-L146

dl8dtl, 2022-04-01:
On a completely different idea: we've got lex and yacc on board anyway. We could create a complete formal language for terminal mode. While that's obviously going to be a full rewrite of terminal mode, it offers quite a bit of flexibility.

MCUdude, 2022-04-01:
That would indeed be the best idea, but I'm not capable of doing a thing. Maybe this should be on the todo-list for future projects?

But as a "short-term fix," it should be possible to tweak `nexttok` so that it's easier to work with strings.

MCUdude, 2022-04-01:
I believe I have a working version of `nexttok` that does what I want it to do.
https://github.com/MCUdude/avrdude/tree/terminal-write-str

I've also added some debug output to `cmd_write` (that can later be rebased away) to show how arguments now is being split.

Here's some output just to show how it works now:
```
>>> write abc 0 1 "hello world!" 1234 0 1 abc 123 "This is a long string" 3.141592 
Arg 0, Len  5, Str: 'write'
Arg 1, Len  3, Str: 'abc'
Arg 2, Len  1, Str: '0'
Arg 3, Len  1, Str: '1'
Arg 4, Len 14, Str: '"hello world!"'
Arg 5, Len  4, Str: '1234'
Arg 6, Len  1, Str: '0'
Arg 7, Len  1, Str: '1'
Arg 8, Len  3, Str: 'abc'
Arg 9, Len  3, Str: '123'
Arg10, Len 23, Str: '"This is a long string"'
Arg11, Len  8, Str: '3.141592'
"abc" memory type not defined for part "ATmega4808"
```

MCUdude, 2022-04-01:
OK, I have just pushed a proof of concept. It works! 🎉
Note that there's still some debug output I'll remove later.
```
>>> read eeprom 

Reading | ################################################## | 100% 1.00s

0000  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0010  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0020  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0030  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0040  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0050  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0060  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0070  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0080  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0090  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
00a0  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
00b0  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
00c0  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
00d0  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
00e0  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
00f0  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|

avrdude> write eeprom 0 'a' 'v' 'r' 0xcafe "Hello world this is a test" 0x01 0x02 0x03
q: write
q: eeprom
q: 0
q: 'a'
q: 'v'
q: 'r'
q: 0xcafe
q: "Hello world this is a test"
q: 0x01
q: 0x02
>>> write eeprom 0 'a' 'v' 'r' 0xcafe "Hello world this is a test" 0x01 0x02 0x03 
Arg 0, Len  5, Str: 'write'
Arg 1, Len  6, Str: 'eeprom'
Arg 2, Len  1, Str: '0'
Arg 3, Len  3, Str: ''a''
Arg 4, Len  3, Str: ''v''
Arg 5, Len  3, Str: ''r''
Arg 6, Len  6, Str: '0xcafe'
Arg 7, Len 28, Str: '"Hello world this is a test"'
Arg 8, Len  4, Str: '0x01'
Arg 9, Len  4, Str: '0x02'
Arg10, Len  4, Str: '0x03'
argv: "Hello world this is a test", malloc: Hello world this is a test

Info: Writing 35 bytes starting from address 0x00

Writing | ################################################## | 100% 1.46s

avrdude> read eeprom
q: read
>>> read eeprom 

Reading | ################################################## | 100% 1.00s

0000  61 76 72 fe ca 48 65 6c  6c 6f 20 77 6f 72 6c 64  |avr..Hello world|
0010  20 74 68 69 73 20 69 73  20 61 20 74 65 73 74 00  | this is a test.|
0020  01 02 03 ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0030  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0040  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0050  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0060  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0070  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0080  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0090  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
00a0  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
00b0  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
00c0  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
00d0  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
00e0  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
00f0  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
```

closed 2022-04-05 by dl8dtl