avrdudes/avrdude #919: Fun fact of the day

stefanrueger, 2022-04-06:
Flash pagesize of `ATtiny20`
 - ATDF puts a value of 128 on this
 - Avr-libc's `iotn20.h` defines SPM_PAGESIZE as 64
 - Page 160 of [datasheet](https://ww1.microchip.com/downloads/en/DeviceDoc/Atmel-8235-8-bit-AVR-Microcontroller-ATtiny20_Datasheet.pdf) says it's 32
 - The current avrdude uses 16

MCUdude, 2022-04-07:
What a mess! Wonder what the _actual_ pagesize is...

BTW did you know that the [AT42QT1070](https://www.microchip.com/en-us/product/AT42QT1070) is actually just an ATtiny20 under the hood? The [AT42QT2120](https://www.microchip.com/en-us/product/AT42QT1070) is just an ATtiny40.

mcuee, 2022-05-20:
The datasheet says that Page Size is 16 words. Maybe that is why avrdude uses 16?
Ref: https://ww1.microchip.com/downloads/en/devicedoc/atmel-8235-8-bit-avr-microcontroller-attiny20_datasheet.pdf

stefanrueger, 2022-05-20:
This is an observation just for the fun of it. I am guessing (but do not know) that the page size doesn't matter for this part. It's programmed by TPI (no page size needed), has no spm opcode (no page size needed), so each instance - including the data sheet(?!) - made up its own favourite number.

I stumbled upon that gem when I compared the signature bytes and main characteristics of the parts as known to avrdude, avr-gcc 4.8.1 to bleeding edge 12.1.0, ATDF files and some selected data sheets when the sources didn't concur. Wanted to know 
  - Is the signature indicative of the part? (No, see below)
  - Is the signature indicative of the flash size? (No: `ATxmega64B3` vs `ATmega48091`)
  - Is the signature indicative of the page size? (No: `ATxmega128A4U` vs `ATxmega128A4`, `AT90USB82` vs `ATA6285`, ...)
  - Do avr-libc and avrdude agree on the signatures of a part? (No, a few examples - reported to the project that had it wrong)

I wanted something in a bootloader that would tell AVRDUDE what the flash size, the EEPROM size, the page size and the number of interrupts the part has. Concluded, I better use something else (and shorter) than the signature... And then I also found out that the [ATtiny1634](https://ww1.microchip.com/downloads/en/DeviceDoc/Atmel-8303-8-bit-AVR-Microcontroller-tinyAVR-ATtiny1634_Datasheet.pdf) has a 4-page spm erase, which means that the bootloader *really* needs to tell AVRDUDE *exactly* which part it's on so AVRDUDE's bootloader programmer can help iron out the quirks of Microchips mcu families.

MCUdude, 2022-06-22:
If the correct flash page size for the ATtiny20 is currently unknown, but it doesn't really matter when uploading anyways, can't we just close this issue as "not planned"?

stefanrueger, 2022-06-22:
Closing as not really a bug, more like a curiosity