avrdudes/avrdude #920: FreeBSD/ATtiny416XNANO vid:pid problem.

cederom, 2022-04-07:
Hello world :-)

I have problem playing with brand new ATtiny416XplainedNano board on FreeBSD where we have 6.3 version of the avrdude.

```
hexagon% avrdude -c xplainedmini -p t416 -P usb:/dev/cuaU0
avrdude: jtag3_open_common(): Did not find any device matching VID 0x03eb and PID list: 0x2145

avrdude done.  Thank you.

hexagon% avrdude -c xplainedmini_updi -p t416 -P usb:/dev/cuaU0
avrdude: jtag3_open_common(): Did not find any device matching VID 0x03eb and PID list: 0x2145

avrdude done.  Thank you.

hexagon% avrdude -c xplainedmini_dw -p t416 -P usb:/dev/cuaU0
avrdude: jtag3_open_common(): Did not find any device matching VID 0x03eb and PID list: 0x2145

avrdude done.  Thank you.
```

Dongle is detected correctly, details below:
```
# uname -a
FreeBSD hexagon 13.1-STABLE FreeBSD 13.1-STABLE #0 stable/13-n250096-4f69c575996: Fri Mar 25 03:50:58 CET 2022

# dmesg | tail
ugen0.8: <ATMEL mEDBG CMSIS-DAP> at usbus0
usbhid4 on uhub15
usbhid4: <ATMEL mEDBG CMSIS-DAP, class 239/2, rev 2.00/10.00, addr 7> on usbus0
hidbus4: <HID bus> on usbhid4
umodem0 on uhub15
umodem0: <ATMEL mEDBG CMSIS-DAP, class 239/2, rev 2.00/10.00, addr 7> on usbus0
umodem0: data interface 2, has CM over data, has break

 # usbconfig -d 0.8 dump_device_desc
ugen0.8: <ATMEL mEDBG CMSIS-DAP> at usbus0, cfg=0 md=HOST spd=FULL
(12Mbps) pwr=ON (100mA)
  bLength = 0x0012
  bDescriptorType = 0x0001
  bcdUSB = 0x0200
  bDeviceClass = 0x00ef  <Miscellaneous device>
  bDeviceSubClass = 0x0002
  bDeviceProtocol = 0x0001
  bMaxPacketSize0 = 0x0020
  idVendor = 0x03eb
  idProduct = 0x2145
  bcdDevice = 0x1000
  iManufacturer = 0x0001  <ATMEL>
  iProduct = 0x0002  <mEDBG CMSIS-DAP>
  iSerialNumber = 0x0003  <ATML2795042700003537>
  bNumConfigurations = 0x0001
```

Very similar problem was reported DIAMEX-AVR-USB interface:
```
> avrdude: usbdev_open(): did not find any (matching) USB device "usb:0" (0x03eb:0x2104)
> avrdude: usbdev_open(): did not find any (matching) USB device "usb:4" (0x03eb:0x2104)
> avrdude: usbdev_open(): did not find any (matching) USB device "usb:" (0x03eb:0x2104)
```

I suspect Diamex interface is not defined and needs to have configuration created. But ATtiny416XNANO is the original board from Microchip/Atmel and it should work?

Any hints welcome :-)
Tomek

dl8dtl, 2022-04-07:
Permission problem.
Your AVRDUDE command is apparently shown as a regular user, but the usbconfig outputs are shown as root.
You need to get dead rules in place.
I'm using this `/usr/local/etc/devd/atmel.conf`:
```
notify 100 {
	match "vendor" "0x03eb";
	match "system" "USB";
	match "subsystem" "DEVICE";
	match "type" "ATTACH";
	action "chmod 0660 /dev/$cdev; chgrp operator /dev/$cdev";
};

nomatch 100 {
	match "vendor" "0x03eb";
	match "bus" "uhub[0-9]+";
	action "logger Atmel USB device attached, product $product, bus $bus";
};
```

cederom, 2022-04-07:
Thank you @cederom for the reply :-) I am working on FreeBSD and devd is set up correctly. I can read/write all other devices. I can attach serial terminal to `/dev/cuaU0` (9600-8-N-1) and I get "nak" response on Enter. Workig as root gives the same result.

```
# avrdude -c xplainedmini -p t416 -P usb:/dev/cuaU0
avrdude: usbhid_open(): No matching device found
avrdude: jtag3_open_common(): Did not find any device matching VID 0x03eb and PID list: 0x2145

avrdude done.  Thank you.
```

There seems to be some sort of problem with devices enumeration in avrdude..?

cederom, 2022-04-08:
Problem seems to be somewhere in `usbhid_open()` and/or `usbdev_open()`.

Below is output with increased verbosity. It finds the device but does not use it for some reason. I will try to provide serial number - is it mandatory?
```
% avrdude -c xplainedmini -p t416 -P usb:/dev/cuaU0 -vvv

avrdude: Version 6.4
         Copyright (c) Brian Dean, http://www.bdmicro.com/
         Copyright (c) Joerg Wunsch

         System wide configuration file is "/usr/local/etc/avrdude.conf"
         User configuration file is "/home/XXX/.avrduderc"

         Using Port                    : usb:/dev/cuaU0
         Using Programmer              : xplainedmini
avrdude: stk500v2_jtag3_open()
avrdude: usbhid_open(): Found mEDBG CMSIS-DAP, serno: ATML2795042700003537
avrdude: usbhid_open(): serial number doesn't match
avrdude: usbhid_open(): No matching device found
avrdude: usbdev_open(): Found mEDBG CMSIS-DAP, serno: ATML2795042700003537
avrdude: usbdev_open(): serial number doesn't match
avrdude: jtag3_open_common(): Did not find any device matching VID 0x03eb and PID list: 0x2145

avrdude done.  Thank you.
```

dl8dtl, 2022-04-08:
> usb:/dev/cuaU0

That does not make any sense at all. `/dev/cuaU0` is taken as a serial number to match. The default of `-P usb` is completely sufficient.

Omit the entire -P option, *unless* you really want to select any particular serial number only (in case you've got more than one EDBG connected).

dl8dtl, 2022-04-08:
> Is there any chance to also accept serial port device

There is no such device for a USB connected programmer.
The serial device you are seeing is for the UART bridge, connecting to the target's UART pins. It is completely unrelated to the actual programming interface. (The device node for the USB connection itself is `/dev/ugen`<something>; you certainly wouldn't want to figure that one out manually each time.)

dl8dtl, 2022-04-08:
As a side note: FreeBSD is my primary development platform, so you can be assured there's never going to be a fundamental issue with its support. ;-)

cederom, 2022-04-08:
Awsome! `-P usb` is enough.. and with valid Interface Programmer configuration it is not even necessary :-)

Just a habit from other utilities where `-p` usually means device where something is attached :-)

FreeBSD ROX! :-)

closed 2022-04-08 by cederom