avrdudes/avrdude #921: ATtiny43U calibration memory size

stefanrueger, 2022-04-08:
The `.conf` file has the calibration memory as 2 bytes, but the  [datasheet of the ATtiny43U](https://ww1.microchip.com/downloads/en/DeviceDoc/doc8048.pdf) only mentions one byte (pages 144, 156).

Probably a copy and paste error... Along with that the SPI command for calibration read ought not to have an address bit in it.

MCUdude, 2022-04-08:
Would be great if you could provide PRs for simple typos like these ones!

dl8dtl, 2022-04-08:
> Along with that the SPI command for calibration read ought not to have an address bit in it.

Which obviously makes those 2 bytes pretty useless. ;-)
Actually, AVRs do have much more calibration bytes internally (the even ones start with the signature bytes, the odd ones with the oscillator calibration value), but the remainder is just used internally, and usually not documented.

stefanrueger, 2022-04-08:
Useful info.

So, is there utility then in pretending a larger calibration memory than documented and throw in a few address bits?

The latter one could do anyway. Specifying too many address bits is commonplace in `.conf` (some 430 bits in 59 parts). Probably not on purpose but by misguided cut and pastes. But won't cause harm either.

I have looked at all 3214 SPI commands that avrdude.conf eventually defines for its 162 SPI programming capable parts. 1563 of them contain one or more address bits. They are read*, write* (including writepage) and loadpage* commands for memories with more than one byte, plus the load_ext_addr command. None of these 1563 addressing commands have a fixed `1` bit in any of the middle 16 command bits; these only ever consist of `x`, `0`, or `a` bits.

They could be replaced by `"a15 a14 a13 a12 a11 a10 a9 a8 a7 a6 a5 a4 a3 a2 a1 a0"` with two exceptions:
 - writepage calls should mask the lower bits that correspond to the word pagesize (flash) or byte pagesize (eeprom) to `0`. I actually suspect(!) this is not necessary. Fun fact: there are also some writepage commands in `.conf` that make 8 parts (`ATmega325*`) *not* mask the lower bits
 - load_ext_addr. However, if one logic shifts right the flash word address by 16 before calling `avr_set_addr()` then you could still use 16 `a` bits  (and cross fingers for Microchip bringing out *that* new part with 8 GB memory and SPI programming interface :)

MCUdude, 2022-05-05:
Fixed in 43ddea9

closed 2022-05-05 by MCUdude