avrdudes/avrdude #923: Xplained104XNANO Programmer for ATtiny10

cederom, 2022-04-09:
Hello world :-)

I have Xplained104XNANO and Xplained416XNANO boards [x104n]. I would like to use one of these to program ATtiny10 MCU [att10].

From develkit schematics [x104ndd] I can see that Xplained104XNANO uses TWI connection to ATtiny104 [att104]. But there is also `DBGPRT` line which is not available in ATtiny10.

I have removed resistors between Programmer MCU and ATtiny104 and conencted them to TWI interface (`RESET`, `TPIDATA`, `TPICLK`, `VCC`, `GND`) of ATtiny10 according to [2].

This does not seem to work out of the box.

Is TWI supported by avrdude? What programmer should I choose or what programmer type should I select to create `xplainedmini_twi`?

Any hints appreciated :-)
Tomek

[x104n] https://www.microchip.com/en-us/development-tool/ATTINY104-XNANO
[att10] https://www.microchip.com/en-us/product/ATtiny10
[att104] https://www.microchip.com/en-us/product/ATtiny104
[x104ndd] https://ww1.microchip.com/downloads/en/DeviceDoc/ATtiny104-Xplained-Nano_Design-Documentation.zip
[avr918] https://ww1.microchip.com/downloads/en/Appnotes/doc8373.pdf

MCUdude, 2022-04-09:
See #822.
Currently, the XNANO-104 can't be used with Avrdude.

cederom, 2022-04-09:
Thank you @MCUdude :-) Looks like TPI needs to be implemented on xplained / jtagice3..

Do you know it `FT2232` or `xplainedpro` driver supports TPI?

MCUdude, 2022-04-09:
Xplained Pro probably uses the same `jtag3` interface, so it probably won't work with TPI as well. I have no experience with FT2232, so I don't know if this can be used for TPI or not.

cederom, 2022-04-09:
I can see some `src/avrftdi_tpi` files so probably STDI can talk TPI :-) Thanks for the hints @MCUdude! :-)

cederom, 2022-04-10:
Okay, I made it work with FT2232H based device called KT-LINK, Pull Request created https://github.com/avrdudes/avrdude/pull/930, details here https://github.com/avrdudes/avrdude/issues/928.

I will close this thread as there is already another one opened for ATtiny104XNANO + ATtiny10 + TPI :-)

Thank you everyone for your time and support! :-)

closed 2022-04-10 by cederom