avrdudes/avrdude #926: portname default for linuxspi programmer not recognized

per1234, 2022-04-10:
A default `portname` value for the `linuxspi` programmer was added in https://github.com/avrdudes/avrdude/commit/dd1255b0c8fcf24e0cf49d04d5cda74785b450c9, and documented:

https://github.com/avrdudes/avrdude/blob/7e26a153752c9ebf346bd4e5db2e7c9b78a8b3c6/src/avrdude.1#L134-L135

However, the command still fails if you omit the `-P` flag:

```text
$ avrdude -vvvv -p atmega328p -c linuxspi -U "flash:w:blink.hex:i"
executable_abspath = /home/per/Documents/deleteme/avrdude/nightly/bin/avrdude
executable_abspath_len = 56
executable_dirpath = /home/per/Documents/deleteme/avrdude/nightly/bin
executable_dirpath_len = 48
sys_config = /usr/local/etc/avrdude.conf
sys_config_found = true

avrdude: Version 6.99
         Copyright (c) Brian Dean, http://www.bdmicro.com/
         Copyright (c) Joerg Wunsch

         System wide configuration file is "/usr/local/etc/avrdude.conf"
         User configuration file is "/home/per/.avrduderc"
         User configuration file does not exist or is not a regular file, skipping

avrdude: no port has been specified on the command line or in the config file
         Specify a port using the -P option and try again
```

MCUdude, 2022-04-10:
[This](https://github.com/avrdudes/avrdude/blob/7e26a153752c9ebf346bd4e5db2e7c9b78a8b3c6/src/linuxspi.c#L141-L143) is probably the problem.

What happens if you use the following code instead?

```c
    if (port == NULL || !strcmp(port, "unknown")) {
        port = port_default;
    }
```

MCUdude, 2022-04-10:
> What happens if you use the following code instead?

Doesn't work, I've tried!

I've found a sort of hacky solution. It works, but ideally

The fix can be found here:
https://github.com/MCUdude/avrdude/tree/linuxspi-default-port-fix

The code snippet looks line this (main.c):

```c
  if (port == NULL) {
    switch (pgm->conntype)
    {
      case CONNTYPE_PARALLEL:
        port = default_parallel;
        break;

      case CONNTYPE_SERIAL:
        port = default_serial;
        break;

      case CONNTYPE_USB:
        port = DEFAULT_USB;
        break;
    }
#ifdef HAVE_LINUXSPI
    if (!strcmp(pgm->type, "linuxspi"))
      port = "unknown";
#endif
  }
```

Most elegant solution, but more than what I'm capable of doing right now:
```c
  if (port == NULL) {
    switch (pgm->conntype)
    {
      case CONNTYPE_PARALLEL:
        port = default_parallel;
        break;

      case CONNTYPE_SERIAL:
        port = default_serial;
        break;

      case CONNTYPE_USB:
        port = DEFAULT_USB;
        break;

      case CONNTYPE_SPI:
        port = default_spi;
        break;
    }
```

@dl8dtl would you accept a PR with a "hacky" solution, or would you like to implement a default SPI port yourself?

MCUdude, 2022-04-11:
This seems to be the issue for `linuxgpio` as well:

```
$ ./build_linux/src/avrdude -v -C build_linux/src/avrdude.conf -c linuxgpio -p atmega1284p

avrdude: Version 6.99-20220405 (7e26a15)
         Copyright (c) Brian Dean, http://www.bdmicro.com/
         Copyright (c) Joerg Wunsch

         System wide configuration file is "build_linux/src/avrdude.conf"
         User configuration file is "/home/pi/.avrduderc"
         User configuration file does not exist or is not a regular file, skipping

avrdude: no port has been specified on the command line or in the config file
         Specify a port using the -P option and try again
```

MCUdude, 2022-04-11:
@per1234 I've come up with a "proper" solution that's way less hacky than my initial fix, see #933. Are you able to build from source on your Raspberry Pi, or do you want me to continue providing binaries for testing?

[Avrdude_linuxspi_default_port.zip](https://github.com/avrdudes/avrdude/files/8467611/Avrdude_linuxspi_default_port.zip)

```
$  ./build_linux/src/avrdude -v -C build_linux/src/avrdude.conf -c linuxspi -p atmega1284p

avrdude: Version 6.99-20220410 (6fceea8)
         Copyright (c) Brian Dean, http://www.bdmicro.com/
         Copyright (c) Joerg Wunsch

         System wide configuration file is "build_linux/src/avrdude.conf"
         User configuration file is "/home/pi/.avrduderc"
         User configuration file does not exist or is not a regular file, skipping

         Using Port                    : unknown
         Using Programmer              : linuxspi
avrdude: defaulting bit clock to 200 kHz
         AVR Part                      : ATmega1284P
         Chip Erase delay              : 55000 us
         PAGEL                         : PD7
         BS2                           : PA0
         RESET disposition             : dedicated
         RETRY pulse                   : SCK
         Serial program mode           : yes
         Parallel program mode         : yes
         Timeout                       : 200
         StabDelay                     : 100
         CmdexeDelay                   : 25
         SyncLoops                     : 32
         PollIndex                     : 3
         PollValue                     : 0x53
         Memory Detail                 :

                                           Block Poll               Page                       Polled
           Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           eeprom                 65    10   128    0 no       4096    8      0  9000  9000 0xff 0xff
           flash                  65    10   256    0 yes    131072  256    512  4500  4500 0xff 0xff
           lock                    0     0     0    0 no          1    1      0  9000  9000 0x00 0x00
           lfuse                   0     0     0    0 no          1    1      0  9000  9000 0x00 0x00
           hfuse                   0     0     0    0 no          1    1      0  9000  9000 0x00 0x00
           efuse                   0     0     0    0 no          1    1      0  9000  9000 0x00 0x00
           signature               0     0     0    0 no          3    1      0     0     0 0x00 0x00
           calibration             0     0     0    0 no          1    1      0     0     0 0x00 0x00

         Programmer Type : linuxspi
         Description     : Use Linux SPI device in /dev/spidev*

avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.00s

avrdude: Device signature = 0x1e9705 (probably m1284p)

avrdude done.  Thank you.
```

per1234, 2022-04-12:
>Are you able to build from source on your Raspberry Pi

I haven't taken the time to give it a try yet. I'm sure I would be able to manage it if required, but I tend to use the RPi only for quick tests since the 3B+ I have is a bit slow to work with compared to my normal machines.

>do you want me to continue providing binaries for testing?

The nice CI/CD system for this project automatically generates builds for every push. So as long as it is pushed to this repo (including PRs), or to a fork with GitHub Actions enabled, the builds are available for anyone to easily test without having to do any toolchain set up.

For example, here are the current builds for #933:
https://github.com/avrdudes/avrdude/actions/runs/2151015230#artifacts

MCUdude, 2022-04-12:
> >Are you able to build from source on your Raspberry Pi

> I haven't taken the time to give it a try yet. I'm sure I would be able to manage it if required, but I tend to use the RPi only for quick tests since the 3B+ I have is a bit slow to work with compared to my normal machines.

Slow you say? I needed a Pi for testing, and couldn't find a spare microSD card, so I ended up using my old Pi model B which supports a full-size SD card. Compiling Avrdude on this takes more than five minutes, but luckily, the build system re-uses the pre-compiled file if there are no changes.

Setting up a build environment using Cmake is _very_ simple, and it makes it way easier to test code if you want to contribute.

> The nice CI/CD system for this project automatically generates builds for every push. 

Totally forgot about this! Not sure if the CI has linuxspi/linuxgpio enabled though. These are features that explicitly need to be enabled.

per1234, 2022-04-12:
>Not sure if the CI has linuxspi/linuxgpio enabled though.

It does:
https://github.com/avrdudes/avrdude/blob/60960ba590cdb70277f40a2a05b374d99d4e9b5c/.github/workflows/build.yml#L105-L106

mcuee, 2022-07-10:
Test results using Raspberry Pi 3B+ and an I/O expansion board and then to connect to an Arduino Uno Clone (ATmega328P with CH340).

The current master without Pull Request #933:

```
pi@raspberrypi:~ $ /usr/local/bin/avrdude -c linuxspi -p m328p 
-P /dev/spidev0.0:/dev/gpiochip0 -U flash:r:readback_m328p_linuxspi:i -v

avrdude: Version 7.0-20220705 (64ee485)
         Copyright (c) Brian Dean, http://www.bdmicro.com/
         Copyright (c) Joerg Wunsch

         System wide configuration file is "/usr/local/bin/../etc/avrdude.conf"
         User configuration file is "/home/pi/.avrduderc"
         User configuration file does not exist or is not a regular file, skipping

         Using Port                    : /dev/spidev0.0:/dev/gpiochip0
         Using Programmer              : linuxspi
avrdude: defaulting bit clock to 200 kHz
         AVR Part                      : ATmega328P
         Chip Erase delay              : 9000 us
         PAGEL                         : PD7
         BS2                           : PC2
         RESET disposition             : dedicated
         RETRY pulse                   : SCK
         Serial program mode           : yes
         Parallel program mode         : yes
         Timeout                       : 200
         StabDelay                     : 100
         CmdexeDelay                   : 25
         SyncLoops                     : 32
         PollIndex                     : 3
         PollValue                     : 0x53
         Memory Detail                 :

                                           Block Poll               Page                       Polled
           Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           eeprom                 65    20     4    0 no       1024    4      0  3600  3600 0xff 0xff
           flash                  65     6   128    0 yes     32768  128    256  4500  4500 0xff 0xff
           lfuse                   0     0     0    0 no          1    1      0  4500  4500 0x00 0x00
           hfuse                   0     0     0    0 no          1    1      0  4500  4500 0x00 0x00
           efuse                   0     0     0    0 no          1    1      0  4500  4500 0x00 0x00
           lock                    0     0     0    0 no          1    1      0  4500  4500 0x00 0x00
           calibration             0     0     0    0 no          1    1      0     0     0 0x00 0x00
           signature               0     0     0    0 no          3    1      0     0     0 0x00 0x00

         Programmer Type : linuxspi
         Description     : Use Linux SPI device in /dev/spidev*

avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.00s

avrdude: Device signature = 0x1e950f (probably m328p)
avrdude: reading flash memory:

Reading | ################################################## | 100% 8.72s

avrdude: writing output file "readback_m328p_linuxspi"

avrdude done.  Thank you.

pi@raspberrypi:~ $ /usr/local/bin/avrdude -c linuxspi -p m328p 
-P /dev/spidev0.0:/dev/gpiochip0 -U flash:v:readback_m328p_linuxspi:i -v

avrdude: Version 7.0-20220705 (64ee485)
         Copyright (c) Brian Dean, http://www.bdmicro.com/
         Copyright (c) Joerg Wunsch

[... 27 lines cut ...]
           flash                  65     6   128    0 yes     32768  128    256  4500  4500 0xff 0xff
           lfuse                   0     0     0    0 no          1    1      0  4500  4500 0x00 0x00
           hfuse                   0     0     0    0 no          1    1      0  4500  4500 0x00 0x00
           efuse                   0     0     0    0 no          1    1      0  4500  4500 0x00 0x00
           lock                    0     0     0    0 no          1    1      0  4500  4500 0x00 0x00
           calibration             0     0     0    0 no          1    1      0     0     0 0x00 0x00
           signature               0     0     0    0 no          3    1      0     0     0 0x00 0x00

         Programmer Type : linuxspi
         Description     : Use Linux SPI device in /dev/spidev*

avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.00s

avrdude: Device signature = 0x1e950f (probably m328p)
avrdude: verifying flash memory against readback_m328p_linuxspi:

Reading | ################################################## | 100% 9.93s

avrdude: 32768 bytes of flash verified

avrdude done.  Thank you.

```

mcuee, 2022-07-10:
> This seems to be the issue for `linuxgpio` as well:

Do you have similar fix to `linuxgpio` like the #933 for the linuxspi?

mcuee, 2022-07-10:
It also works with ATmega2560.
```
pi@raspberrypi:~/Desktop/build/avr/avrdude_hans/build_linux/src $ ./avrdude -C ./avrdude.conf -c linuxspi  -p m2560 -U flash:w:./blinky_m2560.hex:i

avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.00s

avrdude: Device signature = 0x1e9801 (probably m2560)
avrdude: NOTE: "flash" memory has been specified, an erase cycle will be performed
         To disable this feature, specify the -D option.
avrdude: erasing chip
avrdude: reading input file "./blinky_m2560.hex"
avrdude: writing flash (261414 bytes):

Writing | ################################################## | 100% 78.12s

avrdude: 261414 bytes of flash written
avrdude: verifying flash memory against ./blinky_m2560.hex:

Reading | ################################################## | 100% 142.50s

avrdude: 261414 bytes of flash verified

avrdude done.  Thank you.

```

MCUdude, 2022-07-10:
> Do you have similar fix to linuxgpio like the https://github.com/avrdudes/avrdude/pull/933 for the linuxspi?

I may look at a solution for `linuxgpio` as well, but I'd like PR #933 to be reviewed and merged first, preferably by @dl8dtl.

closed 2022-07-18 by stefanrueger