avrdudes/avrdude #927: Incorrect `-b` conversion for linuxspi programmer

per1234, 2022-04-10:
The initial implementation of the `linuxspi` programmer (https://github.com/avrdudes/avrdude/commit/40b0b104d6967ca75929706996b3dede304b9b25) used the `-b` flag as the way for the user to set the SPI clock frequency via the command line.

This usage was deprecated in favor of using the `-B` flag (https://github.com/avrdudes/avrdude/commit/2aee540bfa5429594ca3ebdbf42fab3f3452bbb1), with a conversion of the `-b` flag value accompanied by a warning.

My expectation is that this conversion would cause `-b 2000000` to be equivalent to `-b 2MHz`, but it is dramatically slower:

```text
$ time avrdude -v -p atmega328p -c linuxspi -U "flash:w:chaucer32k.ino.hex:i" -P "/dev/spidev0.0:/dev/gpiochip0" -b 2000000

avrdude: Version 6.99
         Copyright (c) Brian Dean, http://www.bdmicro.com/
         Copyright (c) Joerg Wunsch

         System wide configuration file is "/home/per/Documents/deleteme/avrdude/nightly/bin/../etc/avrdude.conf"
         User configuration file is "/home/per/.avrduderc"
         User configuration file does not exist or is not a regular file, skipping

         Using Port                    : /dev/spidev0.0:/dev/gpiochip0
         Using Programmer              : linuxspi
         Overriding Baud Rate          : 2000000
avrdude: obsolete use of -b <clock> option for bit clock; use -B <clock>
         AVR Part                      : ATmega328P
         Chip Erase delay              : 9000 us
         PAGEL                         : PD7
         BS2                           : PC2
         RESET disposition             : dedicated
         RETRY pulse                   : SCK
         Serial program mode           : yes
         Parallel program mode         : yes
         Timeout                       : 200
         StabDelay                     : 100
         CmdexeDelay                   : 25
         SyncLoops                     : 32
         PollIndex                     : 3
         PollValue                     : 0x53
         Memory Detail                 :

                                           Block Poll               Page                       Polled
           Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           eeprom                 65    20     4    0 no       1024    4      0  3600  3600 0xff 0xff
           flash                  65     6   128    0 yes     32768  128    256  4500  4500 0xff 0xff
           lfuse                   0     0     0    0 no          1    1      0  4500  4500 0x00 0x00
           hfuse                   0     0     0    0 no          1    1      0  4500  4500 0x00 0x00
           efuse                   0     0     0    0 no          1    1      0  4500  4500 0x00 0x00
           lock                    0     0     0    0 no          1    1      0  4500  4500 0x00 0x00
           calibration             0     0     0    0 no          1    1      0     0     0 0x00 0x00
           signature               0     0     0    0 no          3    1      0     0     0 0x00 0x00

         Programmer Type : linuxspi
         Description     : Use Linux SPI device in /dev/spidev*

avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.02s

avrdude: Device signature = 0x1e950f (probably m328p)
avrdude: NOTE: "flash" memory has been specified, an erase cycle will be performed
         To disable this feature, specify the -D option.
avrdude: erasing chip
avrdude: reading input file "chaucer32k.ino.hex"
avrdude: writing flash (31560 bytes):

Writing | ################################################## | 100% 193.02s

avrdude: 31560 bytes of flash written
avrdude: verifying flash memory against chaucer32k.ino.hex:

Reading | ################################################## | 100% 190.86s

avrdude: 31560 bytes of flash verified

avrdude done.  Thank you.

real	6m24.073s
user	0m0.971s
sys	0m5.416s
```
```text
$ time avrdude -v -p atmega328p -c linuxspi -U "flash:w:chaucer32k.ino.hex:i" -P "/dev/spidev0.0:/dev/gpiochip0" -B 2MHz

avrdude: Version 6.99
         Copyright (c) Brian Dean, http://www.bdmicro.com/
         Copyright (c) Joerg Wunsch

         System wide configuration file is "/home/per/Documents/deleteme/avrdude/nightly/bin/../etc/avrdude.conf"
         User configuration file is "/home/per/.avrduderc"
         User configuration file does not exist or is not a regular file, skipping

         Using Port                    : /dev/spidev0.0:/dev/gpiochip0
         Using Programmer              : linuxspi
         Setting bit clk period        : 0.5
         AVR Part                      : ATmega328P
         Chip Erase delay              : 9000 us
         PAGEL                         : PD7
         BS2                           : PC2
         RESET disposition             : dedicated
         RETRY pulse                   : SCK
         Serial program mode           : yes
         Parallel program mode         : yes
         Timeout                       : 200
         StabDelay                     : 100
         CmdexeDelay                   : 25
         SyncLoops                     : 32
         PollIndex                     : 3
         PollValue                     : 0x53
         Memory Detail                 :

                                           Block Poll               Page                       Polled
           Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           eeprom                 65    20     4    0 no       1024    4      0  3600  3600 0xff 0xff
           flash                  65     6   128    0 yes     32768  128    256  4500  4500 0xff 0xff
           lfuse                   0     0     0    0 no          1    1      0  4500  4500 0x00 0x00
           hfuse                   0     0     0    0 no          1    1      0  4500  4500 0x00 0x00
           efuse                   0     0     0    0 no          1    1      0  4500  4500 0x00 0x00
           lock                    0     0     0    0 no          1    1      0  4500  4500 0x00 0x00
           calibration             0     0     0    0 no          1    1      0     0     0 0x00 0x00
           signature               0     0     0    0 no          3    1      0     0     0 0x00 0x00

         Programmer Type : linuxspi
         Description     : Use Linux SPI device in /dev/spidev*

avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.00s

avrdude: Device signature = 0x1e950f (probably m328p)
avrdude: NOTE: "flash" memory has been specified, an erase cycle will be performed
         To disable this feature, specify the -D option.
avrdude: erasing chip
avrdude: reading input file "chaucer32k.ino.hex"
avrdude: writing flash (31560 bytes):

Writing | ################################################## | 100% 2.58s

avrdude: 31560 bytes of flash written
avrdude: verifying flash memory against chaucer32k.ino.hex:

Reading | ################################################## | 100% 1.37s

avrdude: 31560 bytes of flash verified

avrdude done.  Thank you.

real	0m4.101s
user	0m0.268s
sys	0m2.066s
```

Even though `-b` was never used in an official AVRDUDE release, it has been out in the wild for the last ~9 years via @kcuzner's original `linuxspi` implementation, which was later [pulled into Arduino's AVRDUDE builds](https://github.com/facchinm/avrdude/commit/d7eacf839c4e040b155b2903f605d94201fc5463), released in "6.3.0-arduino14" onward.

MCUdude, 2022-04-10:
Great to see you here @per1234!

There is something funny with the whole linuxspi bitclock handling. The `pgm->bitclock` should hold the bit clock period in microseconds. However, when no -B parameter is provided, the `pgm->bitclock` [is set to 5E-6](https://github.com/avrdudes/avrdude/blob/7e26a153752c9ebf346bd4e5db2e7c9b78a8b3c6/src/linuxspi.c#L228-L233).

Is it very slow if you specify the bit clock speed in microseconds, for instance `-B1` (1 us period/1 MHz)?

How fast is it if you don't provide a `-b` or `-B` argument at all?

MCUdude, 2022-04-10:
I figured I'd sooner or later have to dive into this myself, so I dusted off my trusty Raspberry Pi model B, threw in a new image for headless mode use, and built Avrdude using cmake.

I can confirm that the -B option works as it should, but -b doesn't. The following code solves this issue:

```c
pgm->bitclock = 1.0 / pgm->baudrate; // 1.0 instead of 1E6
```

I've attached a binary if you want to test it for yourself and don't want to build the source code. If it works for you I'll provide a PR.

[Avrdude-linuxspi.zip](https://github.com/avrdudes/avrdude/files/8459946/Avrdude-linuxspi.zip)

per1234, 2022-04-11:
Thanks @MCUdude! That works perfectly for me.

MCUdude, 2022-04-11:
Great, I'll submit a PR right away!

PR #931 (2022-05-01): Fix linuxspi baud to clock period calculation
Even though the `-b` option is deprecated, the baud to clock period calculation should work regardless.

Closes #927
closed 2022-05-01 by dl8dtl