avrdudes/avrdude #928: KT-LINK interface support

cederom, 2022-04-10:
Hello world :-)

Because ATtiny416XNANO (missing TPI protocol) nor ATtiny104XNANO (only uses DebugWIRE) that I have at hand can be used to flash ATtiny10 that I need, I have created a definition for KT-LINK interface that is based on FT2232H chip that I also have at hand :-)

I will provide a PR in a moment :-)

Note: In order to use TPI you need to connect both `TDI` and `TDO` over a resistor to `TPIDATA`.

```
TDO --[470R]-- TPIDATA --[470R]-- TDI
```

Just not to spam the PR I will provide example runs below.

```
hexagon% uname -a
FreeBSD hexagon 13.1-STABLE FreeBSD 13.1-STABLE #0 stable/13-n250096-4f69c575996: Fri Mar 25 03:50:58 CET 2022     root@hexagon:/usr/obj/usr/src/amd64.amd64/sys/GENERIC  amd64
```

```
hexagon% avrdude -c ktlink -p t10 -B 100khz

avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.00s

avrdude: Device signature = 0x1e9003 (probably t10)

avrdude done.  Thank you.
```

```
hexagon% avrdude -c ktlink -p t10 -B 100khz -vvv

avrdude: Version 6.4
         Copyright (c) Brian Dean, http://www.bdmicro.com/
         Copyright (c) Joerg Wunsch

         System wide configuration file is "/usr/local/etc/avrdude.conf"
         User configuration file is "/XXX/.avrduderc"

         Using Port                    : usb
         Using Programmer              : ktlink
         Setting bit clk period        : 10.0
I avrftdi_open(713): Using device VID:PID 0403:bbe2 and SN '(null)' on interface A.
D write_flush(478): Setting pin direction (0x0000) and value (0x0000)
I set_frequency(210): Using frequency: 100000
I set_frequency(211): Clock divisor: 0x003b
D avrftdi_check_pins_mpsse(570): Using valid mask mpsse: 0x0000fff8
D avrftdi_check_pins_bb(534): Using valid mask bibanging: 0x0000ffff
I avrftdi_pin_setup(644): Pin direction mask: e523
I avrftdi_pin_setup(645): Pin value mask: 6400
D set_pin(243): Setting pin 15 (ACBUS15) as PGMLED: high (high active)
D write_flush(478): Setting pin direction (0xe523) and value (0xe400)
         AVR Part                      : ATtiny10
         Chip Erase delay              : 0 us
         PAGEL                         : P00
         BS2                           : P00
         RESET disposition             : dedicated
         RETRY pulse                   : SCK
         serial program mode           : yes
         parallel program mode         : yes
         Timeout                       : 0
         StabDelay                     : 0
         CmdexeDelay                   : 0
         SyncLoops                     : 0
         ByteDelay                     : 0
         PollIndex                     : 0
         PollValue                     : 0x00
         Memory Detail                 :
[... 225 lines cut ...]
0280  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
0290  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
02a0  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
02b0  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
02c0  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
02d0  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
02e0  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
02f0  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
0300  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
0310  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
0320  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
0330  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
0340  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
0350  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
0360  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
0370  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
0380  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
0390  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
03a0  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
03b0  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
03c0  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
03d0  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
03e0  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
03f0  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
```

cederom, 2022-04-10:
Okay so the local ~/.avrduderc` part is presented below. Where should I put it in the sources? :-)
```
# Kristech KT-LINK FT2232H interface with IO switching and voltage buffers.
# Created on 20220410 by CeDeROM Tomasz CEDRO (www.cederom.io).
# Interface DataSheet: https://kristech.pl/files/KT-LINK-UM-ENG.pdf
# AVRDUDE FT2232H PIN NUMBER DECODE:
#  | 0      | 1      | .. | 7      | 8      | 9      | .. | 15     |
#  | ADBUS0 | ADBUS1 | .. | ADBUS7 | ACBUS0 | ACBUS1 | .. | ACBUS7 |
# KT-LINK JTAG CONN:
#  1=Vsense(->EXT13), 19=5V(EXT1->EXT3), 20=GND, 3=TPIRST, 9=TPICLK, 7=TPIDATA.
# INTERNALS CONFIGURATION ("~" MEANS ACTIVE LOW):
#  ~TRST_EN=10(ACBUS2), ~CLK_EN=14(ACBUS6), ~MOSI_EN=13(ACBUS5),
#  TMS_SEL=5(ADBUS5), ~TMS_EN=12(ACBUS4), LED=~15(ACBUS7).
# CONNECTION NOTES:
#  * Connect EXT connector pin 1 with 3 to get 5V on JTAG connector pin 19.
#  * Connect JTAG connector pin 1 to 5V (i.e. EXT pin 13 or JTAG pin 19).
#  * For TPI connection use resistors: TDO --[470R]-- TPIDATA --[470R]-- TDI.
#  * Powering target from JTAG pin 19 allows KT-LINK current measurement.
programmer
  id    = "ktlink";
  desc  = "KT-LINK FT2232H interface with IO switching and voltage buffers.";
  type  = "avrftdi";
  connection_type = usb;
  usbvid= 0x0403;
  usbpid= 0xBBE2;
  usbdev= "A";
  reset = 8;
  sck   = 0;
  mosi  = 1;
  miso  = 2;
  buff  = ~10,~14,~13,5;
  rdyled = ~15;
;
```

dl8dtl, 2022-04-10:
> Where should I put it in the sources?

There's already a number of FTDI-based devices present in `avrdude.conf.in`; put it next to them, and submit a PR.

cederom, 2022-04-10:
Aaah sorry and thank you @dl8dtl, I messed up my vim file browser sort order, PR  https://github.com/avrdudes/avrdude/pull/930 created :-)

closed 2022-05-05 by MCUdude

cederom, 2022-05-05:
Thank you @dl8dtl and @MCUdude :-)