avrdudes/avrdude #932: linuxgpio issues

MCUdude, 2022-04-11:
I've just stepped into the linuxspi/linuxgpio land, and the process hasn't been as straightforward as with other programmers.

First, it requires a port passed as a `-P` argument. I can't find any documentation on what this port is supposed to be, but `-B GPIO` seems to do the trick. A default option should be present IMO.

The other issue I've has is that when a session is force-terminated using ctrl+c, the GPIOs remains "occupied", which is a real bummer...

```c
$ sudo ./build_linux/src/avrdude -v -C build_linux/src/avrdude.conf -c linuxgpio -P GPIO -p atmega1284p -Uflash:w:build_linux/src/chaucer32k.hex:i 

avrdude: Version 6.99-20220405 (7e26a15)
         Copyright (c) Brian Dean, http://www.bdmicro.com/
         Copyright (c) Joerg Wunsch

         System wide configuration file is "build_linux/src/avrdude.conf"
         User configuration file is "/root/.avrduderc"
         User configuration file does not exist or is not a regular file, skipping

         Using Port                    : GPIO
         Using Programmer              : linuxgpio
         AVR Part                      : ATmega1284P
         Chip Erase delay              : 55000 us
         PAGEL                         : PD7
         BS2                           : PA0
         RESET disposition             : dedicated
         RETRY pulse                   : SCK
         Serial program mode           : yes
         Parallel program mode         : yes
         Timeout                       : 200
         StabDelay                     : 100
         CmdexeDelay                   : 25
         SyncLoops                     : 32
         PollIndex                     : 3
         PollValue                     : 0x53
         Memory Detail                 :

                                           Block Poll               Page                       Polled
           Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           eeprom                 65    10   128    0 no       4096    8      0  9000  9000 0xff 0xff
           flash                  65    10   256    0 yes    131072  256    512  4500  4500 0xff 0xff
           lock                    0     0     0    0 no          1    1      0  9000  9000 0x00 0x00
           lfuse                   0     0     0    0 no          1    1      0  9000  9000 0x00 0x00
           hfuse                   0     0     0    0 no          1    1      0  9000  9000 0x00 0x00
           efuse                   0     0     0    0 no          1    1      0  9000  9000 0x00 0x00
           signature               0     0     0    0 no          3    1      0     0     0 0x00 0x00
           calibration             0     0     0    0 no          1    1      0     0     0 0x00 0x00

         Programmer Type : linuxgpio
         Description     : Use the Linux sysfs interface to bitbang GPIO lines
         Pin assignment  : /sys/class/gpio/gpio{n}
           RESET   =  25
           SCK     =  11
           MOSI    =  10
           MISO    =  9

avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.00s

avrdude: Device signature = 0x1e9705 (probably m1284p)
avrdude: NOTE: "flash" memory has been specified, an erase cycle will be performed
         To disable this feature, specify the -D option.
avrdude: erasing chip
avrdude: reading input file "build_linux/src/chaucer32k.hex"
avrdude: writing flash (31668 bytes):

Writing | ################################################## | 100% 27.35s

avrdude: 31668 bytes of flash written
avrdude: verifying flash memory against build_linux/src/chaucer32k.hex:

Reading | ##########                                         | 20% 20.27s^C

$ sudo ./build_linux/src/avrdude -v -C build_linux/src/avrdude.conf -c linuxgpio -P GPIO -p atmega1284p -Uflash:w:build_linux/src/chaucer32k.hex:i 

avrdude: Version 6.99-20220405 (7e26a15)
         Copyright (c) Brian Dean, http://www.bdmicro.com/
         Copyright (c) Joerg Wunsch

         System wide configuration file is "build_linux/src/avrdude.conf"
         User configuration file is "/root/.avrduderc"
         User configuration file does not exist or is not a regular file, skipping

         Using Port                    : GPIO
         Using Programmer              : linuxgpio
Can't export GPIO 25, already exported/busy?: Device or resource busyavrdude: opening programmer "linuxgpio" on port "GPIO" failed

```

mcuee, 2022-05-20:
> First, it requires a port passed as a -P argument. I can't find any documentation on what this port is supposed to be, but -B GPIO seems to do the trick. A default option should be present IMO.

I have never tried this option but it seems to me from various website that -P is not required.
Ref: https://learn.adafruit.com/program-an-avr-or-arduino-using-raspberry-pi-gpio-pins?view=all

mcuee, 2022-05-20:
> The other issue I've has is that when a session is force-terminated using ctrl+c, the GPIOs remains "occupied", which is a real bummer...

Do you think moving to libgpiod will help to sort out this issue?
https://github.com/avrdudes/avrdude/issues/831

mcuee, 2022-06-09:
I will try out set up the test setup using linuxspi/linuxgpio as well since I have quite a few Raspberry Pi boards (Raspberry Pi 3B+ and 400) as well as Orange Pi boards.

mcuee, 2022-07-10:
I have followed the follow two guides to get linuxspi working and I will test out linuxgpio as well.

Ref:
https://www.mikrocontroller.net/articles/Raspberry_Pi_als_Universalprogrammer
https://ozzmaker.com/program-avr-using-raspberry-pi-gpio/

mcuee, 2022-07-10:
Somehow I could not even be able to get linuxgpio working.

```
pi@raspberrypi:~/Desktop/build/avr/avrdude/build_linux/src $ sudo ./avrdude -C ./avrdude.conf 
-c linuxgpio -P GPIO -p m328p -v

avrdude: Version 7.0-20220705 (64ee485)
         Copyright (c) Brian Dean, http://www.bdmicro.com/
         Copyright (c) Joerg Wunsch

         System wide configuration file is "./avrdude.conf"
         User configuration file is "/root/.avrduderc"
         User configuration file does not exist or is not a regular file, skipping

         Using Port                    : GPIO
         Using Programmer              : linuxgpio
Can't export GPIO 25, already exported/busy?: Device or resource busy
avrdude: opening programmer "linuxgpio" on port "GPIO" failed

avrdude done.  Thank you.

```

mcuee, 2022-07-10:
Rebooting the Raspberry Pi 3B+ sorted out the issue.

```
pi@raspberrypi:~/Desktop/build/avr/avrdude/build_linux/src $ sudo ./avrdude -C ./avrdude.conf -c linuxgpio -P GPIO -p m328p -U flash:w:./blinky_m328p_with_bootloader.hex:i

avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.00s

avrdude: Device signature = 0x1e950f (probably m328p)
avrdude: NOTE: "flash" memory has been specified, an erase cycle will be performed
         To disable this feature, specify the -D option.
avrdude: erasing chip
avrdude: reading input file "./blinky_m328p_with_bootloader.hex"
avrdude: writing flash (32768 bytes):

Writing | ################################################## | 100% 7.70s

avrdude: 32768 bytes of flash written
avrdude: verifying flash memory against ./blinky_m328p_with_bootloader.hex:

Reading | ################################################## | 100% 6.50s

avrdude: 32768 bytes of flash verified

avrdude done.  Thank you.

```

mcuee, 2022-07-10:
My setup photo.
[image]

mcuee, 2022-07-11:
With the above setup, I can confirm issue #455 is a bug and it applies to both linuxgpio and linuxspi.
* https://github.com/avrdudes/avrdude/issues/455

mcuee, 2023-01-05:
#455 and #1263 have been fixed by PR #1265.
* #455 
* * #1263
* #1265

mcuee, 2023-01-06:
> First, it requires a port passed as a -P argument. I can't find any documentation on what this port is supposed to be, but -P GPIO seems to do the trick. A default option should be present IMO.

I agree, `avrdude.conf.in` needs to be updated to have the defaul `GPIO`.

> The other issue I've had is that when a session is force-terminated using ctrl+c, the GPIOs remain "occupied", which is a real bummer...

This is a real pain. I think we have to move away from sysfs to libgpiod.
* #831

mcuee, 2023-02-04:
#1299 have been created to add libgpiod support.
* #1299

MCUdude, 2023-03-22:
Absolutely!

closed 2023-03-22 by MCUdude