avrdudes/avrdude #935: Programmer DIAMEX-ISP-USB not found

mc3, 2022-04-12:
With a ~/.avrduderc of
```
default_programmer = "diamex-avr-usb";
programmer
 id    = "diamex-avr-usb";
 desc  = "Diamex AVR USB Programmer";
 type  =  "stk500v2";
 connection_type = usb;
 usbvid     = 0x16c0;
 usbpid     = 0x2a9b;
;
```
and a USB device:
```
root@home2l:/usr/local/src/fbsd_local_ports/devel/avrdude # usbconfig -d 0.2 dump_device_desc
ugen0.2: <ERFOS AVR-ISP2> at usbus0, cfg=0 md=HOST spd=FULL (12Mbps) pwr=ON (500mA)

  bLength = 0x0012 
  bDescriptorType = 0x0001 
  bcdUSB = 0x0200 
  bDeviceClass = 0x0002  <Communication device>
  bDeviceSubClass = 0x0000 
  bDeviceProtocol = 0x0000 
  bMaxPacketSize0 = 0x0040 
  idVendor = 0x16c0 
  idProduct = 0x2a9b 
  bcdDevice = 0x4340 
  iManufacturer = 0x0001  <ERFOS>
  iProduct = 0x0002  <AVR-ISP2>
  iSerialNumber = 0x0003  <19331-49101-552>
  bNumConfigurations = 0x0001 
```
I get:
```
root@home2l:/usr/local/src/fbsd_local_ports/devel/avrdude # avrdude -p t85 -vvv -t

avrdude: Version 6.4
         Copyright (c) Brian Dean, http://www.bdmicro.com/
         Copyright (c) Joerg Wunsch

         System wide configuration file is "/usr/local/etc/avrdude.conf"
         User configuration file is "/root/.avrduderc"

         Using Port                    : usb
         Using Programmer              : diamex-avr-usb
avrdude: usbdev_open(): did not find any USB device "usb" (0x03eb:0x2104)
```
Looking at 
usbdev_open, I see:
```
(lldb) p/x pinfo
(pinfo) $1 = {
  baud = 0x00000000210403eb
  usbinfo = (vid = 0x03eb, pid = 0x2104, flags = 0x0000)
```
usbdev_open is called from stk500v2_open, where I see:
```
   pinfo.usbinfo.vid = USB_VENDOR_ATMEL;
    pinfo.usbinfo.pid = USB_DEVICE_AVRISPMKII;
```
This seems to be hard wired for all stk500v2 devices, ignoring my configured usbvid and usbpid.

I can't find a configuration, which allows me to use this device.

Web site of manufacturer:
[DIAMEX USB ISP-Programmer Stick für AVR](https://www.diamex.de/dxshop/DIAMEX-USB-ISP-Programmer-Stick-fuer-AVR)

mc3, 2022-04-13:
It's not such simple. Simply patching USB_VENDOR_ATMEL and USB_DEVICE_AVRISPMKII is not enough:
`
root@home2l:/usr/local/src/fbsd_local_ports/devel/avrdude # avrdude -p t85 -vvv -t

avrdude: Version 6.4
         Copyright (c) Brian Dean, http://www.bdmicro.com/
         Copyright (c) Joerg Wunsch

         System wide configuration file is "/usr/local/etc/avrdude.conf"
         User configuration file is "/root/.avrduderc"

         Using Port                    : usb
         Using Programmer              : diamex-avr-usb
avrdude: usbdev_open(): Found AVR-ISP2, serno: 19331-49101-552
avrdude: max packet size expected 64, but found 8 due to EP 0x82's wMaxPacketSize
avrdude: usbdev_send(): wrote -1 out of 1 bytes, err = Unknown error
avrdude: stk500_send_mk2(): failed to send command to serial port
avrdude: usbdev_recv_frame(): usb_bulk_read(): Unknown error
avrdude: stk500v2_recv_mk2: error in USB receive
avrdude: usbdev_send(): wrote -1 out of 1 bytes, err = Unknown error
avrdude: stk500_send_mk2(): failed to send command to serial port
^C

`

mariusgreuel, 2022-04-13:
> ATMEL-Studio 7: Tools->Add target->Select Tool: STK500, Select serial port: COM...

Judging by their description, the programmer implements a virtual serial port. In that case you do not specify the USB vid:pid, but the serial device via the `-P` option. Meaning you do not need a config, but just use `-c stk500v2`. Also, they claim to implement STK500, not STK500v2.

mc3, 2022-04-14:
avrdude is running in a jail on FreeBSD 13.0-STABLE. 
```
root@home2l:/ #	usbconfig 
ugen0.1: <0x8086 XHCI root HUB> at usbus0, cfg=0 md=HOST spd=SUPER (5.0Gbps) pwr=SAVE (0mA)
ugen0.3: <vendor 0x046a product 0x0023> at usbus0, cfg=0 md=HOST spd=LOW (1.5Mbps) pwr=ON (100mA)
ugen0.4: <Logitech USB Optical Mouse> at usbus0, cfg=0 md=HOST spd=LOW (1.5Mbps) pwr=ON (100mA)
ugen0.2: <ERFOS AVR-ISP2> at usbus0, cfg=0 md=HOST spd=FULL (12Mbps) pwr=ON (500mA)
root@home2l:/ #	ls -l /dev/cua*
crw-rw----  1 uucp  dialer  0x1fc Apr 14 11:12 /dev/cuaU0
crw-rw----  1 uucp  dialer  0x1fd Apr 14 11:12 /dev/cuaU0.init
crw-rw----  1 uucp  dialer  0x1fe Apr 14 11:12 /dev/cuaU0.lock
crw-rw----  1 uucp  dialer   0x4a Apr 14 11:22 /dev/cuau0
crw-rw----  1 uucp  dialer   0x4b Apr 14 09:54 /dev/cuau0.init
crw-rw----  1 uucp  dialer   0x4c Apr 14 09:54 /dev/cuau0.lock
crw-rw----  1 uucp  dialer   0x50 Apr 14 09:54 /dev/cuau1
crw-rw----  1 uucp  dialer   0x51 Apr 14 09:54 /dev/cuau1.init
crw-rw----  1 uucp  dialer   0x52 Apr 14 09:54 /dev/cuau1.lock
```
I have tried to get the serial communication to work with help of some people from the freebsd-hardware list without success. I always get timeouts. 
Without the -i option, I do not get any results. With -i, I see this:
´´´
root@home2l:~ #	avrdude -p t85 -c stk500 -P /dev/cuau0 -t -vvv -i 100

avrdude: Version 6.4
         Copyright (c) Brian Dean, http://www.bdmicro.com/
         Copyright (c) Joerg Wunsch

         System wide configuration file is "/usr/local/etc/avrdude.conf"
         User configuration file is "/root/.avrduderc"
         User configuration file does not exist or is not a regular file, skipping

         Using Port                    : /dev/cuau0
         Using Programmer              : stk500
         Setting isp clock delay        : 100
avrdude: ser_recv(): programmer is not responding
avrdude: stk500v2_ReceiveMessage(): timeout
avrdude: ser_recv(): programmer is not responding
avrdude: stk500v2_ReceiveMessage(): timeout
avrdude: ser_recv(): programmer is not responding
avrdude: stk500v2_ReceiveMessage(): timeout
avrdude: ser_recv(): programmer is not responding
avrdude: stk500v2_ReceiveMessage(): timeout
avrdude: ser_recv(): programmer is not responding
avrdude: stk500v2_ReceiveMessage(): timeout
avrdude: ser_recv(): programmer is not responding
avrdude: stk500v2_ReceiveMessage(): timeout
avrdude: stk500v2_getsync(): timeout communicating with programmer
         AVR Part                      : ATtiny85
         Chip Erase delay              : 4500 us
         PAGEL                         : P00
         BS2                           : P00
         RESET disposition             : possible i/o
         RETRY pulse                   : SCK
         serial program mode           : yes
         parallel program mode         : yes
         Timeout                       : 200
         StabDelay                     : 100
         CmdexeDelay                   : 25
         SyncLoops                     : 32
         ByteDelay                     : 0
         PollIndex                     : 3
         PollValue                     : 0x53
         Memory Detail                 :

                                  Block Poll               Page                       Polled
           Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           eeprom        65     6     4    0 no        512    4      0  4000  4500 0xff 0xff
                                  Block Poll               Page                       Polled
           Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
[... 31 lines cut ...]
avrdude: stk500v2_ReceiveMessage(): timeout
avrdude: ser_recv(): programmer is not responding
avrdude: stk500v2_ReceiveMessage(): timeout
avrdude: ser_recv(): programmer is not responding
avrdude: stk500v2_ReceiveMessage(): timeout
avrdude: ser_recv(): programmer is not responding
avrdude: stk500v2_ReceiveMessage(): timeout
avrdude: ser_recv(): programmer is not responding
avrdude: stk500v2_ReceiveMessage(): timeout
avrdude: ser_recv(): programmer is not responding
avrdude: stk500v2_ReceiveMessage(): timeout
avrdude: ser_recv(): programmer is not responding
avrdude: stk500v2_ReceiveMessage(): timeout
avrdude: stk500v2_getsync(): timeout communicating with programmer

. . .
avrdude: stk500v2_getsync(): timeout communicating with programmer
avrdude: stk500v2_command(): failed miserably to execute command 0x11
avrdude: stk500v2_disable(): failed to leave programming mode

avrdude done.  Thank you.

```
The cuaU device also does not show any results.
I'm out of ideas.

mariusgreuel, 2022-04-14:
The command-line syntax you use looks about right. What does the manual/vendor say about how to connect to the programmer?

> avrdude is running in a jail on FreeBSD 13.0-STABLE

I suppose there is potential for problems. Perhaps you explicitly need to set some permissions? @dl8dtl ?

mc3, 2022-04-15:
> The command-line syntax you use looks about right. What does the manual/vendor say about how to connect to the programmer?

There is no manual. Just one page to download:
[EN: Description](https://www.diamex.de/dxshop/mediafiles//Sonstiges/ENG%20ISP-Programmer%20Stick.pdf)
Or 2 pages in German, which give more details (e.g. adaptive SPI-bitrate adaption):
[DE: Bedienungsanleitung:](https://www.diamex.de/dxshop/mediafiles//Sonstiges/diamex-avr-isp2.pdf)
The target is on a ZIF socket and I have both switches turned on, so the programmer provides 5V to the target.
Unlike written in the German description, the green LED does not stay on after plugging in the programmer.
I'm waiting for email response from the vendor.

> > avrdude is running in a jail on FreeBSD 13.0-STABLE
> 
> I suppose there is potential for problems. Perhaps you explicitly need to set some permissions? @dl8dtl ?

I created the necessary devfs rules, I'm running avrdude as root and, as you can see in the log, the programmer can talk to the cuau0 device.

mc3, 2022-04-27:
Hans Petter Selasky <hps@selasky.org> found cause and work around:
===
I see from the Ellisys .png's what is wrong. The device doesn't support
resetting the data-toogle (CLEAR ENDPOINT HALT) which is a mandatory USB
control endpoint command. So the USB stack tries a couple of times and then
gives up and resets the device!

The device is not USB.org class compliant! :-(

Probably setting:

sysctl hw.usb.no_cs_fail=1

may help.
===
Indeed, it solved the issue.
Vender announced a new device with compliance.
-

closed 2022-04-27 by mc3