avrdudes/avrdude #937: Memory pollindex not used/useful

stefanrueger, 2022-04-16:
Pollindex can be set as a parameter in `avrdude.conf` for the part itself and, *independent of this,* for any of its memories. The part pollindex is documented and finds good use in various programmers, most notably avrftdi, ft232r and stk500v2.

The *memory* pollindex parameter, in contrast,
 - is never set in `avrdude.conf`
 - does not appear to be documented
 - is never used in anger apart from `avr_mem_display()` printing it

The code as is suggests the memory pollindex should be removed, and that `avr_mem_display()` should display the part pollindex as opposed to the uninitialised and unused memory pollindex.

Doing so affects avrpart.c, config_gram.y and libavrdude.h

mcuee, 2022-06-07:
Not sure about the original intention of memory pollindex. It would be good to find the history out before removing.

stefanrueger, 2022-08-06:
Currently, 162 parts in `avrdude.conf` define `pollindex`, all of which set it to 3:
``` sh
avrdude -p*/St | grep pollindex | grep -v 'x.0$' | cut -f1,3- | sort | uniq -c
    162  .pt	pollindex	3
``` 
Only three programmers use `pollindex`:
``` C
$ grep pollindex -rn src | grep -v src/avrdude.conf.in
src/avrftdi.c:879:		if (buf[p->pollindex-1] != p->pollvalue) {
src/lexer.l:194:pollindex        { yylval=NULL; return K_POLLINDEX; }
src/stk500v2.c:1122:  buf[7] = p->pollindex;
src/developer_opts.c:489:  _if_partout(intcmp, "%d", pollindex);
src/developer_opts.c:575:    _if_memout(intcmp, "%d", pollindex);
src/doc/avrdude.texi:1776:    pollindex        = <num> ;
src/avrpart.c:632:              m->mode, m->delay, m->blocksize, m->pollindex,
src/avrpart.c:900:  if(p->pollindex)
src/avrpart.c:901:    fprintf(f, "%sPollIndex                     : %d\n", prefix, p->pollindex);
src/config_gram.y:996:      current_part->pollindex = $3->value.number;
src/config_gram.y:1511:      current_mem->pollindex = $3->value.number;
src/ft245r.c:523:        if (res[p->pollindex-1] == p->pollvalue) return 0;
src/libavrdude.h:240:  int           pollindex;          /* stk500 v2 xml file parameter */
src/libavrdude.h:307:  int pollindex;              /* stk500 v2 xml file parameter */
```
At least `avrftdi.c` and `ft245r.c` do not guard themselves against unsuitable values of `p->pollindex` when they use it as array index. That is potentially a bug when pollindex is undefined. There are only 4 ISP parts that do not define pollindex (`AT90S1200` `AT90S4434` `ATmega406` `ATtiny11` `ATtiny28`), and they might not be programmed with the SPI interface (but I have not checked)

Amongst the atdf files, there are 184 that define the pollindex:
``` sh
$ grep pollindex -ihl atdf/* | wc
    184     184   15371
```
And if they do, it appears to be the same definition for each part:
```
$ grep pollindex -ih atdf/* -h | sort | uniq -c
    366           <property name="IspEnterProgMode_pollIndex" value="3"/>
    366           <property name="IspReadFuse_pollIndex" value="4"/>
    366           <property name="IspReadLock_pollIndex" value="4"/>
    366           <property name="IspReadOsccal_pollIndex" value="4"/>
    366           <property name="IspReadSign_pollIndex" value="4"/>
```
Except for the `ATmega8HVA` and `ATmega16HVA` each of the 184 parts has these 5 pollindex values in the `ISP_INTERFACE` and in the `ISP_INTERFACE_STK600` section.

And it looks like there _is_ a memory pollindex for fuses, lock byte, osccal and signature. AVRDUDE models pollindex for memories, but `avrdude.conf` does not set the memory pollindex for any memory of any part. And the only(!) time AVRDUDE uses `m->pollindex` is in `avrpart.c` for the printout of memory configurations, where it is always 0 (because unset).

For the time being, I leave this issue open
 - Should AVRDUDE model memory pollindex for fuses, lock byte, osccal and signature despite no programmer using it?
 - Should AVRDUDE reflect reality, no longer print m->pollindex (because unset/unused) and stop modelling memory pollindex?

What is relatively uncontroversial is that AVRDUDE should guard itself against unset part pollindex, and hence against referencing `buf[-1]` or `resp[-1]` which might lead to subtle and difficult to debug errors.

dl8dtl, 2022-08-06:
I think it's one of those parameters that are defined in the Atmel XML files for the STK500(v2) protocol.
It seems the FT245 code has been modelled after that as well.

It's probably not useful to have it per-memory.

mcuee, 2022-08-07:
Looks like the Atmel STK500 XML files are used for Atmel Studio 5 and 6. Part description xml files (not as detailed) are used in AVR Studio 4. Then atdf files are used in Microchip Studio 7.

After upgrading the AVR related DFPs in Microchip Studion 7, I got the same results as @stefanrueger.
```
$ grep pollindex -ihl atdf/* | wc
    184     184    3779

$ grep pollindex -ih atdf/* -h | sort | uniq -c
    366           <property name="IspEnterProgMode_pollIndex" value="3"/>
    366           <property name="IspReadFuse_pollIndex" value="4"/>
    366           <property name="IspReadLock_pollIndex" value="4"/>
    366           <property name="IspReadOsccal_pollIndex" value="4"/>
    366           <property name="IspReadSign_pollIndex" value="4"/>

```

stefanrueger, 2022-11-23:
Closing the issue as it's unclear what to do about this - and it doesn't really pose a problem

closed 2022-11-23 by stefanrueger