avrdudes/avrdude #938: Core SPM programmer: request for comments

stefanrueger, 2022-04-18:
This is to start a discussion on avrdude support for bootloaders (ie, SPM programmers).

I propose modelling SPM programmers along existing bootloaders and along the capabilities of the SPM/LPM opcode and EEPROM access (or, in newer parts, along the capabilities of the NVM Controller). This to complement avrdude's modelling of SPI programming, PDI, UPDI, HVPP, JTAG, Debugwire etc.

Some of the peculiarities of SPM programmers: they
 - Work without intermediate programming device
 - Want to be *really, really small* more often than not
 - Hence, *are unlikely to* implement all possible r/w on memories
 - *Do* have a page erase command
 - *Could* perform a read-erase-modify-write page feature for sub-page modifications in flash (but more often than not don't)
 - Operate on byte addresses, even for flash (`RAMPZ` and `Z` are set to *byte* addresses for the SPM and LPM opcodes)
 - Cannot write fuses (in contrast to external programming)
 - Need a reset signal for the part to enter the bootloader code

In essence, SPM programmers allow a higher abstraction level than other modes of programming: the vagaries of the actual to be programmed part are mostly addressed in the bootloader, though avrdude benefits from knowing what the highest flash address is, the flash page size and a few parameters that depend on the *bootloader but not on the part* (for example, which memories the bootloader can read/write and at which granularity, whether the bootloader is a vector bootloader, which interrupt is used for that, how big the bootloader is so avrdude does not overwrite it etc). 

One popular current avrdude SPM programmer is `arduino`, which is used for uploading/downloading applications and EEPROM contents via a bootloader on the part. It builds upon the STK500 protocol that models external programming and behaves as if the bootloader *was* an external programming device. Essentially, for all intents and purposes, the `arduino` programmer *is* an STK500 programmer one minor, but important, difference being that it plucks the DTR line with a view to issue reset on the connected part, so it runs the bootloader code.

In theory, a bootloader *could* emulate an STK500 external programmer, but in practice they *don't* necessarily do this.

I can see the following clear disadvantages of this bootloader-as-an-STK500 approach: `avrdude -c arduino` 
 - Requests SW/HW version from the SPM programmer, which costs code bytes in the bootloader (decode the request send something back) with little benefit
 - Humours not-needed get parameter and set parameter requests; there is no intermediate device with parameters; decoding the requests and ignoring them costs unnecessary bootloader code
 - Expects a chip erase command to be honoured, although such a command is neither needed nor universally implemented in bootloaders (because SPM flash programming *can* erasing a page before writing it)
 - Communicates 16-bit *word* addresses for both EEPROM and flash page reads/writes: the bootloader therefore needs to multiply the word address by two (4 bytes code), and for 128 kB flash parts puts the carry into RAMPZ (more code bytes); bootloaders of larger parts require the decoding (a few handful of code bytes gone) of a specific load extended-byte-of-a-*word-address* command *and* store that upper byte of a three-byte *word address* in a variable to be used later to compute the effective *byte address*. More code bytes gone. For bootloaders this is a clumsy way of learning the needed byte address. A small change of the protocol between avrdude and the bootloader would communicate the *16-bit byte address* of part memories with up to 64 kB flash, and the *24-bit byte address* of part with larger flash. Simples.
 - Does not particularly support vector bootloaders, which want the reset vector to point to them and want another (never otherwise used) vector in (or just behind) the vector table to point to the application; patching the vector table by bootloader itself during avrdude upload/verify is costly (the optiboot source code reckons some 110 bytes), but this is a task that a dedicated SPM programmer could easily do within the proposed SPM programmer and should cost the bootloader *zero* bytes code
 - Identifies the to be programmed part by requesting signature bytes through a protocol command exchange that needs to be decoded and served; again, this puts some strain on the bootloader but there are other mechanisms costing less (or no) bootloader code
 - And then there is Issue #918, where trailing 0xff-suppression in file input are prevented from being properly programmed and verified

So, I suggest writing a *generic core SPM programmer* (I am happy to do so, and have experience in doing so), which supports the following minimum capability of a bootloader
 - Paged flash write at page boundaries

and one or more of the following
 - Buffer flash read at arbitrary byte positions (max size is page size or 256 whichever is greater)
 - Buffer EEPROM read and write at arbitrary byte positions (max 256 buffer size)
 - EEPROM erase
 - Flash erase (apart from obviously the bootloader itself)
 - Vector bootloader

Optiboot is an example of a bootloader that supports flash read/write as above, with some optiboot renderings also supporting EEPROM r/w (it's a compile-time option). The core SPM programmer would emulate the EEPROM erase and/or flash erase writing 0xff to flash/EEPROM as a slower alternative if the bootloader does not provide this functionality.

I suggest using by and large the syntax of STK500 communications with small changes to the protocol to support 2/3-byte addresses and to support the bootloader telling `avrdude` early on which part it is and which subset of the 5 optional capabilities the bootloader has implemented. I suggest using two bytes for this, so this scheme can afford 11 bits for 2048 different parts; currently `avrdude` knows 313 parts (one of which AVR32), avr-libc 293 parts and Microchip's ATDF files know 323 AVR8 parts. The union of these three sources amounts to 374 parts though some of the older/smaller parts are clearly not suitable for bootloader programming at all (no RAM or no SPM).

The remaining parameters (application vector number for vector bootloaders, size of bootloader) can come from a few table bytes at the end of the bootloader, or failing that, from extended `-x` command line options.

For all this to work, `avrdude`'s proposed core SPM programmer might need/benefit from
 - A programmer parameter is_bootloader in `.conf`  (default off for backward compatibility)
 - The ability to switch off the removal of trailing 0xff sequences in the input file or on avr read
 - A readhook function at the *end* of input file reads, so the programmer can patch the vector table on upload and verify; the SPM programmer would insert a function during initialisation.

OK, that's the long of it. Comments appreciated.

stefanrueger, 2022-04-27:
From PR #936:
> I don't think there's a big difference in program size between "upon 'chip erase', loop over all application flash pages, and erase them" vs. "upon 'chip erase', do nothing, but when programming a page, first erase it".

This is not wrong. Below the two scenarios on the back of an SSD packet.

Upon 'chip erase', do nothing, but when programming a page, first erase it:
 - Bootloaders served by `avrdude -c arduino` will ignore a few requests anyway; with code for that already spent (out of necessity) the marginal cost of ignoring chip erase is zero
 - The marginal cost of a page erase before writing a page is 4 bytes (loading the page erase command and calling a subroutine that carries out the SPM opcode and waits/polls until it is finished)

Upon 'chip erase', loop over all application flash pages, and erase them:
 - Compare avrdude's request with chip erase and branch (4 bytes min)
 - Managing the page erase loop: initialise RAMPZ/Z, and increment/decrement that pointer by page size until it hits the bootloader/NULL; depending on part, page size, flash size, how well the compiler optimises (or you write assembler) I'd estimate min 12 bytes, typ 20 bytes, max 30 bytes
 - Page erase 4 bytes (loop contents)
 - Sign off code (min 2 bytes but probably more like 4-6 bytes)
 - Housekeeping (reinstating RAMPZ/Z etc) min zero bytes (but can be more)

So we are looking at approximately 22-44 bytes vs 4 bytes.

One question is whether this difference is big: it can make 10% of the bootloader size and, in many cases, risks pushing the bootloader size over a page boundary. And we are not even talking about reading the actual fuse byte that tells us whether chip erase implies EEPROM erase, and if needed erase that too.

Another is whether the second scenario is really needed.

In my world there is nothing wrong with the first scenario. The unused flash may look messy from previous use, but the suggested SPM programmer could emulate the chip erase by looping *in avrdude* (plenty o' RAM there) to clear the flash (slower, but that's the trade-off to save flash for *every* application). And the user calls the shots by using `-D` or not for explicit chip erase.

I like keeping flash for the application. But I am also partial to bootloaders: I like them for rapid prototyping. And testing. And EEPROM r/w. And for providing a `writepage(flash, ram)` subroutine accessible by the application. And, hence, for the ability to use the flash not used by the application as poor man's EEPROM.  (It is mildly satisfying to write a little sketch for a 3 USD ProMini rig with CR2032 and a temperature sensor that records to flash the 7200 temperature values of a 2 hour washing machine run, the result of which I can read with avrdude using the bootloader).

dl8dtl, 2022-04-27:
I was thinking that for an explicit SPM programmer implementation, it might be possible to offload a "chip erase emulation" (erase everything except the bootloader) into AVRDUDE - but then, only the bootloader knows the size of what constitutes "application flash", and least on small devices where it cannot be deduced by the fuses (which btw. is currently not possible anyway as AVRDUDE has no knowledge about the meaning of each fuse bit in each controller).

So, for an SPM programmer protocol, it would be essential to provide a method where AVRDUDE can query the size of the application section.

dl8dtl, 2022-04-27:
Btw., I tend to transfer this into the discussion area. Objections?

stefanrueger, 2022-04-27:
> Btw., I tend to transfer this into the discussion area. Objections?

Great Idea

> So, for an SPM programmer protocol, it would be essential to provide a method where AVRDUDE can query the size of the application section.

Correct. There are plenty of possibilities of doing so. The way I see it is that we have freedom to determine a) what SPM programmers can and cannot do (ie, how they should behave); b) how they communicate with the host; c) what avrdude promises to provide and what it promises to *not* request (to avoid code in the bootloader that ignores requests). The best outcome for this exercise is something that helps bootloader writers to write efficient bootloaders.

As an example, I have a prototype implementation of the following idea: the bootloader keeps a six-byte table at FLASHEND with
 - its version number (one byte)
 - a capabilities byte (EEPROM r/w, is_vector_bootloader, keep_reset_cause in R2, is_over_the_air bootloader, ...)
 - a two byte rjmp to a writepage(flash, ram) function or a ret if not implemented
 - the number of pages that the bootloader occupies
 - the vector number used for the r/jmp to the application

During handshake the bootloader lets know which part it sits on. Avrdude knows where FLASHEND of the part is and requests the six bytes below FLASHEND from the bootloader using the normal buffer flash read. So, that flash read routine doubles up for this info exchange at no extra cost for the bootloader. At the time I had not thought about bootloaders that do not wish to implement flash read.