avrdudes/avrdude #939: Add Apple M1 silicon to Nightly-Builds

hallard, 2022-04-19:
Hi there,

Tried back from #771, I tried today to build for my Mac M1 and and can't get the compile process going correctly to end on M1 silicon (I tried to investigate but each time I pass a new step, new errors fire until I'm stuck on last error that I can't fix)

Anyway, since I just wanted to try with this non standard baud rate setting on mac OS I gave a try with nightly builds
https://github.com/avrdudes/avrdude/wiki/Getting-Nightly-Builds-for-AVRDUDE
But there is no arm64 architecture (Apple M1) available. As rosetta  can run executables with x86_64 architecture it can't mix it with arm64 lib (which are by default native on my MAC), firing this message which makes sense

```shell
charles@m1-office:avrdude-macos-x86_64$ ./avrdude 
dyld[68388]: Library not loaded: /usr/local/opt/hidapi/lib/libhidapi.0.dylib
  Referenced from: /Users/charles/Downloads/avrdude-macos-x86_64/avrdude
  Reason: tried: '/usr/local/opt/hidapi/lib/libhidapi.0.dylib' (no such file), '/usr/local/lib/libhidapi.0.dylib' (mach-o file, but is an incompatible architecture (have 'arm64', need 'x86_64')), '/usr/lib/libhidapi.0.dylib' (no such file)
Abort trap: 6
```

I succeded to run downloading x86_64 libraries for libhidapi and libftdi and put them into `/usr/local/lib` (I'm lucky the native arm64 lib are not located into this folder so it should not conflict)

Process is not so straighforward, fetch correct architecture
````
brew fetch --force --bottle-tag=arm64_monterey hidapi libftdi
````

untargz them where you want then create symlink (or copy files)

````
sudo ln -s /Users/charles/Downloads/avrdude-macos-x86_64/hidapi/0.11.2/lib/libhidapi.0.dylib /usr/local/lib/
sudo ln -s /Users/charles/Downloads/avrdude-macos-x86_64/libftdi/1.5_2/lib/libftdi1.2.dylib /usr/local/lib/
````

And finally it works
```

charles@m1-office:avrdude-macos-x86_64$ ./avrdude 
Usage: avrdude [options]
Options:
  -p <partno>                Required. Specify AVR device.
  -b <baudrate>              Override RS-232 baud rate.
  -B <bitclock>              Specify JTAG/STK500v2 bit clock period (us).
  -C <config-file>           Specify location of configuration file.
  -c <programmer>            Specify programmer type.
  -D                         Disable auto erase for flash memory
  -i <delay>                 ISP Clock Delay [in microseconds]
  -P <port>                  Specify connection port.
  -F                         Override invalid signature check.
  -e                         Perform a chip erase.
  -O                         Perform RC oscillator calibration (see AVR053). 
  -U <memtype>:r|w|v:<filename>[:format]
                             Memory operation specification.
                             Multiple -U options are allowed, each request
                             is performed in the order specified.
  -n                         Do not write anything to the device.
  -V                         Do not verify.
  -t                         Enter terminal mode.
  -E <exitspec>[,<exitspec>] List programmer exit specifications.
  -x <extended_param>        Pass <extended_param> to programmer.
  -v                         Verbose output. -v -v for more.
  -q                         Quell progress output. -q -q for less.
  -l logfile                 Use logfile rather than stderr for diagnostics.
  -?                         Display this usage.

avrdude version 6.99-20220410 (60960ba), URL: <https://github.com/avrdudes/avrdude>
```

But would be much easier if we could have native build for Apple M1 silicon  :-)

dl8dtl, 2022-04-20:
I'd love to have native M1 builds, but what's the status of the required backend libraries? Are they available for M1 CPUs?

mariusgreuel, 2022-04-22:
Github Actions do not support Apple M1 builds, yet. I have subscribed to this issue and plan to add this to the avrdude builds as soon it is available:
<https://github.com/actions/virtual-environments/issues/2187>

@hallard What was the problem with the native build? I do not have a M1 Mac, but I would assume it to be not much different from the x86 builds.

hallard, 2022-04-22:
@dl8dtl what do you mean by backend libraries are you talking about `libusb` and `hidapi`? if so answer is yes

@mariusgreuel correct I assumed the same finishing deep dive onto makefile changes, filepath changes, going farther each time until I can't find a solution. some description of error [here](https://github.com/avrdudes/avrdude/issues/771) and the end (my message)

mcuee, 2022-05-08:
libusb, libftdi and  hidapi all have no issues with Apple M1. I have the Mac Mini M1 (2020) and I use homebrew.

dl8dtl, 2022-05-08:
Macports works as well, including the existing `build.sh` script.

hallard, 2022-05-08:
@mcuee I use homebrew and also these lib with other build project with no issue on M1, but I'm unable to compile avrdude using wiki instruction (manually or using `build.sh`)  can you confirm a git clone and `build.sh` run out of the box for you?

dl8dtl, 2022-05-08:
> but I'm unable to compile avrdude using wiki instruction

Can you elaborate on that? Error messages, what does "unable to compile" actually mean?

I didn't have any problems whatsoever.

hallard, 2022-05-08:
@dl8dtl sure it's described on the original #771 (at the end)

hallard, 2022-05-08:
And the first issue it can't compile is that `build.sh` assume homebrew located in `/usr/local` and on M1 (and latest homebrew version) it's in `/opt/homebrew/`

dl8dtl, 2022-05-08:
Even with the current codebase?

Strange.

Yes, I can confirm I could run it straight out of a current Git version, using `build.sh`. Prerequisites came from Macports in my case, but Homebrew ought to work as well.

I tried to run the resulting binary on a number of programmers (AtmelICE, USBasp, AVRISPmkII, FT245 board), and only rolled out the v7.0 release after this all went well.

dl8dtl, 2022-05-08:
> And the first issue it can't compile is that `build.sh` assume homebrew located in `/usr/local` and on M1 (and latest homebrew version) it's in `/opt/homebrew/`

Then please provide a patch for that. Shouldn't be all that hard. `build.sh` already contains the logic to probe for different locations, just extend it.

Sorry, I don't use Homebrew myself.

hallard, 2022-05-08:
Yeah I'm sure binary is working, but I'm not able to build nor get it on nighty builds so if you have can can attach it on I would love to try it :-)

hallard, 2022-05-08:
> Then please provide a patch for that. Shouldn't be all that hard. build.sh already contains the logic to probe for different locations, just extend it.

Would like to but for now I can't get compilation process to works on M1 with homebrew, so patch would be useless

mcuee, 2022-05-08:
> @mcuee I use homebrew and also these lib with other build project with no issue on M1, but I'm unable to compile avrdude using wiki instruction (manually or using `build.sh`) can you confirm a git clone and `build.sh` run out of the box for you?

I am following the instruction here using CMake (just change `/usr/local` to `/opt/homebrew`)
https://github.com/avrdudes/avrdude/wiki/Building-AVRDUDE-for-macOS#build-instructions

Build log:
```
mcuee@mcuees-Mac-mini avrdude % cmake -D CMAKE_C_FLAGS=-I/opt/homebrew/include -D CMAKE_EXE_LINKER_FLAGS=-L/opt/homebrew/Cellar -D CMAKE_BUILD_TYPE=RelWithDebInfo -B build_macos
-- The C compiler identification is AppleClang 13.1.6.13160021
-- The CXX compiler identification is AppleClang 13.1.6.13160021
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /Library/Developer/CommandLineTools/usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /Library/Developer/CommandLineTools/usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found Git: /opt/homebrew/bin/git (found version "2.36.0") 
-- Found FLEX: /usr/bin/flex (found version "2.6.4") 
-- Found BISON: /usr/bin/bison (found version "2.3")
-- Looking for libelf.h
-- Looking for libelf.h - not found
-- Looking for libelf/libelf.h
-- Looking for libelf/libelf.h - found
-- Looking for usb.h
-- Looking for usb.h - found
-- Looking for lusb0_usb.h
-- Looking for lusb0_usb.h - not found
-- Looking for libusb.h
-- Looking for libusb.h - not found
-- Looking for libusb-1.0/libusb.h
-- Looking for libusb-1.0/libusb.h - found
-- Looking for hidapi/hidapi.h
-- Looking for hidapi/hidapi.h - found
-- Looking for ftdi_tcioflush
-- Looking for ftdi_tcioflush - found
-- Configuration summary:
-- ----------------------
-- DO HAVE    libelf
-- DO HAVE    libusb
-- DO HAVE    libusb_1_0
-- DO HAVE    libhidapi
-- DO HAVE    libftdi (but prefer to use libftdi1)
-- DO HAVE    libftdi1
-- DISABLED   doc
-- DISABLED   parport
-- DISABLED   linuxgpio
-- DISABLED   linuxspi
-- ----------------------
-- Configuring done
-- Generating done
-- Build files have been written to: /Users/mcuee/build/avr/avrdude/build_macos
mcuee@mcuees-Mac-mini avrdude % make
make: *** No targets specified and no makefile found.  Stop.
mcuee@mcuees-Mac-mini avrdude % cmake --build build_macos
[  1%] [FLEX][Parser] Building scanner with flex 2.6.4
[  3%] [BISON][Parser] Building parser with bison 2.3
[  4%] Building C object src/CMakeFiles/libavrdude.dir/arduino.c.o
[  6%] Building C object src/CMakeFiles/libavrdude.dir/avr.c.o
[  8%] Building C object src/CMakeFiles/libavrdude.dir/avr910.c.o
[  9%] Building C object src/CMakeFiles/libavrdude.dir/avrftdi.c.o
[ 11%] Building C object src/CMakeFiles/libavrdude.dir/avrftdi_tpi.c.o
[ 13%] Building C object src/CMakeFiles/libavrdude.dir/avrpart.c.o
[ 14%] Building C object src/CMakeFiles/libavrdude.dir/bitbang.c.o
[ 16%] Building C object src/CMakeFiles/libavrdude.dir/buspirate.c.o
[ 18%] Building C object src/CMakeFiles/libavrdude.dir/butterfly.c.o
[ 19%] Building C object src/CMakeFiles/libavrdude.dir/config.c.o
[... 32 lines cut ...]
[ 67%] Building C object src/CMakeFiles/libavrdude.dir/stk500v2.c.o
[ 68%] Building C object src/CMakeFiles/libavrdude.dir/stk500generic.c.o
[ 70%] Building C object src/CMakeFiles/libavrdude.dir/teensy.c.o
[ 72%] Building C object src/CMakeFiles/libavrdude.dir/updi_link.c.o
[ 73%] Building C object src/CMakeFiles/libavrdude.dir/updi_nvm.c.o
[ 75%] Building C object src/CMakeFiles/libavrdude.dir/updi_readwrite.c.o
[ 77%] Building C object src/CMakeFiles/libavrdude.dir/updi_state.c.o
[ 78%] Building C object src/CMakeFiles/libavrdude.dir/usbasp.c.o
[ 80%] Building C object src/CMakeFiles/libavrdude.dir/usb_hidapi.c.o
[ 81%] Building C object src/CMakeFiles/libavrdude.dir/usb_libusb.c.o
[ 83%] Building C object src/CMakeFiles/libavrdude.dir/usbtiny.c.o
[ 85%] Building C object src/CMakeFiles/libavrdude.dir/update.c.o
[ 86%] Building C object src/CMakeFiles/libavrdude.dir/wiring.c.o
[ 88%] Building C object src/CMakeFiles/libavrdude.dir/xbee.c.o
[ 90%] Building C object src/CMakeFiles/libavrdude.dir/__/lexer.c.o
[ 91%] Building C object src/CMakeFiles/libavrdude.dir/__/config_gram.c.o
[ 93%] Linking C static library libavrdude.a
[ 93%] Built target libavrdude
[ 95%] Building C object src/CMakeFiles/avrdude.dir/main.c.o
[ 96%] Building C object src/CMakeFiles/avrdude.dir/term.c.o
[ 98%] Building C object src/CMakeFiles/avrdude.dir/whereami.c.o
[100%] Linking C executable avrdude
[100%] Built target avrdude

```

mcuee, 2022-05-08:
As for build.sh, it will fail as it does not detect Homebrew M1 properly.

```
   darwin)
	# determine whether we are running using Mac Ports
	# if not, assume Mac Brew
	if [ -f /opt/local/bin/port ]
	then
	    build_flags="${build_flags} -D CMAKE_C_FLAGS=-I/opt/local/include -D CMAKE_EXE_LINKER_FLAGS=-L/opt/local/lib"
	else
	    build_flags="${build_flags} -D CMAKE_C_FLAGS=-I/usr/local/include -D CMAKE_EXE_LINKER_FLAGS=-L/usr/local/Cellar"
	fi
	;;
```

hallard, 2022-05-08:
you saved my day,  now it works, not sure what changed and where (may be an update on my mac) but now it works, trust me I spent a couple of hours last time on my 2 macs and it's not my first time compiling

````shell
charles@mini-m1:devt$ git clone https://github.com/avrdudes/avrdude/
Clonage dans 'avrdude'...
remote: Enumerating objects: 12203, done.
remote: Counting objects: 100% (2624/2624), done.
remote: Compressing objects: 100% (756/756), done.
remote: Total 12203 (delta 2088), reused 2095 (delta 1864), pack-reused 9579
Réception d'objets: 100% (12203/12203), 6.22 Mio | 11.75 Mio/s, fait.
Résolution des deltas: 100% (8450/8450), fait.
````

````shell
charles@mini-m1:devt$ cd avrdude/
charles@mini-m1:avrdude$ cmake -D CMAKE_C_FLAGS=-I/opt/homebrew/include -D CMAKE_EXE_LINKER_FLAGS=-L/opt/homebrew/Cellar -D CMAKE_BUILD_TYPE=RelWithDebInfo -B build_macos
-- The C compiler identification is AppleClang 13.1.6.13160021
-- The CXX compiler identification is AppleClang 13.1.6.13160021
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /Library/Developer/CommandLineTools/usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /Library/Developer/CommandLineTools/usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found Git: /opt/homebrew/bin/git (found version "2.35.1") 
-- Found FLEX: /usr/bin/flex (found version "2.6.4") 
-- Found BISON: /usr/bin/bison (found version "2.3")
-- Looking for libelf.h
-- Looking for libelf.h - found
-- Looking for libelf/libelf.h
-- Looking for libelf/libelf.h - found
-- Looking for usb.h
-- Looking for usb.h - found
-- Looking for lusb0_usb.h
-- Looking for lusb0_usb.h - not found
-- Looking for libusb.h
-- Looking for libusb.h - not found
-- Looking for libusb-1.0/libusb.h
-- Looking for libusb-1.0/libusb.h - found
-- Looking for hidapi/hidapi.h
-- Looking for hidapi/hidapi.h - found
-- Looking for ftdi_tcioflush
-- Looking for ftdi_tcioflush - found
-- Configuration summary:
-- ----------------------
-- DO HAVE    libelf
-- DO HAVE    libusb
-- DO HAVE    libusb_1_0
-- DO HAVE    libhidapi
-- DO HAVE    libftdi (but prefer to use libftdi1)
-- DO HAVE    libftdi1
-- DISABLED   doc
-- DISABLED   parport
-- DISABLED   linuxgpio
-- DISABLED   linuxspi
-- ----------------------
-- Configuring done
-- Generating done
-- Build files have been written to: /Users/charles/devt/avrdude/build_macos
````

````shell

charles@mini-m1:avrdude$ cmake --build build_macos
[  1%] [FLEX][Parser] Building scanner with flex 2.6.4
[  3%] [BISON][Parser] Building parser with bison 2.3
[... 74 lines cut ...]
  -B <bitclock>              Specify JTAG/STK500v2 bit clock period (us).
  -C <config-file>           Specify location of configuration file.
  -c <programmer>            Specify programmer type.
  -D                         Disable auto erase for flash memory
  -i <delay>                 ISP Clock Delay [in microseconds]
  -P <port>                  Specify connection port.
  -F                         Override invalid signature check.
  -e                         Perform a chip erase.
  -O                         Perform RC oscillator calibration (see AVR053). 
  -U <memtype>:r|w|v:<filename>[:format]
                             Memory operation specification.
                             Multiple -U options are allowed, each request
                             is performed in the order specified.
  -n                         Do not write anything to the device.
  -V                         Do not verify.
  -t                         Enter terminal mode.
  -E <exitspec>[,<exitspec>] List programmer exit specifications.
  -x <extended_param>        Pass <extended_param> to programmer.
  -v                         Verbose output. -v -v for more.
  -q                         Quell progress output. -q -q for less.
  -l logfile                 Use logfile rather than stderr for diagnostics.
  -?                         Display this usage.

avrdude version 7.0-20220508 (4601bee), URL: <https://github.com/avrdudes/avrdude>
````

hallard, 2022-05-08:
interesting that I also modified script (like I already done before) also works now :-)

```shell

    darwin)
        # determine whether we are running using Mac Ports
        # if not, assume Mac Brew
        if [ -f /opt/local/bin/port ]
        then
            build_flags="${build_flags} -D CMAKE_C_FLAGS=-I/opt/local/include -D CMAKE_EXE_LINKER_FLAGS=-L/opt/local/lib"
        else
            # Apple M1 (may be new version of homebrew also)
            if [ -d /opt/homebrew ]  
            then
                build_flags="${build_flags} -D CMAKE_C_FLAGS=-I/usr/homebrew/include -D CMAKE_EXE_LINKER_FLAGS=-L/usr/homebrew/Cellar"
            else
                build_flags="${build_flags} -D CMAKE_C_FLAGS=-I/usr/local/include -D CMAKE_EXE_LINKER_FLAGS=-L/usr/local/Cellar"
            fi
        fi
        ;;
```
Thanks for your support.

mcuee, 2022-05-08:
There is a minor issue here.
```
-- DO HAVE    libftdi (but prefer to use libftdi1)
-- DO HAVE    libftdi1
```
There is no libftdi here at all, rather the detection is a bit off. It should probably detect libftdi1 first. If libftdi1 exists then should skip libftdi-0.1 detection.

mcuee, 2022-05-09:
> interesting that I also modified script (like I already done before) also works now :-)
>
> ```shell
>     darwin)
>         # determine whether we are running using Mac Ports
>         # if not, assume Mac [...]

Nice patch. This change of build.sh should probably be merged.

hallard, 2022-05-09:
> Nice patch. This change of build.sh should probably be merged.

thanks done with #950

mcuee, 2022-05-09:
BTW, not so sure if someone can create a pull request to Homebrew to add --HEAD option. In that case, Homebrew users can easily install the git head version.
https://github.com/Homebrew/homebrew-core/blob/master/Formula/avrdude.rb

mcuee, 2022-06-01:
#950 has been merged. Hopefully github action for Apple ARM64 (M1 or later) can be added.

mcuee, 2022-06-18:
> BTW, not so sure if someone can create a pull request to Homebrew to add --HEAD option. In that case, Homebrew users can easily install the git head version. 
> https://github.com/Homebrew/homebrew-core/blob/master/Formula/avrdude.rb

Homebrew has updated the formula for avrdude and now they use libusb-1.0 and libftdi1 API. Somehow they dropped libusb-compat-0.1 which I think is incorrect.
https://github.com/Homebrew/homebrew-core/commit/1adc2774df32e62bc32e153bf32a8a05093284b4

mcuee, 2022-07-01:
Just FYI, homebrew formula has been fixed.
https://github.com/Homebrew/homebrew-core/pull/103964

mcuee, 2022-07-29:
The infrasture mentioned earlier is still not ready.
https://github.com/actions/virtual-environments/issues/2187

mcuee, 2022-07-31:
For those who are using Homebrew and wants to use the latest, you can use the following command one time.

```
brew install avrdude (install 7.0 release in case you need it)
brew unlink avrdude 
brew install avrdude --HEAD
```

After that, you can use the following command to be always on the latest.
```
brew reinstall avrdude
```

Example:
```
mcuee@mcuees-Mac-mini avrdude_sr % brew reinstall avrdude
==> Cloning https://github.com/avrdudes/avrdude.git
Updating /Users/mcuee/Library/Caches/Homebrew/avrdude--git
From https://github.com/avrdudes/avrdude
   c81f52f..affe4cb  main       -> origin/main
==> Checking out branch main
Already on 'main'
Your branch is behind 'origin/main' by 73 commits, and can be fast-forwarded.
  (use "git pull" to update your local branch)
HEAD is now at affe4cb Merge pull request #1034 from stefanrueger/config_file
==> Reinstalling avrdude 
==> cmake -S . -B build/shared -DCMAKE_INSTALL_SYSCONFDIR=/opt/homebrew/etc -DBUILD_SHARED_LIBS=ON -DCMAKE_INSTALL_RPATH=@loader_path/../lib -DCMAKE_SHA
==> cmake --build build/shared
==> cmake --install build/shared
==> cmake -S . -B build/static -DCMAKE_INSTALL_SYSCONFDIR=/opt/homebrew/etc
==> cmake --build build/static
🍺  /opt/homebrew/Cellar/avrdude/HEAD-affe4cb_2: 13 files, 1.5MB, built in 7 seconds
==> Running `brew cleanup avrdude`...
Disable this behaviour by setting HOMEBREW_NO_INSTALL_CLEANUP.
Hide these hints with HOMEBREW_NO_ENV_HINTS (see `man brew`).
```

mcuee, 2023-06-21:
Reference:

* https://github.com/github/roadmap/issues/528

mcuee, 2023-10-26:
github has sorted out the issue.
https://github.blog/2023-10-02-introducing-the-new-apple-silicon-powered-m1-macos-larger-runner-for-github-actions/

mcuee, 2023-10-26:
I tried to use my avrdude repo to test out the public beta but it does not seem to work.

mcuee, 2024-02-07:
Close this one and convert it into a discussion.