avrdudes/avrdude #941: `iseavrprog` has too long description string

stefanrueger, 2022-04-28:
`src/avrdude.conf.in` contains the following description longer than the buffer can stomach
```
  desc = "USBtiny-based USB programmer, https://github.com/IowaScaledEngineering/ckt-avrprogrammer";
```

Hence, `avrdude -c \? ` cuts short the URL possibly causing mild disappointment in users who try to follow the link.

I can see a range of obvious solutions
1. Remove substring ` USB programmer` from the description
2. Increase `PGM_DESCLEN` to accommodate this string
3. Warn in `config_gram.y` whenever a string is parsed that's too long for the fixed-length buffer
4. Change the desc components of programmer, part and memory structures	from fixed-length arrays to `malloc()`-ated pointers. Would reduce memory footprint, too, but requires a small amount of rewrite in `config_gram.y` and	`pickit2.c` alongside careful checking of some 572 lines where `desc` components are used.
5. Do 3. and 1.
6. Do 3. and 2.

Thoughts?

Either way, happy to do the PR once the couple of currently waiting PRs that affect `avrdude.conf.in` and `config_gram.y` are out of the way.

dl8dtl, 2022-04-28:
I'd vote for 5
Limiting strings to about 80 characters still seems feasible to me – so we could bump `PGM_DESCLEN` to that value, but that wouldn't suffice for this string (88 chars).

dl8dtl, 2022-04-28:
option 7: use a tinyURL shortcut ;-)
just kidding

stefanrueger, 2022-04-28:
> Limiting strings to about 80 characters still seems feasible to me

`PGM_DESCLEN` is already at 80 characters, so just clipping 9 characters off the end of the URL.  Not worth increasing more.

PR #1000 (2022-06-19): Reduce programmer desc string length in avrdude.conf to < 80 characters
I'm currently doing a bit of house cleaning to get rid of some "easy-fix" issues.

Closes #941
closed 2022-06-19 by dl8dtl