avrdudes/avrdude #942: Treat `x` bits in SPI commands as if they were '0'

stefanrueger, 2022-04-28:
While the `x` don't care bit description in `.conf` SPI programming instructions by its very definition means that this bit *does not need* to be initialised, I suggest clearing `x` bits in `avr_set_bits()` nevertheless.

Rationale: Traditionally, there have been quite a few errors in `.conf` by users making mistakes in the SPI command descriptions. This intervention makes `x` bits behave like `0` and removes a potential pitfall. In case the part cares about the value of `x` bits (and, hence, the command description in `.conf` was wrong) the suggested change makes avrdude behave predictably and therefore eases debugging. Some of the 48 calls of `avr_set_bits()` set bits on uninitialised memory leaving `x` bits in a truly unknown state (mostly in `stk500.c` and `stk500v2.c`), thereby giving utility to clear the `x` bits of SPI commands in `avr_set_bits()`.

mcuee, 2022-06-07:
Will setting the other bits as zero fix this issue?
https://github.com/avrdudes/avrdude/blob/main/src/avrpart.c#L79

stefanrueger, 2022-06-28:
closed with PR #943

closed 2022-06-28 by stefanrueger