avrdudes/avrdude #944: Micronucleus read error

MCUdude, 2022-04-30:
When trying to upload a program to an ATtiny85 running the Micronucleus bootloader, I'm getting the following error:

```
$ ./avrdude -C avrdude.conf -c micronucleus -p attiny85 -Uflash:w:/var/folders/6l/ypg6qbw172v1s4vtt6g990tw0000gn/T/arduino_build_489940/Blink.ino.hex:i

avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.00s

avrdude: Device signature = 0x1e930b (probably t85)
avrdude: NOTE: "flash" memory has been specified, an erase cycle will be performed
         To disable this feature, specify the -D option.
avrdude: erasing chip
avrdude: reading input file "/var/folders/6l/ypg6qbw172v1s4vtt6g990tw0000gn/T/arduino_build_489940/Blink.ino.hex"
avrdude: writing flash (542 bytes):

Writing | ################################################## | 100% 0.14s

avrdude: 542 bytes of flash written
avrdude: verifying flash memory against /var/folders/6l/ypg6qbw172v1s4vtt6g990tw0000gn/T/arduino_build_489940/Blink.ino.hex:

Reading | ######                                             | 11% 0.00savrdude: Unsupported memory type: flash
avr_read(): error reading address 0x0000
    read operation not supported for memory "flash"
avrdude: failed to read all of flash memory, rc=-2

avrdude done.  Thank you.
```

Since the Micronucleus bootloader doesn't support flash reading at all, why is Avrdude trying to read the flash by default? Wouldn't it be better if Avrdude simply skipped reading the flash by default when using the micronucleus programmer option instead of "manually" adding the -V flag every time?

MCUdude, 2022-04-30:
Another thing, probably a question for you @mariusgreuel.

When the user program is running on the microcontroller, and I'm trying to upload using Avrdude, I get the following error. I have to unplug the board, execute Avrdude and then insert the board again for it to work.

```
$ ./avrdude -C avrdude.conf -c micronucleus -p attiny85 -x wait

avrdude: WARNING: Failed to get bootloader info block: Input/output error
avrdude: initialization failed, rc=-5
         Double check connections and try again, or use -F to override
         this check.

```

However, with the "original" micronucleus upload tool I can execute the uploader even though the board is running the user application, and it will just wait until I re-connect the board.

``` 
# Board is connected and running the user application
$ ./micronucleus /var/folders/6l/ypg6qbw172v1s4vtt6g990tw0000gn/T/arduino_build_489940/Blink.ino.hex
> Please plug in the device ... 
# I unplug the device
#
# I re-conect the board again
> Device is found!
connecting: 40% complete
> Device has firmware version 2.5
> Device signature: 0x1e930b 
> Available space for user applications: 6586 bytes
> Suggested sleep time between sending pages: 7ms
> Whole page count: 103  page size: 64
> Erase function sleep duration: 721ms
parsing: 60% complete
> Erasing the memory ...
erasing: 80% complete
> Starting to upload ...
writing: 100% complete
>> Micronucleus done. Thank you!
```

It would be awesome if Avrdude could mimic the Micronuceus upload tool behaviour!

dl8dtl, 2022-05-01:
> why is Avrdude trying to read the flash by default?

Because it has never been designed with the intention of a (kinda) crippled bootloader in mind. Verification of the upload seemed to be crucial and is thus the default.

Actually, I think the need to add `-V` manually even has some merit: it reminds you that Avrdude cannot do anything on behalf of your data integrity in that environment.

MCUdude, 2022-05-02:
> Actually, I think the need to add -V manually even has some merit: it reminds you that Avrdude cannot do anything on behalf of your data integrity in that environment.

Well, that kinda makes sense. 

The other issue where Avrdude return `initialization failed, rc=-5` is more annoying I think and should be looked into if the goal is that Avrdude should be _just as good_ as the original micronucleus upload tool.

stefanrueger, 2022-05-02:
Bootloaders are a bit like octogenarians. They do whatever they like, and there is little one can do about it (this is the *one* thing I actually look forward to at old age).

In fairness, AVRDUDE's documentation invites the user to type `-V` for micronucleus. Having said this, I prefer if AVRDUDE *just works*:tm:. There is value in the programmer code being able to tell AVRDUDE's `update` that it shouldn't verify when the programmer knows the bootloader cannot read. Why not head over to Discussion #940 and make suggestions for a better modelling of SPM programmers?

mariusgreuel, 2022-05-04:
> Since the Micronucleus bootloader doesn't support flash reading at all, why is Avrdude trying to read the flash by default?

The primary reason was that I could not figure out how to turn that off without touching the rest of the code base. Currently, avrdude expects certain read functions to be implemented, which I would have otherwise just marked as NULL. So, I went with `-V`, and then kinda liked it for the same reasons Jörg mentioned. But I agree that the users experience could be better. For instance, we could just print a warning that the programmer does not support reading.

> When the user program is running on the microcontroller, and I'm trying to upload using Avrdude, I get the following error.

I suspect that the `usb_open()` call succeeds despite the fact that the USB bootloader is disconnected. The reason being, a Micronucleus bootloader has a non-conformant USB hardware and software implementation (i.e. V-USB), and it probably does not disconnect properly. The macOS implementation may be a little different from Windows/Linux, and it seems it still enumerates a disconnected device.

I don't know how macOS works, but perhaps there is some tool that shows the state of an USB device? I am curious what macOS reports when the Micronucleus bootloader exited and the user-program runs. In this case the USB device should report some disconnected/unresponsive state.

If we know what to look for we could perhaps examine the state and skip the device if necessary. Another idea is to send a dummy request in `micronucleus_open()` and skip the device if the request fails. For instance, we might as well call `micronucleus_get_bootloader_info()` in `micronucleus_open()`.

MCUdude, 2022-05-04:
> But I agree that the users experience could be better. For instance, we could just print a warning that the programmer does not support reading.

Personally, I would prefer a nicely formatted warning instead when not using the -V flash. I don't think the Micronucleus bootloader will support read functionality any time soon anyway. 

> I don't know how macOS works, but perhaps there is some tool that shows the state of a USB device?

I couldn't find a built-in tool that outputs a different status from when in bootloader mode and when it runs the user program. However, when it is in this "unresponsive" state, `usb_control_msg` returns -5 in the function `micronucleus_get_bootloader_info_v2()`. I couldn't find the source code for `usb_control_msg`, but -5 has to mean something useful that we can utilize.

https://github.com/avrdudes/avrdude/blob/a4c9fcdcd7746f2aafae1ae5155d561a2af07cca/src/micronucleus.c#L232-L247

A potential solution would be to require a disconnect/reconnect if -5 is returned, and not print any scary errors or warnings.
I guess this applies for `micronucleus_get_bootloader_info_v1()` as well?

MCUdude, 2022-05-04:
According to the libusb documentation I just found, -5 means `LIBUSB_ERROR_NOT_FOUND`.

```c
enum | libusb_error {   LIBUSB_SUCCESS = 0, LIBUSB_ERROR_IO = -1, LIBUSB_ERROR_INVALID_PARAM = -2, LIBUSB_ERROR_ACCESS = -3,   LIBUSB_ERROR_NO_DEVICE = -4, LIBUSB_ERROR_NOT_FOUND = -5, LIBUSB_ERROR_BUSY = -6, LIBUSB_ERROR_TIMEOUT = -7,   LIBUSB_ERROR_OVERFLOW = -8, LIBUSB_ERROR_PIPE = -9, LIBUSB_ERROR_INTERRUPTED = -10, LIBUSB_ERROR_NO_MEM = -11,   LIBUSB_ERROR_NOT_SUPPORTED = -12, LIBUSB_ERROR_OTHER = -99 }
```
https://libusb.sourceforge.io/api-1.0/group__libusb__misc.html#gab2323aa0f04bc22038e7e1740b2f29ef

MCUdude, 2022-05-05:
@mariusgreuel do you think it's possible to come up with a solution for the "-5 error" before 7.0 is released? I've tried playing around with some of the USB-related code, but I'm just getting segmentation faults.

mariusgreuel, 2022-05-05:
@MCUdude I don't have the possibility to do any testing right now, so here is a compile-check only patch. Could you please try this? If it works, I can submit the PR.

<https://github.com/mariusgreuel/avrdude-fork/commit/3bdf138721bd4a08ed743312e11a95aa8a2ee892>

MCUdude, 2022-05-05:
Thanks, @mariusgreuel for taking a look at the problem. Unfortually, 

```
$ ./avrdude -v -C avrdude.conf -c micronucleus -p attiny85 -x wait

avrdude: Version 6.99-20211218
         Copyright (c) Brian Dean, http://www.bdmicro.com/
         Copyright (c) Joerg Wunsch

         System wide configuration file is "avrdude.conf"
         User configuration file is "/Users/hans/.avrduderc"
         User configuration file does not exist or is not a regular file, skipping

Hello
         Using Port                    : usb
         Using Programmer              : micronucleus
avrdude: Found device with Micronucleus V2.5, bus:device: 020:016
avrdude: WARNING: Failed to get bootloader info block: Input/output error
avrdude: WARNING: Failed to probe device (error -5), skipping...
avrdude: No device found, waiting for device to be plugged in...
avrdude: Press CTRL-C to terminate.
avrdude: Found device with Micronucleus V2.5, bus:device: 020:016
avrdude: WARNING: Failed to get bootloader info block: Input/output error
avrdude: WARNING: Failed to probe device (error -5), skipping...
avrdude: Found device with Micronucleus V2.5, bus:device: 020:016
avrdude: WARNING: Failed to get bootloader info block: Input/output error
avrdude: WARNING: Failed to probe device (error -5), skipping...
avrdude: Found device with Micronucleus V2.5, bus:device: 020:016
avrdude: WARNING: Failed to get bootloader info block: Input/output error
avrdude: WARNING: Failed to probe device (error -5), skipping...
avrdude: Found device with Micronucleus V2.5, bus:device: 020:016
avrdude: WARNING: Failed to get bootloader info block: Input/output error
avrdude: WARNING: Failed to probe device (error -5), skipping...
avrdude: Found device with Micronucleus V2.5, bus:device: 020:016
avrdude: WARNING: Failed to get bootloader info block: Input/output error
avrdude: WARNING: Failed to probe device (error -5), skipping...
avrdude: Found device with Micronucleus V2.5, bus:device: 020:016
avrdude: WARNING: Failed to get bootloader info block: Input/output error
avrdude: WARNING: Failed to probe device (error -5), skipping...
^C
```

It does, however, work if I disconnect and conect it again:

```
$ ./avrdude -v -C avrdude.conf -c micronucleus -p attiny85 -x wait

avrdude: Version 6.99-20211218
         Copyright (c) Brian Dean, http://www.bdmicro.com/
         Copyright (c) Joerg Wunsch

         System wide configuration file is "avrdude.conf"
         User configuration file is "/Users/hans/.avrduderc"
         User configuration file does not exist or is not a regular file, skipping

Hello
         Using Port                    : usb
         Using Programmer              : micronucleus
avrdude: Found device with Micronucleus V2.5, bus:device: 020:017
avrdude: WARNING: Failed to get bootloader info block: Input/output error
avrdude: WARNING: Failed to probe device (error -5), skipping...
avrdude: No device found, waiting for device to be plugged in...
avrdude: Press CTRL-C to terminate.
avrdude: Found device with Micronucleus V2.5, bus:device: 020:017
avrdude: WARNING: Failed to get bootloader info block: Input/output error
avrdude: WARNING: Failed to probe device (error -5), skipping...
avrdude: Found device with Micronucleus V2.5, bus:device: 020:017
avrdude: WARNING: Failed to get bootloader info block: Input/output error
avrdude: WARNING: Failed to probe device (error -5), skipping...
avrdude: Found device with Micronucleus V2.5, bus:device: 020:017
[... 32 lines cut ...]
           lock                    0     0     0    0 no          1    1      0  9000  9000 0x00 0x00
           lfuse                   0     0     0    0 no          1    1      0  9000  9000 0x00 0x00
           hfuse                   0     0     0    0 no          1    1      0  9000  9000 0x00 0x00
           efuse                   0     0     0    0 no          1    1      0  9000  9000 0x00 0x00
           calibration             0     0     0    0 no          1    1      0     0     0 0x00 0x00

         Programmer Type : Micronucleus V2.0
         Description     : Micronucleus Bootloader

avrdude: Bootloader version: 2.5
avrdude: Available flash size: 6586
avrdude: Page size: 64
avrdude: Bootloader start: 0x19C0
avrdude: Write sleep: 7ms
avrdude: Erase sleep: 721ms
avrdude: Signature1: 0x93
avrdude: Signature2: 0x0B
avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.00s

avrdude: Device signature = 0x1e930b (probably t85)

avrdude done.  Thank you.
```

mariusgreuel, 2022-05-05:
I take it that it works, beside the shitload of messages, correct?

MCUdude, 2022-05-05:
> I take it that it works, beside the shitload of messages, correct?

Correct. It also works like it should with `-x wait=[timeout]`. So a nice message that tells the user what to do, printed once, would make sense? Perhaps something like: 
```
avrdude: Warning: Unresponsive Micronucleus device detected, please reconnect...
```

Or similar.

mcuee, 2022-05-25:
> The reason being, a Micronucleus bootloader has a non-conformant USB hardware and software implementation (i.e. V-USB).

That is a very good point. That being said, V-USB seems to be popular among non-USB avr devices, including the tools like usbasp and usbtinyisp.
* https://www.fischl.de/usbasp/
* https://learn.adafruit.com/usbtinyisp

mariusgreuel, 2022-05-25:
Although the V-USB devices do not meet the specs, they often just work. I have not seen this issue with the usbasp and usbtinyisp devices I own, but only with certain Digispark clones. They somehow mucked up the hardware, so that certain hubs do not detect a device disconnect properly. I was actually able to reproduce that issue also on Windows, when the right hub and port was used.

mcuee, 2022-05-26:
> but only with certain Digispark clones. They somehow mucked up the hardware, so that certain hubs do not detect a device disconnect properly

I am facing this issue now with two MH-Tiny ATtiny88 boards. The on-board Micronucleus bootloader works very poorly. They worked once for my laptop and now do not seem to work anymore.

Ref: I will first try different version of the bootloader.
https://github.com/SpenceKonde/ATTinyCore/tree/master/avr/bootloaders/micronucleus

Or I will just give up and the use optiboot and USB-TTL converter.

mcuee, 2022-05-26:
Flashing new FW here seems to sort out the issues.
https://github.com/micronucleus/micronucleus/blob/master/firmware/releases/t88_default.hex

mcuee, 2022-05-28:
Interesting Arduino [ATTinyCore](https://github.com/SpenceKonde/ATTinyCore) has the same thing -- I need to plug out and then plug in to get the programming working with the micronucleus bootloader for the ATtiny88.
```
Sketch uses 468 bytes (6%) of program storage space. Maximum is 6780 bytes.
Global variables use 9 bytes (1%) of dynamic memory, leaving 503 bytes for local variables. Maximum is 512 bytes.
> Please plug in the device (will time out in 60 seconds) ... 
> Device is found!
connecting: 16% complete
connecting: 22% complete
connecting: 28% complete
connecting: 33% complete
> Device has firmware version 2.6
> Device signature: 0x1e9311 
> Available space for user applications: 6716 bytes
> Suggested sleep time between sending pages: 7ms
> Whole page count: 105  page size: 64
> Erase function sleep duration: 735ms
parsing: 50% complete
> Erasing the memory ...
erasing: 55% complete
erasing: 60% complete
erasing: 65% complete
> Starting to upload ...
writing: 70% complete
writing: 75% complete
writing: 80% complete
> Starting the user app ...
running: 100% complete
>> Micronucleus done. Thank you!

```

PR #945 (2022-06-08): Fix micronucleus bootloader to check for unresponsive USB devices
Fixes #944, pending approval of @MCUdude.