avrdudes/avrdude #952: Unable to build both static and shared libraries with the new CMake buildsystem

subnut, 2022-05-09:
Previously, `./configure` would build both static and shared libraries in one go. 
Now, using the new CMake build, I am unable to do so without having to build twice.

mcuee, 2022-05-09:
This was used as a reason against CMake in the libusb build system discussion.
https://github.com/libusb/libusb/pull/1134

But I believe something can be done as in the libftdi example.
http://developer.intra2net.com/git/?p=libftdi

mcuee, 2022-05-09:
@subnut Are you able to build the share library? Somehow I have no issues to build the static library under MSYS2 (mingw32 and mingw64) but not able to do the same for the shared library.

subnut, 2022-05-09:
@mcuee Yes, I am. But I am on Linux, and I do not have a Windows system nearby to test.

mcuee, 2022-05-09:
> @mcuee Yes, I am. But I am on Linux, and I do not have a Windows system nearby to test.

Thanks. I will try your patch as well to see if that helps. Or maybe I will try to fix my MSY2 MinGW32/64 build.

subnut, 2022-05-09:
@mcuee I don't think my patch is going to fix that.  

https://stackoverflow.com/a/57932764 -

> The reason why this worked in Linux and not in Windows is that **DLL needs to have all external dependencies resolved during linking**, while in Linux SO can have undefined symbols, which are resolved only when the library is used.

mariusgreuel, 2022-05-09:
@dl8dtl Building the avrdude libraries is getting quite a bit attention. I spend very little time looking at this because I was under the impression that the avrdude library is not really used. Also, the avrdude code base does not really seem to be prepared for building libraries (i.e. external references such as `progname`, exposure of internal data structures, lack of versioning, etc.).

What's your view on this? Are libraries part of the avrdude delivery, if so, both static and dynamic libraries?

dl8dtl, 2022-05-09:
Basically, the separation of the library backend from the CLI frontend has been a long-standing wish, but it's still kinda work in progress, see for example issue #774 
The idea is that a completely separated backend would make it much easier to provide a GUI frontend as an alternative to the traditional CLI frontend. All existing GUI attempts wrap around the CLI which is merely a crock only.
For the CLI, it's probably fine to just build upon the static library, but for the idea of a possible GUI frontend, at least under non-Windows systems, a shared library has some merit.

mariusgreuel, 2022-05-09:
I guess my question is whether we accept PRs for this, or just call the libraries 'deprecated' until there is a real need for it. Right now, I see very little value in PRs trying to build avrdude dummy DLLs, just for the sake of building it. It would be a whole different matter if there was a real project behind this.

mcuee, 2022-06-11:
Pull requrst #960 was closed -- but it actually helps to fix the issue under macOS even though it can not help on Windows.

mcuee, 2023-01-14:
Right now macOS homebrew is using a trick to build shared library.

```
cmake -S . -B build/shared -DCMAKE_INSTALL_SYSCONFDIR=/opt/homebrew/etc -DBUILD_SHARED_LIBS=ON -DCMAKE_INSTALL_RPATH=@loader_path/../lib -DCMAKE_SHARED_LINKER_FLAGS=-Wl,-undefined,dynamic_lookup   -D CMAKE_C_FLAGS=-I/opt/homebrew/include -D CMAKE_EXE_LINKER_FLAGS=-L/opt/homebrew/Cellar
cmake --build build/shared
#cmake --install build/shared
cmake -S . -B build/static -DCMAKE_INSTALL_SYSCONFDIR=/opt/homebrew/etc -D CMAKE_C_FLAGS=-I/opt/homebrew/include -D CMAKE_EXE_LINKER_FLAGS=-L/opt/homebrew/Cellar
cmake --build build/static
```

mcuee, 2023-01-14:
As of now, there is no way to build shared libavrdude.dll under Windows, either with MinGW or MSVC.

mcuee, 2023-01-14:
Under Linux, there is no issues build the static link and dynamic link libraries.

The default `build.sh` will build static link libarvdude.a.
```
mcuee@UbuntuSwift3:~/build/avr/avrdude_main$ sudo cmake --build build_linux --target install
[sudo] password for mcuee: 
[ 89%] Built target libavrdude
[ 98%] Built target avrdude
[100%] Built target conf
Install the project...
-- Install configuration: "RelWithDebInfo"
-- Installing: /usr/local/bin/avrdude
-- Set runtime path of "/usr/local/bin/avrdude" to ""
-- Installing: /usr/local/lib/libavrdude.a
-- Up-to-date: /usr/local/include/libavrdude.h
-- Installing: /usr/local/etc/avrdude.conf
-- Up-to-date: /usr/local/share/man/man1/avrdude.1
```

But it is also easy to build shared library `libavrdude.so`.
```
mcuee@UbuntuSwift3:~/build/avr/avrdude_main$ cat build_shared.sh 
#!/bin/sh
#
# Build script for Unix-like systems, using the CMake subsystem

cmake -D CMAKE_BUILD_TYPE=RelWithDebInfo -D DEBUG_CMAKE=1 -D USE_STATIC_LIBS=OFF -DBUILD_SHARED_LIBS=ON -D CMAKE_VERBOSE_MAKEFILE=ON  -B build_shared -D CMAKE_EXE_LINKER_FLAGS="-pthread -ludev -lz" 

cmake --build build_shared

mcuee@UbuntuSwift3:~/build/avr/avrdude_main$ sudo cmake --build build_shared --target install
/usr/bin/cmake -S/home/mcuee/build/avr/avrdude_main -B/home/mcuee/build/avr/avrdude_main/build_shared --check-build-system CMakeFiles/Makefile.cmake 0
/usr/bin/cmake -E cmake_progress_start /home/mcuee/build/avr/avrdude_main/build_shared/CMakeFiles /home/mcuee/build/avr/avrdude_main/build_shared/CMakeFiles/progress.marks
/usr/bin/make -f CMakeFiles/Makefile2 all
make[1]: Entering directory '/home/mcuee/build/avr/avrdude_main/build_shared'
/usr/bin/make -f src/CMakeFiles/libavrdude.dir/build.make src/CMakeFiles/libavrdude.dir/depend
make[2]: Entering directory '/home/mcuee/build/avr/avrdude_main/build_shared'
cd /home/mcuee/build/avr/avrdude_main/build_shared && /usr/bin/cmake -E cmake_depends "Unix Makefiles" /home/mcuee/build/avr/avrdude_main /home/mcuee/build/avr/avrdude_main/src /home/mcuee/build/avr/avrdude_main/build_shared /home/mcuee/build/avr/avrdude_main/build_shared/src /home/mcuee/build/avr/avrdude_main/build_shared/src/CMakeFiles/libavrdude.dir/DependInfo.cmake --color=
make[2]: Leaving directory '/home/mcuee/build/avr/avrdude_main/build_shared'
/usr/bin/make -f src/CMakeFiles/libavrdude.dir/build.make src/CMakeFiles/libavrdude.dir/build
make[2]: Entering directory '/home/mcuee/build/avr/avrdude_main/build_shared'
make[2]: Nothing to be done for 'src/CMakeFiles/libavrdude.dir/build'.
make[2]: Leaving directory '/home/mcuee/build/avr/avrdude_main/build_shared'
[ 89%] Built target libavrdude
/usr/bin/make -f src/CMakeFiles/avrdude.dir/build.make src/CMakeFiles/avrdude.dir/depend
make[2]: Entering directory '/home/mcuee/build/avr/avrdude_main/build_shared'
cd /home/mcuee/build/avr/avrdude_main/build_shared && /usr/bin/cmake -E cmake_depends "Unix Makefiles" /home/mcuee/build/avr/avrdude_main /home/mcuee/build/avr/avrdude_main/src /home/mcuee/build/avr/avrdude_main/build_shared /home/mcuee/build/avr/avrdude_main/build_shared/src /home/mcuee/build/avr/avrdude_main/build_shared/src/CMakeFiles/avrdude.dir/DependInfo.cmake --color=
make[2]: Leaving directory '/home/mcuee/build/avr/avrdude_main/build_shared'
/usr/bin/make -f src/CMakeFiles/avrdude.dir/build.make src/CMakeFiles/avrdude.dir/build
make[2]: Entering directory '/home/mcuee/build/avr/avrdude_main/build_shared'
make[2]: Nothing to be done for 'src/CMakeFiles/avrdude.dir/build'.
make[2]: Leaving directory '/home/mcuee/build/avr/avrdude_main/build_shared'
[ 98%] Built target avrdude
/usr/bin/make -f src/CMakeFiles/conf.dir/build.make src/CMakeFiles/conf.dir/depend
make[2]: Entering directory '/home/mcuee/build/avr/avrdude_main/build_shared'
cd /home/mcuee/build/avr/avrdude_main/build_shared && /usr/bin/cmake -E cmake_depends "Unix Makefiles" /home/mcuee/build/avr/avrdude_main /home/mcuee/build/avr/avrdude_main/src /home/mcuee/build/avr/avrdude_main/build_shared /home/mcuee/build/avr/avrdude_main/build_shared/src /home/mcuee/build/avr/avrdude_main/build_shared/src/CMakeFiles/conf.dir/DependInfo.cmake --color=
make[2]: Leaving directory '/home/mcuee/build/avr/avrdude_main/build_shared'
/usr/bin/make -f src/CMakeFiles/conf.dir/build.make src/CMakeFiles/conf.dir/build
make[2]: Entering directory '/home/mcuee/build/avr/avrdude_main/build_shared'
make[2]: Nothing to be done for 'src/CMakeFiles/conf.dir/build'.
make[2]: Leaving directory '/home/mcuee/build/avr/avrdude_main/build_shared'
[100%] Built target conf
make[1]: Leaving directory '/home/mcuee/build/avr/avrdude_main/build_shared'
/usr/bin/cmake -E cmake_progress_start /home/mcuee/build/avr/avrdude_main/build_shared/CMakeFiles 0
/usr/bin/make -f CMakeFiles/Makefile2 preinstall
make[1]: Entering directory '/home/mcuee/build/avr/avrdude_main/build_shared'
make[1]: Nothing to be done for 'preinstall'.
make[1]: Leaving directory '/home/mcuee/build/avr/avrdude_main/build_shared'
Install the project...
/usr/bin/cmake -P cmake_install.cmake
-- Install configuration: "RelWithDebInfo"
-- Up-to-date: /usr/local/bin/avrdude
-- Up-to-date: /usr/local/lib/libavrdude.so.1.0.0
-- Up-to-date: /usr/local/lib/libavrdude.so.1
-- Up-to-date: /usr/local/lib/libavrdude.so
-- Up-to-date: /usr/local/include/libavrdude.h
-- Up-to-date: /usr/local/etc/avrdude.conf
-- Up-to-date: /usr/local/share/man/man1/avrdude.1

```

mcuee, 2023-06-20:
Just want to mention this issue is still there.

```
[68/70] cmd.exe /C "cd . && C:\msys64\mingw64\bin\cc.exe -O2 -g -DNDEBUG  -Wl,-undefined,dynamic_lookup -shared -o src\libavrdude.dll -Wl,--out-implib,src\liblibavrdude.dll.a -Wl,--major-image-version,1,--minor-image-version,0 src/CMakeFiles/libavrdude.dir/arduino.c.obj src/CMakeFiles/libavrdude.dir/avr.c.obj src/CMakeFiles/libavrdude.dir/avr910.c.obj src/CMakeFiles/libavrdude.dir/avrcache.c.obj src/CMakeFiles/libavrdude.dir/avrftdi.c.obj src/CMakeFiles/libavrdude.dir/avrftdi_tpi.c.obj src/CMakeFiles/libavrdude.dir/avrintel.c.obj src/CMakeFiles/libavrdude.dir/avrpart.c.obj src/CMakeFiles/libavrdude.dir/bitbang.c.obj src/CMakeFiles/libavrdude.dir/buspirate.c.obj src/CMakeFiles/libavrdude.dir/butterfly.c.obj src/CMakeFiles/libavrdude.dir/ch341a.c.obj src/CMakeFiles/libavrdude.dir/config.c.obj src/CMakeFiles/libavrdude.dir/confwin.c.obj src/CMakeFiles/libavrdude.dir/crc16.c.obj src/CMakeFiles/libavrdude.dir/dfu.c.obj src/CMakeFiles/libavrdude.dir/dryrun.c.obj src/CMakeFiles/libavrdude.dir/fileio.c.obj src/CMakeFiles/libavrdude.dir/flip1.c.obj src/CMakeFiles/libavrdude.dir/flip2.c.obj src/CMakeFiles/libavrdude.dir/ft245r.c.obj src/CMakeFiles/libavrdude.dir/jtagmkI.c.obj src/CMakeFiles/libavrdude.dir/jtagmkII.c.obj src/CMakeFiles/libavrdude.dir/jtag3.c.obj src/CMakeFiles/libavrdude.dir/linuxgpio.c.obj src/CMakeFiles/libavrdude.dir/linuxspi.c.obj src/CMakeFiles/libavrdude.dir/lists.c.obj src/CMakeFiles/libavrdude.dir/micronucleus.c.obj src/CMakeFiles/libavrdude.dir/par.c.obj src/CMakeFiles/libavrdude.dir/pgm.c.obj src/CMakeFiles/libavrdude.dir/pgm_type.c.obj src/CMakeFiles/libavrdude.dir/pickit2.c.obj src/CMakeFiles/libavrdude.dir/pindefs.c.obj src/CMakeFiles/libavrdude.dir/ppi.c.obj src/CMakeFiles/libavrdude.dir/ppiwin.c.obj src/CMakeFiles/libavrdude.dir/serbb_posix.c.obj src/CMakeFiles/libavrdude.dir/serbb_win32.c.obj src/CMakeFiles/libavrdude.dir/ser_avrdoper.c.obj src/CMakeFiles/libavrdude.dir/ser_posix.c.obj src/CMakeFiles/libavrdude.dir/ser_win32.c.obj src/CMakeFiles/libavrdude.dir/serialupdi.c.obj src/CMakeFiles/libavrdude.dir/stk500.c.obj src/CMakeFiles/libavrdude.dir/stk500v2.c.obj src/CMakeFiles/libavrdude.dir/stk500generic.c.obj src/CMakeFiles/libavrdude.dir/strutil.c.obj src/CMakeFiles/libavrdude.dir/teensy.c.obj src/CMakeFiles/libavrdude.dir/term.c.obj src/CMakeFiles/libavrdude.dir/updi_link.c.obj src/CMakeFiles/libavrdude.dir/updi_nvm.c.obj src/CMakeFiles/libavrdude.dir/updi_readwrite.c.obj src/CMakeFiles/libavrdude.dir/updi_state.c.obj src/CMakeFiles/libavrdude.dir/urclock.c.obj src/CMakeFiles/libavrdude.dir/usbasp.c.obj src/CMakeFiles/libavrdude.dir/usb_hidapi.c.obj src/CMakeFiles/libavrdude.dir/usb_libusb.c.obj src/CMakeFiles/libavrdude.dir/usbtiny.c.obj src/CMakeFiles/libavrdude.dir/update.c.obj src/CMakeFiles/libavrdude.dir/wiring.c.obj src/CMakeFiles/libavrdude.dir/xbee.c.obj src/CMakeFiles/libavrdude.dir/__/lexer.c.obj src/CMakeFiles/libavrdude.dir/__/config_gram.c.obj  -lm  C:/msys64/mingw64/lib/libelf.a  C:/msys64/mingw64/lib/libusb.dll.a  C:/msys64/mingw64/lib/libusb-1.0.dll.a  C:/msys64/mingw64/lib/libhidapi.dll.a  C:/msys64/mingw64/lib/libftdi1.dll.a  C:/msys64/mingw64/lib/libreadline.dll.a  C:/msys64/mingw64/lib/libncurses.a  -lsetupapi  -lhid  -lws2_32  -lkernel32 -luser32 -lgdi32 -lwinspool -lshell32 -lole32 -loleaut32 -luuid -lcomdlg32 -ladvapi32 && cd ."
FAILED: src/libavrdude.dll src/liblibavrdude.dll.a
cmd.exe /C "cd . && C:\msys64\mingw64\bin\cc.exe -O2 -g -DNDEBUG  -Wl,-undefined,dynamic_lookup -shared -o src\libavrdude.dll -Wl,--out-implib,src\liblibavrdude.dll.a -Wl,--major-image-version,1,--minor-image-version,0 src/CMakeFiles/libavrdude.dir/arduino.c.obj src/CMakeFiles/libavrdude.dir/avr.c.obj src/CMakeFiles/libavrdude.dir/avr910.c.obj src/CMakeFiles/libavrdude.dir/avrcache.c.obj src/CMakeFiles/libavrdude.dir/avrftdi.c.obj src/CMakeFiles/libavrdude.dir/avrftdi_tpi.c.obj src/CMakeFiles/libavrdude.dir/avrintel.c.obj src/CMakeFiles/libavrdude.dir/avrpart.c.obj src/CMakeFiles/libavrdude.dir/bitbang.c.obj src/CMakeFiles/libavrdude.dir/buspirate.c.obj src/CMakeFiles/libavrdude.dir/butterfly.c.obj src/CMakeFiles/libavrdude.dir/ch341a.c.obj src/CMakeFiles/libavrdude.dir/config.c.obj src/CMakeFiles/libavrdude.dir/confwin.c.obj src/CMakeFiles/libavrdude.dir/crc16.c.obj src/CMakeFiles/libavrdude.dir/dfu.c.obj src/CMakeFiles/libavrdude.dir/dryrun.c.obj src/CMakeFiles/libavrdude.dir/fileio.c.obj src/CMakeFiles/libavrdude.dir/flip1.c.obj src/CMakeFiles/libavrdude.dir/flip2.c.obj src/CMakeFiles/libavrdude.dir/ft245r.c.obj src/CMakeFiles/libavrdude.dir/jtagmkI.c.obj src/CMakeFiles/libavrdude.dir/jtagmkII.c.obj src/CMakeFiles/libavrdude.dir/jtag3.c.obj src/CMakeFiles/libavrdude.dir/linuxgpio.c.obj src/CMakeFiles/libavrdude.dir/linuxspi.c.obj src/CMakeFiles/libavrdude.dir/lists.c.obj src/CMakeFiles/libavrdude.dir/micronucleus.c.obj src/CMakeFiles/libavrdude.dir/par.c.obj src/CMakeFiles/libavrdude.dir/pgm.c.obj src/CMakeFiles/libavrdude.dir/pgm_type.c.obj src/CMakeFiles/libavrdude.dir/pickit2.c.obj src/CMakeFiles/libavrdude.dir/pindefs.c.obj src/CMakeFiles/libavrdude.dir/ppi.c.obj src/CMakeFiles/libavrdude.dir/ppiwin.c.obj src/CMakeFiles/libavrdude.dir/serbb_posix.c.obj src/CMakeFiles/libavrdude.dir/serbb_win32.c.obj src/CMakeFiles/libavrdude.dir/ser_avrdoper.c.obj src/CMakeFiles/libavrdude.dir/ser_posix.c.obj src/CMakeFiles/libavrdude.dir/ser_win32.c.obj src/CMakeFiles/libavrdude.dir/serialupdi.c.obj src/CMakeFiles/libavrdude.dir/ [... cut ...]
[69/70] C:\msys64\mingw64\bin\cc.exe -DCONFIG_DIR="\"C:/Program Files (x86)/avrdude/etc\"" -IC:/work/avr/avrdude_test/avrdude_main/build_shared/src -IC:/work/avr/avrdude_test/avrdude_main -IC:/work/avr/avrdude_test/avrdude_main/build_shared -IC:/msys64/mingw64/include/libusb-compat -IC:/work/avr/avrdude_test/avrdude_main/src -O2 -g -DNDEBUG -std=gnu11 -Wall -Wextra -Wno-unused-parameter -MD -MT src/CMakeFiles/avrdude.dir/developer_opts.c.obj -MF src\CMakeFiles\avrdude.dir\developer_opts.c.obj.d -o src/CMakeFiles/avrdude.dir/developer_opts.c.obj -c C:/work/avr/avrdude_test/avrdude_main/src/developer_opts.c
ninja: build stopped: subcommand failed.
```

mcuee, 2023-06-21:
Change this to bug for Windows.

It seems to me that most CMake based project will require seperate build for static and dynamic library. So the original request can be closed as not planned.

But the Windows dynamic library build is a bug to me.

mcuee, 2023-06-21:
But the Windows dynamic library build is a bug to me. So I will create a new issue for that.

closed 2023-06-21 by mcuee

mcuee, 2023-10-28:
Full build log with latest git main and MSYS2 mingw64.

<details>
<summary> click for the full log </summary>

```
 MINGW64 /c/work/avr/avrdude_test/avrdude
$ cmake -D CMAKE_BUILD_TYPE=RelWithDebInfo -D DEBUG_CMAKE=1 -D USE_STATIC_LIBS=OFF -DBUILD_SHARED_LIBS=ON -D CMAKE_VERBOSE_MAKEFILE=ON  -B build_shared -DCMAKE_SHARED_LINKER_FLAGS=-Wl,-undefined,dynamic_lookup
-- Building for: Ninja
-- The C compiler identification is GNU 13.1.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: C:/msys64/mingw64/bin/cc.exe - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Found Git: C:/msys64/usr/bin/git.exe (found version "2.40.1")
-- Found FLEX: C:/msys64/usr/bin/flex.exe (found version "2.6.4")
-- Found BISON: C:/msys64/usr/bin/bison.exe (found version "3.8.2")
-- Looking for libelf.h
-- Looking for libelf.h - found
-- Looking for libelf/libelf.h
-- Looking for libelf/libelf.h - found
-- Looking for usb.h
-- Looking for usb.h - not found
-- Looking for lusb0_usb.h
-- Looking for lusb0_usb.h - found
-- Looking for libusb.h
-- Looking for libusb.h - not found
-- Looking for libusb-1.0/libusb.h
-- Looking for libusb-1.0/libusb.h - found
-- Looking for hidapi/hidapi.h
-- Looking for hidapi/hidapi.h - found
-- Looking for ftdi_tcioflush
-- Looking for ftdi_tcioflush - found
-- Configuration summary:
-- ----------------------
-- CMAKE_HOST_SYSTEM: Windows-10.0.22621
-- CMAKE_SYSTEM: Windows-10.0.22621
-- CMAKE_FIND_ROOT_PATH:
-- CMAKE_C_COMPILER: C:/msys64/mingw64/bin/cc.exe
-- CONFIG_DIR: C:/Program Files (x86)/avrdude/etc
-- AVRDUDE_FULL_VERSION: 7.2-20231024 (d6c61f5f)
-- USE_EXTERNAL_LIBS: OFF
-- USE_LIBUSBWIN32: OFF
-- HAVE_LIBELF: C:/msys64/mingw64/lib/libelf.a
-- HAVE_LIBUSB: C:/msys64/mingw64/lib/libusb.dll.a
-- HAVE_LIBUSB_1_0: C:/msys64/mingw64/lib/libusb-1.0.dll.a
-- HAVE_LIBUSB_WIN32: C:/msys64/mingw64/lib/libusb0.dll.a
-- HAVE_LIBHIDAPI: C:/msys64/mingw64/lib/libhidapi.dll.a
-- HAVE_LIBFTDI: HAVE_LIBFTDI-NOTFOUND
-- HAVE_LIBFTDI1: C:/msys64/mingw64/lib/libftdi1.dll.a
-- HAVE_LIBREADLINE: C:/msys64/mingw64/lib/libreadline.dll.a
-- HAVE_LIBSERIALPORT: 1
-- HAVE_LIBELF_H: 1
-- HAVE_LIBELF_LIBELF_H: 1
-- HAVE_USB_H: 1
-- HAVE_LUSB0_USB_H: 1
-- HAVE_LIBUSB_H:
-- HAVE_LIBUSB_1_0_LIBUSB_H: 1
-- HAVE_HIDAPI_HIDAPI_H: 1
-- LIBUSB_COMPAT_DIR: C:/msys64/mingw64/include/libusb-compat
-- HAVE_LIBGPIOD:
-- ----------------------
-- DO HAVE    libelf
-- DO HAVE    libusb
-- DO HAVE    libusb_1_0
-- DO HAVE    libhidapi
-- DON'T HAVE libftdi
-- DO HAVE    libftdi1
-- DO HAVE    libreadline
[... 112 lines cut ...]
C:/msys64/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/13.1.0/../../../../x86_64-w64-mingw32/bin/ld.exe: src/CMakeFiles/libavrdude.dir/avrcache.c.obj:C:/work/avr/avrdude_test/avrdude/src/avrcache.c:302: more undefined references to `avrdude_message2' follow
C:/msys64/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/13.1.0/../../../../x86_64-w64-mingw32/bin/ld.exe: src/CMakeFiles/libavrdude.dir/dryrun.c.obj:dryrun.c:(.rdata$.refptr.partdesc[.refptr.partdesc]+0x0): undefined reference to `partdesc'
C:/msys64/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/13.1.0/../../../../x86_64-w64-mingw32/bin/ld.exe: src/CMakeFiles/libavrdude.dir/fileio.c.obj: in function `fileio_fmt_autodetect_fp':
C:/work/avr/avrdude_test/avrdude/src/fileio.c:1412: undefined reference to `avrdude_message2'
C:/msys64/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/13.1.0/../../../../x86_64-w64-mingw32/bin/ld.exe: src/CMakeFiles/libavrdude.dir/fileio.c.obj: in function `fileio_fmt_autodetect':
C:/work/avr/avrdude_test/avrdude/src/fileio.c:1422: undefined reference to `avrdude_message2'
C:/msys64/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/13.1.0/../../../../x86_64-w64-mingw32/bin/ld.exe: src/CMakeFiles/libavrdude.dir/fileio.c.obj: in function `segment_normalise':
C:/work/avr/avrdude_test/avrdude/src/fileio.c:1469: undefined reference to `avrdude_message2'
C:/msys64/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/13.1.0/../../../../x86_64-w64-mingw32/bin/ld.exe: C:/work/avr/avrdude_test/avrdude/src/fileio.c:1460: undefined reference to `avrdude_message2'
C:/msys64/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/13.1.0/../../../../x86_64-w64-mingw32/bin/ld.exe: src/CMakeFiles/libavrdude.dir/fileio.c.obj: in function `fileio_setparms':
C:/work/avr/avrdude_test/avrdude/src/fileio.c:1309: undefined reference to `avrdude_message2'
C:/msys64/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/13.1.0/../../../../x86_64-w64-mingw32/bin/ld.exe: src/CMakeFiles/libavrdude.dir/fileio.c.obj:C:/work/avr/avrdude_test/avrdude/src/fileio.c:1233: more undefined references to `avrdude_message2' follow
C:/msys64/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/13.1.0/../../../../x86_64-w64-mingw32/bin/ld.exe: src/CMakeFiles/libavrdude.dir/ft245r.c.obj:ft245r.c:(.rdata$.refptr.progbuf[.refptr.progbuf]+0x0): undefined reference to `progbuf'
C:/msys64/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/13.1.0/../../../../x86_64-w64-mingw32/bin/ld.exe: src/CMakeFiles/libavrdude.dir/jtagmkI.c.obj: in function `jtagmkI_resync':
C:/work/avr/avrdude_test/avrdude/src/jtagmkI.c:247: undefined reference to `avrdude_message2'
C:/msys64/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/13.1.0/../../../../x86_64-w64-mingw32/bin/ld.exe: C:/work/avr/avrdude_test/avrdude/src/jtagmkI.c:255: undefined reference to `avrdude_message2'
[... cut ...]
C:/msys64/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/13.1.0/../../../../x86_64-w64-mingw32/bin/ld.exe: C:/work/avr/avrdude_test/avrdude/src/jtagmkI.c:258: undefined reference to `avrdude_message2'
C:/msys64/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/13.1.0/../../../../x86_64-w64-mingw32/bin/ld.exe: src/CMakeFiles/libavrdude.dir/jtagmkI.c.obj:C:/work/avr/avrdude_test/avrdude/src/jtagmkI.c:259: more undefined references to `avrdude_message2' follow
collect2.exe: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
```

</details>

ndim, 2024-08-11:
Regarding `_avrdude_message2` and the linking problems, see also https://github.com/avrdudes/avrdude/pull/1874#issuecomment-2275666777 and some of the follow-on comments.